merbanan/rtl_433 #2814: New device Bresser HCHO/VOC sensor, PN 7009978

Mart124, 2024-02-02:
Hello,

Bresser does a HCHO/VOC sensor, PN [7009978](https://www.bresser.de/en/Weather-Time/BRESSER-HCHO-VOC-Sensor.html), [user manual](https://www.bresser.de/out/media/8b90e446809ad4799c34f577ee2db342.pdf).

Would be really nice to support it, in addition to the (same form-factor) Bresser [PM2.5 / PM10](https://github.com/merbanan/rtl_433/issues/2693) sensor and Bresser [CO2](https://github.com/merbanan/rtl_433/issues/2813) sensor.

Here are then some samples from it : [7009978.zip](https://github.com/merbanan/rtl_433/files/14145559/7009978.zip) :
- g001_868.34M_250k.cu8 : 5 ppb / 5 leaves
- g002_868.34M_250k.cu8 : 5 ppb / 5 leaves
- g001_868.34M_1000k.cu8 : 5 ppb / 4 leaves
- g002_868.34M_1000k.cu8 : 5 ppb / 4 leaves
- g003_868.34M_1000k.cu8 : 8 ppb / 2 leaves
- g004_868.34M_1000k.cu8 : 4 ppb / 1 leaves

Could we then try to decode it please ?
Of course, feel free if needed !

Thank you very much 👍

ProfBoc75, 2024-02-02:
Hi @Mart124 

What I already found is that the signal is like other Bresser devices,
you can try with :

live
`rtl_433 -f 868.34M -X 'n=Bresser,m=FSK_PCM,s=124,l=124,r=1000,preamble=aaaa2dd4'`

or replay 
`rtl_433 -X 'n=Bresser,m=FSK_PCM,s=124,l=124,r=1000,preamble=aaaa2dd4' -r *.cu8`

Then the data needs to be decoded, looking into the other Bresser sensors, we need to fallow a specific data processing to get the figures. I'm working on it and let you know.

it's the same for #2813, I guess we can combine both in the same decoder.

ProfBoc75, 2024-02-03:
Good progress for me, I found that the Bresser 7in1 should match, it found s_type=10 (CO2 / 7009977) and 11 (HCHO/VOC / 7009978) not yet decoded, where s_type=8 is already a AIR_PM sensor.

To be continued ....

Mart124, 2024-02-03:
Wow, good news @ProfBoc75 👍
Really glad you took a look at this, and managed to get good progress !
Thankfully we can benefit from existing Bresser devices.
These 2 devices will really be nice additions for air quality monitoring.
Of course feel free if you need additional samples at different rates etc... from the devices.
Thank you again !

ProfBoc75, 2024-02-03:
@Mart124 : here my findings and taken into account in the device updates :

```
Air Quality Sensor CO2 (PN 7009977) : issue #2813

From user manual , co2 ppm is from 400 to 5000 ppm, so it's 16 bits coded.

Samples :
Raw :
              SType Startup & Channel
                  | |
{207}dab6d782acd9 a 1 ad9aad9aad9aaaaaaaaaaaaaaaaae99aaaaa00 Type = 0xa = 10, Startup = 0, ch = 1
{207}04a9d782acd8 a 1 ad9aad9aad9aaaaaaaaaaaaaaaaae99aaaaa00 Type = 0xa = 10, Startup = 0, ch = 1
{207}04a9d782acd8 a 1 ad9aad9aad9aaaaaaaaaaaaaaaaae99aaaaa00 Type = 0xa = 10, Startup = 0, ch = 1
{207}0dd1d782b8ee a 1 ad9aad9aad9aaaaaaaaaaaaaaaaae99aaaaa00 Type = 0xa = 10, Startup = 0, ch = 1

Data layout raw : 
DIGEST:16h ID:16h 8x8x STYPE:4h STARTUP:1b CH:3d 8x8x8x8x8x8x8x8x8x8x8x8x8x8x8x8x8x8x TRAILER:8x

XOR / de-whitened :

      0 1  2 3  4 5  6 7 8 9101112131415161718192021222324
   DIGEST   ID  ppm                  bat
        |    |    |                    |
{200}701c 7d28 0673 0b073007300730000000000000000043300000 [ XOR from g001_868.34M_1000k.cu8 co2 ppm  673]
{200}ae03 7d28 0672 0b073007300730000000000000000043300000 [ XOR from g001_868.34M_250k.cu8  co2 ppm  672]
{200}ae03 7d28 0672 0b073007300730000000000000000043300000 [ XOR from g002_868.34M_1000k.cu8 co2 ppm  672]
{200}a77b 7d28 1244 0b073007300730000000000000000043300000 [ XOR from g002_868.34M_250k.cu8  co2 ppm 1244]

Data layout de-whitened : 
DIGEST:16h ID:16h PPM:16h 8x8x8x8x8x8x8x8x8x8x4x BATT:1b 3x8x8x8x8x8x8x TRAILER:16x

Air Quality Sensor HCHO/VPC (PN 7009978) : issue #2814

From user manual , hcho ppb is from 0 to 1000 ppm, so it's 16 bits coded.
              and voc level is from 1 (bad air quality) to 5 (good air quality), so it's 4 bits coded.

Samples:
Raw :
              SType Startup & Channel
                  | |
{207}3f2dc4a5aaaf b 1 aaa8aaa8aaa8aaaaaaaaaaaaaaaae9feaaaa00 Type = 0xb = 11, Startup = 0, ch = 1
{207}0c1cc4a5aaaf b 1 aaa8aaa8aaa8aaaaaaaaaaaaaaaae9ffaaaa00 Type = 0xb = 11, Startup = 0, ch = 1
{207}3f2dc4a5aaaf b 1 aaa8aaa8aaa8aaaaaaaaaaaaaaaae9feaaaa00 Type = 0xb = 11, Startup = 0, ch = 1
{207}0c1cc4a5aaaf b 1 aaa8aaa8aaa8aaaaaaaaaaaaaaaae9ffaaaa00 Type = 0xb = 11, Startup = 0, ch = 1
{207}61afc4a5aaa2 b 9 aaa8aaa8aaa9aaaaaaaaaaaaaaaae9f8aaaa00 Type = 0xb = 11, Startup = 1, ch = 1
{207}ecddc4a5aaae b 9 aaa8aaa8aaa9aaaaaaaaaaaaaaaae9fbaaaa00 Type = 0xb = 11, Startup = 1, ch = 1

Data layout raw : 
DIGEST:16h ID:16h 8x8x STYPE:4h STARTUP:1b CH:3d 8x8x8x8x8x8x8x8x8x8x8x8x8x8x8x8x8x8x TRAILER:8x

XOR / de-whitened :

      0 1  2 3  4 5  6 7 8 9101112131415161718192021 22 2324
   DIGEST   ID  ppb                  bat            voc
        |    |    |                    |              |
{200}9587 6e0f 0005 1b0002000200020000000000000000435 4 0000 [XOR from g001_868.34M_1000k.cu8 hcho_ppb 5 voc_level 4]
{200}a6b6 6e0f 0005 1b0002000200020000000000000000435 5 0000 [XOR from g001_868.34M_250k.cu8  hcho_ppb 5 voc_level 5]
{200}9587 6e0f 0005 1b0002000200020000000000000000435 4 0000 [XOR from g002_868.34M_1000k.cu8 hcho_ppb 5 voc_level 4]
{200}a6b6 6e0f 0005 1b0002000200020000000000000000435 5 0000 [XOR from g001_868.34M_250k.cu8  hcho_ppb 5 voc_level 5]
{200}cb05 6e0f 0008 130002000200030000000000000000435 2 0000 [XOR from g003_868.34M_1000k.cu8 hcho_ppb 8 voc_level 2]
{200}4677 6e0f 0004 130002000200030000000000000000435 1 0000 [XOR from g004_868.34M_1000k.cu8 hcho_ppb 4 voc_level 1]

Data layout de-whitened : 
DIGEST:16h ID:16h PPB:16h 8x8x8x8x8x8x8x8x8x8x4x BATT:1b 3x8x8x8x8x8x4x VOC:4h TRAILER:16x
```

I will push a PR with the bresser_7n1.c updates very soon.

closed 2024-02-03 by zuckschwerdt

Mart124, 2024-02-06:
_Added some info in first post related to sensors used, so that one can judge about this device quality._

farmabep, 2025-07-14:
First of all, thanks for the quick reply,
but the link you sent me (https://share.google/yapICMpT1xWFZCxGs)explains doesn't open.