merbanan/rtl_433 #2816: Bresser Air Quality sensors, ignore first packet

Mart124, 2024-02-04:
Hello,

There's a tiny issue with the following 3 sensors :
Bresser CO2 sensor, PN 7009977
Bresser HCHO/VOC sensor, PN 7009978
Bresser PM2.5/PM10 sensor, PN 7009970
(I do not own other Bresser sensors, so can only report for these 3).

When they start, they all send a first packet with their values set to 15.
CO2, HCHO, VOC, PM2.5, PM10 : all set to 15.
Of course these are dummy values, certainly because these devices send a first packet before their sensors had time to compute a real value.

Could we then ignore these "15" values please ?
So that they do not mess-up graphs etc... at presentation level (Home Assistant for instance).

Thank you very much 👍

ProfBoc75, 2024-02-04:
Hi @Mart124 : 

Thanks to report this use case.
It's normal behavior, as such sensors require preheating to generate the correct calibrated value, so the first one is not good and/or not yet filled.

Yes, for me their is 2 possible approaches around this use case:

- First one, we want from rtl_433 to alter the result as less as possible, so we would like to let any filtering of values outside rtl_433.

- Second one, if we can identify a flag present only during this first message then we can take it into account and not provide any value. For that, we need you to capture the codes at this exact moment of the first signal is sent, then, we may found into the codes (raw MSG and de-whitened XOR) this specific flag (like startup flag is already existing) that could help us identify this 15 "wrong" value, then ignore it with a simple DATA_COND and provide only this init situation, like "init = 1".

if you captured already this moment, replay it and provide both MSG and XOR codes :
`rtl_433 *.cu8 -R 173:vv`

sample result :
[image]

or redo it in live :
`rtl_433 -f 868.34M -R 173:vv`

In that situation, the sensor ID may change too, but it will definitely help us to get the good figures.

Thanks for your contribution.

zuckschwerdt, 2024-02-05:
I suspect that the indicator is a raw `0xF` in the BCD values. We can DATA_COND those values, e.g. `int co2_ok = msg[5] & 0x0f == 0x0f;`

ProfBoc75, 2024-02-05:
@zuckschwerdt : you're right , I forgot it was coded in BCD, so yes, easy to filter out.

Mart124, 2024-02-05:
I'll be able to capture these first packets later tonight, I'll come back here ASAP 👍

ProfBoc75, 2024-02-05:
> I'll be able to capture these first packets later tonight, I'll come back here ASAP 👍

Thanks, samples are always welcome to deep dive and later support.
In the meantime, we updated rtl_433 to take into account this scenario, you should be able to test it and waiting after your feedback to close this issue.

Thx.

Mart124, 2024-02-09:
Sorry for the delay, but here are the startup moments for every device : [2816.zip](https://github.com/merbanan/rtl_433/files/14218350/2816.zip)
I captured each device two times.
Note that 7009978 (HCHO/VOC) seems to send 2 "empty" frames (at 1 minute interval) before displaying proper counters (hence the 4 dump files for this device).

ProfBoc75, 2024-02-09:
Hi @Mart124 : Thanks a lot ! 
I will have to correct the device since, it's ok for HCHO/VOC and CO2 but not for the PM2.5/PM10, still showing wrong value.
Anyway, we have all the kind of cu8 files to test accordingly.

Thanks again, and I will push a PR very soon with the correction.

Mart124, 2024-02-10:
Thank you very much for this PR / bugfix @ProfBoc75 !

ProfBoc75, 2024-02-10:
@Mart124 , if all is fine, you can close this issue 😉

Mart124, 2024-02-10:
Great, thank you very much to both of you @ProfBoc75 @zuckschwerdt 👍

closed 2024-02-10 by Mart124

zuckschwerdt, 2024-02-10:
@ProfBoc75 can you send me a private email to coordinate project things?