merbanan/rtl_433 #2819: Integrate: TST-507 TPMS sensors

jaysmith9, 2024-02-05:
Would it be possible to include the TST-507 TPMS sensors for this? There is a website (https://github.com/cterwilliger/tst_tpms) that contains the details that they were able to decode for the sensor. I'd be happy to test as needed.

The sensors transmit data periodically on 443.92Mhz using OOK modulation. The 8 byte data packet contains a checksum, sensor ID, and measurement data. The data packet is encoded with Manchester I (IEEE 802.3).

32 bits of preamble (1010...)
checksum of 0xE5 (byte 0)
sensor ID of 0x563ED9 (bytes 1-3)
data of 0x2C4B0001
Decoding the data portion we get:

pressure = 2nd nibble of byte 7 + byte 4 = 0x012C
0x012C / 0.4 = 750 kPa
750 / 6.895 = 108.8 psi
temperature = byte 5 = 0x4B
0x4B - 50 = 25 deg C
(25 * 1.8) + 32 = 77 deg F
checksum = (sum of bytes 1-7) % 256
Flags:

Byte 6
0x10 = Fast Leak
0x20 = Low Battery

ProfBoc75, 2024-02-06:
I found old commits for rtl_433_tests and cu8 samples from the same author https://github.com/merbanan/rtl_433_tests/tree/06e96e6a36b44e0a22c8c23c88738fe4cea67987/tests/tst507

I'll analyze them and come back here with my findings.

ProfBoc75, 2024-02-06:
Sounds to be already known under EezTire-E618 device.

    [241]  EezTire E618, Carchet TPMS

Samples from my previous post regarding the tst-507 :

[image]

So, @jaysmith9 , waiting after your own cu8 samples to confirm that, and we will just update the supported TPMS model "TST-507" for rtl_433 and/or update the code too if not working properly as expected.

Thx.

jaysmith9, 2024-02-06:
Hi ProfBoc75,

Thank you or the very quick response and investigation on this. I am trying to run this on a new install. I used "apt-get install rtl-433" to install the package and it runs fine (and displays various temperature sensors). However if I try to use it for that specicific sensor ("rtl_433 -R 241"), it gives the error message that there are only 223 protocols available. How can I add this protocol so I can test if it works as is? I had thought that "apt-get install rtl-433" would automatically install the latest version.

Please forgive me as I am not that well versed on Linux.

ProfBoc75, 2024-02-06:
@jaysmith9 : Understand, the linux repository is not up to date, and apt-get will not install the last one which explain that it's not working.

You have to carefully follow this [build guide](https://github.com/merbanan/rtl_433/blob/master/docs/BUILDING.md)

3 major steps extract from this guide :
- Install prerequisites,
- git clone rtl_433,
- then cmake/make/make install

#### Install all the required libraries to be able to build the program, it will depend on your Linux distribution, the first part of the guide (Debian / Ubuntu) :

````
sudo apt-get install libtool libusb-1.0-0-dev librtlsdr-dev rtl-sdr build-essential cmake pkg-config libssl-dev
````
#### Clone rtl_433

    git clone https://github.com/merbanan/rtl_433.git

#### CMake / Make / Make install

Installation using CMake and Make (commonly available):

    cd rtl_433/
    mkdir build
    cd build
    cmake ..
    make
    sudo make install

This will take some time, but they are simple tasks.
You can interrupt make by Ctrl + C and restart it, it will continue from the break. 
Depends on your system, you can speed up the make by adding -j option followed by the number of processors/threads:

    make -j 4
    sudo make install

From here, you should be able to use rtl_433 like it was installed through apt-get.

#### New Build if new updates:

With the above approach, the next updates will be faster. Just git pull to update the code then make and make install

Go back the rtl_433 folder :

    cd rtl_433/
    git pull
    cd build
    make -j 4
    sudo make install

jaysmith9, 2024-02-06:
I ended up manually downloading and rebuilding and now it seems to pull in all of the latest sensors, but doesn't seem to recognize the TST-507 sensors.

I am in a rather "noisy" area so if I run "rtl_433 -S unknown" it picks up quite a few unknown signals. Is there a way to identify which files might be the right ones so I can filter them out for you?

ProfBoc75, 2024-02-07:
You can drag and drop file here https://triq.org/pdv and look if you see the signal.

You can also try with -R 241:vv to get the debug information, why it's not decoding the data.

    rtl_433 -r gxxxxx.cu8 -R 241:vv

You can check the size of the files, they should be around 130 KB.

Depends on the number of files, you can drag and drop here few files, you can compress them into a single zip and upload here.

jaysmith9, 2024-02-07:
Thank you for your ongoing help with this. I have attached several captures and I am hoping that at least one of them might be from the TST-507. Can you determine if any of these looks correct?

When I try "rtl_433 -r gxxxxx.cu8 -R 241:vv" on these files, it seemed that #12 and 15 had some results, but I'm not sure how to interpret it.

[cu8 Files.zip](https://github.com/merbanan/rtl_433/files/14188604/cu8.Files.zip)

jaysmith9, 2024-02-07:
@ProfBoc75: Good news - I've got it going! The key was the increased sample rate (-s 1024K). It does not recognize the sensors at the default sample rate, but once I increased the sample rate to 1024k, it was pulling in the TST-507 sensors and providing the correct data. 

I greatly appreciate all of your help with this. You may want to update the file to show support for the TPMS model "TST-507.

Let me know if you would like anything further from me.

ProfBoc75, 2024-02-07:
Hi @jaysmith9 : Good to know ! We will update the supported model TST-507.

ProfBoc75, 2024-02-10:
Sorry, I made a little mistake that opened 3 branch patches.
Now, it's ok, the 3 commits in the same PR and my same branch patch-2.

closed 2024-02-22 by ProfBoc75