merbanan/rtl_433 #2821: BMW Gen5 TPMS support

petrjac, 2024-02-06:
Hi,

I'm trying to decode **BMW Gen5 TPMS** sensors.

I recorded several samples using _Airspy R2_ SDR - https://github.com/merbanan/rtl_433_tests/pull/463/files .

I was able to read raw data using _URH_ and resolved _CRC-8_ as `poly=0x2f  init=0x40`:
```
0xfff20323e136a14a3e016b686b
0xfff20323e136a1343d017468cf
0xfff20323e137a81b3e005a68dc
```

However with _rtl_433_ I'm getting different results:
```
❯ rtl_433 -r tests/BMW_G5_TPMS/g018_433.92M_2500k.cs16 -R 0 -X 'n=tpms_bmw,m=FSK_MC_ZEROBIT,s=25,l=25,r=100' -Y minmax
rtl_433 version 23.11-19-g83afc7b4 branch master at 202312252023 inputs file rtl_tcp RTL-SDR SoapySDR with TLS
Disabling all device decoders.
[Input] Test mode active. Reading samples from file: tests/BMW_G5_TPMS/g018_433.92M_2500k.cs16
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : @0.002138s
model     : tpms_bmw     count     : 1             num_rows  : 1             rows      : 
len       : 104          data      : 00020323e136a14a3e016b686b
codes     : {104}00020323e136a14a3e016b686b
```

Binary data started with `0x000` instead of `0xfff`.

Could please someone help with verifying data before proceeding?

zuckschwerdt, 2024-02-06:
It's basically the same. MC can't reliably discern the timing of the short transitions until the first long one.

petrjac, 2024-02-06:
Does it make sense then calculated CRC for stripped data:
```
❯ bin/i386-linux/reveng -w 8 -s 0323e136a14a3e016b686b 0323e136a1343d017468cf 0323e137a81b3e005a68dc 0323e136a1403e016f6830 0323e136a1343d017468cf
width=8  poly=0x2f  init=0xaa  refin=false  refout=false  xorout=0x00  check=0x2a  residue=0x00  name=(none)
```
?

petrjac, 2024-02-06:
With [bitbench](https://triq.net/bitbench#c=fff20323e137a82f3e005268f7&c=fff20323e137a82a3e00546898&c=fff20323e137a8263e00566874&c=fff20323e136a1343d017468cf&f=24h%20%7C%20SENSOR_ID%20%3D%2032h%20%7C%20PRESS%20%3D%208d%20%7C%20TEMP%20%3D%208d%20%7C%208h%20%7C%20SEQ_ID%20%3D%208h%20%7C%208h%20%7C%20CRC%20%3D%208h%20&cw=4) I am able to guess:
 - sensor ID
 - pressure [bar]
 - nominal (target?) pressure [bar]
 - temperate [°C]
 - CRC

Should I investigate more or could I try to prepare decoder?

zuckschwerdt, 2024-02-06:
Looks good already. You might be missing the alarm flags like rapid deflate, low battery and diagnostics as spinning, triggered, ...

petrjac, 2024-02-06:
I tried to guess couple more columns - [bitbench](https://triq.net/bitbench#c=fff20323e137a82f3e005268f7&c=fff20323e137a82a3e00546898&c=fff20323e137a8263e00566874&c=fff20323e136a1343d017468cf&f=16h%20%7C%20MANUFACTURER%20%3D%208h%20%7C%20SENSOR_ID%20%3D%2032h%20%7C%20PRESS%20%3D%208d%20%7C%20TEMP%20%3D%208d%20%7C%20WARN%20%3D%208h%20%7C%20SEQ_ID%20%3D%208h%20%7C%20TARGET_PRESS%20%3D%208d%20%7C%20CRC%20%3D%208h%20&cw=4), but some of them are hard for me to verify.

I'm using diagnostic tool as reference:
[image]

What should be the next steps?

ProfBoc75, 2024-02-07:
> Binary data started with `0x000` instead of `0xfff`.
> 
> Could please someone help with verifying data before proceeding?

Hi @petrjac  : try to increase the pulse to 27 instead of 25 and you will get the 0xffff, if you reduce to 24 you will get the 0x0000

zuckschwerdt, 2024-02-07:
> increase the pulse to 27 instead of 25 and you will get the 0xfff, if you reduce to 24 you will get the 0x000

That's unexpected. For MC we don't need an exact timing and the slicer should always match. Does this slight change somehow match/ignore the first "bit"? That would change the assumption about 000 vs fff.

petrjac, 2024-02-07:
- `s=25` → `{104}00020323e136a14a3e016b686b`
- `s=26` → `{104}40020323e136a14a3e016b686b`
- `s=27` → `{104}7ff20323e136a14a3e016b686b`

When trying with `tests/BMW_G5_TPMS/g018_433.92M_2500k.cs16`
```
 ❯ rtl_433 -r tests/BMW_G5_TPMS/g018_433.92M_2500k.cs16 -R 0 -X 'n=tpms_bmw,m=FSK_MC_ZEROBIT,s=27,l=27,r=100' -Y minmax
rtl_433 version 23.11-19-g83afc7b4 branch master at 202312252023 inputs file rtl_tcp RTL-SDR SoapySDR with TLS
Disabling all device decoders.
[Input] Test mode active. Reading samples from file: tests/BMW_G5_TPMS/g018_433.92M_2500k.cs16
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : @0.002138s
model     : tpms_bmw     count     : 1             num_rows  : 1             rows      : 
len       : 104          data      : 7ff20323e136a14a3e016b686b
codes     : {104}7ff20323e136a14a3e016b686b
```

zuckschwerdt, 2024-02-07:
Thanks for confirming! That does not seem right, we need to look into that. I can try to rewrite the MC slicer with that nice example we have here.
(Idea: on leading short pulses push to a counter and only add the bits when the position later is confirmed.)

ProfBoc75, 2024-02-08:
Hi @zuckschwerdt , @petrjac : 

FYI, I drafted a decoder, and started listening to the cars passing on my street.

Sounds like the bitbench is not exactly as expected, I need to consolidate my findings in a table and back here soon.

About the rf decode part, I'm using FSK_PCM, s=l=25, r=100, then looking for preamble = 0Xaa59 then MC decode + Invert.

Data Layout :
```
Brand / Manuf   : 8h, I found at least 3 Brand : 0x03 (=3),  0x23 (=35) and 0x80 (=128)
Sensor ID       : 32h,  4 byte Sensor ID
Pressure        : 8h, purpose to be confirmed, formula not guess yet
Temp            : 8h, Temp sound to be OK , `Temperature_C - 50`   (with this formula temp - 50 , I have some around 12 or 13°C matching my local temp and other more around 17 to 19°C, may be cars coming from garage ?)
Warn / Tags 1   : 8h, purpose to be confirmed, some incremental values here
SED_ID / Tags 2 : 8h, purpose to be confirmed,
Target Pressure : 8h, purpose to be confirmed, formula not guess yet, could be a third tag.
CRC             : 8h, CRC-8 poli 0x2f  , init 0xaa
```

The pressures values, I'm not able yet to find a simple formula to convert value into kPa or PSI need more samples for that.
For tags and warning did not match the bitbench and I don't have the valves, so I can't confirm.

Notice that the captures from Airspy @ 10MHz are not readable by rtl_433 (no error, but only partial value), only the 2.5MHz give expected results with -Y minmax.

petrjac, 2024-02-08:
@ProfBoc75 you are faster than me :), I was planning to write decoder on the weekend.

I'l be happy to help you at least with my findings:
- `Brand / Manuf` - considering it as _unknown_
- `Pressure` - could by `DEC * 0,0245` to get pressure in _bars_
  - *Ref. press [bar]* is value from diagnostic software
 
  | HEX  | DEC | Ref. press [bar] | round(DEC * 0.0245, 2) |
  | ------- | ------ | --------------------- | ------------------------------- |
  | 2F | 47 | 1.15 | 1.15 |
  | 2A | 42 | 1,03 | 1.03 | 
  | 26 | 38 | 0.93 | 0.93 |
  | 1B | 27 | 0.65 | _0.66_ | 
- `Temp` - could be `DEC - 52`, would need more data
    | HEX  | DEC | Ref. temp [°C] | DEC - 52 |
    | ------- | ------ | --------------------- | ------------------------------- |
    | 3E | 62 | 10 | 10 |
- `Warn / Tags` - _unknown_
-  `SEQ_ID / Tags 2` - this could be some _message id_ increment by one. I saw always 3 exactly same messages with same `SEQ_ID` followed by next 3 messages with `SEQ_ID` incremented by `0x01`.
- `Target Pressure` - _unknown_, but interestingly:
  `0x68 = 104`, `ROUND(104 * 0.0245, 2) = 2.55` is similar to `2.53` reported by diagnostic software 
- `CRC` - should be OK

ProfBoc75, 2024-02-08:
I noticed also that , depends on the first byte , Manuf/Brand, the coding is not the same, And for 0x80 (128) sounds to be Continental model.

There is another discussion [here](https://github.com/merbanan/rtl_433_tests/issues/457) where the data packet is also 11 byte 
Starting with a Brand ID (0x80) then the Sensor ID (4 byte) , 01 or 02, then temp, pressure and or tags information then exactly the same CRC-8 poli 0x2F, init 0xaa.

The RF part is not the same (not Manchester coded), but the data formatting sounds to be very close, just not information in the same order as Brand HUF (0x03).

if we guess all the formula and with DATA_COND, we may be able to handle these 2 Brands.

ProfBoc75, 2024-02-08:
FYI, it takes time to collect metrics and to deep dive, since I'm trying to manage the 3 brands in the same decoder, and not so much cars in my one way street.

At least the brand 0x03 / Huf Gen 5 is ok for me without the flags decoding which require some work, we don't know which bit is related to battery, fast deflating and so on ...

petrjac, 2024-02-08:
In diagnostic sw data I have found more manufacturer for TPMS:
- `Autonet`
- `HUF`
- `Continental`
- `TRW`
- `Sensata`
- `Unknown manufacturer`

However I cannot be sure that list is correct/complete.

On the weekend I'll try to receive signals from _HUF_ sensors to confirm _temperate_, _pressure_ and maybe also alarm tags.

Interestingly the _RDCi gen 5_ sensors should also send tire size ...

ProfBoc75, 2024-02-09:
Hi, I collected some data from the street :

I kept the original order to get the sequence number increasing if any.
Notice a new detect Brand 0x88, only one message captured.
All of them are similar for the 4 first data: Brand, ID, Pressure and Temp
The 3 last bytes (except CRC) have not the same behavior, some are fixed, other are variable, but they reflect the reality from the street.
So I tried to guess the data layout :

```

_______________________________________________________________________________________________________________________
|                               |                                              |                    | Decimal Values  |
| Msg                           | Data layout                                  | Comments           |  B7 |  B8 |  B9 |
| BB IIIIIIII PP TT F0 F1 F2 CC |                                              |                    |  F0 |  F1 |  F2 |
| 03 23b3dd0f 6d 43 57 71 e8 ae | Brand ID Pressure Temp SEQ Flags1 Target CRC |                    |  87 | 113 | 232 |
| 03 23b3dd0f 6d 43 6b 71 e8 e9 | Brand ID Pressure Temp SEQ Flags1 Target CRC | SEQ = previous +20 | 107 | 113 | 232 |
| 80 33ca4f80 6c 3e 31 0e 09 74 | Brand ID Pressure Temp Flags0 Flags1 SEQ CRC |                    |  49 |  14 |   9 |
| 80 33ca4f80 6c 3e 31 0e 0a 05 | Brand ID Pressure Temp Flags0 Flags1 SEQ CRC | SEQ = previous +1  |  49 |  14 |  10 |
| 80 33ca4f80 6c 3e 31 0e 0b 2a | Brand ID Pressure Temp Flags0 Flags1 SEQ CRC | SEQ = previous +1  |  49 |  14 |  11 |
| 80 33ca5530 6c 3e 32 1c 0e 23 | Brand ID Pressure Temp Flags0 Flags1 SEQ CRC |                    |  50 |  28 |  14 |
| 80 33ca5530 6c 3e 32 1c 0f 0c | Brand ID Pressure Temp Flags0 Flags1 SEQ CRC | SEQ = previous +1  |  50 |  28 |  15 |
| 80 33ca63b0 6a 3f 31 08 07 ce | Brand ID Pressure Temp Flags0 Flags1 SEQ CRC |                    |  49 |   8 |   7 |
| 23 501a8308 63 44 63 67 c5 ab | Brand ID Pressure Temp SEQ Flags1 Target CRC |                    |  99 | 103 | 197 |
| 23 5258ba4b 6c 45 49 71 e1 c3 | Brand ID Pressure Temp SEQ Flags1 Target CRC |                    |  73 | 113 | 225 |
| 23 5258ba4b 6c 45 5e 71 e1 09 | Brand ID Pressure Temp SEQ Flags1 Target CRC | SEQ = previous +21 |  94 | 113 | 225 |
| 23 5258ba4b 6c 45 83 71 e1 93 | Brand ID Pressure Temp SEQ Flags1 Target CRC | SEQ = previous +37 | 131 | 113 | 225 |
| 23 5258c379 6f 45 70 71 e1 59 | Brand ID Pressure Temp SEQ Flags1 Target CRC |                    | 112 | 113 | 225 |
| 23 5258c379 6f 45 84 71 e1 52 | Brand ID Pressure Temp SEQ Flags1 Target CRC | SEQ = previous +20 | 132 | 113 | 225 |
| 80 30b30da4 5d 3f 6a 21 09 2c | Brand ID Pressure Temp Flags0 Flags1 SEQ CRC |                    | 106 |  33 |   9 |
| 80 30b30da4 5d 3f 6a 21 0a 5d | Brand ID Pressure Temp Flags0 Flags1 SEQ CRC | SEQ = previous +1  | 106 |  33 |  10 |
| 80 30b30da4 5d 3f 6a 21 0b 72 | Brand ID Pressure Temp Flags0 Flags1 SEQ CRC | SEQ = previous +1  | 106 |  33 |  11 |
| 80 33ca4f80 70 44 31 29 0e e9 | Brand ID Pressure Temp Flags0 Flags1 SEQ CRC |                    |  49 |  41 |  14 |
| 80 33ca4f80 70 44 31 29 0f c6 | Brand ID Pressure Temp Flags0 Flags1 SEQ CRC | SEQ = previous +1  |  49 |  41 |  15 |
| 80 33ca63b0 6a 41 31 16 0e a7 | Brand ID Pressure Temp Flags0 Flags1 SEQ CRC |                    |  49 |  22 |  14 |
| 80 30b30da4 5d 3d 6a 15 0a a2 | Brand ID Pressure Temp Flags0 Flags1 SEQ CRC |                    | 106 |  21 |  10 |
| 80 30b30dbf 69 3f 6a 0a 05 63 | Brand ID Pressure Temp Flags0 Flags1 SEQ CRC |                    | 106 |  10 |   5 |
| 80 385135e3 5b 42 21 0d 09 50 | Brand ID Pressure Temp Flags0 Flags1 SEQ CRC |                    |  33 |  13 |   9 |
| 80 385135e3 5b 42 21 0d 0a 21 | Brand ID Pressure Temp Flags0 Flags1 SEQ CRC | SEQ = previous +1  |  33 |  13 |  10 |
| 80 385135e3 5b 42 21 0d 0b 0e | Brand ID Pressure Temp Flags0 Flags1 SEQ CRC | SEQ = previous +1  |  33 |  13 |  11 |
| 80 385135de 5b 41 1f 0d 09 74 | Brand ID Pressure Temp Flags0 Flags1 SEQ CRC |                    |  31 |  13 |   9 |
| 80 385135de 5b 41 1f 0d 0a 05 | Brand ID Pressure Temp Flags0 Flags1 SEQ CRC | SEQ = previous +1  |  31 |  13 |  10 |
| 80 385135de 5b 41 1f 0d 0b 2a | Brand ID Pressure Temp Flags0 Flags1 SEQ CRC | SEQ = previous +1  |  31 |  13 |  11 |
| 80 35ac8aa2 5e 3e 38 21 0d ea | Brand ID Pressure Temp Flags0 Flags1 SEQ CRC |                    |  56 |  33 |  13 |
| 80 35ac8aa2 5e 3e 38 21 0e 9b | Brand ID Pressure Temp Flags0 Flags1 SEQ CRC | SEQ = previous +1  |  56 |  33 |  14 |
| 80 35ac8aa2 5e 3e 38 21 0f b4 | Brand ID Pressure Temp Flags0 Flags1 SEQ CRC | SEQ = previous +1  |  56 |  33 |  15 |
| 23 235d2558 63 3e 51 78 e9 f9 | Brand ID Pressure Temp SEQ Flags1 Target CRC |                    |  81 | 120 | 233 |
| 23 235d2558 63 3e 65 78 e9 ce | Brand ID Pressure Temp SEQ Flags1 Target CRC | SEQ = previous +20 | 101 | 120 | 233 |
| 23 235d2558 63 3e 79 78 e9 66 | Brand ID Pressure Temp SEQ Flags1 Target CRC | SEQ = previous +20 | 121 | 120 | 233 |
| 80 3edad6e7 71 0e 18 12 09 61 | Brand ID Pressure Temp Flags0 Flags1 SEQ CRC |                    |  24 |  18 |   9 |
| 80 3edad6e7 71 0e 18 12 0b 3f | Brand ID Pressure Temp Flags0 Flags1 SEQ CRC | SEQ = previous +2  |  24 |  18 |  11 |
| 80 3ec301f1 74 10 18 0d 0a 44 | Brand ID Pressure Temp Flags0 Flags1 SEQ CRC |                    |  24 |  13 |  10 |
| 80 3ec301f1 74 10 18 0d 0b 6b | Brand ID Pressure Temp Flags0 Flags1 SEQ CRC | SEQ = previous +1  |  24 |  13 |  11 |
[... cut ...]
| 03 23561022 5e 3f 01 74 e9 62 | Brand ID Pressure Temp SEQ Flags1 Target CRC |                    |   1 | 116 | 233 |
| 03 23561022 5e 3f 01 74 e9 62 | Brand ID Pressure Temp SEQ Flags1 Target CRC | SEQ = previous     |   1 | 116 | 233 |
| 03 23560f93 64 3f 65 67 e9 ee | Brand ID Pressure Temp SEQ Flags1 Target CRC |                    | 101 | 103 | 233 |
| 03 23560f93 64 3f 79 67 e9 46 | Brand ID Pressure Temp SEQ Flags1 Target CRC | SEQ = previous +20 | 121 | 103 | 233 |
| 80 3177a541 5e 45 2c 09 06 85 | Brand ID Pressure Temp Flags0 Flags1 SEQ CRC |                    |  44 |   9 |   6 |

```

petrjac, 2024-02-11:
Hi,
I collected some signal and have some notes/questions:

1. Shouldn't we create separate issue for `FSK_MC_ZEROBIT` as mentioned in [comment](https://github.com/merbanan/rtl_433/issues/2821#issuecomment-1932814033) from @zuckschwerdt ?
2. Not all _BMW gen5_ signals are correctly recognized, e.g. [g082_433.92M_2500k.cs16](https://github.com/petrjac/rtl_433_tests/blob/1e49119a7997d3917b9e6597811a7bb3754540f6/tests/BMW_G5_TPMS/2/g082_433.92M_2500k.cs16)?
3. Would it be possible to somehow automatically add `-Y minmax` option for _BMW Gen5 TPMS_? I understand that this option is necessary.
4. `DEC - 52` for temperature seems to be right:
    - `059 ~ 7°C`
    - `060 ~ 8°C`
5. Pressure reading however does not exactly match with diagnostics readings, e.g. `27 * 0,0245 = 0,66 != 0,65`. Some more examples in [bitbench](https://triq.net/bitbench#c=0323e1416f3c3b004568d7&c=0323e1416f343b004768d1&c=0323e1416f343b00486895%20%5B1.3%20bar%2C%207C%2C%20WARN%20ACTIVE%5D&c=0323e1416f343b00486895&c=0323e1416f343b00486895&c=0323e1416f1e3b005168cf%20%5B0.75%20bar%2C%207C%2C%20WARN%20ACTIVE%5D&c=0323e1416f153b00586880%20%5B0.53%20bar%2C%207C%2C%20WARN%20ACTIVE%5D&c=0323e1416f513b005068fa%20%5B2.03%20bar%2C%207C%2C%20WARN%20ACTIVE%5D&c=0323e1416f763c005d6896%20%5B2.95%20bar%2C%208C%2C%20WARN%20INACTIVE%5D&c=0323e1416f693c004268e7%20%5B2.63%20bar%2C%208C%2C%20WARN%20INACTIVE%5D&f=MANUFACTURER%20%3D%208h%20%7C%20SENSOR_ID%20%3D%2032h%20%7C%20PRESS%20%3D%208d%20%7C%20TEMP%20%3D%208d%20%7C%20TAGS%20%3D%208h%20%7C%20SEQ_ID%20%3D%208d%20%7C%20TARGET_PRESS%20%3D%208h%20%7C%20CRC%20%3D%208h%20&cw=4).
6. I feel confident about `SEQ_ID`. I recorded many samples and the pattern is:
    - 3x same data message with same `SEQ_ID` → would it be possible to ignore repeated occurancy?
    - some separator? ([g069_433.92M_2500k.cs16](https://github.com/petrjac/rtl_433_tests/blob/1e49119a7997d3917b9e6597811a7bb3754540f6/tests/BMW_G5_TPMS/2/g069_433.92M_2500k.cs16), [g073_433.92M_2500k.cs16](https://github.com/petrjac/rtl_433_tests/blob/1e49119a7997d3917b9e6597811a7bb3754540f6/tests/BMW_G5_TPMS/2/g073_433.92M_2500k.cs16), [g081_433.92M_2500k.cs16](https://github.com/petrjac/rtl_433_tests/blob/1e49119a7997d3917b9e6597811a7bb3754540f6/tests/BMW_G5_TPMS/2/g081_433.92M_2500k.cs16))
    - 3x same data message with same `SEQ_ID + 0x01`
7. _Nominal pressure_ is not correct name for last data _byte_. I always recorded `0x68` but diagnostic report another values (2,48 bar, 2,45 bar, 2,53 bar).
8. I recorded some strange unknown signal, which looks similar to _BMW gen5_. Based on signal intensity it could be from same TPMS sensor.
    - `ffe40647c282de9a76009ed0a6` - [g080_433.92M_2500k.cs16](https://github.com/petrjac/rtl_433_tests/blob/1e49119a7997d3917b9e6597811a7bb3754540f6/tests/BMW_G5_TPMS/2/g080_433.92M_2500k.cs16)

zuckschwerdt, 2024-02-11:
I'm already looking into `FSK_MC_ZEROBIT`. It's really a problem with the bad input signal.

The minmax discriminator is default on 868 MHz where most TPMS are. It does not work too well for 433.92 MHz, usually OOK.

Try to convert some of your signals to OOK (with FSK proper name would be mark/space not on/off...) and look how noisy the resulting pulses are. `rtl_433 -Y minmax -w out.ook file.cs16`
We'd want to only see short and long pulses (~25 and ~50 µs). I guess the slicer is working surprisingly well given that input :)

petrjac, 2024-02-11:
OK, thank you for clarification:). This is probably due unappropriate Soapy params for _Airspy R2_ `rtl_433 -d driver=airspy -g "LNA=8,MIX=8,VGA=11" -s 2.5M`. I'm gonna experiment with different params or try sdr without _433M_ antenna.

ProfBoc75, 2024-02-11:
>2. Not all _BMW gen5_ signals are correctly recognized, e.g. [g082_433.92M_2500k.cs16](https://github.com/petrjac/rtl_433_tests/blob/1e49119a7997d3917b9e6597811a7bb3754540f6/tests/BMW_G5_TPMS/2/g082_433.92M_2500k.cs16)?

Checked with :

    rtl_433 g082_433.92M_2500k.cs16 -Y minmax -R 252:vv

After MC decoded , missing 1 bit to be a valid message, expected {88} bits, 11x8 , so the message is too short and excluded.

[image]

I tried to go further and the CRC failed.

    {87}fcdc1ebe90aec4ffaf9704 MC decoded
    {87}0323e1416f513b0050687d Inverted, last byte is wrong and CRC failed, expected 0xfa not 0x7d

>4. `DEC - 52` for temperature seems to be right:
>        
>        * `059 ~ 7°C`
>        * `060 ~ 8°C`

Good for brand 0x03, HUF, sounds like brand 0x80, Continental, the temp is DEC - 50,  we need feedback from Continental valve owners to confirm that.

> 5. Pressure reading however does not exactly match with diagnostics readings, e.g. `27 * 0,0245 = 0,66 != 0,65`. Some more examples in [...]

From your bitbench, the pressure is clearly  = DEC * 2.5  (kPA)  or DEC * 0.025 (Bar)  , as most of TPMS sensors.

We saw that from other TPMS decoders , value can be coded like that :  (DEC - x ) * 2.5 to get the figures.
From your sample 27  = 0.65 Bar, could be (27 - 1 ) * 0.025 = 26 * 0.025 = 0.65 Bar

> 6. I feel confident about `SEQ_ID`. I recorded many samples and the pattern is:
>
>        * 3x same data message with same `SEQ_ID` → would it be possible to ignore repeated occurancy?
>        * [...]

rtl_433 is receiving 3 signals, so each one are decoded.
Did you test the valves in static or rolling situations ? If you look at my [table ](https://github.com/merbanan/rtl_433/issues/2821#issuecomment-1936392670) from rolling cars from my street, it's clear that the behavior is not the same, the reason why I kept the 3 flags as is to let external decoding and deep dive. TPMS sensor can report some moving / centrifugal acceleration situation, the opposite of storage state.
 
>7. _Nominal pressure_ is not correct name for last data _byte_. I always recorded `0x68` but diagnostic report another values (2,48 bar, 2,45 bar, 2,53 bar).

From the RDC tool guide, the temp should be taken into account in the normal pressure, + 0.1 bar / + 10 °C  , so may be the temp is not same too ?
 
> 8. I recorded some strange unknown signal, which looks similar to _BMW gen5_. Based on signal intensity it could be from same TPMS sensor.
>
>        * `ffe40647c282de9a76009ed0a6` - [...]

I don't understand your point, it's properly decoded with `rtl_433 g080*.cs16 -Y minmax -R 252:vv`
[image]

luki343, 2024-02-12:
What bmw part number TPMS this refer?

petrjac, 2024-02-13:
For _Huf gen. 5_ the P/N is _36106877937_.

petrjac, 2024-03-30:
Hello again,

I recorded several samples with BMW **Gen4 sensors**. The message format looks similar to Gen5 sensors and also samples collected by @ProfBoc75.
I pushed the samples here - https://github.com/merbanan/rtl_433_tests/pull/466/files

Here is [bitbench](https://triq.net/bitbench#c=fff28030c448e756472601a1ec&c=fff28030c448e73e462601a374&c=fff28030c448e71e462601a1b7&c=&c=fff28030c448e71b462601a1f0%20%5B%3F17C%2C%200.7bar%5D&c=fff28030c448e71b462601a3ae%20%5B18C%2C%200.7bar%5D&c=fff28030c448e73e462601a374%20%5B18C%2C%201.5bar%5D&c=fff28030c448e756472601a3b2%20%5B19C%2C%202.1bar%5D&c=&c=fff28036bcdadc57452601a395&f=16h%20%7C%20MANUFACTURER%20%3D%208h%20%7C%20SENSOR_ID%20%3D%2032d%20%7C%20PRESS%20%3D%208d%20%7C%20TEMP%20%3D%208d%20%7C%20UKN%20%3D%208h%20%7C%20UKN%20%3D%208h%20%7C%20UKN%20%3D%208h%20%7C%20CRC%20%3D%208h%20&z=1&cw=4) with decoded data.

I was also able to compare data with official diagnostics.
[image]

Notes:
- I was not able to decode signals using `rtl_433 version 23.11-51-g2d63f306`, I was using _Universal Radio Hacker_ to decode. This could be due weak signal or other issue related to my setup. Can someone please validate?
- The manufacturer `0x80` is really _Continental_.
- The sensor ID is interpreted as **decimal** (unlike _gen5_ sensors).
- Pressure [bar] is probably `round((RAW - 1) * 0,025, 1)`. Displayed as rounded to 1 decimal places (_gen5_ showed 2 decimal places).
- Temperature is `RAW - 52`.
- There is no _sequence ID_.
- Last two bytes is also _CRC-8_ with `poly=0x2f  init=0xaa`.

ProfBoc75, 2024-03-30:
Hi @petrjac: Thanks for this feedback, can you try with this option `-R 252:vv` to see why it's not decoded.

I'm not able to decode your last 4 x CS16 files.

petrjac, 2024-03-30:
@ProfBoc75 I tried this... would this be useful?
```
> rtl_433 -r g097_433.92M_2500k.cs16 -R 252 -vvvv
rtl_433 version 23.11-51-g2d63f306 branch master at 202402101513 inputs file rtl_tcp RTL-SDR SoapySDR with TLS
[Protocols] Registered 1 out of 252 device decoding protocols
[Input] Test mode active. Reading samples from file: g097_433.92M_2500k.cs16
[Input] Input format "CS16 IQ (2ch int16)"
[Baseband] low pass filter for 2500000 Hz at cutoff 250000 Hz, 4.0 us
Pulse data: 1 pulses
[  0] Pulse: 13804, Gap: 138041, Period: 151845
[Input] Test mode file issued 3 packets
```

Demodulated signal looks like this in _URH_:
[image]

Raw hex is `aaaaaa5995555a55a5656595a96a6669656a59695556995a9a59f8` for *g097*, `fff28030c448e756472601a3b2` after *MC*.

ProfBoc75, 2024-03-30:
@petrjac:

I fact, I wanted you to run rtl_433 in live to see if you are able to get some answers, and I put verbose information in the decoder itself, like data before and after MC, so we can see where it failed and it could help me understand.

This option `-R 252:vv` will enable only the BMW gen5 decoder with verbose level 2. (correct your Airspy options if I'm wrong)

    rtl_433 -R 252:vv -d driver=airspy -g "LNA=8,MIX=8,VGA=11" -s 2.5M -Y minmax

you may test with auto gain with `-g ""`

From your 4 samples, rtl_433 is not able to decode, `preamble not found` and raw `codes` are not well decoded. May be a bad airspy options (signal is overloaded/clipping, to much gain):

    rtl_433 -R 252:vv *.cs16 -Y minmax
[image]

But from your last raw hex information, it looks like it should work, we can replay with rtl_433 like this:

    rtl_433 -y aaaaaa5995555a55a5656595a96a6669656a59695556995a9a59f8

[image]

petrjac, 2024-03-31:
@ProfBoc75 Thank you for your comment. You are right, the signal is obviously clipping:).

I've been adjusting the gain settings and also forgot to set `-Y minmax`, but I was finally able to decode using the *252* decoder, so this decoder also works for *gen4 BMW TPMS*.

Should I upload some samples or try something else?

ProfBoc75, 2024-03-31:
> @ProfBoc75 Thank you for your comment. You are right, the signal is obviously clipping:).
>
> I've been adjusting the gain settings and also forgot to set `-Y minmax`, but I was finally able to decode [...]

Yes, you can upload few other samples into rtl_433_tests with your other gen4 files. Update the README file with pressure and temp details for each file. I would like to check if we can identify the gen 4 from them compare to other gen 5  files.

petrjac, 2024-04-01:
@ProfBoc75 I [pushed](https://github.com/petrjac/rtl_433_tests/commit/35c80bc2b2a6c50e0db10143e8a607fd26a8185d) some new samples which can be read (the signal is still noisy and weak though). I also added _JSON_ with data from diagnostics.

The `FLAGS3` could be some _sefuence number_ for _gen4 continental. I've observed consecutive values of `0xa1`, `0xa2`, `0xa3`, followed by repetition.

ProfBoc75, 2024-04-02:
@Billymazze : For BMW you need to set modulation to FSK and pulse at 25 µs. For Abarth same FSK but pulse is 52 µs.

Billymazze, 2024-04-02:
Could clarify where I need set those values? Do you mean on the flipper zero or in URH recorder? Sorry I'm idiot.

ProfBoc75, 2024-04-02:
> Could clarify where I need set those values? Do you mean on the flipper zero or in URH recorder? Sorry I'm idiot.

I don't know for flipper zero but for URH you should have modulation option = FSK and the samples/symbol to be adjusted until you have some codes starting with aaaaaa59 or 5555555. I guess you are not so far since the length is little more than 208 bits (after aaaaaa59 preamble) like from your samples.

Billymazze, 2024-04-02:
Alright, I'll give that a shot. Any chance the issue is my recoding settings on URH? I recorded these signals with an RTL-SDR (Realtek RTL2838UHIDIR) @433.92mhz with a sample rate of 1.0M, bandwidth of 1.0M, gain of 49, bias tree disabled, and direct sampling disabled. Having issues installing rtl-433, but maybe it's necessary that I get it working.

Billymazze, 2024-04-03:
Also forgot to mention these signals I've captured are from a Gen 3 bmw tpms system, 2011 bmw x5 35d.

petrjac, 2024-04-03:
@Billymazze the protocol for *Gen 3* could be different. Are you able to get TPMS reference data from diagnostics (ISTA)?

Can you provide some samples recorded using *URH*?

Billymazze, 2024-04-03:
I don't currently have access to ISTA D, but I will soon. Currently there's a google drive link in my first post, [here](https://drive.google.com/drive/folders/1ZM9ZmDwIUaQ74j1viKW6knrmj_pdXmgD?usp=drive_link). I collected signal with URH and With the flipper zero. I'm not 100% sure my recoding settings are correct, especially gain level. But I can replay the signals and trigger the TPMS warning light with the flipper zero.

petrjac, 2024-04-03:
It looks that *gen3* protocol differ from *gen4/5*. I'm not able to find any sense in there ...

The *ISTA/D* would help, you could also try Android [*Ediabas*](https://github.com/uholeschak/ediabaslib) app.

Since the protocol for *gen3* differs you can consider open separate issue.

Billymazze, 2024-04-04:
Alright, I'll create a new issue.

Gucioo, 2024-04-08:
At first great job guys! 👍  I recently retrofitted car with Autel MX-2 TPMS sensors and started playing with Noelec RTL-SDR.
Listening for the packets using:
rtl_433 -s 1000000 -X "name=new,m=FSK_PCM,s=25,l=25,r=100" -S All           (-s 250000 not enough)
I got similar data to BMW Gen5 and I manage to rewrite decoder to suit Audi etron (Autel, but original is Huf).
[https://triq.net/bitbench#c=BMW%20Huf%20Gen5&c=555554b2aab4b2b552acb4d332accb32b552aaacd334d32ad334&c=555554b2aab4b2b552acb4d332acb4cab54caaacd4cad32b4b55&c=Audi%20Huf&c=rtl_433%20-s%201000000%20-X%20%22name%3Dnew%2Cm%3DFSK_PCM%2Cs%3D25%2Cl%3D25%2Cr%3D100%22%20-S%20All&c=%7B167%7D5555552caaaaacaad2aab532b3352aab32d34cacfe&c=%7B149%7D2aaaaa59a5565955a5556a65666a9655a969f8%20-----Invalid&c=%7B166%7D5555552caaaaacaad2aab532b3352b54b2d352ad7c&c=%7B166%7D2aaaaa5955555955a5556a65666a56aa65a65999fc&c=%7B165%7D2aaaaa5955555955a5556a65666a56aa65a65999f8&c=%7B167%7D5555552caaaaacaad2aab532b3352b5532d32cccfe&c=%7B166%7D2aaaaa5955555955a5556a65666a56aa65a65999fc&c=%7B151%7D5555552cd2ab2caad2aab532b3354b2ad4b4fe%20-----Invalid&c=%7B166%7D2aaaaa5955555955a5556a65666a555665a69959fc&c=%7B167%7D5555552caaaaacaad2aab532b3352aab32d34cacfe&f=BRAND%20%3D%208h%20%7C%20SENSOR_ID%20%3D%2032h%20%20%20%20%20%20%7C%20PRESS%20%3D%208d%20%20%7C%20TEMP%20%3D%208d%20%20%7C%20FLAGS1%20%3D%208h%20%7C%20FLAGS2%20%3D%208h%20%7C%20FLAGS3%20%3D%208d%20%20%7C%20CRC%20%3D%208h%0A%0ABRAND%20%3D%208h%20%7C%20SENSOR_ID%20%3D%2032h%20%20%20%20%20%20%7C%20PRESS%20%3D%208d%20%20%7C%20TEMP%20%3D%208d%20%20%7C%20CRC%20%3D%208h&z=1&a=Preamble&m=aa59&d=MC&cw=4](url)

Would you like to rewrite BMW_Gen5 to include Audi (with if, else, case) or it is better to create new device for this brand?
[https://github.com/Gucioo/rtl_433/blob/master/src/devices/tpms_audi.c](url)

[image]

ProfBoc75, 2024-04-09:
@Gucioo: Thanks for your finding !

Yes, I can update later this week the existing decoder to take into account the specific brand with shorter message without the 3 flags.

ProfBoc75, 2024-04-10:
Done, last version should decode Audi TPMS, don't hesitate to give feedback here.

Gucioo, 2024-04-10:
I wasn't quick enough to get back to you with the latest finding, but Audi seems to be a bit specific with TPMS implementation.

BMW and Audi responds different on different trigger.

[https://imgur.com/a/9JH5Iir](https://imgur.com/a/9JH5Iir
)
BMW:
sends 26 hexes and ID=03 regardless how it was triggered. [11 bytes after preamble]
Audi:
1 - If was triggered by Autel Read it sends the same 26 hexes with MM id=88 [11 bytes]
2 - when triggered by inflate/deflate, it sends only 20 hexes with MM id=00 [8 bytes]

```
21:28:04.271 -> Timings....
21:28:04.271 -> CD_Width=0
21:28:04.271 -> TimingsIndex=158
21:28:04.271 -> Checking complete. Bitcount: 202  StartDataIndex: 26
21:28:04.366 ->  [88]  [20]  [C0]  [74]  [57]  [1]  [4C]  [E3]  [9]  [9A]  [29] 
21:28:04.557 ->  [88]  [20]  [C0]  [74]  [57]  [1]  [4C]  [E3]  [9]  [9A]  [29] 
21:28:04.605 -> CRC Check OK
21:28:04.605 -> Pos:    Manuf ID: Undefined(0x88)   ID: 20C07457   Flags1: 0xE3   Flags2: 0x9   Flags3: 0x9A   Temperature: 24.00   Tyre Pressure: 0.36 (psi)  2.45 (kpa)
21:28:04.605 ->   Low Pressure warning.  Limit: 30.00 [30.41]  Measured: 0.36
```
During normal operation (in motion) Audi sends 11 bytes with id 88, when it gets sudden pressure loss it sends 11 bytes with id 88. Some exception would be needed to cover this changing length.

When there is sudden pressure loss, it burst frames each second, maybe thats why they decided to reduce message length, to save on energy, but who knows?!
[https://imgur.com/a/wKUPGlo](https://imgur.com/a/wKUPGlo)

I've also noticed the same sensor is used with A3, may be worth adding to header.
```
// Audi E-tron 01/2018-12/2023(433Mhz)
// Audi A3 06/2012-12/2023(433Mhz)8L
// OE No 2N0907251 - Huf/Beru RDE018
```

I would like to thank [JSMSolns](https://www.hackster.io/jsmsolns) for his help to implement this protocol that he made under his Arduino TPMS Tyre Pressure Display.

Gucioo, 2024-04-10:
> Done, last version should decode Audi TPMS, don't hesitate to give feedback here.

It works fine now. **-s 1000000** is required to decode properly, on standard sampling rate is not working.

The only thing, it would require ID update according to my previous comment:
0x88 brand: 136  - Audi in motion [11 bytes]
0x00 brand: 0      - Audi/possible others? in sudden deflate/burst mode [8 bytes]

```
rtl_433 version 23.11-104-g2c1967f4 branch master at 202404102250 inputs file rtl_tcp RTL-SDR
MQTT: Publishing MQTT data to 192.168.999.999 port 1883
MQTT: Publishing events info to MQTT topic "rtl_433[/model][/id]".
Use "-F log" if you want any messages, warnings, and errors in the console.
Detached kernel driver
Found Rafael Micro R820T tuner
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
{"time" : "2024-04-10 21:26:49", "model" : "BMW-GEN5", "type" : "TPMS", "brand" : 136, "id" : "20c07457", "pressure_kPa" : 2.450, "temperature_C" : 25.000, "flags1" : 228, "flags2" : 9, "flags3" : 154, "msg" : "8820c07457014de4099ab6", "mic" : "CRC"}
{"time" : "2024-04-10 21:27:05", "model" : "BMW-GEN5", "type" : "TPMS", "brand" : 136, "id" : "20c07457", "pressure_kPa" : 2.450, "temperature_C" : 25.000, "flags1" : 228, "flags2" : 9, "flags3" : 154, "msg" : "8820c07457014de4099ab6", "mic" : "CRC"}
{"time" : "2024-04-10 21:27:54", "model" : "Audi", "type" : "TPMS", "brand" : 0, "id" : "20c07457", "pressure_kPa" : 26.950, "temperature_C" : 25.000, "msg" : "0020c074570b4dda", "mic" : "CRC"}
{"time" : "2024-04-10 21:27:55", "model" : "Audi", "type" : "TPMS", "brand" : 0, "id" : "20c07457", "pressure_kPa" : 29.400, "temperature_C" : 25.000, "msg" : "0020c074570c4d1b", "mic" : "CRC"}
{"time" : "2024-04-10 21:27:56", "model" : "Audi", "type" : "TPMS", "brand" : 0, "id" : "20c07457", "pressure_kPa" : 26.950, "temperature_C" : 25.000, "msg" : "0020c074570b4dda", "mic" : "CRC"}
{"time" : "2024-04-10 21:27:57", "model" : "Audi", "type" : "TPMS", "brand" : 0, "id" : "20c07457", "pressure_kPa" : 26.950, "temperature_C" : 25.000, "msg" : "0020c074570b4dda", "mic" : "CRC"}
{"time" : "2024-04-10 21:27:58", "model" : "Audi", "type" : "TPMS", "brand" : 0, "id" : "20c07457", "pressure_kPa" : 26.950, "temperature_C" : 25.000, "msg" : "0020c074570b4dda", "mic" : "CRC"}
{"time" : "2024-04-10 21:27:59", "model" : "Audi", "type" : "TPMS", "brand" : 0, "id" : "20c07457", "pressure_kPa" : 19.600, "temperature_C" : 25.000, "msg" : "0020c07457084dce", "mic" : "CRC"}
{"time" : "2024-04-10 21:28:00", "model" : "Audi", "type" : "TPMS", "brand" : 0, "id" : "20c07457", "pressure_kPa" : 4.900, "temperature_C" : 25.000, "msg" : "0020c07457024db6", "mic" : "CRC"}
{"time" : "2024-04-10 21:28:01", "model" : "Audi", "type" : "TPMS", "brand" : 0, "id" : "20c07457", "pressure_kPa" : 2.450, "temperature_C" : 25.000, "msg" : "0020c07457014da2", "mic" : "CRC"}
{"time" : "2024-04-10 21:28:02", "model" : "Audi", "type" : "TPMS", "brand" : 0, "id" : "20c07457", "pressure_kPa" : 2.450, "temperature_C" : 25.000, "msg" : "0020c07457014da2", "mic" : "CRC"}
{"time" : "2024-04-10 21:28:03", "model" : "Audi", "type" : "TPMS", "brand" : 0, "id" : "20c07457", "pressure_kPa" : 2.450, "temperature_C" : 25.000, "msg" : "0020c07457014da2", "mic" : "CRC"}
{"time" : "2024-04-10 21:28:06", "model" : "Audi", "type" : "TPMS", "brand" : 0, "id" : "20c07457", "pressure_kPa" : 2.450, "temperature_C" : 25.000, "msg" : "0020c07457014da2", "mic" : "CRC"}
```

ProfBoc75, 2024-04-10:
@Gucioo : Very interesting ! My update is still valid but yes for this specific situation of fast deflating, and probably related to the behavior of this kind of sensor than to Audi.

In my [street scanning](https://github.com/merbanan/rtl_433/issues/2821#issuecomment-1936392670), I found one message with 0x88, unknown until now, so I will update the decoder with this brand link to Audi.

I will check also JSMSolns projects to see if we have some other findings there and may be some answers related to the flags we didn't guess for BMW.

Gucioo, 2024-04-14:
> @Gucioo:
> 
> To summarize:
> 
> * Short Message = Audi Pressure Alert, sudden pressure increase or decrease, Brand ID = 0x00, send every seconde.
> * Long Message , no Change, just Brand ID = 0x88 = Audi
> 
> I'm preparing an update to the decoder and will commit soon.

Yes, that's correct.

luki343, 2024-05-06:
Does this sensor need to receive any data to send? 
I'm thinking about making simple arduino board to receive data from tpms and send it via CAN to my standalone ECU to have possibility to log this data.

luki343, 2024-05-12:
@ProfBoc75 
I know it depends on the sensor, which is why I'm asking about this particular TPMS. I thought maybe someone who sniffed this sensor found something sent from the car to the sensor. 

I need to adapt some TPMS to my "race car" and send it to ECU via CANbus, so the easiest way will be to use one of the currently decoded sensors, but if it needs to receive some message from car, I'm lost. 
On second car (daily, renault), I have factory TPMS (0435r) and from what I sniff myself it just sends data and nothing receive, it don't even need to be activated after mount in wheel, it activate automatically after some distance and you only need to code ID to car and thats all.  
I would prefer to use this BMW sensor (continental version more precisely) instead of Renault.

I will definitely check out this project, thx

gdt, 2024-06-03:
What's the status and plan?  Is there a decoder in tree, or does something need to happen?  If so, is anyone working on it?

ProfBoc75, 2024-06-04:
> What's the status and plan? Is there a decoder in tree, or does something need to happen? If so, is anyone working on it?

Hi @gdt:

- The decoder is working fine, no known issue.
- Remaining the decoding part of 3 unknown flags. Flags behave differently from one brand to another or from id range to another range.
- Based on our findings and expected information, they could contain battery level, target/nominal pressure, tire details (size, pressure...) , synectic/motion status, abnormal pressure increase/decrease....
- At my level I can help only if someone shares more samples since I don't have these TPMS sensors.
- However, rtl_433 already provides the 3 flags as 3 raw bytes which can be decoded externally.

Said that, this issue can be closed.

gdt, 2024-06-04:
Great, thanks.

(FWIW, I am a fan of putting unknown bits in the output labeled as unknown bits/bytes, as it seems the path to improvement.)

closed 2024-06-04 by gdt

run-mike, 2024-08-02:
Hi!
First of all: Awesome work!
Decoder works fine on my Gen5 BMW TPMS sensors.

@luki343 : 
Have you been able to determine how these sensors can be forced to communicate without request / demand from the car?
Tried it by running a few laps in the garage with a spare tire to assume movement / spinning, but that didn't yield a result.
Tried to "activate" with a 433.92 MHz remote before spinning, not good enough. Guess there has to be knowledge of the communication protocol?

Thanks in advance

Mike

run-mike, 2024-08-02:
Thanks for your fast response!
I was actually looking for a solution to get read-outs in non-emergency mode.
I'll give ISTA a try, but don't expect much. Will have to build some sniffing device to see the request signal.
In the description there is talk of a 150kHz signal for the US / Canada model.
Not sure if that applies to the European version either.

Mike

run-mike, 2024-08-04:
Being a noob to all this, I would like to discuss the BMW Gen5 TPMS Sensors a little further. 
I guess I could even contribute to decode them further. Is that ok to do here or should I open a new thread?

Mike

zuckschwerdt, 2024-08-04:
Unless there is an actionable todo it is preferred to continue here, where all involved people already listen :)

run-mike, 2024-08-04:
Ok. thanks for your response!
I own two sets of original BMW alloy wheels. One set (delivered with the car) has an estimated sensor programming date of 02/2018. The winter set was produced in 08/2015; so probably programmed in 11/2015.
The read-out "Flags1" on the first set is 42 / 43. Read-out on the winter set is 15.
I contacted a specialist (works in a BMW service/repair shop) and asked him, what kind of parameter he is able to determine with the "RDC-Tool" on the F45/F46 models.
Answer: 
- Battery level
- Sensor generation
- Sensor manufacturer
- Temperatur
- Sensor-ID
- Actual pressure

When ISTA is connected to my car, none of the ABL (Ablauf = sequence of events) concerning TPMS are able to be carried out. Probably due to "No error / fault code". So no activation of transmitter ...
But then again I get a different, more specific FUB (Funktionsbeschreibung = function description).
And one of those bullet points is: "Restliche Lebensdauer der Batterie" (remaining battery lifetime).
So I am quite sure, that the data "Flags1" is the remaining battery time in months. At least it fits my two wheel sets.
My other findings have to be chross-checked and confirmed yet.

Mike

ProfBoc75, 2024-08-04:
@run-mike : 

what is the brand reported for your sensors ?

As I'm also pretty sure that the behavior/ purpose of flags is different between brands. I listened to the sensors passing in my street and tracked them in an excel table. For each transmission there are 3 identical messages except for 1 flag which increments and depending on the brand it is not the same flag.

run-mike, 2024-08-05:
Today I did that test-drive for confirmation of suspected speed numbers.
Sadly I have to report, that the data in "Flags2" does not represent the speed in kph. It is definitely speed dependent, but reaches 255 (decimal) at a speed of 100 kph. It is not an acceleration value either. I tried to keep the different speed-levels for as long as traffic would allow it.
Another thing that I can confirm now, is that after stopping (and ignition "off") the sensors pop up again for a last round of messages after about 4.5 minutes.

I attached the logfile so that everyone can conclude for themselves ...

Mike

[218d_TPMS_testing_4.txt](https://github.com/user-attachments/files/16499705/218d_TPMS_testing_4.txt)

ProfBoc75, 2024-08-05:
Very interesting.

Notice also the range for the flags3.
With low speed the range is 5/6/7 , then 9/10/11, then 13/14/15.
And stopped position (flags2=1), the flags3 values are 129/130/131.
I noticed that also from the street, so into the flags3 and brand 0x80 we have sub flags linked to the speed/status of the sensor.
Only the last 2 bits are seq numbers, the 6 other bits are linked to speed or static position.

ProfBoc75, 2024-08-06:
So we can update the comment part with the data layout of the 3 flags.

Continental model: (brand id = 0x80 = 128)

    F1: Battery remaining months
    F2: 0x01 = static/parked, all values > 0x01 = moving, 0xFF when speed > = 100 kmh
    F3: divided into 2 parts, the last 2 bit x x, for sequence number, 1, 2 and 3, the other bit depend on the speed.
        bit position                          1 2 3 4  5 6 7 8
        F3 = 0x05/06/07 low speed (starting), 0 0 0 0  0 1 x x
        F3 = 0x09/0a/0b around 40 kmh,        0 0 0 0  1 0 x x
        F3 = 0x0d/0e/0f > 40 kmh,             0 0 0 0  1 1 x x
        F3 = 0x81/82/83 , stopped,            1 0 0 0  0 0 x x
        F3 = 0xA1/A2/A3 , stopped,            1 0 1 0  0 0 x x
        F3 = 0xC1/C2/C3 , driving + Flag      1 1 0 0  0 0 x x may be an alert like fast deflate / inflate ?
                                                SEQ Number x x = 01 / 10 / 11

ProfBoc75, 2024-08-06:
>Btw: Are we sure, that the temperature readings are correct?

I saw that also from my street, where temperature is much less than the air but this could be explained if the car just exit from a garage with a colder temp, depends also if the wheel rim is aluminum or steel, this could help also to dissipate heat ? I can't confirm the temp value as I don't have the sensor at all. Only the tpms display or the diagnostic tool can confirm the temp.

run-mike, 2024-08-07:
> Only the tpms display or the diagnostic tool can confirm the temp

Unfortunately my car doesn't show the temperatures on the TPMS display; only tire pressures.
But I have posted that question on the F45/F46-forum. Still waiting for a response ...
Meanwhile I exposed the "winter-wheel" to direct sunlight. I will let it bake there for  a few hours and then measure temperature of tire, rim, valve and flowing-out air while doing a "fast deflate".
Hopefully that will give me more insight (pun intended ...).

Mike

[6 more replies]