merbanan/rtl_433 #2823: `rtl_443 --help` output barely readable on bright colour scheme terminals.

dreirund, 2024-02-07:
When I run `rtl_443 --help`, I get the following output (screenshot):

---

[image]

---

The bright colour on the white background is barely readable.

Please use either monochrome by default, or some colouring that works both on dark and bright terminals.

Regards!

zuckschwerdt, 2024-02-07:
This is a know shortcoming. We don't know how to determine if you use a bright or dark theme.

A workaround for now is e.g. `rtl_433 -h | cat`

Tested ideas would be appreciated, but for now this is not a todo for us.

closed 2024-02-07 by zuckschwerdt

dreirund, 2024-02-07:
On Wed, 07 Feb 2024 07:55:40 -0800, "Christian W. Zuckschwerdt"
***@***.***> wrote:

> A workaround for now is e.g. `rtl_433 -h | cat`

Interestingly, it works, but only for recognised argument (`-h`).

For output to standard error (e.g. `rtl_433 --help | cat` -- `--help`
is not recognised and so nothing is piped through `cat`) it does not
work.

> Tested ideas would be appreciated,

* Use no colouring by default (use a command line switch for colouring),
* use dark green instead of bright yellow.

Regards!

zuckschwerdt, 2024-02-07:
We could maybe use an env var like `RTL433_COLOR=[auto|never|always]`

zuckschwerdt, 2024-02-10:
I'm looking into this, and I remember explicitly using the non-bright colors with bold to suit light themes.
Does your terminal perhaps not support bold and converts dull bold to bright colors? I get e.g. these colors:
[image]

For comparison this is without using bold:
[image]

You can test this by removing `;1` here https://github.com/merbanan/rtl_433/blob/2d63f30681dd527955542edea4b7a570bacf8a6f/src/term_ctl.c#L291

zuckschwerdt, 2024-03-26:
Thanks for the feedback. Lets merge #2864 -- your suggested change then is automatic for light backgrounds.

dreirund, 2024-03-26:
> *your suggested change then is automatic for light backgrounds.*

Only if something sets `COLORFGBG`.