merbanan/rtl_433 #2824: Terminal output ignores terminal capabilities (termcap).

dreirund, 2024-02-07:
When I run `TERM=dump rtl_443 --help`, there should be no colouring at all, since dump terminals do not support this.

But there still is coloured output, see screenshot below.

Please consider making coloured output according to the terminal capabilities, preferably using the termcap library.

Regards!

---

[image]

---

zuckschwerdt, 2024-02-07:
We won't link to external libs like e.g. termcap. If there is another way to cheaply accomplish this please advice. For now this is not a todo for us.

closed 2024-02-07 by zuckschwerdt

dreirund, 2024-02-07:
> We won't link to external libs like e.g. termcap. If there is another
> way to cheaply accomplish this please advice. For now this is not a
> todo for us.

OK. But then this would break in some cases, e.g. also some serial
terminals.

Maybe you then add a command line option like `-nc | --no-colour`?

Regards!

zuckschwerdt, 2024-02-10:
This should be addressed by 0443488 now. `RTL433_COLOR=always/never/auto`
Can you check if it works for you?