merbanan/rtl_433 #2826: Build of LaTeX documentation fails with `! Dimension too large.` in `bitbuffer_8h.tex`.

dreirund, 2024-02-07:
After workarounding #2825, build of LaTeX documentation fails:

```
[131]) (./baseband_8h_source.tex [132] [133]) (./bitbuffer_8h.tex
<bitbuffer_8h__incl.eps>
! Dimension too large.
<to be read again> 
                   \relax 
l.22 ...hics[width=350pt]{bitbuffer_8h__dep__incl}
```

Build procedure:

```
git clone https://github.com/merbanan/rtl_433.git
cd rtl_433
```

Patch `Doxyfile.in` with  
```patch
diff -rU1 rtl_433.orig/Doxyfile.in rtl_433/Doxyfile.in
--- rtl_433.orig/Doxyfile.in	2024-02-07 15:59:47.959870493 +0100
+++ rtl_433/Doxyfile.in	2024-02-07 16:00:06.606536622 +0100
@@ -793,3 +793,3 @@
 
-WARN_AS_ERROR          = YES
+WARN_AS_ERROR          = NO

```

```
cmake -DBUILD_DOCUMENTATION=ON .
make
cd doc
make
```

More output of the last `make` command:

```
[...]
Package longtable Warning: Column widths have changed
(longtable)                in table 6.3 on input line 319.

Underfull \hbox (badness 10000) detected at line 320
[][][]
[131]) (./baseband_8h_source.tex [132] [133]) (./bitbuffer_8h.tex
<bitbuffer_8h__incl.eps>
! Dimension too large.
<to be read again> 
                   \relax 
l.22 ...hics[width=350pt]{bitbuffer_8h__dep__incl}
                                                  
?  
```

zuckschwerdt, 2024-02-07:
I have now cleaned out some of the more unexpected characters. The only remaining chars that upset LaTeX are now `≤`, `₀`, `π` in acurite.c and tpms_renault_0435r.c. They look nice, but I guess need replacing to.

dreirund, 2024-02-07:
> I have now cleaned out some of the more unexpected characters. The
> only remaining chars that upset LaTeX are now `≤`, `₀`, `π` in
> acurite.c and tpms_renault_0435r.c. They look nice, but I guess need
> replacing to.

I don't know by myself, but there are LaTeX variants (LuaLaTeX? XeTeX?)
that can natively handle UTF-8 input.

Maybe switching to that?

Anyway, using ASCII-only in code (and comments) is always a good idea
for portability reasons :-).

zuckschwerdt, 2024-06-03:
Fixed.

closed 2024-06-03 by zuckschwerdt

dreirund, 2024-06-03:
I still get  
```
[140]) (./baseband_8h_source.tex [141]) (./bit__util_8h.tex
<bit__util_8h__incl.eps> [142]
! Dimension too large.
<to be read again> 
                   \relax 
l.22 ...hics[width=350pt]{bit__util_8h__dep__incl}
```

zuckschwerdt, 2024-06-03:
Strange. The only errors I get are the unicode characters in acurite.c and tpms_renault_0435r.c
Maybe we should just disable LaTeX generation, nobody ever cared about that AFAICS?