merbanan/rtl_433 #2828: IDEA: rtl_433 device database

billchurch, 2024-02-07:
Sometimes finding what sensor uses what protocol can be tricky as there is a bunch of white label manufacturers and it depends on who was first to documenting the protocol as to what the "device" is called. 

I was thinking it might be good to maintain a database of FCC IDs (or something else regulatory) to the protocols they go with. I'm willing to take this on but wanted to see if something like this already exists before I go through the trouble... That or, if I'm just approaching this the wrong way.

zuckschwerdt, 2024-02-07:
We have a script to extract meta-data and doc-comments from the decoders. This is e.g. displayed at https://triq.org/explorer/

Adding more device info in some fixed format to the decoders would be great!

psa-jforestier, 2024-04-24:
look a little like https://github.com/merbanan/rtl_433/issues/2611

zuckschwerdt, 2024-04-24:
I'm again working on this. With users comments, a rating system and such and I hope to offer something in a few months.

gdt, 2024-06-03:
Calling this a dup of #2611.

closed 2024-06-03 by gdt