merbanan/rtl_433 #2838: Problem using mqtts in 21.12 on ubuntu LTS 22.04 solved by building and running 23.11

jatmega, 2024-02-14:
OS: ubuntu 22.04 running as LCX under PROXMOX 8.1
rtl_version: 21.12 -not working and 23.11 - working

Just want to share how I got mqtt logon with certificates working since it is not to well documented and we have to help each other if we want the world to become better :)

When trying to use certificates (TLS) for authenticating to a mqtt-broker, I just got the "mqtts (TLS) not available" message.
Ubuntu 22.04 installed rtl_433 version 21.12 with apt install. Following the steps to build and install the latest version (at the time 23.11) making sure to include the library for tls-connection, and it just worked :) The exact same line of "rtl_433"-code just worked with version 23.11 :)
Link to how to build it page : https://github.com/merbanan/rtl_433/blob/master/docs/BUILDING.md

The format of my rtl_433 call, in case it is usefull to anyone:

rtl_433 -C si -F "mqtts://mqtt.mydomain.no:1883,tls_cert=/home/user/mqtt_certs/rtl_433.crt ,tls_key=/home/user/mqtt_certs/rtl_433.key,tls_ca_cert=/home/user/mqtt_certs/ca.crt,retain=1,devices=testing/devices[/type][/model][/subtype][/channel][/id]"

The things for TLS is the "mqtts" instead of "mqtt" in the start, and tls_cert, tls_key, and tls_ca_cert parameter. You will of course need to change the paths for these parameters to point to where you have stored the certificates and the key. This mqtt-server does not require username and password when using certificates, if your setup require that, you might need to supply additional parameter. 
See https://github.com/merbanan/rtl_433/issues/1633

There are a lot of good guides on how to set up certificate autentication for MQTT, and with a guide like that and the knowledge herer and there : https://github.com/merbanan/rtl_433/issues/1633, you might have a chance :)

It would be nice with some examples of mqtts configurations for documentation. I do not feel proficient enough to provide them possibly missguiding others (there is so much crappy guides that do not work on the internet), at least not without someone doing quality-assurance.

Thankyou to all of you who make this and other open projects possible; you make world move foreward!

gdt, 2025-07-04:
@zuckschwerdt Were you still intending to apply a doc patch?

zuckschwerdt, 2025-07-04:
Ah yes, I'll do that shortly.

zuckschwerdt, 2025-07-04:
With c385970 there is now a short note on MQTT TLS in the help texts.