merbanan/rtl_433 #2840: paho-mqtt 2.x is out, breaking compatibility with rtl_433_mqtt_hass.py

deviantintegral, 2024-02-17:
The current directions don't specify a version of paho-mqtt to install, which means users will now be getting the 2.x branch. I know for sure this breaks rtl_433_mqtt_hass.py, and it wouldn't surprise me if it [examples/rtl_433_mqtt_relay.py](https://github.com/merbanan/rtl_433/blob/5786c01c8ca89d689dbaf6ffba279dfd4812d19b/examples/rtl_433_mqtt_relay.py#L11) too.

```
Starting rtl_433_mqtt_hass.py...
[2024-02-17T10:53:24+0100] INFO:root:Enabling debug logging
[2024-02-17T10:53:24+0100] INFO:root:Discovering all devices
Exception ignored in: <function Client.__del__ at 0x7fb05d0900>
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/paho/mqtt/client.py", line 874, in __del__
    self._reset_sockets()
  File "/usr/local/lib/python3.12/site-packages/paho/mqtt/client.py", line 1133, in _reset_sockets
    self._sock_close()
  File "/usr/local/lib/python3.12/site-packages/paho/mqtt/client.py", line 1119, in _sock_close
    if not self._sock:
           ^^^^^^^^^^
AttributeError: 'Client' object has no attribute '_sock'
Traceback (most recent call last):
  File "/rtl_433_mqtt_hass.py", line 990, in <module>
    run()
  File "/rtl_433_mqtt_hass.py", line 917, in run
    rtl_433_bridge()
  File "/rtl_433_mqtt_hass.py", line 888, in rtl_433_bridge
    mqttc = mqtt.Client()
            ^^^^^^^^^^^^^
TypeError: Client.__init__() missing 1 required positional argument: 'callback_api_version'
```

https://eclipse.dev/paho/files/paho.mqtt.python/html/migrations.html has upgrading directions.

We fixed this at https://github.com/pbkhrv/rtl_433-hass-addons/pull/180 by pinning the old version for now.

zuckschwerdt, 2024-02-17:
Thanks! We should check if we can use 2.x by just using `mqttc = mqtt.Client(mqtt.CallbackAPIVersion.VERSION1)`

zuckschwerdt, 2024-05-02:
Closed by #2916

closed 2024-05-02 by zuckschwerdt