merbanan/rtl_433 #2843: Sensor 30.3161 is not decoded correctly

Heimetli, 2024-02-20:
The sensor 30.3161 is part of the TFA Monsun package (rain gauge)

A comment in inovalley-kw9015b.c claims that this file can decode the sensor, but rtl_433 never recognized the sensor. It was wrongly identified as "TFA-TwinPlus". The temperature was correct, but the humidity was way out of range.

The data layout described in the comment seems correct (I did not verify the checksum)

short_width and long_width given in the file are near the values seen with the -A switch:

https://triq.org/pdv/#AAB02F040701B807D40FE823EC8182818281818181818181818182818281818282818181818282818181818181828281818355+AAB02C040101B807D40FE823EC8182818281818181818181818182818281818282818181818282818181818181828355

I don't need a fix, but it might help other users.

ProfBoc75, 2024-02-20:
Hi @Heimetli : It's normal behavior.

Your device, Temp + Rain gauge, `[37]* Inovalley kw9015b, TFA Dostmann 30.3161 (Rain and temperature sensor)` is disabled by default, notice the star close to the device number. 

    * Disabled by default, use -R n or a conf file to enable

And in the same time the other device `[31]  TFA-Twin-Plus-30.3049, Conrad KW9010, Ea2 BL999` is active by default ,Temp + Humidity.

So your rain gauge value is decoded as humidity by protocol 31 and it's wrong value for sure.
To avoid that, you need enable protocol 37 like this :

```
rtl_433 -R 37
```

This will enable only it, all the others protocols are ignored.
If you need more protocols, just add them like this :

```
rtl_433 -R 37 -R xx -R yy  -R zz    
```
Replace the xx yy and zz by your other device numbers, you can add as much as devices you need.
You can also enable using a conf file.

Heimetli, 2024-02-21:
You are right, with -R 37 it works.

Thanks

closed 2024-02-21 by ProfBoc75