merbanan/rtl_433 #2844: Hangs on "Allocating 15 zero-copy buffers"

krikey, 2024-02-21:
I've install rtl_433 on my main PC running POP OS and it seems to work well when running rtl_433 and I get an reading from my energy monitor.  Here's the output:

[image]

When I plug the same USB device into the Raspberry Pi and run rtl_433, it seems to hang on "Allocating 15 zero-copy buffers" and nothing else is shown.

Here' the output from the Rpi:

[image]

ProfBoc75, 2024-02-22:
Hi @krikey : 

Could be an issue with your old version of rtl_433 installed on your Rpi, may be old libraries (like librtl-sdr) are not aligned nor up to date.

Try to update first your Rpi :

```
sudo apt-get update
sudo apt-get upgrade
```

Then, try to update rtl_433 to the last versions, follow the [Build guide](https://github.com/merbanan/rtl_433/blob/master/docs/BUILDING.md) , look at my [post here](https://github.com/merbanan/rtl_433/issues/2819#issuecomment-1930967730) , I explained the same.

krikey, 2024-02-22:
Hi @ProfBoc75 ,

I ensured that my instance was updated as mentioned above and then build a new version of rtl_433 from the GIT repo. I'm now getting a version number when running plus some nice colour but still theres no sign of any signals at the default 433.92Mhz on my RPi, but there is on my other linux laptop.

Heres the command I'm running: 

rtl_433 -vvv

and heres the new output:

[image]

[image]

ProfBoc75, 2024-02-22:
Hi @krikey : 

Play with `rtl_sdr -f 433920000 -` to see if you have some raw signal or the same error.

`Allocating 15 zero-copy buffers` is related to librtlsdr and not rtl_433, so the efforts should be at this level, try to uninstall / reinstall this part (librtlsdr-dev rtl-sdr ) , then recompile rtl_433.

Try also to exclude / detach by default your rtl dongle, this is not the case since you have the `Detached kernel driver` message.

Extract from the [rtl_433 home page at the bottom](https://github.com/merbanan/rtl_433?tab=readme-ov-file#troubleshooting) :

## Troubleshooting

If you see this error:

    Kernel driver is active, or device is claimed by second instance of librtlsdr.
    In the first case, please either detach or blacklist the kernel module
    (dvb_usb_rtl28xxu), or enable automatic detaching at compile time.

then

    sudo rmmod rtl2832_sdr dvb_usb_rtl28xxu rtl2832

or add

    blacklist dvb_usb_rtl28xxu

to /etc/modprobe.d/blacklist.conf

zuckschwerdt, 2024-02-22:
Two additional things come to mind:
- Ensure a resonable distance between the Pi and the SDR, always use an extension cable.
- Fitipower receivers might have trouble with gain. Workaround is to add `-Y autolevel -M level -M noise`