merbanan/rtl_433 #2845: rtl_433 sometimes works with Adalm-Pluto, but most of the time doesn't

imca-potato, 2024-02-22:
Greetings!

I am currently working on using the rtl_433 on an Adalm-pluto. Whenever I try to do so, there are times where the software pushes through and manages to decode signals. However, most of the time, it fails to connect with the pluto and even disconnects it from the computer entirely. It displays the following error.

[image]

I'd like to ask for help with regards to this. For context, I'm using a raspberry pi 4 with Ubuntu Mate 22.04 OS. I have installed SoapySDR, but I'd like to note that I wasn't able to install pothosware since its repositories are not compatible with the Ubuntu version that I have. Thank you in advance for your help!

zuckschwerdt, 2024-02-22:
Pothos is not needed, just SoapySDR and SoapyPlutoSDR. The error looks like connection problems. I'll have to run some tests. Also rtl_433 should be able to run natively on the Pluto. https://github.com/triq-org/PlutoSDR-Apps/ -- but I have not checked that in a long time ;)

imca-potato, 2024-02-22:
Ok. I'll try the one in the link. But I'd like to ask if the rtl_433 software and the other packages, like the SoapySDR and SoapyPlutoSDR, need to be uninstalled first before extracting/unpacking the release? Or I can just let them be and install the release anyway? If it's the former case, how would I go about uninstalling them if I built them from source? I apologize if this is a basic question, I am relatively new to Linux systems in general.

ProfBoc75, 2024-02-22:
Hi , I guess that the sample rate and the bandwidth are really too high for the Pi4 and rtl_433.
My advice is to stay around 1MHz for the sample rate with option `-s 1000K` same for bandwidth  `-t "bandwidth=1M"`

zuckschwerdt, 2024-02-22:
> if the rtl_433 software and the other packages, like the SoapySDR and SoapyPlutoSDR, need to be uninstalled first before extracting/unpacking the release?

The files from that Pluto Apps package don't need to be installed, you could run them from a connect usb stick To avoid confusion: the apps need to be run on the Pluto, not e.g. a Raspberry Pi.

imca-potato, 2024-02-23:
> Hi , I guess that the sample rate and the bandwidth are really too high for the Pi4 and rtl_433. My advice is to stay around 1MHz for the sample rate with option `-s 1000K` same for bandwidth `-t "bandwidth=1M"`

Thank you! I will heed this advice as well.

zuckschwerdt, 2024-02-23:
I've checked both older firmware 0.35 and current 0.38 and both work fine using a USB connection to the Pluto.
I'm using `rtl_433 -v -Y autolevel -M level -M noise -d ""` but `rtl_433 -d ""` should work the same (but with less debug info).
I'd guess it really is connection or power problems.

ProfBoc75, 2024-02-23:
I've also encountered the same, poor USB ground isolation and interference problems with Pi4, known issue, all are linked to hmdi, wifi and bluetooth from PI4, all around the 2.4 Ghz. If not needed, the best is to disable them and prefer hmdi2 from hmdi1 output too, or just unplug hdmi if not used.

Good power supply + good cables (power and usb cables) , put the pluto far from the pi4, all should be better.

zuckschwerdt, 2024-02-23:
There is a new release supporting the current 0.38 firmware now: https://github.com/triq-org/PlutoSDR-Apps/releases/tag/2024-02-23
Frustrating complicated to get running as the ARM toolchain with already very old GCC 8.2.0 was downgraded to the ancient and deprecated Linaro with GCC 7.3 in some recent firmware.
See [Using](https://github.com/triq-org/PlutoSDR-Apps?tab=readme-ov-file#using) in the readme to get going.

imca-potato, 2024-02-25:
> I've checked both older firmware 0.35 and current 0.38 and both work fine using a USB connection to the Pluto. I'm using `rtl_433 -v -Y autolevel -M level -M noise -d ""` but `rtl_433 -d ""` should work the same (but with less debug info). I'd guess it really is connection or power problems.

I see. Maybe I should try upgrading the firmware, since the one I am using is still on 0.30.

imca-potato, 2024-03-13:
Greetings!

As an update to this thread, I would like to highlight the things I've done/not done so far as well as some things I've noticed:

> My advice is to stay around 1MHz for the sample rate with option `-s 1000K` same for bandwidth `-t "bandwidth=1M"`

Setting the sample rate to 1MHz worked for me, but for some reason the bandwidth setting incurs errors, so I simply omitted this

> Good power supply + good cables (power and usb cables) , put the pluto far from the pi4, all should be better.

Not entirely sure what is meant by this, but I assumed it is having connections on both the power and the USB port of the Pluto. I attempted this but unfortunately it did not work. The Pluto just disconnects from the raspberry pi when an additional power supply is connected. For cables, I use the one that comes with the Pluto.

Stuff I have not tried:

- upgrading the firmware
- using plutoSDR apps

I've decided to not do these yet since the software still works despite the pluto having an outdated firmware.

Some thing to note:

Whenever I try to terminate the program and re-start it immediately, the error occurs. It needs some time (around 10 seconds when I was testing it) before calling the program again in order for it to not error.

Summary: It works for now!

I'd like to thank you two for your responses. :>
Kudos!

closed 2024-03-21 by imca-potato