merbanan/rtl_433 #2848: "No supported tuner found" occasionally with RTL2838 DVB-T tuner

coldib, 2024-02-23:
Hello, I use a RTL2838 DVB-T on a Raspberry PI 3. 

When using the command sudo rtl_433 -R 141 -R 56 every now and then I receive "No supported tuner found". 

Things I've done so far:

Blacklis unnecessary modules
```
vim /etc/modprobe.d/rtl-sdr-blacklist.conf 
	blacklist dvb_usb_rtl28xxu
	blacklist e4000
	blacklist rtl2832
```
Checking running process when it's not working (no process hangs)

`sudo ps aux | grep rtl_433`

Most of the time after a reboot it works again

Here is a test output:

```
user@hostname:~ $ sudo rtl_433 -R 141 -R 56
rtl_433 version 23.11-55-g5786c01c branch master at 202402160905 inputs file rtl_tcp RTL-SDR
Found Rafael Micro R820T tuner
[SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
^CSignal caught, exiting!

user@hostname:~ $ sudo rtl_433 -R 141 -R 56
rtl_433 version 23.11-55-g5786c01c branch master at 202402160905 inputs file rtl_tcp RTL-SDR
Found Rafael Micro R820T tuner
[SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
^CSignal caught, exiting!

user@hostname:~ $ sudo rtl_433 -R 141 -R 56
rtl_433 version 23.11-55-g5786c01c branch master at 202402160905 inputs file rtl_tcp RTL-SDR
Found Rafael Micro R820T tuner
[SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
^CSignal caught, exiting!

user@hostname:~ $ sudo rtl_433 -R 141 -R 56
rtl_433 version 23.11-55-g5786c01c branch master at 202402160905 inputs file rtl_tcp RTL-SDR
No supported tuner found
Enabled direct sampling mode, input 1
[SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 250000.000414 Hz
Allocating 15 zero-copy buffers
Signal caught, exiting!

user@hostname:~ $ sudo rtl_433 -R 141 -R 56
rtl_433 version 23.11-55-g5786c01c branch master at 202402160905 inputs file rtl_tcp RTL-SDR
No supported tuner found
Enabled direct sampling mode, input 1
[SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 250000.000414 Hz
Allocating 15 zero-copy buffers
^CSignal caught, exiting!

user@hostname:~ $ sudo rtl_433 -R 141 -R 56
rtl_433 version 23.11-55-g5786c01c branch master at 202402160905 inputs file rtl_tcp RTL-SDR
No supported tuner found
Enabled direct sampling mode, input 1
[SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 250000.000414 Hz
Allocating 15 zero-copy buffers
^CSignal caught, exiting!

user@hostname:~ $ sudo rtl_433 -R 141 -R 56
rtl_433 version 23.11-55-g5786c01c branch master at 202402160905 inputs file rtl_tcp RTL-SDR
Found Rafael Micro R820T tuner
[SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
```

And some additional info: With `sudo rtl_433 -R 141 -R 56 -F csv` it seems to never work.

```
user@hostname:~ $ sudo rtl_433 -R 141 -R 56 -F csv
rtl_433 version 23.11-55-g5786c01c branch master at 202402160905 inputs file rtl_tcp RTL-SDR
Use "-F log" if you want any messages, warnings, and errors in the console.
time,msg,codes,model,id,channel,battery_ok,temperature_C,humidity,button,mic
No supported tuner found
Enabled direct sampling mode, input 1
Exact sample rate is: 250000.000414 Hz
Allocating 15 zero-copy buffers
^CSignal caught, exiting!

user@hostname:~ $ sudo rtl_433 -R 141 -R 56 -F csv
rtl_433 version 23.11-55-g5786c01c branch master at 202402160905 inputs file rtl_tcp RTL-SDR
Use "-F log" if you want any messages, warnings, and errors in the console.
time,msg,codes,model,id,channel,battery_ok,temperature_C,humidity,button,mic
No supported tuner found
Enabled direct sampling mode, input 1
Exact sample rate is: 250000.000414 Hz
Allocating 15 zero-copy buffers
^CSignal caught, exiting!

user@hostname:~ $ sudo rtl_433 -R 141 -R 56 -F csv
rtl_433 version 23.11-55-g5786c01c branch master at 202402160905 inputs file rtl_tcp RTL-SDR
Use "-F log" if you want any messages, warnings, and errors in the console.
time,msg,codes,model,id,channel,battery_ok,temperature_C,humidity,button,mic
No supported tuner found
Enabled direct sampling mode, input 1
Exact sample rate is: 250000.000414 Hz
Allocating 15 zero-copy buffers
^CSignal caught, exiting!

user@hostname:~ $ sudo rtl_433 -R 141 -R 56 -F csv
rtl_433 version 23.11-55-g5786c01c branch master at 202402160905 inputs file rtl_tcp RTL-SDR
Use "-F log" if you want any messages, warnings, and errors in the console.
time,msg,codes,model,id,channel,battery_ok,temperature_C,humidity,button,mic
No supported tuner found
Enabled direct sampling mode, input 1
Exact sample rate is: 250000.000414 Hz
Allocating 15 zero-copy buffers
^CSignal caught, exiting!
```

zuckschwerdt, 2024-02-23:
That's actually a librtlsdr error message and points to unreliable USB communication or the RTL-SDR in a broken state.
Nothing much we can do, check cables, power supply, worst case buy another RTL-SDR :/
https://github.com/osmocom/rtl-sdr/blob/master/src/librtlsdr.c#L1630

closed 2024-02-23 by zuckschwerdt

merbanan, 2024-02-23:
For some reason either the chip gets into a broken state that results in hangs on the i2c bus or the tuner chip stops responding. This is unrelated to rtl_433 and is most likely caused by a combination of the software and hardware. It is very unlikely to be fixed by someone who has the hardware and the skills to debug the issue.

Your best bet is to work around this with another dongle or some change in the hardware configuration.

coldib, 2024-03-08:
Thanks for that hint!