merbanan/rtl_433 #2857: Wmbus device type 104, using different frequency and different transmission speed

jacksonlancaster, 2024-03-01:
I have a wireless mbus device that uses 915MHz and 240kbps transmission rate. It sends data in C1 format. Since it's non standard freq and speed, how would I read this? Any parameters I can change or code I can modify to accommodate this?

zuckschwerdt, 2024-03-01:
Frequency is no problem, just use a fitting `-f`.

The rate is pretty high, you'd need to adjust short and long here: https://github.com/merbanan/rtl_433/blob/master/src/devices/m_bus.c#L1217 and then use a high sample rate.

jacksonlancaster, 2024-03-05:
I changed the short width. Any idea what the long_width should be set to in the case of 250kbps?

.short_width = 25,  // Bit rate: 250kb/s
.long_width  = 10,  // NRZ encoding (bit width = pulse width)

ProfBoc75, 2024-03-05:
Hi @jacksonlancaster : this is less, at 250kb/s the pulse duration is 4 µs, and because of PCM , short = long = 4
it's very short, as @zuckschwerdt  said you have to increase the sample rate too with `-s 2048k` for example.

ProfBoc75, 2024-03-05:
Yes, you can reduce with r = 200 to keep same ratio. Up to you to update the value.

If you want to test, you can use the flex decoder until you have a good result then you apply your findings into the decoder.
Because each time you update the c file you need to compile, then to test, correct again , compile , test, .... Lot of time to get the figures.

jacksonlancaster, 2024-03-05:
Thank you guys so much. I was able to get it to read!

closed 2024-03-05 by jacksonlancaster