merbanan/rtl_433 #2858: Windows binary or compile guide?

bkenobi, 2024-03-02:
I am hoping to be able to use rtl_433 to receive the TPMS sensors on my 2006 Ford Escape.  The easiest way to do that would be with my Windows laptop rather than a RPi.  Perhaps I could plug an RPi in next to the vehicle and use the laptop to ssh in and that will be my backup.  Is there a current precompiled Windows binary or compile guide available?  I found one compiled binary dated from 2019 but I'm not sure if the Ford TPMS would be usable or if I need a newer release because I haven't found a date that protocol 89 was released.  I only have one SDR and it's on my home automation setup so I'd rather not pull it until I have a known working binary that has the correct protocol.

ProfBoc75, 2024-03-02:
Hi @bkenobi : You can follow the [Build](https://github.com/merbanan/rtl_433/blob/master/docs/BUILDING.md#windows) guide to compile the last version in Windows.

zuckschwerdt, 2024-03-02:
Also see https://github.com/merbanan/rtl_433/releases

closed 2024-03-02 by zuckschwerdt

bkenobi, 2024-03-03:
I don't know why I didn't see the windows binary when I looked before, but yes there are versions already in the release path.  This should work great.  Thanks!