merbanan/rtl_433 #2862: rtl_433 stopped receiving all of my sensors

bkenobi, 2024-03-04:
I have a SDR connected to my RPi3B that has been receiving several sensors correctly for months without issue.  I noticed my Acurite 5in1 weather station stopped receiving recently for some reason.  The batteries went dead last month but after replacing the data has only been sporadic.  I had another issue unrelated and was going to use the SDR to test but when I connected it back to the RPi and restarted the service, I am now only getting one or two Acurite tower sensors sporadically (I have 9 Acurite TX592, a Acurite 5in1 weather station, 3 Govee water leak, and a few other misc).

I suppose the 5in1 is several years old and changing the batteries could have cause some strange issue, but I doubt it.  Also, that doesn't explain why all of the other sensors disappeared.

My configuration is:
RPi3B running Raspbian Buster

rtl_433 version 23.11-12-g8950dc41 branch master at 202312052040 inputs file rtl_tcp RTL-SDR
Found Fitipower FC0013 tuner
[SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: , "Generic RTL2832U OEM"
Exact sample rate is: 250000.000414 Hz

The command line is:
ExecStart=/usr/local/bin/rtl_433 -R 231 -R 40 -R 176:17104 -M utc -F mqtt://localhost:1883,events=rtl_433[/model][/id]

I tested the device/signal by connecting the SDR to SDRSharp with the rtl_433 plugin.  While running I could see spikes for different sensors and depending on how close I was to them I could see the waterfall plot change.  The plugin decoded several sensors but it wasn't everything though I don't know if the sdrsharp+rtl_433 plugin are the same as running rtl_433 directly.

If there are any details I can provide to help diagnose what happened please let me know.  This has been a very stable setup when running but for some reason it's stopped working almost entirely when the SDR was removed which makes no sense.

bkenobi, 2024-03-04:
I stopped the service to collect the information above and when I restarted the service it appears that the SDR is receiving most of the sensors again.  I wonder if this is an auto-gain issue?

zuckschwerdt, 2024-03-04:
Or maybe frequency drift due to temperature. You can look at the input rtl_433 sees with the -F rtl_tcp  output. Use a SDR gui with rtl_tcp support to connect to rtl_433 then.

bkenobi, 2024-03-04:
I did notice the center of the waterfall was left of all sensor signals. Since the width of the lower strength signals was very narrow, perhaps rtl-433 didnt detect them? Is there a way to have the sdr accept a wider frequency band?

merbanan, 2024-03-04:
> Is there a way to have the sdr accept a wider frequency band?
Higher sample rate.

bkenobi, 2024-03-04:
In SDRsharp I had read that using the default sample rate of 2.4 MSPS might cause the plugin to crash so I down selected to 0.9 MSPS.  When I did that I was getting data but only when close to the sensor.  I will try changing that and testing again.

However, that doesn't explain why rtl_433 was getting 1 or 2 sensors for a couple days and then all of a sudden all of my sensors are back.  There was an issue a couple years ago (issue 1669) that appeared to be resolved with a change to the auto-gain setting.  Perhaps the change didn't fix it completely?

https://github.com/merbanan/rtl_433/issues/1669#issuecomment-1741743959

I'm not sure if it is related, but I did remove the code that initialized the SDR 10x before actually launching rtl_433 for real late last year when this fix was applied.  I can try adding it back in if the issue is auto-gain setting again.

bkenobi, 2024-03-11:
I had the power fail over the weekend due to trees in the lines.  When power was restored, the sensors were not receiving again.  Instead of tinkering with things like i did last time around, I tried adding in the initialization lines that made the rtl_433 service very stable and on first restart of the service the sensors immediately were seen.  I don't know if there is an actionable item, but in my experience the fix I linked from issue 1669 did not help in my case.

For reference, the lines that make my system work are the following (10x prime lines prior to actually launching).

```ExecStartPre=/usr/local/bin/rtl_433 -T 3
ExecStartPre=/usr/local/bin/rtl_433 -T 3
ExecStartPre=/usr/local/bin/rtl_433 -T 3
ExecStartPre=/usr/local/bin/rtl_433 -T 3
ExecStartPre=/usr/local/bin/rtl_433 -T 3
ExecStartPre=/usr/local/bin/rtl_433 -T 3
ExecStartPre=/usr/local/bin/rtl_433 -T 3
ExecStartPre=/usr/local/bin/rtl_433 -T 3
ExecStartPre=/usr/local/bin/rtl_433 -T 3
ExecStartPre=/usr/local/bin/rtl_433 -T 3

ExecStart=/usr/local/bin/rtl_433 -R 231 -R 40 -R 176:17104 -M utc -F mqtt://localhost:1883,events=rtl_433[/model][/id]```

gdt, 2024-06-03:
Where are we on this?  seems like the only thing that is left is a "device needs priming sometimes" problem, but this issue doesn't really say that, and there's a lot of history.   Should we
  1) close
  2) open a new issue about needing priming
  3) retitle and restarte this one?

I vote for 2, and @bkenobi if you are ok with that, maybe you could try to open an issue trying hard to have it be something useful for people to scan over the years until it is fixed, sort of all the needed info and otherwise terse.   I am too fuzzy on what's up to try to write that.

bkenobi, 2024-06-03:
I can open a new issue if that's preferred.  This setup is pretty hands off and only gets noticed when data stops collecting.  I can make a note to regularly check if that's needed for testing.

gdt, 2024-06-03:
If there are N maintainers, there are 2N+3 opinions...  But my take is that there was a lot of back and forth while things got figured out, and now in your head is a much crisper description of what is now wrong, and if you were to open a fresh issue that describes your observed facts and theories, and cross refs, written with the benefit of hindsight, dropping talking all the stuff that's now fixed on master, that can be easier for someone reviewing issues to understand.  It does feel like something is still wrong.

gdt, 2024-06-23:
@bkenobi Could you open a fresh issue written from your current understanding?  I find this one too confusing to read over, compared to a new cleaner one.   Also please update to current git master.

gdt, 2024-07-29:
This issue appears to have gone dormant.  Please see my comment from june 23.

closed 2025-07-04 by gdt