merbanan/rtl_433 #2865: adding attributes to output

nebulous, 2024-03-06:
I have multiple instances of rtl_433 running with multiple sdr devices - all of which publish to mqtt with -F mqtt. This works fabulously, but it would be nice to be able to add an attribute to the payload to identify which device a given message was received on. Apologies if this is documented someplace already, if so I've been unable to find it.

contrived example:
two receivers, one located on the north side and other on the south side of a building. Either or both of which could receive the same TPMS signals, but it would provide useful context to know which one did the receiving.

zuckschwerdt, 2024-03-06:
```
$ rtl_433 -K
…
		= Data tags option =
  [-K FILE | PATH | <tag> | <key>=<tag>] Add an expanded token or fixed tag to every output line.
	If <tag> is "FILE" or "PATH" an expanded token will be added.
	The <tag> can also be a GPSd URL, e.g.
		"-K gpsd,lat,lon" (report lat and lon keys from local gpsd)
		"-K loc=gpsd,lat,lon" (report lat and lon in loc object)
		"-K gpsd" (full json TPV report, in default "gps" object)
		"-K foo=gpsd://127.0.0.1:2947" (with key and address)
		"-K bar=gpsd,nmea" (NMEA default GPGGA report)
		"-K rmc=gpsd,nmea,filter='$GPRMC'" (NMEA GPRMC report)
	Also <tag> can be a generic tcp address, e.g.
		"-K foo=tcp:localhost:4000" (read lines as TCP client)
		"-K bar=tcp://127.0.0.1:3000,init='subscribe tags\r\n'"
		"-K baz=tcp://127.0.0.1:5000,filter='a prefix to match'"
```

closed 2024-03-06 by zuckschwerdt

zuckschwerdt, 2024-03-06:
No worries, what little help text we have is rather compact and not easily "discoverable".