merbanan/rtl_433 #2867: Ecowitt WS-68:  Apparently I gave y'all bad data for windspeed

nitrogen76, 2024-03-09:
So uh.
I got a ecowitt panel, because i noticed the windspeed i THOUGHT i gave y'all was good data, but apparently not.

I have no idea what the windspeed actually is, but it's faster than what the value is.

It seems to be around 2.75 times faster in km/h than what it should be.  I'm not exactly sure what it's measuring, so I apologize for giving y'all bad information previously.

zuckschwerdt, 2024-03-09:
This was #2790

2.75 seems odd. The raw values then is neither km/h or m/s or mi/h.

nitrogen76, 2024-03-09:
i'm going to see if i can decompile the firmware and figure out what the heck it is

nitrogen76, 2024-03-09:
yeah the decompiling wasn't helpful, unfortunately, so I sat here and watched the panel and the output of the WS68 sensor.

It's SUPER weird.  The multiplication factor varies between:

2.786885245901639
2.773722627737226
2.772277227722772

The panel seems to get km/h data down to the tenths decimal point, so here's some examples from the output from the station vs what's reported under km/h on the panel, along with "codes"

Reported / actual / codes

25.0 / 9  / codes     : {197}aaaaaaaaaaa2dd468000bd61fcf5c0fffff19101f32142e210
15 / 5.4/ codes     : {197}aaaaaaaaaaa2dd468000bd61fc75c2fffff0f511a3294f8210
21 / 7.6 / codes     : {196}aaaaaaaaaaa2dd468000bd61f945c2fffff15621f32095642
18 / 6.5 / codes     : {197}aaaaaaaaaaa2dd468000bd61f935c2fffff125b1f320a4c210
24 / 8.6 / codes     : {197}aaaaaaaaaaa2dd468000bd61f7c5c2fffff182b1f32d1d2210

I can get you anything else you need.  The panel I have also outputs longs that it saves to SD card

Apologies again for the previous bad data.

zuckschwerdt, 2024-03-09:
@ProfBoc75 looked into this previously, perhaps he has any ideas?

ProfBoc75, 2024-03-10:
Hi @nitrogen76 : 

From my side, I can't see any error, the signal is correctly decoded (I checked with rtl_433 -y yourcodes), wind speed and wind gust values are in km/h and coherent, like Gust always > Speed. CRC is good, same the LUX around 81K Lux and UV index =5 are aligned.

    rtl_433 -y  {197}aaaaaaaaaaa2dd468000bd61fcf5c0fffff19101f32142e210

[image]

What is your display panel model ?
Sounds like your panel is not showing live values in km/h but some average in m/s (daily / hourly ?? ) then values are smaller.
This could be a display mode with m/s unit.

Quick table from your 5 codes: Each time the closest value matches the m/s unit.

km/h | mph | knots | m/s | ft/s | Beaufort Scale | Your Display | Comments
| -:| -:| -:| -:| -:| -:| -:| -:|
25 | 15.5 | 13.5 | 6.9 | 22.8 | 4 | 9 | The closest value is in m/s
15 | 9.3 | 8.1 | 4.2 | 13.7 | 3 | 5.4 | The closest value is in m/s
21 | 13 | 11.3 | 5.8 | 19.1 | 4 | 7.6 | The closest value is in m/s
18 | 11.2 | 9.7 | 5 | 16.4 | 3 | 6.5 | The closest value is in m/s
24 | 14.9 | 13 | 6.7 | 21.9 | 4 | 8.6 | The closest value is in m/s

nitrogen76, 2024-03-10:
The panel is an Ecowitt HP2560.

I don't think that's whats happening.

I have acquired a WS90 as well, so i will crosscheck the values with that and see what I get.  (I am using multiple weather stations to test some logging software i am writing as a python learning project)

The values of both are quite far apart, as seen here

> time      : 2024-03-10 12:08:44                    Protocol  : 244
> model     : Fineoffset-WS90                        ID        : 004c79
> Battery   : 1.000        Battery Voltage: 3280 mV  Temperature: 15.1 C       Humidity  : 36 %          Wind direction: 133
> Wind speed: 1.7 m/s      Gust speed: 2.0 m/s       UVI       : 6.0           Light     : 95380.0 lux   Flags     : 8a
> Total Rain: 65.2 mm      Supercap Voltage: 5.3 V   Firmware Version: 138     Extra Data: 3fff6bd084------b44fff7ff60000
> Integrity : CRC
> _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
> time      : 2024-03-10 12:08:46                    Protocol  : 113
> model     : EcoWitt-WS68 id        : 3030
> Battery Raw: 92          Battery OK: 1             Lux       : 84740 lux     Wind Speed: 10.0 km/h     Wind Gust : 10.0 km/h
> UVI       : 6            Wind dir  : 106           Extra Data: 420           Integrity : CRC
> 

and

> time      : 2024-03-10 12:12:04                    Protocol  : 113
> model     : EcoWitt-WS68 id        : 3030
> Battery Raw: 92          Battery OK: 1             Lux       : 85630 lux     Wind Speed: 6.0 km/h      Wind Gust : 10.0 km/h
> UVI       : 6            Wind dir  : 101           Extra Data: 420           Integrity : CRC
> _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
> time      : 2024-03-10 12:12:05                    Protocol  : 244
> model     : Fineoffset-WS90                        ID        : 004c79
> Battery   : 1.000        Battery Voltage: 3280 mV  Temperature: 15.3 C       Humidity  : 35 %          Wind direction: 9
> Wind speed: 0.9 m/s      Gust speed: 1.3 m/s       UVI       : 7.0           Light     : 95790.0 lux   Flags     : 8a
> Total Rain: 65.2 mm      Supercap Voltage: 5.3 V   Firmware Version: 138     Extra Data: 3fff6bd084------b44fff7ff70000
> Integrity : CRC
> ^CSignal caught, exiting!

ProfBoc75, 2024-03-11:
Hi @nitrogen76 : I find the issue, it's just very simple, and you were very close to the result.

From your 5 codes:

    9 / 25 = 0.36
    5.4 / 15 = 0.36
    7.6 / 21 = 0.36
    6.5 / 18 = 0.36
    8.6 /24 = 0.36

And 0.36 is exactly the ratio between km/h and m/s scale 10.

Back to your 5 codes 

    0x19 = 25 = 2.5 m/s x 3.6 = 9 km/h
    0x0f  = 15 = 1.5 m/s x 3.6 = 5.4 km/h
    0x15 =  21 = 2.1 m/s x 3.6 = 7.6 km/h
    0x12 = 18 = 1.8 m/s x 3.6 = 6.5 km/h
    0x18 = 24 = 2.4 m/s x 3.6 = 8.6 km/h

And you were close, because 1 / 0.36 = 2.77777777777 , then the round value and you have the above explanation.

Now, we have an issue, since the user guide specification said that the rand is from 0 to 50 m/s , so raw value from 0 to 500 , and we have only 8 bit, this means that the 9th bit is somewhere in another flag, like into the battery byte ?

ProfBoc75, 2024-03-11:
@nitrogen76 : Would be nice if you can capture codes with lot of wind  (  > 92 km/h , > 25.5 m/s ). May be try with a strong fan , cold air. In order to identity the 9th bit position and get the good data layout.

nitrogen76, 2024-03-11:
let me see what i can do, and will get back to you.

ProfBoc75, 2024-03-11:
Good idea, but not enough, in fact the wind speed/gust > 92 km/h or > 25.5 m/s reported by your panel since the current values from rtl_433 are false (for the moment) and 10x m/s.

So the gust at 128.0 km/h, in reality it's 12.8 m/s ..., same wind speed at 50.0 km/h = 5 m/s 

From rtl_433 point of view, track any Wind Gust at 255 km/h (so 25.5 m/s in reality) ... the objective is to pass the 0xFF (LSB) to force the MSB to be set to 1 somewhere in the data layout, but the best will be from your display panel to be sure that the windows speed is > 92 km/h.

If we compare the current ws68 data layout with the FineOffset ws90 (very close to ws68 sensor):

- Wind Dir MSB : 1 bit from b[7], remaining other bits for Wind Speed MSB and Wind Gust MSB, and if you look into the FineOffset ws90, we may have very good candidate here since it's already identified.

Both sensor should have the same data layout for the same wind metrics.

My proposal:

    TYPE:8h ?8h ID:16h LUX:16h BATT:8d ?1b WGUST_MSB:1b WDIR_MSB:1b WSPEED_MSB:1b ?4h8h8h WSPEED_LSB:8d WDIR_LSB:8h WGUST_LSB:8d UVI:8h CRC:8h SUM:8h ?8h4h

I'm preparing a PR, and waiting for your attempts to confirm.

nitrogen76, 2024-03-11:
I couldn't get iot to go fast enough, but my panel WILL report in  m/s.

The meters /secvond is 10x what's reported. (so if it reports 156 , it's really 15.6 m/s

same for gusts

nitrogen76, 2024-03-12:
Fastest I could get (and i needed a dremel to get it going this fast)

_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-03-11 18:45:26                    Protocol  : 113
model     : EcoWitt-WS68 id        : 3030
Battery Raw: 92          Battery OK: 1             Lux       : 10 lux        Wind Speed: 237.0 km/h    Wind Gust : 144.0 km/h
UVI       : 0            Wind dir  : 83            Extra Data: 840           Integrity : CRC
[pulse_slicer_pcm] Ambient Weather WH31E Thermo-Hygrometer Sensor, EcoWitt WH40 rain gauge, WS68 weather station

_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-03-11 18:50:23                    Protocol  : 113
model     : EcoWitt-WS68 id        : 3030
Battery Raw: 92          Battery OK: 1             Lux       : 10 lux        Wind Speed: 156.0 km/h    Wind Gust : 184.0 km/h
UVI       : 0            Wind dir  : 6             Extra Data: 210           Integrity : CRC
[pulse_slicer_pcm] Ambient Weather WH31E Thermo-Hygrometer Sensor, EcoWitt WH40 rain gauge, WS68 weather station
codes     : {197}aaaaaaaaaaa2dd468000bd600015c0fffff9c06b8000916210

For some reason i couldn't reliably get a good gust; I had to use a dremel attached to the stalk of the anemometer to get it to even go this fast.

ProfBoc75, 2024-03-12:
Wind Speed: 237.0 km/h Wind Gust : 144.0 km/h  : In reality, it's not possible to get speed > gust, and I can confirm that the Wind_Gust MSB is set to 1 , see the 0x4 before the 0xfffff into the code

Sounds good for me, at least the Wind Gust MSB matches with ws90.
Real value are:
- Wind Speed = 23.7 m/s , 
- and Wind Gust = 0x190 = 400 = 40 m/s. ( instead of 0x90 = 14.4 m/s without the MSB)

I pushed the PR to let you try.

nitrogen76, 2024-03-12:
the PR you posted I built and it looks good so far