merbanan/rtl_433 #2869: 'heap-use-after-free" issue with ASAN

gvanem, 2024-03-09:
I compiled `rtl_433.exe` on Windows-10 using clang-cl ver.17 and `-fsanitize=address -fsanitize-recover=address` .

Running this ` bin\rtl_433.exe -s 2M -f 1090M -M time:usec -R 0 -Y autolevel -Y magest -A -v`, 
and pressing `^C`, causes *ASAN* to kick in with this, but only **sometimes**:
```
< lot of details snipped >
==9996==ERROR: AddressSanitizer: heap-use-after-free on address 0x12d39c7a69b8 at pc 0x7ff6fc9de385 bp 0x00db25983bc0 
sp 0x00db25983bc8
READ of size 8 at 0x12d39c7a69b8 thread T0
    #0 0x7ff6fc9de384 in sdr_stop F:\gv\dx-radio\RTL_433\src\sdr.c:1777
    #1 0x7ff6fc9d901f in sdr_handler F:\gv\dx-radio\RTL_433\src\rtl_433.c:1438
    #2 0x7ff6fc9a647d in mg_mgr_handle_ctl_sock F:\gv\dx-radio\RTL_433\src\mongoose.c:4167  << mg_broadcast in another thread
    #3 0x7ff6fc9a6bc0 in mg_socket_if_poll F:\gv\dx-radio\RTL_433\src\mongoose.c:4311
    #4 0x7ff6fc98db20 in mg_mgr_poll F:\gv\dx-radio\RTL_433\src\mongoose.c:2671
    #5 0x7ff6fc98d6a2 in mg_mgr_free F:\gv\dx-radio\RTL_433\src\mongoose.c:2641
    #6 0x7ff6fc9cc61f in r_free_cfg F:\gv\dx-radio\RTL_433\src\r_api.c:244
    #7 0x7ff6fc9db671 in main F:\gv\dx-radio\RTL_433\src\rtl_433.c:2059
    #8 0x7ff6fca5db83 in invoke_main D:\a\_work\1\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl:78
    #9 0x7ff6fca5db83 in __scrt_common_main_seh D:\a\_work\1\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl:288
    #10 0x7ffc75b17343  (C:\Windows\System32\KERNEL32.DLL+0x180017343)
    #11 0x7ffc763426b0  (C:\Windows\SYSTEM32\ntdll.dll+0x1800526b0)

0x12d39c7a69b8 is located 120 bytes inside of 168-byte region [0x12d39c7a6940,0x12d39c7a69e8)
freed by thread T0 here:
    #0 0x7ffc230b23f8 in free D:\a\_work\1\s\src\vctools\asan\llvm\compiler-rt\lib\asan\asan_malloc_win.cpp:230
    #1 0x7ff6fc9dc1f8 in sdr_close F:\gv\dx-radio\RTL_433\src\sdr.c:1132
    #2 0x7ff6fc9cc2e3 in r_free_cfg F:\gv\dx-radio\RTL_433\src\r_api.c:211
    #3 0x7ff6fc9db671 in main F:\gv\dx-radio\RTL_433\src\rtl_433.c:2059
    #4 0x7ff6fca5db83 in invoke_main D:\a\_work\1\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl:78
    #5 0x7ff6fca5db83 in __scrt_common_main_seh D:\a\_work\1\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl:288
    #6 0x7ffc75b17343  (C:\Windows\System32\KERNEL32.DLL+0x180017343)
    #7 0x7ffc763426b0  (C:\Windows\SYSTEM32\ntdll.dll+0x1800526b0)

previously allocated by thread T0 here:
    #0 0x7ffc230b22ba in calloc D:\a\_work\1\s\src\vctools\asan\llvm\compiler-rt\lib\asan\asan_malloc_win.cpp:276
    #1 0x7ff6fc9df830 in sdr_open_rtl F:\gv\dx-radio\RTL_433\src\sdr.c:395
    #2 0x7ff6fc9dbf89 in sdr_open F:\gv\dx-radio\RTL_433\src\sdr.c:1086
    #3 0x7ff6fc9d90ce in start_sdr F:\gv\dx-radio\RTL_433\src\rtl_433.c:1463
    #4 0x7ff6fc9db4fd in main F:\gv\dx-radio\RTL_433\src\rtl_433.c:2015
    #5 0x7ff6fca5db83 in invoke_main D:\a\_work\1\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl:78
    #6 0x7ff6fca5db83 in __scrt_common_main_seh D:\a\_work\1\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl:288
    #7 0x7ffc75b17343  (C:\Windows\System32\KERNEL32.DLL+0x180017343)
    #8 0x7ffc763426b0  (C:\Windows\SYSTEM32\ntdll.dll+0x1800526b0)

SUMMARY: AddressSanitizer: heap-use-after-free F:\gv\dx-radio\RTL_433\src\sdr.c:1777 in sdr_stop
...
```

Without any `-v` or `-vv`, I do not see ASAN kicking in (some timing issue). 
This is also OK: `bin\rtl_433.exe -s 2M -f 1090M -T 10 -A -v`.

But the core issue seems to be that `cfg->dev` never becomes `NULL`.
So should it be rewritten into `int sdr_stop(sdr_dev_t **dev)` and use 
`pthread_mutex_lock()` / `pthread_mutex_unlock()` around the `free()`?
I'm not so into these details.

zuckschwerdt, 2024-03-09:
Interesting. `sdr_stop()` should not free, the dev could be reused. The `sdr_close()` does already free but is only called when the `cfg` is destructed. A free is missing in `start_sdr()` at least (called from `timer handler` on `DEVICE_MODE_RESTART`)

This should go to the beginning of `start_sdr()` (rtl_433.c line 1444)
```C
    if (cfg->dev) {
        r = sdr_close(cfg->dev);
        if (r < 0) {
            print_logf(LOG_WARNING, "Input", "Error closing SDR (%d)", r);
        }
    }

```

But that's not what you saw. Did you maybe press Ctrl-C twice when the exit got stuck? The second press is a hard kill without cleanup.

gvanem, 2024-03-10:
>  Did you maybe press Ctrl-C twice when the exit got stuck? 

No, I did not. But could you try with ASAN and the same command?

zuckschwerdt, 2024-03-11:
About to give up at try 30 or 40 I got it too.
Using `rtl_433 -s 2M  -M time:usec -Y autolevel -Y magest -vv`

The trace shows that we are nearly done with exit and calling `r_free_cfg()` which as a last step calls `mg_mgr_free()` but that does one last poll with `mg_mgr_poll()` and using the cfg after free trips ASAN. In both our cases `sdr_handler()` shouldn't run that late but all we can do is safe-guard the callback.

closed 2024-03-11 by zuckschwerdt

gvanem, 2024-03-11:
Good job. Work fine with ASAN now.

My motivation behind the initial command was to decode ADSB using the flex-decoder. It seems impossible since pulse-widths
in ADSB is 0.5 usec, but the minimum in rtl_433 is 1.0 usec. Can this be fixed?

zuckschwerdt, 2024-03-11:
The demod and slicers work in sample counts, e.g. https://github.com/merbanan/rtl_433/blob/master/src/pulse_slicer.c#L67
So with a high enough sample rate the 0.5 should work. I don't think 1 sample per bit (2M rate) would work well with our code, you should go much higher.

zuckschwerdt, 2024-03-12:
Yes, you are right. I wasn't aware that `atoi()` was still lurking there.

zuckschwerdt, 2024-03-12:
> uses `atoi()` to set `dev->short_width` . So to set `0.5` as float, shouldn't `arg_float()` be used instead?

done with 06170b2

commit a8fe08cd (2024-03-11): Fix event callback on shutdown (closes #2869)