merbanan/rtl_433 #2870: End of signal classification?

frevoli, 2024-03-11:
Hello community.

I have a key fob which repeats a signal 8 times in a burst.

In my ook capture file this is recorded as 8 individual captures, each with their own timestamp.

My question is - how long does rtl_433 wait after a signal, before moving onto what it thinks is a new one? And, can this time period be altered?

It would be really handy to have all 8 grouped as one capture so that I can accurately identify the time period between them.

Thank you.

Matt

zuckschwerdt, 2024-03-11:
Gap or pulse length more than `long_width` × `PD_MAX_GAP_RATIO` (=10) will end the detection. It's not recommended to change that.

A default frame of our SDR input is 128kB, 64k samples in cu8. You'll always get whole frames in the raw output. For parsed output like ook frames are not considered. Maybe use `-S` or even `-w` with `-T` then measure the timing in https://triq.org/pdv/

rtl_433 will report that it saw different transmissions when the gaps between packets are wide enough that they can't be part of the coding (e.g. a long pause symbol).

If you want to deduce, let's say "less than 2 packets: interference, more then 15 packets: jamming, between 3 to 14 packets: valid trigger" then you need to code that into the recipient or use a bridge script, see https://github.com/merbanan/rtl_433/tree/master/examples

closed 2024-03-11 by zuckschwerdt

frevoli, 2024-03-11:
Thank you that's great

frevoli, 2024-03-11:
Oh I nearly forgot to ask - can the timestamps show fractions of a second in addition to HH:MM:SS
Thanks again

zuckschwerdt, 2024-03-11:
Yes, see `rtl_433 -M`

frevoli, 2024-03-11:
Thank you sir

frevoli, 2024-03-17:
-M time:usec has worked perfectly to show microseconds in the terminal, thank you.

But the ook capture file still records time as HH:MM:SS.

Is there a way to record microseconds in the ook file too?

Thanks

zuckschwerdt, 2024-03-17:
Sorry, no. But you can change `format_time_str` to `usecs_time_str` here https://github.com/merbanan/rtl_433/blob/master/src/pulse_data.c#L190

frevoli, 2024-03-17:
Thank you, that will do! 
Do I need to run any commands or recompile after making that change, before it will take effect?

zuckschwerdt, 2024-03-17:
A recompile is needed as the change is to source code.

frevoli, 2024-03-17:
Best command to recompile?
Sorry - I haven't made local changes before.
Thanks again

zuckschwerdt, 2024-03-17:
https://triq.org/rtl_433/BUILDING.html#cmake

frevoli, 2024-03-17:
I did it with git diff, then the updater script