merbanan/rtl_433 #2872: "clang static analyzer" issues

gvanem, 2024-03-12:
Compiling using `clang-cl --analyze`, shows some statements that puzzles me.
E.g. in `src/rtl_433.c`:
```c
                if (in_file != stdin)
                    fclose(in_file = stdin);  // line 1899

                continue;
```
causes this warning:
```c
src/rtl_433.c(1899,28): warning: Although the value stored to 'in_file' is used in the enclosing expression, 
the value is never actually read from 'in_file' [deadcode.DeadStores]
 1899 |                     fclose(in_file = stdin);
      |                            ^         ~~~~~
```

I guess  the intention here is:
```c
            if (in_file != stdin) {
                fclose(in_file);
                in_file = stdin;
            }
```

But as the program is about to quit, `in_file` is no longer needed. Is that the case?
It seems the case on line 1968 anyway.

Besides there are many more `--analyze` warnings (mostly  `[deadcode.DeadStores]`): 
[clang-cl-analyze-report.txt](https://github.com/merbanan/rtl_433/files/14571099/clang-cl-analyze-report.txt)

Especially in `src/mongoose.c`. Time to update to ver. 7.3?

zuckschwerdt, 2024-03-12:
I suspect a copy-paste typo, I wouldn't write an expression like that intentionally ;)
The other warnings, esp. with `data = ...` are valuable and I probably need to fix all that.

Mongoose has changed a lot, it would be a bigger project to upgrade.

zuckschwerdt, 2024-03-12:
I've addressed the worst offenders and I'll close the rest soon.

commit 8786d5b6 (2024-03-12): minor: Cleanup clang analyze warnings (closes #2872)
closed 2024-03-12 by zuckschwerdt