merbanan/rtl_433 #2873: Issue with "-M time:usec"

gvanem, 2024-03-12:
Using the options `-F syslog:127.0.0.1:514 -M time:usec -v`, shows the 1st message received by my syslog-daemon like this:
```
2024.12.03-12:10:00 127.0.0.1, 1 2024-03-12T11:10:00Z AMD-PC rtl_433 - - - 
{"time":"1970-01-01 01:00:00.000000","src":"SDR","lvl":2,"msg":"Using device 0: Realtek, RTL2838-Green, SN: 00000000, 
\"Generic RTL2832U OEM\""}
```

All on 1 line.  And all subsequent lines have correct time-stamps.
Dropping `:usec` (options `-F syslog:127.0.0.1:514 -M time -v`) all OK with the first line:
```
2024.12.03-12:27:11 127.0.0.1, 1 2024-03-12T11:27:11Z AMD-PC rtl_433 - - - 
{"time":"2024-03-12 12:27:11","src":"SDR","lvl":2,"msg":"Using device 0: Realtek, RTL2838-Green, SN: 00000000, 
\"Generic RTL2832U OEM\""}
```

This start-up `"time"` and `"Using device .."` is coming from `sdr.c`:
`print_logf(LOG_CRITICAL, "SDR", "Using device %u: %s,...`. But where and how should the `:usec` time get filled?

zuckschwerdt, 2024-03-12:
That's from `usecs_time_str()` in r_util.c, called from `time_pos_str()` in r_api.c -- not sure why the timestamp would be still be zero.

gvanem, 2024-03-12:
>  not sure why the timestamp would be still be zero.

Perhaps `ago.tv_sec = ago.tv_usec = 0;` the 1st time in:
```c
        if (cfg->report_time_hires)
            return usecs_time_str(buf, format, cfg->report_time_tz, &ago);
```

since `sdr_callback()` wasn't called?

gvanem, 2024-03-12:
I did this that seems to fix it:
```diff
--- a/src/r_api.c 2024-03-11 13:20:59
+++ b/src/r_api.c 2024-03-12 14:19:42
@@ -188,6 +188,7 @@
     baseband_init();

     time(&cfg->frames_since);
+    get_time_now(&cfg->demod->now);

     list_ensure_size(&cfg->demod->r_devs, 100);
     list_ensure_size(&cfg->demod->dumper, 32);
```

zuckschwerdt, 2024-03-12:
Worse even, it seems that logging always uses the last sdr callback timestamp and not the actual time. It shouldn't be off more than a few hundred ms though.
But if we change demod->now for the logging then we'll likely break the sdr callback's time keeping.
Your change is the best we can do for now.

commit 5d03ee19 (2024-03-12): minor: Fix issue with first log message timestamp (closes #2873)
closed 2024-03-12 by zuckschwerdt