merbanan/rtl_433 #2877: Support for Thermopro TP-28B

joshpearce, 2024-03-14:
My best guess from the following is, `rtl_433 -f 915M -X "n=name,m=FSK_PCM,s=100,l=100,r=4096"`. The `r=4096` is a just something I put to make the error go away. I have no idea how to figure out `reset`.

```
rtl_433 -f 915M -A

Detected FSK package	2024-03-14 12:41:00
Analyzing pulses...
Total count:   90,  width: 51.69 ms		(51688 S)
Pulse width distribution:
 [ 0] count:    1,  width:    0 us [0;0]	(   0 S)
 [ 1] count:    5,  width:    2 us [2;2]	(   2 S)
 [ 2] count:    2,  width:    8 us [8;9]	(   8 S)
 [ 3] count:    2,  width:    5 us [5;6]	(   5 S)
 [ 4] count:    1,  width:    3 us [3;3]	(   3 S)
 [ 5] count:    3,  width:    1 us [1;1]	(   1 S)
 [ 6] count:   61,  width:  105 us [104;110]	( 105 S)
 [ 7] count:   12,  width:  212 us [210;215]	( 212 S)
 [ 8] count:    3,  width:  319 us [319;320]	( 319 S)
Gap width distribution:
 [ 0] count:    1,  width:    2 us [2;2]	(   2 S)
 [ 1] count:    5,  width:    3 us [3;3]	(   3 S)
 [ 2] count:    2,  width:    1 us [1;1]	(   1 S)
 [ 3] count:    2,  width:    5 us [5;5]	(   5 S)
 [ 4] count:   55,  width:  107 us [105;112]	( 107 S)
 [ 5] count:    8,  width:  214 us [214;216]	( 214 S)
 [ 6] count:    3,  width:  320 us [320;321]	( 320 S)
 [ 7] count:    6,  width:  504 us [429;536]	( 504 S)
 [ 8] count:    2,  width: 1922 us [1817;2028]	(1922 S)
 [ 9] count:    2,  width:  642 us [640;644]	( 642 S)
 [10] count:    1,  width: 4881 us [4881;4881]	(4881 S)
 [11] count:    1,  width: 18733 us [18733;18733]	(18733 S)
 [12] count:    1,  width: 1267 us [1267;1267]	(1267 S)
Pulse period distribution:
 [ 0] count:    1,  width:    2 us [2;2]	(   2 S)
 [ 1] count:    4,  width:    5 us [5;6]	(   5 S)
 [ 2] count:    2,  width:    9 us [9;9]	(   9 S)
 [ 3] count:    1,  width:   14 us [14;14]	(  14 S)
 [ 4] count:    1,  width:    7 us [7;7]	(   7 S)
 [ 5] count:    1,  width:    3 us [3;3]	(   3 S)
 [ 6] count:    1,  width:  109 us [109;109]	( 109 S)
 [ 7] count:   49,  width:  213 us [209;217]	( 213 S)
 [ 8] count:    9,  width:  320 us [317;325]	( 320 S)
 [ 9] count:    8,  width:  430 us [424;461]	( 430 S)
 [10] count:    3,  width:  641 us [639;644]	( 641 S)
 [11] count:    4,  width:  854 us [854;855]	( 854 S)
 [12] count:    2,  width: 2135 us [2028;2243]	(2135 S)
 [13] count:    1,  width: 4987 us [4987;4987]	(4987 S)
 [14] count:    1,  width: 18839 us [18839;18839]	(18839 S)
 [15] count:    1,  width: 1268 us [1268;1268]	(1268 S)
Pulse timing distribution:
 [ 0] count:    1,  width:    0 us [0;0]	(   0 S)
 [ 1] count:    7,  width:    2 us [2;2]	(   2 S)
 [ 2] count:    2,  width:    8 us [8;9]	(   8 S)
 [ 3] count:    4,  width:    5 us [5;6]	(   5 S)
 [ 4] count:    6,  width:    3 us [3;3]	(   3 S)
 [ 5] count:    5,  width:    1 us [1;1]	(   1 S)
 [ 6] count:  116,  width:  106 us [104;112]	( 106 S)
 [ 7] count:   20,  width:  213 us [210;216]	( 213 S)
 [ 8] count:    6,  width:  319 us [319;321]	( 319 S)
 [ 9] count:    6,  width:  504 us [429;536]	( 504 S)
 [10] count:    2,  width: 1922 us [1817;2028]	(1922 S)
 [11] count:    2,  width:  642 us [640;644]	( 642 S)
 [12] count:    1,  width: 4881 us [4881;4881]	(4881 S)
 [13] count:    1,  width: 18733 us [18733;18733]	(18733 S)
 [14] count:    1,  width: 1267 us [1267;1267]	(1267 S)
Level estimates [high, low]:   2565,      5
RSSI: -8.1 dB SNR: 27.1 dB Noise: -35.2 dB
Frequency offsets [F1, F2]:   -3151,   3303	(-48.1 kHz, +50.4 kHz)
Guessing modulation: Non Return to Zero coding (Pulse Code)
Attempting demodulation... short_width: 1, long_width: 1, reset_limit: 1024, sync_width: 0
Use a flex decoder with -X 'n=name,m=FSK_PCM,s=1,l=1,r=1024'
```

joshpearce, 2024-03-14:
Reading some other threads. I think reset should be 220 now.

zuckschwerdt, 2024-03-14:
It should be somewhat longer that the longest gap you expect. If this is 100 µs Manchester then the longest gap ist 200 --use 250, then anything shorter than 250 will be accepted as "200". But if this isn't MC but raw PCM then you need 100 x maximum consecutive 0-bits, maybe 1000 or even 2000.

Try to get a cleaner recording, the very short pulses/gaps need to go away, then you get a PDV link to view the code.
Compare https://triq.org/rtl_433/PULSE_FORMATS.html

joshpearce, 2024-03-14:
I've been fiddling with the gain setting to try to get a cleaner signal, but I can't find a value between 1-50 that does anything other than toggle the initial read of background noise.

joshpearce, 2024-03-14:
[g029_915M_1000k.cs16.zip](https://github.com/merbanan/rtl_433/files/14607582/g029_915M_1000k.cs16.zip)

joshpearce, 2024-03-14:
I did find [this](https://github.com/merbanan/rtl_433/issues/1366) issue and tried these args `-d driver=rtlsdr -t digital_agc=true -g TUNER=25` which seemed to give me better control of the gain

ProfBoc75, 2024-03-14:
Need more samples to found data layout, and can you share the temp values for each sample to help decode it.

I got codes with your sample and found the classic preamble 0xd2aa2dd4:

    rtl_433 -X "n=tp28b,m=FSK_PCM,s=105,l=105,r=5500,preamble=d2aa2dd4" *.cs16

klohner, 2024-03-17:
It might be a coincidence, but after the first two bits, every 9th bit is a "0".

[BitBench](https://triq.net/bitbench#c=%7B349%7Daaaaaaaaaaaaaab4aa8b75333309c1800015809581800028a41b5c22010000000000002aaa80000000000000%20%5BTP-28B%20g029_915M_1000k.cs16%5D&c=%5B%5D&c=%5B%5D&f=bb%208hv8hv8hv8hv%208hv8hv8hv8hv%208hv8hv8hv8hv%208hv8hv8hv8hv%208hv8hv8hv8hv%208hv8hv8hv8hv%208hv8hv8hv8hv%208hv8hv8hv8hv%20%0A&a=Preamble&m=%7B32%7Dd2aa2dd4&cw=4)

joshpearce, 2024-03-18:
I have it figured out. Thanks for the help @zuckschwerdt. I will submit a PR a little later.

joshpearce, 2024-03-19:
I guess I should have said I figured out the temps.

Here is what I have so far. 

- I could use some help with the checksum, as I have no experience in that area. 
- I haven't tried low battery. I should have saved some old AAAs.
- I feel I'm close on the flags, but can't quite make them work out.
- Where is the ID? Is it aaaa?

```
/**
ThermoPro TP28b Super Long Range Wireless Meat Thermometer for Smoker BBQ Grill.

Example data:

    rtl_433 -f 915M -F json -X "n=tp28b,m=FSK_PCM,s=105,l=105,r=5500,preamble=d2aa2dd4" | jq --unbuffered -r '.codes[0]' 
    (spaces below added manually)

    {259}2802 0626 0000 2802 1107 0000 a290 6d70 a702 000000000000 aaaa 0000000000000 [over1: 261C, Temp1: 23C, over2: 71C, Temp2: 23C]
    {259}2217 0626 0000 3102 1107 0000 a290 6d70 bf02 000000000000 aaaa 0000000000000 [over1: 261C, Temp1: 172, over2: 71C, Temp2: 23C]
    {259}4421 1026 9009 3002 1012 4410 a298 6d70 5a03 000000000000 aaaa 0000000000000 [over1: 261C, Temp1: 214, over2: 121C, Temp2: 23C]

Data layout:

    Bit bench format: A_TEMP:hhhh A_HI:hhhh A_LO:hhhh B_TEMP:hhhh B_HI:hhhh B_LO:hhhh FLAGS:hhhh UNK:hhhh CHK:hhhh hhhhhhhhhhhh hhhh hhhhhhhhhhhhh

    [p1_temp] [p1_set_hi] [p1_set_lo] [p2_temp] [p2_set_hi] [p2_set_lo] [status] [unk] [cksum] 000000000000 aaaa 0000000000000

    Temps are 16 bit Binary Coded Decimals (BCD), LLHH. They are always in Celsius. 
    Example: 2821, 
        28 => 2.8 deg C
        21 => 210 deg C
        210 + 2.8 = 212.8 C (displayed rounded to 213)

    - p1_temp: probe 1 current temp. 16 bit BCD
    - p1_set_hi: probe 1 high alarm temp. 16 bit BCD
    - p1_set_lo: probe 1 low alarm temp. 16 bit BCD
    - p2_temp: probe 2 current temp. 16 bit BCD
    - p2_set_hi: probe 2 high alarm temp. 16 bit BCD
    - p2_set_lo: probe 2 low alarm temp. 16 bit BCD

    Below is some work on status/alarm flags, but I can't quite make sense of them all.

    - status: 16 bit status flags 

        02d8 => F,  p1: in-range,    p2: in-range
        02f9 => F,  p1: low,         p2: in-range
        02dd => F,  p1: in-range,    p2: low
        02de => F,  p1: in-range,    p2: hi
        02fa => F,  p1: hi,          p2: in-range
        86f9 => F,  p1: low,         p2: low
        82f9 => F,  p1: low,         p2: low        ack'd
        a2f9 => C,  p1: low,         p2: low        ack'd
        a6f9 => C,  p1: low,         p2: low        unack'd

        - status & 0x2000 => Display in Celcius, otherwise Fahrenheit
        - status & 0x0400 => Alarm unacknowledged, otherwise acknowledged
        - status & 0x0020 => P1 in alarm, otherwise normal
        - status & 0x0004 => P2 in alarm, otherwise normal
        - status & 0x0001 => P2 in alarm low

    - unk: 16 bit unknown
    - cksum: 16 bit checksum

*/
```

And some data:

```
[... 100 lines cut ...]
{259}14023012900907093012900902fd6d70b803000000000000aaaa0000000000000
{259}14023012900907093012900902fd6d70b803000000000000aaaa0000000000000
{259}14023012900908093012900902fd6d70b903000000000000aaaa0000000000000
{310}14023012900909093012900902fd6d70ba03000000000000aaaa00000000000000000000000000
{259}14023012900910093012900902fd6d70c103000000000000aaaa0000000000000
{261}14023012900910093012900902fd6d70c103000000000000aaaa00000000000000
{259}14023012900910093012900902fd6d70c103000000000000aaaa0000000000000
{308}14023012900910093012900902fd6d70c103000000000000aaaa0000000000000000000000000
{259}14023012900910093012900902fd6d70c103000000000000aaaa0000000000000
{259}12023012900911093012900902fd6d70c003000000000000aaaa0000000000000
{259}14023012900912093012900902fd6d70c303000000000000aaaa0000000000000
{259}14023012900911093012900902fd6d70c203000000000000aaaa0000000000000
{259}12023012900913093012900902fd6d70c203000000000000aaaa0000000000000
{259}12023012900913093012900902fd6d70c203000000000000aaaa0000000000000
{259}14023012900911093012900902fd6d70c203000000000000aaaa0000000000000
{258}14023012900901103012900902f96d70b50300000000000155540000000000000
{259}16023012900948103012900902f96d70fe03000000000000aaaa0000000000000
{259}16023012900953103012900902f96d700904000000000000aaaa0000000000000
{259}16023012900954103012900902f96d700a04000000000000aaaa0000000000000
{299}00133012900951103012900902fa6d700304000000000000aaaa00000000000000000000000
{259}61153012900949103012900902fa6d705e04000000000000aaaa0000000000000
{259}92093012900945103012900902d86d705d04000000000000aaaa0000000000000
{259}92073012900943103012900902f96d707a04000000000000aaaa0000000000000
{303}66063012900938103012900902f96d704204000000000000aaaa000000000000000000000000
```

zuckschwerdt, 2024-03-19:
Everything after the checksum is junk.

The checksum is just an addition over all preceeding bytes, written as little-endian number.
E.g. `5a04` -> the sum is 0x45a

In BitBench use the Calculation row with "ADD", offset 0, length 128, width 8.

PR #2882 (2024-04-02): Add ThermoPro TP28b
resolve #2877
closed 2024-04-02 by zuckschwerdt