merbanan/rtl_433 #2879: Support for Thermor DG950 weather station

NicolasGagne, 2024-03-18:
[Data 1024k.zip](https://github.com/merbanan/rtl_433/files/14640459/Data.1024k.zip)

# Thermor DG950

FCC test reports are available at: https://fccid.io/S24DG950R
The manual is available at: https://www.manualslib.com/manual/806371/Thermor-Home-Weather-Station.html
Review of the station: https://www.home-weather-stations-guide.com/thermor-weather-station.html

# Data capture log
Thanks zuckschwerdt for the help.

Note: 
Sync recording are when the system start to set the ID to the receiving station. It change every time the transmitter is reset ID on this capture is 231
All recording in normal mode include Wind direction, Wind Speed, Temperature, Rainfall
The captured file are available in the "data 1024k.zip" file at the top.

FileNumber	               -Mode    -WindDir   -WindSpeed	  -Temp	  -Rainfall

g001_433.89M_1024k.cu8	  SYNC 		  
g002_433.89M_1024k.cu8	  SYNC      
g003_433.89M_1024k.cu8	  SYNC
g004_433.89M_1024k.cu8	  SYNC  
g005_433.89M_1024k.cu8	  SYNC
g006_433.89M_1024k.cu8	  SYNC
g007_433.89M_1024k.cu8	  SYNC
g008_433.89M_1024k.cu8	  NORMAL    Unknow    Unknow        19.3	     0
g009_433.89M_1024k.cu8	  NORMAL    22	      0	            19.2	     0
g010_433.89M_1024k.cu8	  NORMAL    22	      0	            19.2	     0
g011_433.89M_1024k.cu8	  NORMAL    22	      0	            19.2	     0.1
g012_433.89M_1024k.cu8	  NORMAL    22	      0	            19.1	     0.1
g013_433.89M_1024k.cu8	  NORMAL    22	      0	            19.2	     0.2
g014_433.89M_1024k.cu8	  NORMAL    22	      0	            19.1	     0.4
g015_433.89M_1024k.cu8	  NORMAL    22	      0	            19.1	     0.4
g016_433.89M_1024k.cu8	  NORMAL    22	      0	            23.4	     0.4
g017_433.89M_1024k.cu8	  NORMAL    22	      0	            24.1	     0.4
g018_433.89M_1024k.cu8	  NORMAL    22	      0	            22.2	     0.4
g019_433.89M_1024k.cu8	  NORMAL    135	      0	            21.1	     0.4
g020_433.89M_1024k.cu8	  NORMAL    202	      0	            21.6          0.4
g021_433.89M_1024k.cu8	  NORMAL    225	      0	            22	     0.4
g022_433.89M_1024k.cu8	  NORMAL    135	      0	            20.1	     0.4
g023_433.89M_1024k.cu8	  NORMAL    135	      3.6	            20.2	     0.4
g024_433.89M_1024k.cu8	  NORMAL    135	      8.5	            20.5	     0.4
g025_433.89M_1024k.cu8	  NORMAL    135	      6.5	            20.5	     0.4
g026_433.89M_1024k.cu8	  NORMAL    292	      0	            20.5	     0.4
g027_433.89M_1024k.cu8	  NORMAL    292	      6.5	            20.8	     0.4
g028_433.89M_1024k.cu8	  NORMAL    90	      5.4	            20.9	     0.5
g029_433.89M_1024k.cu8	  NORMAL    45	      6.2	            20.8	     0.6
g030_433.89M_1024k.cu8	  NORMAL    202	      3.6	            20.9	     0.6

## Analysis

The station consists of a outside wind speed, wind direction, rainfall and temperature sensor. 
All outside sensors are connected to a wireless transission unit. 
The transmission frequency is stated as 433.92 MHz.

There is an old github No affiliation, no idea is it correct.
https://github.com/gbThreepwood/Thermor-DG950

zuckschwerdt, 2024-03-18:
Interesting signal. Note the actual carrier seems to be at 433.99 MHz, and some samples show the carrier at 434.01 MHz, later some at 433.86 M and 433.81 M. The frequency seems to be very unstable (also within a signal).

There is a preamble of 53 x 1438 / 1438 µs.
Then packets of 9 symbols separated by 5381 µs gap.
The symbols are 2128 / 744 µs and 744 / 2128 µs (PWM).

This can be read as PCM
`-X 'n=name,m=OOK_PCM,s=750,l=750,g=3000,r=8000'`
or
`-X 'n=name,m=OOK_PCM,s=750,l=750,r=8000'`

or PWM:
`-X 'n=name,m=OOK_PWM,s=750,l=2128,y=1438,g=3000,r=8000'`

Ignoring the sync and breaking up the packets with
`rtl_433 -R 0 -X 'n=name,m=OOK_PWM,s=750,l=2128,y=1438,g=3000,r=8000' *.cu8`
we get:
```
g001_433.89M_1024k.cu8: {9}7f8, {9}0c0, {9}0c0, {9}460, {9}0c0, {9}0c0, {9}0c0, {9}0c0, {9}0c0, {9}290, {9}0c0, {9}0f8, {9}440
g002_433.89M_1024k.cu8: {9}7f8, {9}0c0, {9}0c0, {9}460, {9}0c0, {9}0c0, {9}0c0, {9}0c0, {9}0c0, {9}290, {9}0c0, {9}0f8, {9}440
g003_433.89M_1024k.cu8: {9}7f8, {9}0c0, {9}0c0, {9}460, {9}0c0, {9}0c0, {9}0c0, {9}0c0, {9}0c0, {9}290, {9}0c0, {9}0f8, {9}440
g004_433.89M_1024k.cu8: {9}7f8, {9}0c0, {9}0c0, {9}460, {9}0c0, {9}0c0, {9}0c0, {9}0c0, {9}0c0, {9}290, {9}0c0, {9}0f8, {9}440
g005_433.89M_1024k.cu8: {9}7f8, {9}0c0, {9}0c0, {9}460, {9}0c0, {9}0c0, {9}0c0, {9}0c0, {9}0c0, {9}290, {9}0c0, {9}0f8, {9}440
g007_433.89M_1024k.cu8: {9}7f8, {9}0c0, {9}0c0, {9}460, {9}0c0, {9}0c0, {9}0c0, {9}0c0, {9}0c0, {9}290, {9}0c0, {9}0f8, {9}440
g008_433.89M_1024k.cu8: {9}0c0, {9}358, {9}7f8, {9}798, {9}7f8, {9}7f8, {9}7f8, {9}7f8, {9}7f8, {9}7f8, {9}0f8, {9}0f8, {9}0f8
g009_433.89M_1024k.cu8: {9}0c0, {9}358, {9}7f8, {9}398, {9}570, {9}570, {9}7f8, {9}7f8, {9}458, {9}458, {9}778, {9}0f8, {9}0f8
g010_433.89M_1024k.cu8: {9}0c0, {9}358, {9}7f8, {9}398, {9}570, {9}570, {9}7f8, {9}7f8, {9}258, {9}258, {9}778, {9}0f8, {9}0f8
g011_433.89M_1024k.cu8: {9}0c0, {9}358, {9}3f8, {9}398, {9}570, {9}570, {9}7f8, {9}7f8, {9}258, {9}258, {9}778, {9}0f8, {9}778
g012_433.89M_1024k.cu8: {9}0c0, {9}358, {9}3f8, {9}598, {9}570, {9}570, {9}7f8, {9}7f8, {9}258, {9}258, {9}378, {9}0f8, {9}778
g013_433.89M_1024k.cu8: {9}0c0, {9}358, {9}5f8, {9}398, {9}570, {9}570, {9}7f8, {9}7f8, {9}258, {9}258, {9}778, {9}0f8, {9}378
g014_433.89M_1024k.cu8: {9}0c0, {9}358, {9}1f8, {9}598, {9}570, {9}570, {9}7f8, {9}7f8, {9}258, {9}258, {9}378, {9}0f8, {9}578
g015_433.89M_1024k.cu8: {9}0c0, {9}358, {9}7f8, {9}598, {9}570, {9}570, {9}7f8, {9}7f8, {9}258, {9}258, {9}378, {9}0f8, {9}0f8
g016_433.89M_1024k.cu8: {9}0c0, {9}2d8, {9}7f8, {9}158, {9}570, {9}570, {9}7f8, {9}7f8, {9}258, {9}258, {9}4f8, {9}0f8, {9}0f8
g017_433.89M_1024k.cu8: {9}0c0, {9}6d8, {9}7f8, {9}258, {9}570, {9}570, {9}7f8, {9}7f8, {9}258, {9}258, {9}378, {9}0f8, {9}0f8
g018_433.89M_1024k.cu8: {9}0c0, {9}4d8, {9}7f8, {9}458, {9}570, {9}570, {9}7f8, {9}7f8, {9}258, {9}258, {9}778, {9}0f8, {9}0f8
g019_433.89M_1024k.cu8: {9}0c0, {9}0d8, {9}7f8, {9}798, {9}270, {9}270, {9}7f8, {9}7f8, {9}258, {9}258, {9}378, {9}0f8, {9}0f8
g020_433.89M_1024k.cu8: {9}0c0, {9}0d8, {9}7f8, {9}158, {9}1f0, {9}1f0, {9}7f8, {9}7f8, {9}258, {9}258, {9}6f8, {9}0f8, {9}0f8
g021_433.89M_1024k.cu8: {9}0c0, {9}0d8, {9}7f8, {9}0d8, {9}0f0, {9}0f0, {9}7f8, {9}7f8, {9}258, {9}258, {9}7f8, {9}0f8, {9}0f8
g022_433.89M_1024k.cu8: {9}0c0, {9}758, {9}7f8, {9}398, {9}270, {9}270, {9}7f8, {9}7f8, {9}258, {9}258, {9}378, {9}0f8, {9}0f8
g023_433.89M_1024k.cu8: {9}0c0, {9}758, {9}7f8, {9}798, {9}270, {9}270, {9}4b8, {9}7f8, {9}258, {9}1e8, {9}778, {9}0f8, {9}0f8
g024_433.89M_1024k.cu8: {9}0c0, {9}758, {9}7f8, {9}258, {9}270, {9}270, {9}618, {9}7f8, {9}258, {9}348, {9}2f8, {9}0f8, {9}0f8
g025_433.89M_1024k.cu8: {9}0c0, {9}758, {9}7f8, {9}258, {9}270, {9}270, {9}458, {9}7f8, {9}658, {9}528, {9}2f8, {9}0f8, {9}0f8
g026_433.89M_1024k.cu8: {9}0c0, {9}758, {9}7f8, {9}658, {9}7f0, {9}7f0, {9}7f8, {9}7f8, {9}258, {9}258, {9}6f8, {9}0f8, {9}0f8
g027_433.89M_1024k.cu8: {9}0c0, {9}758, {9}7f8, {9}558, {9}7f0, {9}7f0, {9}658, {9}7f8, {9}258, {9}328, {9}5f8, {9}0f8, {9}0f8
g028_433.89M_1024k.cu8: {9}0c0, {9}758, {9}5f8, {9}358, {9}770, {9}770, {9}6d8, {9}7f8, {9}258, {9}3a8, {9}3f8, {9}0f8, {9}378
g029_433.89M_1024k.cu8: {9}0c0, {9}758, {9}3f8, {9}558, {9}470, {9}470, {9}158, {9}7f8, {9}258, {9}728, {9}5f8, {9}0f8, {9}778
g030_433.89M_1024k.cu8: {9}0c0, {9}758, {9}7f8, {9}358, {9}1f0, {9}1f0, {9}4b8, {9}7f8, {9}258, {9}1e8, {9}3f8, {9}0f8, {9}0f8
```

Or [this BitBench](https://triq.net/bitbench#c=%7B9%7D7f8%7B9%7D0c0%7B9%7D0c0%7B9%7D460%7B9%7D0c0%7B9%7D0c0%7B9%7D0c0%7B9%7D0c0%7B9%7D0c0%7B9%7D290%7B9%7D0c0%7B9%7D0f8%7B9%7D440%20%5Bg001%5D&c=%7B9%7D7f8%7B9%7D0c0%7B9%7D0c0%7B9%7D460%7B9%7D0c0%7B9%7D0c0%7B9%7D0c0%7B9%7D0c0%7B9%7D0c0%7B9%7D290%7B9%7D0c0%7B9%7D0f8%7B9%7D440%20%5Bg002%5D&c=%7B9%7D7f8%7B9%7D0c0%7B9%7D0c0%7B9%7D460%7B9%7D0c0%7B9%7D0c0%7B9%7D0c0%7B9%7D0c0%7B9%7D0c0%7B9%7D290%7B9%7D0c0%7B9%7D0f8%7B9%7D440%20%5Bg003%5D&c=%7B9%7D7f8%7B9%7D0c0%7B9%7D0c0%7B9%7D460%7B9%7D0c0%7B9%7D0c0%7B9%7D0c0%7B9%7D0c0%7B9%7D0c0%7B9%7D290%7B9%7D0c0%7B9%7D0f8%7B9%7D440%20%5Bg004%5D&c=%7B9%7D7f8%7B9%7D0c0%7B9%7D0c0%7B9%7D460%7B9%7D0c0%7B9%7D0c0%7B9%7D0c0%7B9%7D0c0%7B9%7D0c0%7B9%7D290%7B9%7D0c0%7B9%7D0f8%7B9%7D440%20%5Bg005%5D&c=%7B9%7D7f8%7B9%7D0c0%7B9%7D0c0%7B9%7D460%7B9%7D0c0%7B9%7D0c0%7B9%7D0c0%7B9%7D0c0%7B9%7D0c0%7B9%7D290%7B9%7D0c0%7B9%7D0f8%7B9%7D440%20%5Bg007%5D&c=%7B9%7D0c0%7B9%7D358%7B9%7D7f8%7B9%7D798%7B9%7D7f8%7B9%7D7f8%7B9%7D7f8%7B9%7D7f8%7B9%7D7f8%7B9%7D7f8%7B9%7D0f8%7B9%7D0f8%7B9%7D0f8%20%5Bg008%5D&c=%7B9%7D0c0%7B9%7D358%7B9%7D7f8%7B9%7D398% [... cut ...] D328%7B9%7D5f8%7B9%7D0f8%7B9%7D0f8%20%5Bg027%5D&c=%7B9%7D0c0%7B9%7D758%7B9%7D5f8%7B9%7D358%7B9%7D770%7B9%7D770%7B9%7D6d8%7B9%7D7f8%7B9%7D258%7B9%7D3a8%7B9%7D3f8%7B9%7D0f8%7B9%7D378%20%5Bg028%5D&c=%7B9%7D0c0%7B9%7D758%7B9%7D3f8%7B9%7D558%7B9%7D470%7B9%7D470%7B9%7D158%7B9%7D7f8%7B9%7D258%7B9%7D728%7B9%7D5f8%7B9%7D0f8%7B9%7D778%20%5Bg029%5D&c=%7B9%7D0c0%7B9%7D758%7B9%7D7f8%7B9%7D358%7B9%7D1f0%7B9%7D1f0%7B9%7D4b8%7B9%7D7f8%7B9%7D258%7B9%7D1e8%7B9%7D3f8%7B9%7D0f8%7B9%7D0f8%20%5Bg030%5D&f=9b%20&cw=4).

zuckschwerdt, 2024-03-18:
The bits seem to be reflected and the fourth byte looks like the temperature: try `1x^8d ` in the BitBench.
Or some guesses like `TYPE?1x^8d 1x^8d 1x^8d TEMP?1x^8d 1x^8d 1x^8d SPEED?1x^8d 1x^8d 1x^8d SPEED?1x^8d 1x^8d FIX?1x^8d ?1x^8d `

ProfBoc75, 2024-03-19:
Hi All,

Based on this [ThermorWeatherRx DG950R arduino code](https://github.com/kayno/ThermorWeatherRx/blob/edc747e3b4aab6420295f2fc1046bb538c56a277/ThermorWeatherRx.pde)

Data layout could be:

    there are 4 known packet types - sync, wind, temp and rain:

    PACKET_TYPE_SYNC:
    ssssssss     pppp
    10000000 11000000 01000100 00000010 00000000 00000000 00000010 00001110 

    PACKET_TYPE_TEMP:
                            72        6        72.6 - 41 (temp offset) = 31.6 degrees celcius
    ssssssss     pppp tttttttt tttttttt
    10000000 11010010 01001000 00000110 01010011 01111111 11111111 11111111

    PACKET_TYPE_WIND:
                      wwwwwwww wwwwwwww              wwww
    ssssssss     pppp ssssssss ssssssss              dddd
    10000000 11010001 00000000 00000000 00100011 00001101 10000001 1111111

    PACKET_TYPE_RAIN:
    ssssssss     pppp rrrrrrrr rrrrrrrr
    10000000 11010011 11001110 10111011 00000000 00100110 01111111 11111111

    s = station id
    p = packet type
    t = outside temp
    ws = wind speed
    wd = wind direction
    r = rain

But I'm not sure of this, since the result with the flex decoder is 13 bytes where from above link, expected are 8 bytes (64 bites), sounds like is not exactly the same or I'm wrong.

ProfBoc75, 2024-03-19:
I tried to understand how to at least get the data from the ThermorWeatherRX project in arduino.

- Preamble is 100 short pulses then the message.
- Sounds to be Manchester encoded:

    // incoming pulses are a valid bit stream, manchester encoded. starting with a zero bit, the next bit will be the same as the 
    // previous bit if there are two short pulses, or the bit will swap if the pulse is long

- Then, depends on the packet type, temp, wind and rain could be decoded.

zuckschwerdt, 2024-03-19:
That description doesn't seem to fit? The preamble is ~53 medium pulses. And I don't see MC anywhere. Strange.

ProfBoc75, 2024-03-19:
Strange, because from Pulsedata we have more pulses at the start of the signal, with 24 x nibble 0, so 24 x 4 = 96 pulses.

[image]

zuckschwerdt, 2024-03-19:
You are right, I miscounted. But the length is really different (medium 1438 µs) compared to the short / long  744 / 2128 µs. And those can't be MC with a pulse/gap ratio of 1 : 3, with MC we'd see 1:1 and 1:2. And then the 8 vs 13 bytes issue you mentioned.

klohner, 2024-03-19:
I decoded with:

`rtl_433 -R 0 -X n=name,m=OOK_PCM,s=730,l=730,t=10,r=8000,PREAMBLE=3300 *.cu8` and got this data:

[BitBench](https://triq.net/bitbench#c=%7B549%7De8888888803bbba23bb80eeee88eee03a3bba23b80eeee88eee03bbba23bb80eeee88eee03bbba23bb80eeee88eee03ba3a3ba380eeee88eee03bbba222200e8eee8eee000%20%5Bg001%5D&c=%7B549%7De8888888803bbba23bb80eeee88eee03a3bba23b80eeee88eee03bbba23bb80eeee88eee03bbba23bb80eeee88eee03ba3a3ba380eeee88eee03bbba222200e8eee8eee000%20%5Bg002%5D&c=%7B549%7De8888888803bbba23bb80eeee88eee03a3bba23b80eeee88eee03bbba23bb80eeee88eee03bbba23bb80eeee88eee03ba3a3ba380eeee88eee03bbba222200e8eee8eee000%20%5Bg003%5D&c=%7B549%7De8888888803bbba23bb80eeee88eee03a3bba23b80eeee88eee03bbba23bb80eeee88eee03bbba23bb80eeee88eee03ba3a3ba380eeee88eee03bbba222200e8eee8eee000%20%5Bg004%5D&c=%7B549%7De8888888803bbba23bb80eeee88eee03a3bba23b80eeee88eee03bbba23bb80eeee88eee03bbba23bb80eeee88eee03ba3a3ba380eeee88eee03bbba222200e8eee8eee000%20%5Bg005%5D&c=%7B549%7De8888888803bbba23bb80eeee88eee03a3bba23b80eeee88eee03bbba23bb80eeee88eee03bbba23bb80eeee88eee03ba3a3ba380eeee88eee03bbba222200e8eee8eee000%20%5Bg007%5D&c=%7B547%7Deeee88eee03ba23a3a200e8888888803a2223ba200e8888888803a222222200e8888888803a222222200e8888888803a222222200eeee8888803bbba222200eeee8888800%20%5Bg008%5D&c=%7B547%7Deeee88eee03ba23a3a200e8888888803ba223ba200e8e8e888e03a3a3a22380e8888888803a222222200e8eee8e8803a3bba3a200e888e888803bbba222200eeee8888800%20%5Bg009%5D&c=%7B547%7Deeee88eee03ba23a3a200e8888888803ba223ba200e8e8e888e03a3a3a22380e8888888803a222222200ee8ee8e8803ba3ba3a200e888e888803bbba222200eeee8888800%20%5Bg010%5D&c=%7B547%7Deeee88eee03ba23a3a200ee888888803ba223ba200e8e8e888e03a3a3a22380e8888888803a222222200ee8ee8e8803ba3ba3a200e888e888803bbba222200e888e888800%20%5Bg011%5D&c=%7B547%7Deeee88eee03ba23a3a200ee888888803a3a23ba200e8e8e888e03a3a3a22380e8888888803a222222200ee8ee8e8803ba3ba3a200ee88e888803bbba222200e888e888800%20%5Bg012%5D&c=%7B547%7Deeee88eee03ba23a3a200e8e88888803ba223ba200e8e8e888e03a3a3a22380e8888888803a222222200ee8ee8e8803ba3ba3a200e888e888803bbba222200ee88e888800%20%5Bg013%5D&c=%7B547%7Deeee88eee03ba23a3a200eee88888803a3a23ba200e8e8e888e03a3a3a22380e8888888803a222222200ee8ee8e8803ba3ba3a200ee88e888803bbba222200e8e8e888800%20%5Bg014%5D&c=%7B547%7Deeee88eee03ba23a3a200e8888888803a3a23ba200e8e8e888e03a3a3a22380e8888888803a222222200ee8ee8e8803ba3ba3a200ee88e888803bbba222200eeee8888800%20%5Bg015%5D&c=%7B547%7Deeee88eee03ba3a23a200e8888888803bba3a3a200e8e8e888e03a3a3a22380e8888888803a222222200ee8ee8e8803ba3ba3a200e8ee8888803bbba222200eeee8888800%20%5Bg016%5D&c=%7B547%7Deeee88eee03a23a23a200e8888888803ba3ba3a200e8e8e888e03a3a3a22380e8888888803a222222200ee8ee8e8803ba3ba3a200ee88e888803bbba222200eeee8888800%20%5Bg017%5D&c=%7B547%7Deeee88eee03a3ba23a200e8888888803a3bba3a200e8e8e888e03a3a3a22380e8888888803a222222200ee8ee8e8803ba3ba3a200e888e888803bbba222200eeee8888800%20%5Bg018%5D&c=%7B547%7Deeee88eee03bbba23a200e8888888803a2223ba200ee8ee888e03ba3ba22380e8888888803a222222200ee8ee8e8803ba3ba3a200ee88e888803bbba222200eeee8888800%20%5Bg019%5D&c=%7B547%7Deeee88eee03bbba23a200e8888888803bba3a3a200eee88888e03bba2222380e8888888803a222222200ee8ee8e8803ba3ba3a200e88e8888803bbba222200eeee8888800%20%5Bg020%5D&c=%7B547%7Deeee88eee03bbba23a200e8888888803bbba23a200eeee8888e03bbba222380e8888888803a222222200ee8ee8e8803ba3ba3a200e8888888803bbba222200eeee8888800%20%5Bg021%5D&c=%7B547%7Deeee88eee03a223a3a200e8888888803ba223ba200ee8ee888e03ba3ba22380e8888888803a222222200ee8ee8e8803ba3ba3a200ee88e888803bbba222200eeee8888800%20%5Bg022%5D&c=%7B547%7Deeee88eee03a223a3a200e8888888803a2223ba200ee8ee888e03ba3ba22380e8ee8e88803a222222200ee8ee8e8803bba2223a00e888e888803bbba222200eeee8888800%20%5Bg023%5D&c=%7B547%7Deeee88eee03a223a3a200e8888888803ba3ba3a200ee8ee888e03ba3ba22380e88eeee8803a222222200ee8ee8e8803ba23a3ba00ee8e8888803bbba222200eeee8888800%20%5Bg024%5D&c=%7B547%7Deeee88eee03a223a3a200e8888888803ba3ba3a200ee8ee888e03ba3ba22380e8eee8e8803a222222200e88ee8e8803a3a3ba3a00ee8e8888803bbba222200eeee8888800%20%5Bg025%5D&c=%7B547%7Deeee88eee03a223a3a200e8888888803a23ba3a200e8888888e03a222222380e8888888803a222222200ee8ee8e8803ba3ba3a200e88e8888803bbba222200eeee8888800%20%5Bg026%5D&c=%7B547%7Deeee88eee03a223a3a200e8888888803a3a3a3a200e8888888e03a222222380e88ee8e8803a222222200ee8ee8e8803ba23ba3a00e8e88888803bbba222200eeee8888800%20%5Bg027%5D&c=%7B547%7Deeee88eee03a223a3a200e8e88888803ba23a3a200e888e888e03a223a22380e88e88e8803a222222200ee8ee8e8803ba223a3a00ee888888803bbba222200ee88e888800%20%5Bg028%5D&c=%7B547%7Deeee88eee03a223a3a200ee888888803a3a3a3a200e8eee888e03a3bba22380eee8e8e8803a222222200ee8ee8e8803a223ba3a00e8e88888803bbba222200e888e888800%20%5Bg029%5D&c=%7B547%7Deeee88eee03a223a3a200e8888888803ba23a3a200eee88888e03bba2222380e8ee8e88803a222222200ee8ee8e8803bba2223a00ee888888803bbba222200eeee8888800%20%5Bg030%5D&f=36h6x%2036h6x%2036h6x%2036h6x%2036h6x%2036h6x%2036h6x%2036h6x%2036h6x%2036h6x%2036h6x%2036h6x%2036h6x%2032x&cw=4)

There seems to be 13 sets of 9 4-symbol-wide PWM data bits separated by 6-symbol wide gaps.  Throwing away those gaps between the 9 bits of logical data and converting logical bits, I get:

[BitBench](https://triq.net/bitbench#c=y011111111000011000000011000010001100000011000000011000000011000000011000000011000001010010000011000000011111010001000%20%5Bg001%5D&c=y011111111000011000000011000010001100000011000000011000000011000000011000000011000001010010000011000000011111010001000%20%5Bg002%5D&c=y0111111110000110000000110000 [... cut ...] 1000011111000011111%20%5Bg027%5D&c=y000011000011101011010111111001101011011101110011101110011011011011111111001001011001110101001111111000011111001101111%20%5Bg028%5D&c=y000011000011101011001111111010101011010001110010001110000101011011111111001001011011100101010111111000011111011101111%20%5Bg029%5D&c=y000011000011101011011111111001101011000111110000111110010010111011111111001001011000111101001111111000011111000011111%20%5Bg030%5D&f=xhh%20xhhxhh%20xhhxhh%20xhhxhh%20xhhxhh%20xhhxhh%20xhhxhh&cw=4)

zuckschwerdt, 2024-03-19:
That matches the BitBench from PWM data I posted above. Those 13 bytes seem like plausible data. Now we need to find the values.

NicolasGagne, 2024-03-19:
I have slightly change the format string from Klohner and I get the proper Station ID as observer on the receiving unit: 231

x~d xhhxhh xhhxhh xhhxhh xhhxhh xhhxhh xhhxhh

ProfBoc75, 2024-03-20:
At least I found the Wind direction, some data are repeated in the message.
I guessed the positions of Temp, Wind Speed and Rain, but not yet found the conversion to get figures.

[bitbench](https://triq.net/bitbench#c=%7B213%7D0000000000000000000000000c1adfef37fbfdfeff7fbfc3e1f0f8%20%5B%20g008_433.89M_1024k.cu8%20NORMAL%20Unknow%20Unknow%2019.3%200%5D&c=%7B213%7D0000000000000000000000000c1adfe73572b9feff45a2dde1f0f8%20%5B%20g009_433.89M_1024k.cu8%20NORMAL%2022%200%2019.2%200%20%5D&c=%7B213%7D0000000000000000000000000c1adfe73572b9feff2592dde1f0f8%20%5B%20g010_433.89M_1024k.cu8%20NORMAL%2022%200%2019.2%200%20%5D&c=%7B213%7D0000000000000000000000000c1acfe73572b9feff2592dde1f778%20%5B%20g011_433.89M_1024k.cu8%20NORMAL%2022%200%2019.2%200.1%20%5D&c=%7B213%7D0000000000000000000000000c1acfeb3572b9feff2592cde1f778%20%5B%20g012_433.89M_1024k.cu8%20NORMAL%2022%200%2019.1%200.1%20%5D&c=%7B213%7D0000000000000000000000000c1ad7e73572b9feff2592dde1f378%20%5B%20g013_433.89M_1024k.cu8%20NORMAL%2022%200%2019.2%200.2%20%5D&c=%7B213%7D0000000000000000000000000c1ac7eb3572b9feff2592cde1f578%20%5B%20g014_433.89M_1024k.cu8%20NORMAL%2022%200%2019.1%200.4%20%5D&c=%7B213%7D0000000000000000000000000c1adfab3572b9feff2592cde1f0f8%20%5B%20g015_433.89M_1024k.cu8%20NORMAL%2022%200%2019.1%200.4%20%5D&c=%7B213%7D0000000000000000000000000c16dfe2b572b9feff2592d3e1f0f8%20%5B%20g016_433.89M_1024k.cu8%20NORMAL%2022%200%2023.4%200.4%20%5D&c=%7B213%7D0000000000000000000000000c36dfe4b572b9feff2592cde1f0f8%20%5B%20g017_433.89M_1024k.cu8%20NORMAL%2022%200%2024.1%200.4%20%5D&c=%7B213%7D0000000000000000000000000c26dfe8b572b9feff2592dde1f0f8%20%5B%20g018_433.89M_1024k.cu8%20NORMAL%2022%200%2022.2%200.4%20%5D&c=%7B213%7D0000000000000000000000000c06dfef327139feff2592cde1f0f8%20%5B%20g019_433.89M_1024k.cu8%20NORMAL%20135%200%2021.1%200.4%20%5D&c=%7B213%7D0000000000000000000000000c06dfe2b1f0f9feff2592dbe1f0f8%20%5B%20g020_433.89M_1024k.cu8%20NORMAL%20202%200%2021.6%200.4%20%5D&c=%7B213%7D0000000000000000000000000c06dfe1b0f079feff2592dfe1f0f8%20%5B%20g021_433.89M_1024k.cu8%20NORMAL%20225%200%2022%200.4%20%5D&c=%7B213%7D0000000000000000000000000c3adfe7327139feff2592cde1f0f8%20%5B%20g022_433.89M_1024k.cu8%20NORMAL%20135%200%2020.1%200.4%20%5D&c=%7B213%7D0000000000000000000000000c3adfef3271392eff258f5de1f0f8%20%5B%20g023_433.89M_1024k.cu8%20NORMAL%20135%203.6%2020.2%200.4%20%5D&c=%7B213%7D0000000000000000000000000c3adfe4b2713986ff259a4be1f0f8%20%5B%20g024_433.89M_1024k.cu8%20NORMAL%20135%208.5%2020.5%200.4%20%5D&c=%7B213%7D0000000000000000000000000c3adfe4b2713916ff65a94be1f0f8%20%5B%20g025_433.89M_1024k.cu8%20NORMAL%20135%206.5%2020.5%200.4%20%5D&c=%7B213%7D0000000000000000000000000c3adfecb7f3f9feff2592dbe1f0f8%20%5B%20g026_433.89M_1024k.cu8%20NORMAL%20292%200%2020.5%200.4%20%5D&c=%7B213%7D0000000000000000000000000c3adfeab7f3f996ff259957e1f0f8%20%5B%20g027_433.89M_1024k.cu8%20NORMAL%20292%206.5%2020.8%200.4%20%5D&c=%7B213%7D0000000000000000000000000c3ad7e6b773b9b6ff259d4fe1f378%20%5B%20g028_433.89M_1024k.cu8%20NORMAL%2090%205.4%2020.9%200.5%20%5D&c=%7B213%7D0000000000000000000000000c3acfeab4723856ff25b957e1f778%20%5B%20g029_433.89M_1024k.cu8%20NORMAL%2045%206.2%2020.8%200.6%20%5D&c=%7B213%7D0000000000000000000000000c3adfe6b1f0f92eff258f4fe1f0f8%20%5B%20g030_433.89M_1024k.cu8%20NORMAL%20202%203.6%2020.9%200.6%20%5D&f=x%20ID%3A~8d%20x%20%3F%3A~8d%20x%20%3F%3A~8d%20x%20T_DEC%3F%3A%5E8d%20x%20WDIR%3A4d%20FLAG%3A~4h%20x%20WDIR%3A4d%20FLAG%3A~4h%20x%20WSPEED%3A8d%20x%20%3F%3A~8d%20x%20%3F%3A~8d%20x%20%3F%3A~8d%20x%20TEMP%3F%3A~4d%20F%3A4d%20x%20%3F%3A~8d%20x%20RAIN%3A%20~%5E4d%20%3F%3A4h&a=Preamble&m=000000000000000000000000&cw=4)

Win Direction table: (based on arduino code)

Wind Dir ° |  In Sample  | Wind Dir Name | Map Dec | Mac Hexa
-- | -- | -- | -- | --
0 |   | N | 11 | B
22 | x | NNE | 10 | A
45 | x | NE | 8 | 8
67 |   | ENE | 12 | C
90 | x | E | 14 | E
112 |   | ESE | 6 | 6
135 |   | SE | 4 | 4
157 |   | SSE | 0 | 0
180 |   | S | 2 | 2
202 | x | SSW | 3 | 3
225 | x | SW | 1 | 1
247 |   | WSW | 5 | 5
270 |   | W | 7 | 7
292 | x | WNW | 15 | F
315 |   | NW | 13 | D
337 |   | NNW | 9 | 9

ProfBoc75, 2024-03-20:
> For Rain In my data recording I have noted how many click of the pendulum I done, and it match with a base of 7. it just transmit the number of click of the pendulum form the last transmission. Example 9 - 7 = 2 pendulum move. The calculation must be made in the receiver unit for the amount of rain fallen.

Yes, this is also my understanding from the arduino code where it could be a problem is you miss a transmit.

ProfBoc75, 2024-03-21:
Little update, I found the whole temp, at the second position after the ID, it's a 8 bit , reflect and -195 offset, in ° C.

This is align with a max temp of 60 ° C from the user guide, if a raw value = 255 --> - 195 = 60 ° C
And min raw value = 155 --> -195 = -40 ° C

NicolasGagne, 2024-03-22:
Quick update for position
https://triq.net/bitbench#c=%7B213%7D0000000000000000000000000c1adfef37fbfdfeff7fbfc3e1f0f8%20%5B%20g008_433.89M_1024k.cu8%20NORMAL%20Unknow%20Unknow%2019.3%200%5D&c=%7B213%7D0000000000000000000000000c1adfe73572b9feff45a2dde1f0f8%20%5B%20g009_433.89M_1024k.cu8%20NORMAL%2022%200%2019.2%200%20%5D&c=%7B213%7D0000000000000000000000000c1adfe73572b9feff2592dde1f0f8%20%5B%20g010_433.89M_1024k.cu8%20NORMAL%2022%200%2019.2%200%20%5D&c=%7B213%7D0000000000000000000000000c1acfe73572b9feff2592dde1f778%20%5B%20g011_433.89M_1024k.cu8%20NORMAL%2022%200%2019.2%200.1%20%5D&c=%7B213%7D0000000000000000000000000c1acfeb3572b9feff2592cde1f778%20%5B%20g012_433.89M_1024k.cu8%20NORMAL%2022%200%2019.1%200.1%20%5D&c=%7B213%7D0000000000000000000000000c1ad7e73572b9feff2592dde1f378%20%5B%20g013_433.89M_1024k.cu8%20NORMAL%2022%200%2019.2%200.2%20%5D&c=%7B213%7D0000000000000000000000000c1ac7eb3572b9feff2592cde1f578%20%5B%20g014_433.89M_1024k.cu8%20NORMAL%2022%200%2019.1%200.4%20%5D&c=%7B213%7D0000000000000000000000000c1adfab3572b9feff2592cde1f0f8%20%5B%20g015_433.89M_1024k.cu8%20NORMAL%2022%200%2019.1%200.4%20%5D&c=%7B213%7D0000000000000000000000000c16dfe2b572b9feff2592d3e1f0f8%20%5B%20g016_433.89M_1024k.cu8%20NORMAL%2022%200%2023.4%200.4%20%5D&c=%7B213%7D0000000000000000000000000c36dfe4b572b9feff2592cde1f0f8%20%5B%20g017_433.89M_1024k.cu8%20NORMAL%2022%200%2024.1%200.4%20%5D&c=%7B213%7D0000000000000000000000000c26dfe8b572b9feff2592dde1f0f8%20%5B%20g018_433.89M_1024k.cu8%20NORMAL%2022%200%2022.2%200.4%20%5D&c=%7B213%7D0000000000000000000000000c06dfef327139feff2592cde1f0f8%20%5B%20g019_433.89M_1024k.cu8%20NORMAL%20135%200%2021.1%200.4%20%5D&c=%7B213%7D0000000000000000000000000c06dfe2b1f0f9feff2592dbe1f0f8%20%5B%20g020_433.89M_1024k.cu8%20NORMAL%20202%200%2021.6%200.4%20%5D&c=%7B213%7D0000000000000000000000000c06dfe1b0f079feff2592dfe1f0f8%20%5B%20g021_433.89M_1024k.cu8%20NORMAL%20225%200%2022%200.4%20%5D&c=%7B213%7D0000000000000000000000000c3adfe7327139feff2592cde1f0f8%20%5B%20g022_433.89M_1024k.cu8%20NORMAL%20135%200%2020.1%200.4%20%5D&c=%7B213%7D0000000000000000000000000c3adfef3271392eff258f5de1f0f8%20%5B%20g023_433.89M_1024k.cu8%20NORMAL%20135%203.6%2020.2%200.4%20%5D&c=%7B213%7D0000000000000000000000000c3adfe4b2713986ff259a4be1f0f8%20%5B%20g024_433.89M_1024k.cu8%20NORMAL%20135%208.5%2020.5%200.4%20%5D&c=%7B213%7D0000000000000000000000000c3adfe4b2713916ff65a94be1f0f8%20%5B%20g025_433.89M_1024k.cu8%20NORMAL%20135%206.5%2020.5%200.4%20%5D&c=%7B213%7D0000000000000000000000000c3adfecb7f3f9feff2592dbe1f0f8%20%5B%20g026_433.89M_1024k.cu8%20NORMAL%20292%200%2020.5%200.4%20%5D&c=%7B213%7D0000000000000000000000000c3adfeab7f3f996ff259957e1f0f8%20%5B%20g027_433.89M_1024k.cu8%20NORMAL%20292%206.5%2020.8%200.4%20%5D&c=%7B213%7D0000000000000000000000000c3ad7e6b773b9b6ff259d4fe1f378%20%5B%20g028_433.89M_1024k.cu8%20NORMAL%2090%205.4%2020.9%200.5%20%5D&c=%7B213%7D0000000000000000000000000c3acfeab4723856ff25b957e1f778%20%5B%20g029_433.89M_1024k.cu8%20NORMAL%2045%206.2%2020.8%200.6%20%5D&c=%7B213%7D0000000000000000000000000c3adfe6b1f0f92eff258f4fe1f0f8%20%5B%20g030_433.89M_1024k.cu8%20NORMAL%20202%203.6%2020.9%200.6%20%5D&f=x%20ID%3A~8d%20x%20%3F%3A~8d%20x%20%3F%3A~8d%20x%20T_INT%3F%3A%5E8d%20x%20WDIR%3A4d%20FLAG%3A~4h%20x%20WDIR%3A4d%20FLAG%3A~4h%20x%20%3F%3A8d%20x%20%3F%3A~8d%20x%20T_DEC%3A~8d%20x%20%3F%3A~8d%20x%20%3F%3A~4d%20F%3A4d%20x%20%3F%3A~8d%20x%20RAIN%3A%20~%5E4d%20%3F%3A4h&z=1&a=Preamble&m=000000000000000000000000&cw=4

Missing: Wind speed

Position:
- 1    - ID
- 2    - Unknow
- 3    - Unknow
- 4    - Temperature Integer - 195 offset
- 5    - Wind Direction (table above)
- 6    - Flag
- 7    - Wind Direction (table above)
- 8    - Flag
- 9    - Unknow
- 10  - Unknow
- 11  - Temperature Decimal  -245 offset
- 12  - Unknow
- 13  - Unknow
- 14  - Flag
- 15  - Unknow
- 16  - Rain -7 offset
- 17  - Unknow

ProfBoc75, 2024-03-22:
> Quick update for position [bitbench](https://triq.net/bitbench#c=%7B213%7D0000000000000000000000000c1adfef37fbfdfeff7fbfc3e1f0f8%20%5B%20g008_433.89M_1024k.cu8%20NORMAL%20Unknow%20Unknow%2019.3%200%5D&c=%7B213%7D0000000000000000000000000c1adfe73572b9feff45a2dde1f0f8%20%5B%20g009_433.89M_1024k.cu8%20NORMAL%2022%200%2019.2%200%20%5D&c=%7B213%7D0000000000000000000000000c1adfe73572b9feff2592dde1f0f8%20%5B%20g010_433.89M_1024k.cu8%20NORMAL%2022%200%2019.2%200%20%5D&c=%7B213%7D0000000000000000000000000c1acfe73572b9feff2592dde1f778%20%5B%20g011_433.89M_1024k.cu8%20NORMAL%2022%200%2019.2%200.1%20%5D&c=%7B213%7D0000000000000000000000000c1acfeb3572b9feff2592cde1f778%20%5B%20g012_433.89M_1024k.cu8%20NORMAL%2022%200%2019.1%200.1%20%5D&c=%7B213%7D0000000000000000000000000c1ad7e73572b9feff2592dde1f378%20%5B%20g013_433.89M_1024k.cu8%20NORMAL%2022%200%2019.2%200.2%20%5D&c=%7B213%7D0000000000000000000000000c1ac7eb3572b9feff2592cde1f578%20%5B%20g014_433.89M_1024k.cu8%20NORMAL%2022%200%2019.1%200.4%20%5D&c=%7B213%7D0000000000000000000000000c1adfab3572b9feff2592cde1f0f8%20%5B%20g015_433.89M_1024k.cu8%20NORMAL%2022%200%2019.1%200.4%20%5D&c=%7B213%7D0000000000000000000000000c16dfe2b572b9feff2592d3e1f0f8%20%5B%20g016_433.89M_1024k.cu8%20NORMAL%2022%200%2023.4%200.4%20%5D&c=%7B213%7D0000000000000000000000000c36dfe4b572b9feff2592cde1f0f8%20%5B%20g017_433.89M_1024k.cu8%20NORMAL%2022%200%2024.1%200.4%20%5D&c=%7B213%7D0000000000000000000000000c26dfe8b572b9feff2592dde1f0f8%20%5B%20g018_433.89M_1024k.cu8%20NORMAL%2022%200%2022.2%200.4%20%5D&c=%7B213%7D0000000000000000000000000c06dfef327139feff2592cde1f0f8%20%5B%20g019_433.89M_1024k.cu8%20NORMAL%20135%200%2021.1%200.4%20%5D&c=%7B213%7D0000000000000000000000000c06dfe2b1f0f9feff2592dbe1f0f8%20%5B%20g020_433.89M_1024k.cu8%20NORMAL%20202%200%2021.6%200.4%20%5D&c=%7B213%7D0000000000000000000000000c06dfe1b0f079feff2592dfe1f0f8%20%5B%20g021_433.89M_1024k.cu8%20NORMAL%20225%200%2022%200.4%20%5D&c=%7B213%7D0000000000000000000000000c3adfe7327139feff2592cde1f0f8%20%5B%20g022_433.89M_1024k.cu8%20NORMAL%20135%200%2020.1%200.4%20%5D&c=%7B213%7D0000000000000000000000000c3adfef3271392eff258f5de1f0f8%20%5B%20g023_433.89M_1024k.cu8%20NORMAL%20135%203.6%2020.2%200.4%20%5D&c=%7B213%7D0000000000000000000000000c3adfe4b2713986ff259a4be1f0f8%20%5B%20g024_433.89M_1024k.cu8%20NORMAL%20135%208.5%2020.5%200.4%20%5D&c=%7B213%7D0000000000000000000000000c3adfe4b2713916ff65a94be1f0f8%20%5B%20g025_433.89M_1024k.cu8%20NORMAL%20135%206.5%2020.5%200.4%20%5D&c=%7B213%7D0000000000000000000000000c3adfecb7f3f9feff2592dbe1f0f8%20%5B%20g026_433.89M_1024k.cu8%20NORMAL%20292%200%2020.5%200.4%20%5D&c=%7B213%7D0000000000000000000000000c3adfeab7f3f996ff259957e1f0f8%20%5B%20g027_433.89M_1024k.cu8%20NORMAL%20292%206.5%2020.8%200.4%20%5D&c=%7B213%7D0000000000000000000000000c3ad7e6b773b9b6ff259d4fe1f378%20%5B%20g028_433.89M_1024k.cu8%20NORMAL%2090%205.4%2020.9%200.5%20%5D&c=%7B213%7D0000000000000000000000000c3acfeab4723856ff25b957e1f778%20%5B%20g029_433.89M_1024k.cu8%20NORMAL%2045%206.2%2020.8%200.6%20%5D&c=%7B213%7D0000000000000000000000000c3adfe6b1f0f92eff258f4fe1f0f8%20%5B%20g030_433.89M_1024k.cu8%20NORMAL%20202%203.6%2020.9%200.6%20%5D&f=x%20ID%3A~8d%20x%20%3F%3A~8d%20x%20%3F%3A~8d%20x%20T_INT%3F%3A%5E8d%20x%20WDIR%3A4d%20FLAG%3A~4h%20x%20WDIR%3A4d%20FLAG%3A~4h%20x%20%3F%3A8d%20x%20%3F%3A~8d%20x%20T_DEC%3A~8d%20x%20%3F%3A~8d%20x%20%3F%3A~4d%20F%3A4d%20x%20%3F%3A~8d%20x%20RAIN%3A%20~%5E4d%20%3F%3A4h&z=1&a=Preamble&m=000000000000000000000000&cw=4)
> 
> Missing: Wind speed
> 
> Position:
> 
>     * 1    - ID
> 
>     * 2    - Unknow
> 
>     * 3    - Unknow
> 
>     * 4    - Temperature Integer - 195 offset
> 
>     * 5    - Wind Direction (table above)
> 
>     * 6    - Flag
> 
>     * 7    - Wind Direction (table above)
> 
>     * 8    - Flag
> 
>     * 9    - Unknow
> 
>     * 10  - Unknow
> 
>     * 11  - Temperature Decimal  -245 offset
> 
>     * 12  - Unknow
> 
>     * 13  - Unknow
> 
>     * 14  - Flag
> 
>     * 15  - Unknow
> 
>     * 16  - Rain -7 offset
> 
>     * 17  - Unknow

Not exactly that since we have 13 bytes in total, let me few minutes to update the data layout since I have other findings.

ProfBoc75, 2024-03-22:
Here my [bitbench](https://triq.net/bitbench#c=%7B213%7D0000000000000000000000000c1adfe73572b9feff45a2dde1f0f8%20%5B%20g009_433.89M_1024k.cu8%20NORMAL%2022%200%2019.2%200%20%5D&c=%7B213%7D0000000000000000000000000c1adfe73572b9feff2592dde1f0f8%20%5B%20g010_433.89M_1024k.cu8%20NORMAL%2022%200%2019.2%200%20%5D&c=%7B213%7D0000000000000000000000000c1acfe73572b9feff2592dde1f778%20%5B%20g011_433.89M_1024k.cu8%20NORMAL%2022%200%2019.2%200.1%20%5D&c=%7B213%7D0000000000000000000000000c1acfeb3572b9feff2592cde1f778%20%5B%20g012_433.89M_1024k.cu8%20NORMAL%2022%200%2019.1%200.1%20%5D&c=%7B213%7D0000000000000000000000000c1ad7e73572b9feff2592dde1f378%20%5B%20g013_433.89M_1024k.cu8%20NORMAL%2022%200%2019.2%200.2%20%5D&c=%7B213%7D0000000000000000000000000c1ac7eb3572b9feff2592cde1f578%20%5B%20g014_433.89M_1024k.cu8%20NORMAL%2022%200%2019.1%200.4%20%5D&c=%7B213%7D0000000000000000000000000c1adfab3572b9feff2592cde1f0f8%20%5B%20g015_433.89M_1024k.cu8%20NORMAL%2022%200%2019.1%200.4%20%5D&c=%7B213%7D0000000000000000000000000c16dfe2b572b9feff2592d3e1f0f8%20%5B%20g016_433.89M_1024k.cu8%20NORMAL%2022%200%2023.4%200.4%20%5D&c=%7B213%7D0000000000000000000000000c36dfe4b572b9feff2592cde1f0f8%20%5B%20g017_433.89M_1024k.cu8%20NORMAL%2022%200%2024.1%200.4%20%5D&c=%7B213%7D0000000000000000000000000c26dfe8b572b9feff2592dde1f0f8%20%5B%20g018_433.89M_1024k.cu8%20NORMAL%2022%200%2022.2%200.4%20%5D&c=%7B213%7D0000000000000000000000000c06dfef327139feff2592cde1f0f8%20%5B%20g019_433.89M_1024k.cu8%20NORMAL%20135%200%2021.1%200.4%20%5D&c=%7B213%7D0000000000000000000000000c3adfe7327139feff2592cde1f0f8%20%5B%20g022_433.89M_1024k.cu8%20NORMAL%20135%200%2020.1%200.4%20%5D&c=%7B213%7D0000000000000000000000000c06dfe2b1f0f9feff2592dbe1f0f8%20%5B%20g020_433.89M_1024k.cu8%20NORMAL%20202%200%2021.6%200.4%20%5D&c=%7B213%7D0000000000000000000000000c06dfe1b0f079feff2592dfe1f0f8%20%5B%20g021_433.89M_1024k.cu8%20NORMAL%20225%200%2022%200.4%20%5D&c=%7B213%7D0000000000000000000000000c3adfecb7f3f9feff2592dbe1f0f8%20%5B%20g026_433.89M_1024k.cu8%20NORMAL%20292%200%2020.5%200.4%20%5D&c=%7B213%7D0000000000000000000000000c3adfef3271392eff258f5de1f0f8%20%5B%20g023_433.89M_1024k.cu8%20NORMAL%20135%203.6%2020.2%200.4%20%5D&c=%7B213%7D0000000000000000000000000c3adfe6b1f0f92eff258f4fe1f0f8%20%5B%20g030_433.89M_1024k.cu8%20NORMAL%20202%203.6%2020.9%200.6%20%5D&c=%7B213%7D0000000000000000000000000c3ad7e6b773b9b6ff259d4fe1f378%20%5B%20g028_433.89M_1024k.cu8%20NORMAL%2090%205.4%2020.9%200.5%20%5D&c=%7B213%7D0000000000000000000000000c3acfeab4723856ff25b957e1f778%20%5B%20g029_433.89M_1024k.cu8%20NORMAL%2045%206.2%2020.8%200.6%20%5D&c=%7B213%7D0000000000000000000000000c3adfe4b2713916ff65a94be1f0f8%20%5B%20g025_433.89M_1024k.cu8%20NORMAL%20135%206.5%2020.5%200.4%20%5D&c=%7B213%7D0000000000000000000000000c3adfeab7f3f996ff259957e1f0f8%20%5B%20g027_433.89M_1024k.cu8%20NORMAL%20292%206.5%2020.8%200.4%20%5D&c=%7B213%7D0000000000000000000000000c3adfe4b2713986ff259a4be1f0f8%20%5B%20g024_433.89M_1024k.cu8%20NORMAL%20135%208.5%2020.5%200.4%20%5D&f=x%20ID%3A%20~8d%20x%20TEMP_INT%20%5E8d%20x%20RAIN_RATE%20~%5E4d%20%3F%20~4d%20x%20TEMP_CHILL%3F%20%5E8d%20x%20WDIR%204d%20F%20~4h%20x%20WDIR%204d%20F%20~4h%20x%20A%20%5E~8d%20x%20%3F~%5E8d%20x%20B%20~%5E8d%20x%20%3D%20A%2BB%20~%5E8d%20x%20TEMP_DEC%20%5E8d%20x%20%3F8d%20x%20RAIN_RATE%20~%5E4d%20%3F4h%0A&a=Preamble&m=000000000000000000000000&cw=4)

```
Byte position             0          1           2          3            4          5           6          7          8          9         10         11          12
bit position     0 12345678 0 12345678 0 1234 5678 0 12345678  0 1234 5678 0 1234 5678 0 12345678 0 12345678 0 12345678 0 12345678 0 12345678 0 12345678 0 1234 5678
Data             X IIIIIIII X TEMP_INT X RT_2 ???? X TEMPCHILL X WDIR FLAG X WDIR FLAG X AAAAAAAA X ???????? X BBBBBBBB X CCCCCCCC X TEMP_DEC X ???????? X RT12 ????

Except ID and WDIR are inversed, all other nibbles or bytes are reflected

-I:{8}         ID
-TEMP_INT:{8}  temperature interger part, offset 195, °C
-TEMP_DEC:{8}  temperature decimal part, offset 245, scale 10
-RT_2:{4}      rain rate in 0.1 mm
-RT12:{4}      rain rate in 0.1 mm offset 7
-TEMPCHILL:{8} may be temperature chill, to be confirmed, need more capture / sample to confirm the decode formula
-WDIR:{4}      wind direction, map table to be used.
-FLAG:{4}      if 1  = valid win dir, if 0 wind dir = unknown.
-A B C:{8}     values related to Wind Speed, C = A + B, not yet decoded.
-?:{8} or {4}  unknown values, fixed.
-X:{1}         Bit start 0 is ignore

```

I tried to calculate the wind speed in km/h, with ax+b, where `a = 0.129` and `b = -5` but not very good values
I converted your Wind speed values in m/s and mp/h to compare.

A | B | C | A+B | km/h | m/s | mp/h | Calc
-- | -- | -- | -- | -- | -- | -- | --
22 | 45 | 67 | 67 | 3.6 | 1 | 2.24 | 3.576
60 | 45 | 105 | 105 | 8.5 | 2.4 | 5.28 | 8.44
46 | 44 | 90 | 90 | 6.5 | 1.8 | 4.04 | 6.52
44 | 45 | 89 | 89 | 6.5 | 1.8 | 4.04 | 6.392
36 | 45 | 81 | 81 | 5.4 | 1.5 | 3.36 | 5.368
43 | 45 | 88 | 88 | 6.2 | 1.7 | 3.85 | 6.264
22 | 45 | 67 | 67 | 3.6 | 1 | 2.24 | 3.576

ProfBoc75, 2024-03-22:
I will start to draft a decoder with current information, @NicolasGagne , if you can get more samples please, don't need the cu8 just the codes with weather station information, temp, wind speed, wind dir, temp chill, rain .....

try with:
`rtl_433 -f 433.89M -R 0 -X 'n=thermor,m=OOK_PWM,s=750,l=2128,y=1438,g=3000,r=8000,get=byte:@1:{8}:%x'`

NicolasGagne, 2024-03-23:
I had to modified the command for the following: 
<code class="notranslate">rtl_433 -f 433.89M -s 1024k -R 0 -X 'n=thermor,m=OOK_PWM,s=750,l=1913,y=1485,g=3000,r=7492,get=byte:@1:{8}:%x'
</code>

I get the follwing result, 
codes     : {9}5e0, {9}358, {9}7f8, {9}598, {9}3f0, {9}3f0, {9}7f8, {9}7f8, {9}258, {9}258, {9}378, {9}0f8, {9}0f8

Is that what you are looking for? 
(I will recode the data tomorrow)

ProfBoc75, 2024-03-23:
> I had to modified the command for the following: `rtl_433 -f 433.89M -s 1024k -R 0 -X 'n=thermor,m=OOK_PWM,s=750,l=1913,y=1485,g=3000,r=7492,get=byte:@1:{8}:%x' `
> 
> I get the follwing result, codes : {9}5e0, {9}358, {9}7f8, {9}598, {9}3f0, {9}3f0, {9}7f8, {9}7f8, {9}258, {9}258, {9}378, {9}0f8, {9}0f8
> 
> Is that what you are looking for? (I will recode the data tomorrow)

Yes this "codes" line, with the 13 x 9 bit data , it can be imported in bitbench

NicolasGagne, 2024-03-23:
I didn't recorde the ID for the sample I gave you yesterday, but it change each time the transmitter unit is reset.

I also note that the freq is drifting; I have seen up to 435.2Mhz, doesn't seem to affect the receiver unit.
I have also recorded more data (I don't think all of those are transmitted)

**"u" are for unknow; not display

Id | Wind Dir | Wind speed | Outdoor Temp | Railfall | Rail fall click| humidity | Wind chill | pressure | | codes
201 | u | u | 16.2 | 0 | 0 | 36 | u | 993.27 |   | codes     : {9}360, {9}658, {9}7f8, {9}698, {9}7f8, {9}7f8, {9}7f8, {9}7f8, {9}7f8, {9}7f8, {9}778, {9}0f8, {9}0f8
201 | 225 | 0 | 17.9 | 0 | 0 | 36 | u | 991,81 |   | codes     : {9}360, {9}158, {9}7f8, {9}658, {9}0f0, {9}0f0, {9}7f8, {9}7f8, {9}458, {9}458, {9}3f8, {9}0f8, {9}0f8
201 | 225 | 0 | 17.7 | 0.2 | 3 | 36 | u | 991.81 |   | codes     : {9}360, {9}158, {9}1f8, {9}458, {9}0f0, {9}0f0, {9}7f8, {9}7f8, {9}258, {9}258, {9}1f8, {9}0f8, {9}578
201 | 225 | 0 | 17.3 | 0.3 | 2 | 34 | u | 990.21 |   | codes     : {9}360, {9}158, {9}5f8, {9}598, {9}0f0, {9}0f0, {9}7f8, {9}7f8, {9}058, {9}058, {9}0f8, {9}0f8, {9}378
29 | u | u | 20.1 | 0 | 0 | 37 | u | 983.83 |   | codes     : {9}238, {9}758, {9}7f8, {9}398, {9}7f8, {9}7f8, {9}7f8, {9}7f8, {9}7f8, {9}7f8, {9}378, {9}0f8, {9}0f8
29 | 67 | 0 | 20.1 | 0 | 0 | 37 | u | 985.43 |   | codes     : {9}238, {9}758, {9}7f8, {9}398, {9}670, {9}670, {9}7f8, {9}7f8, {9}258, {9}258, {9}378, {9}0f8, {9}0f8
29 | 90 | 0 | 21.1 | 0.1 | 2 | 36 | u | 989.29 |   | codes     : {9}238, {9}0d8, {9}5f8, {9}798, {9}770, {9}770, {9}7f8, {9}7f8, {9}258, {9}258, {9}378, {9}0f8, {9}378
29 | 0 | 4 | 16.7 | 0.1 | 0 | 34 | u | 985.43 |   | codes     : {9}238, {9}658, {9}7f8, {9}058, {9}5f0, {9}5f0, {9}538, {9}7f8, {9}458, {9}768, {9}1f8, {9}0f8, {9}0f8
29 | 67 | 8.2 | -2.1 | 0.1 | 0 | 36 | -6.7 | 983.83 |   | codes     : {9}238, {9}018, {9}7f8, {9}7e8, {9}670, {9}670, {9}518, {9}7f8, {9}258, {9}0c8, {9}3f8, {9}0f8, {9}0f8
29 | 135 | 5.7 | -3.9 | 0.1 | 0 | 36 | -3.9 | 983.83 |   | codes     : {9}238, {9}7e8, {9}7f8, {9}368, {9}270, {9}270, {9}558, {9}7f8, {9}798, {9}528, {9}378, {9}0f8, {9}0f8
29 | 135 | 5.7 | -3.7 | 0.1 | 0 | 35 | u | 985.43 |   | codes     : {9}238, {9}7e8, {9}7f8, {9}0e8, {9}270, {9}270, {9}438, {9}7f8, {9}258, {9}168, {9}0f8, {9}0f8, {9}0f8
29 | 112 | 7.1 | -3.6 | 0.1 | 0 | 35 | -7.4 | 983.83 |   | codes     : {9}238, {9}7e8, {9}7f8, {9}4e8, {9}370, {9}370, {9}398, {9}7f8, {9}258, {9}428, {9}4f8, {9}0f8, {9}0f8
29 | 112 | 4.2 | -3.5 | 0.1 | 0 | 37 | u | 983.83 |   | codes     : {9}238, {9}7e8, {9}7f8, {9}2e8, {9}370, {9}370, {9}638, {9}7f8, {9}258, {9}368, {9}2f8, {9}0f8, {9}0f8

I will try to get more data later today or tomorrow.

ProfBoc75, 2024-03-24:
Little update, the supposed TEMPCHILL is not ... but it's a temp checksum of temp_int and temp_dec.

So:

    TEMP_CHECK = ((TEMP_INT + TEMP_DEC) & 0xFF) + 1

I still not found the exact relationship between the A, B, C and the Wind Speed, but here also, we have a checksum. C = A+B
Same, we have 2 times the wind direction. So, at least 5 controls can be made to validate the message:
- Start bit = 0 for each 13 byte
- Temp_checksum
- Wind dir twice
- Rain rate twice
- A + B = C

I started to draft the decoder and will open a PR soon after I fully guess the Wind Speed...

NicolasGagne, 2024-03-24:
Some other data

Id | Wind Dir | Wind speed | Outdoor Temp | Railfall | railnfall move | humidity | Wind chill | pressure |   |  
76 | 135 | 0 | 20.7 | 0 | 0 | 36 | u | 982.37 |   | codes     : {1}8, {1}8, {61}fffffffffffffff8, {7}fe, {9}668, {9}758, {9}7f8, {9}158, {9}270, {9}270, {9}7f8, {9}7f8, {9}258, {9}258, {9}1f8, {9}0f8, {9}0f8
76 | 90 | 6.5 | 0.4 | 0 | 0 | 36 | -2.2 | 984.5 |   | codes     : {9}668, {9}618, {9}7f8, {9}5e8, {9}770, {9}770, {9}258, {9}7f8, {9}658, {9}328, {9}4f8, {9}0f8, {9}0f8
76 | 112 | 0 | -0.7 | 0 | 0 | 34 | u | 984.5 |   | codes     : {9}668, {9}218, {9}7f8, {9}6e8, {9}370, {9}370, {9}7f8, {9}7f8, {9}258, {9}258, {9}0f8, {9}0f8, {9}0f8
76 | 67 | 2.9 | -1.9 | 0 | 0 | 35 | u | 984.5 |   | codes     : {9}668, {9}418, {9}7f8, {9}0e8, {9}670, {9}670, {9}1b8, {9}7f8, {9}258, {9}7e8, {9}378, {9}0f8, {9}0f8
76 | 67 | 4.8 | -1.3 | 0 | 0 | 35 | u | 984.5 |   | codes     : {9}668, {9}418, {9}7f8, {9}3e8, {9}670, {9}670, {9}7d8, {9}7f8, {9}258, {9}268, {9}1f8, {9}0f8, {9}0f8
76 | 135 | 2.9 | -2.7 | 0 | 0 | 35 | u | 984.5 |   | codes     : {9}668, {9}018, {9}7f8, {9}4e8, {9}270, {9}270, {9}1b8, {9}7f8, {9}258, {9}7e8, {9}0f8, {9}0f8, {9}0f8
76 | 180 | 1.8 | -2.3 | 0 | 0 | 35 | u | 984.5 |   | codes     : {9}668, {9}018, {9}7f8, {9}5e8, {9}170, {9}170, {9}278, {9}7f8, {9}258, {9}518, {9}1f8, {9}0f8, {9}0f8
76 | 0 | 1.8 | -3.7 | 0 | 0 | 36 | u | 984.5 |   | codes     : {9}668, {9}7e8, {9}7f8, {9}0e8, {9}5f0, {9}5f0, {9}278, {9}7f8, {9}258, {9}518, {9}0f8, {9}0f8, {9}0f8
76 | 337 | 1.8 | -3.4 | 0 | 0 | 36 | u | 984.5 |   | codes     : {9}668, {9}7e8, {9}7f8, {9}6e8, {9}4f0, {9}4f0, {9}678, {9}7f8, {9}258, {9}318, {9}6f8, {9}0f8, {9}0f8
76 | 135 | 1.4 | -4.9 | 0 | 0 | 35 | u | 984.5 |   | codes     : {9}668, {9}3e8, {9}7f8, {9}568, {9}270, {9}270, {9}178, {9}7f8, {9}398, {9}618, {9}378, {9}0f8, {9}0f8
76 | 135 | 0 | -4.6 | 0 | 0 | 37 | u | 984.5 |   | codes     : {9}668, {9}3e8, {9}7f8, {9}0e8, {9}070, {9}070, {9}3f8, {9}7f8, {9}258, {9}458, {9}4f8, {9}0f8, {9}0f8
76 | 180 | 22.7 | 3.8 | 0 | 0 | 35 | -1.2 | 984.5 |   | codes     : {9}668, {9}318, {9}7f8, {9}118, {9}170, {9}170, {9}760, {9}7f8, {9}258, {9}280, {9}5f8, {9}0f8, {9}0f8
76 | 220 | 7.4 | 4.1 | 0 | 0 | 34 | 2.2 | 984.5 |   | codes     : {9}668, {9}718, {9}7f8, {9}3e8, {9}1f0, {9}1f0, {9}698, {9}7f8, {9}258, {9}3c8, {9}378, {9}0f8, {9}0f8
76 | 202 | 6.8 | 6.5 | 0 | 0 | 34 | 5.7 | 984.5 |   | codes     : {9}668, {9}498, {9}7f8, {9}118, {9}1f0, {9}1f0, {9}058, {9}7f8, {9}258, {9}628, {9}2f8, {9}0f8, {9}0f8

Let me know if you need something else.

ProfBoc75, 2024-03-24:
From my side, I just need Wind speed information, if you can try to get more than 30 km/h it helps to see if the byte after "A" (position 7, may be LSB) is no more = 0 and it could be the MSB of the wind speed.
 
All the rest is already ok and exactly decoded. Another missing information, is the battery level, but I did not see such information from the user guide. May be fews codes of the sync/set ID, it could help also to confirm that the ID is sent into the message.

NicolasGagne, 2024-03-24:
I try to get above 30, best I got is 28,4, sorry.
As for the battery it not mention anywere in the documentation so I don't think there is a flag.

Here is the data

Id | Wind Dir | Wind speed | Outdoor Temp | Railfall | railnfall move | humidity | Wind chill | pressure |   |  
82 | 90 | 0 | 18.5 | 0 | 0 | 42 | u | 993.27 |   | codes     : {9}5a8, {9}558, {9}7f8, {9}058, {9}770, {9}770, {9}7f8, {9}7f8, {9}258, {9}258, {9}2f8, {9}0f8, {9}0f8
82 | 180 | 0 | 18.7 | 0 | 0 | 39 | u | 993.27 |   | codes     : {9}5a8, {9}558, {9}7f8, {9}258, {9}170, {9}170, {9}7f8, {9}7f8, {9}258, {9}258, {9}1f8, {9}0f8, {9}0f8
82 | 270 | 16.3 | 18.6 | 0 | 0 | 39 | u | 993.27 |   | codes     : {9}5a8, {9}558, {9}7f8, {9}458, {9}3f0, {9}3f0, {9}508, {9}7f8, {9}258, {9}0d0, {9}6f8, {9}0f8, {9}0f8
82 | 67 | 21.2 | 18.6 | 0 | 0 | 40 | u | 993.27 |   | codes     : {9}5a8, {9}558, {9}7f8, {9}458, {9}670, {9}670, {9}710, {9}7f8, {9}258, {9}2c0, {9}6f8, {9}0f8, {9}0f8
82 | 45 | 25 | 18.4 | 0 | 0 | 40 | u | 993.27 |   | codes     : {9}5a8, {9}558, {9}7f8, {9}798, {9}470, {9}470, {9}020, {9}7f8, {9}658, {9}178, {9}4f8, {9}0f8, {9}0f8
82 | 67 | 28 | 18.2 | 0 | 0 | 40 | u | 993.27 |   | codes     : {9}5a8, {9}558, {9}7f8, {9}598, {9}670, {9}670, {9}200, {9}7f8, {9}258, {9}558, {9}778, {9}0f8, {9}0f8
82 | 90 | 25.8 | 19.2 | 0 | 0 | 40 | u | 993.27 |   | codes     : {9}5a8, {9}358, {9}7f8, {9}558, {9}770, {9}770, {9}740, {9}7f8, {9}258, {9}2b8, {9}3f8, {9}0f8, {9}0f8
82 | 67 | 28 | 19.6 | 0 | 0 | 41 | u | 991.81 |   | codes     : {1}8, {9}5a8, {9}358, {9}7f8, {9}258, {9}670, {9}670, {9}400, {9}7f8, {9}258, {9}158, {9}6f8, {9}0f8, {9}0f8
82 | 67 | 27.7 | 19.6 | 0 | 0 | 37 | u | 993.27 |   | codes     : {9}5a8, {9}358, {9}7f8, {9}258, {9}670, {9}670, {9}100, {9}7f8, {9}258, {9}758, {9}6f8, {9}0f8, {9}0f8
82 | 270 | 28.4 | 19.1 | 0 | 0 | 41 | u | 995.67 |   | codes     : {9}5a8, {9}358, {9}7f8, {9}598, {9}3f0, {9}3f0, {9}5f8, {9}3f8, {9}258, {9}798, {9}378, {9}0f8, {9}0f8
82 | 270 | 28.2 | 21 | 0 | 0 | 39 | u | 994.07 |   | codes     : {9}5a8, {9}758, {9}7f8, {9}758, {9}3f0, {9}3f0, {9}3f8, {9}3f8, {9}258, {9}058, {9}7f8, {9}0f8, {9}0f8

ProfBoc75, 2024-03-24:
Very good , this confirmed that the A value, is with LSB (position 6) and MSB (position 7) , and A_LSB + A_MSB + B = C.

I'm still working on the wind speed decode ...

ProfBoc75, 2024-03-24:
Update of the Data Layout:

    Byte position             0          1           2          3            4           5          6          7          8          9         10         11          12
    bit position     0 12345678 0 12345678 0 1234 5678 0 12345678 0 1234 5678 0 1234 5678 0 12345678 0 12345678 0 12345678 0 12345678 0 12345678 0 12345678 0 1234 5678
    Data             X IIIIIIII X TEMP_INT X RT_2 ???? X TEMP_CHK X WDIR FLAG X WDIR FLAG X AAAA_LSB X AAAA_MSB X BBBBBBBB X CCCC_CHK X TEMP_DEC X ???????? X RT12 ????

About the Wind Speed checksum (CCCC_CHK = byte position 9) :

    C = (A_LSB + A_MSB + B) & 0xFF

ProfBoc75, 2024-03-25:
Just to show you how it's difficult to have the good formula for the Wind Speed.

The ratio between your speed values and the Wind Speed Raw value is not linear.

[image]

Especially with raw values less than 50.
Above 50 the curve is more linear with a ratio going down from around 0.14 to 0.11

I guess that the B value is a duration and this value is 45 most of the time, except little +/- 4 and ratio is altered if not 45:

A | B | km/h | Ratio
-- | -- | -- | --
0 | 45 | 0 | 0
1 | 45 | 0 | 0.00
11 | 49 | 1.4 | 0.127
12 | 45 | 1.8 | 0.150
13 | 45 | 1.8 | 0.138
19 | 45 | 2.9 | 0.153
22 | 45 | 3.6 | 0.164
26 | 46 | 4.0 | 0.154
28 | 45 | 4.2 | 0.150
30 | 45 | 4.5 | 0.150
32 | 45 | 4.8 | 0.150
36 | 45 | 5.7 | 0.158
42 | 48 | 5.7 | 0.136
43 | 45 | 6.2 | 0.144
44 | 45 | 6.5 | 0.148
45 | 44 | 6.5 | 0.144
46 | 44 | 6.5 | 0.141
47 | 45 | 6.8 | 0.145
49 | 45 | 7.1 | 0.145
52 | 45 | 7.4 | 0.142
58 | 45 | 8.2 | 0.141
60 | 45 | 8.5 | 0.142
122 | 45 | 16.3 | 0.134
184 | 45 | 21.2 | 0.115
200 | 45 | 22.7 | 0.114
223 | 44 | 25 | 0.112
232 | 45 | 25.8 | 0.111
251 | 45 | 27.7 | 0.110
253 | 45 | 28 | 0.111
254 | 45 | 28 | 0.110
257 | 45 | 28.2 | 0.110
258 | 45 | 28.4 | 0.110

ProfBoc75, 2024-03-25:
@NicolasGagne : can you test my PR, and let me know please ? The Wind Speed is estimated using an adaptive formula. You may have a gap between rl_433 and your display. I've tried to keep the gap as small as possible, I've had to make choices/compromises and I know it's not perfect. Let me know if it's acceptable and I'll merge the request.

NicolasGagne, 2024-03-25:
I will, I am away for work, I will do it when I get back home.

NicolasGagne, 2024-03-30:
I made my comment on the PR 2886,

Work great thanks

closed 2024-03-30 by ProfBoc75