merbanan/rtl_433 #2891: R900 Non-BC Wrapp bits in Unknown3

dockSquadron, 2024-04-03:
My Water meter was recently changed from a R900BCD to a R900M V4 Non-BCD. I've been analyzing my neighborhood's water meters and identified the following:

- When UNKN_ONE:161, it appears that Consumption is BCD coded
- When UNKN_ONE:163, it appears that Consumption is Non-BCD coded
- UNKN_THREE has been renamed to WRAP in the data below and appears to be the upper 3 bits of Consumption for Non-BCD data. 

I found that meters in my area have a 50 Gallon resolution. To decode properly, I identified the following formula: decimal(WRAP << 24b + CONSUMPTION) / 10.

There are a few meters nearby that UNKN_TWO = 4 and NOUSE = 1, and I'm not sure what that means. Also, one where WRAP = 7, and that seems to be an error of some kind. I also see that the LEAK fields don't seem correct for the BCD meters. Thoughts? I'd like to write a pull request for the WRAP bits. Do you think it holds beyond just my neighborhood? I couldn't tell what the WRAP bits do for the BCD meters.

ID: 142xxxx224 UNKN_ONE:161 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 02564640 WRAP: 1 LEAK: 7 LEAKNOW: 3 EXTRA: 8d9693   CONSUMPTION=25646400
ID: 142xxxx224 UNKN_ONE:161 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 02564640 WRAP: 1 LEAK: 7 LEAKNOW: 3 EXTRA: 8d9693   CONSUMPTION=25646400
ID: 146xxxx676 UNKN_ONE:161 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 02331792 WRAP: 1 LEAK: 7 LEAKNOW: 3 EXTRA: ba14b8   CONSUMPTION=23317920
ID: 146xxxx676 UNKN_ONE:161 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 02331797 WRAP: 1 LEAK: 7 LEAKNOW: 3 EXTRA: e0a025   CONSUMPTION=23317970
ID: 146xxxx834 UNKN_ONE:161 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 02521216 WRAP: 1 LEAK: 7 LEAKNOW: 3 EXTRA: 849bd0   CONSUMPTION=25212160
ID: 146xxxx834 UNKN_ONE:161 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 02521216 WRAP: 1 LEAK: 7 LEAKNOW: 3 EXTRA: 849bd0   CONSUMPTION=25212160
ID: 146xxxx614 UNKN_ONE:161 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 02687328 WRAP: 1 LEAK: 7 LEAKNOW: 3 EXTRA: b3a695   CONSUMPTION=26873280
ID: 146xxxx614 UNKN_ONE:161 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 02687328 WRAP: 1 LEAK: 7 LEAKNOW: 3 EXTRA: b3a695   CONSUMPTION=26873280
ID: 146xxxx622 UNKN_ONE:161 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 02701408 WRAP: 1 LEAK: 7 LEAKNOW: 3 EXTRA: e3052b   CONSUMPTION=27014080
ID: 146xxxx622 UNKN_ONE:161 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 02701413 WRAP: 1 LEAK: 7 LEAKNOW: 3 EXTRA: b9b1b6   CONSUMPTION=27014130
ID: 148xxxx904 UNKN_ONE:161 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 04657664 WRAP: 1 LEAK: 7 LEAKNOW: 3 EXTRA: 8f9926   CONSUMPTION=46576640
ID: 148xxxx904 UNKN_ONE:161 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 04657664 WRAP: 1 LEAK: 7 LEAKNOW: 3 EXTRA: 8f9926   CONSUMPTION=46576640
ID: 148xxxx978 UNKN_ONE:161 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 04264336 WRAP: 1 LEAK: 7 LEAKNOW: 3 EXTRA: bdcb44   CONSUMPTION=42643360
ID: 148xxxx978 UNKN_ONE:161 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 04264341 WRAP: 1 LEAK: 7 LEAKNOW: 3 EXTRA: e77fd9   CONSUMPTION=42643410
ID: 148xxxx608 UNKN_ONE:161 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 03545344 WRAP: 1 LEAK: 7 LEAKNOW: 3 EXTRA: 942fcf   CONSUMPTION=35453440
ID: 148xxxx608 UNKN_ONE:161 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 03545344 WRAP: 1 LEAK: 7 LEAKNOW: 3 EXTRA: 942fcf   CONSUMPTION=35453440
ID: 154xxxx088 UNKN_ONE:163 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 09239068 WRAP: 2 LEAK: 0 LEAKNOW: 0 EXTRA: 09c260   CONSUMPTION=4279350
ID: 154xxxx088 UNKN_ONE:163 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 09239568 WRAP: 2 LEAK: 0 LEAKNOW: 0 EXTRA: 92f4f1   CONSUMPTION=4279400
ID: 154xxxx936 UNKN_ONE:163 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 07961284 WRAP: 1 LEAK: 0 LEAKNOW: 0 EXTRA: 1ebb1d   CONSUMPTION=2473850
ID: 154xxxx936 UNKN_ONE:163 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 07961284 WRAP: 1 LEAK: 0 LEAKNOW: 0 EXTRA: 1ebb1d   CONSUMPTION=2473850
ID: 154xxxx642 UNKN_ONE:163 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 08049284 WRAP: 1 LEAK: 0 LEAKNOW: 0 EXTRA: a48233   CONSUMPTION=2482650
ID: 154xxxx412 UNKN_ONE:163 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 04998852 WRAP: 3 LEAK: 0 LEAKNOW: 0 EXTRA: f2d307   CONSUMPTION=5533050
ID: 154xxxx412 UNKN_ONE:163 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 04999352 WRAP: 3 LEAK: 0 LEAKNOW: 0 EXTRA: ad5cf0   CONSUMPTION=5533100
ID: 154xxxx888 UNKN_ONE:163 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 07826068 WRAP: 2 LEAK: 0 LEAKNOW: 0 EXTRA: b67fce   CONSUMPTION=4138050
ID: 154xxxx888 UNKN_ONE:163 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 07826068 WRAP: 2 LEAK: 0 LEAKNOW: 0 EXTRA: b67fce   CONSUMPTION=4138050
ID: 154xxxx976 UNKN_ONE:163 UNKN_TWO: 4 NOUSE: 01 CONSUMPTION: 00352241 WRAP: 0 LEAK: 0 LEAKNOW: 0 EXTRA: 772279   CONSUMPTION=35224.1
ID: 154xxxx976 UNKN_ONE:163 UNKN_TWO: 4 NOUSE: 01 CONSUMPTION: 00352550 WRAP: 0 LEAK: 0 LEAKNOW: 0 EXTRA: 180b6a   CONSUMPTION=35255
ID: 156xxxx314 UNKN_ONE:163 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 08549568 WRAP: 2 LEAK: 0 LEAKNOW: 0 EXTRA: cee213   CONSUMPTION=4210400
ID: 156xxxx314 UNKN_ONE:163 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 08549568 WRAP: 2 LEAK: 0 LEAKNOW: 0 EXTRA: cee213   CONSUMPTION=4210400
ID: 156xxxx904 UNKN_ONE:163 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 07322284 WRAP: 1 LEAK: 0 LEAKNOW: 0 EXTRA: a6cbbd   CONSUMPTION=2409950
ID: 156xxxx904 UNKN_ONE:163 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 07322284 WRAP: 1 LEAK: 0 LEAKNOW: 0 EXTRA: a6cbbd   CONSUMPTION=2409950
ID: 156xxxx230 UNKN_ONE:163 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 04352284 WRAP: 1 LEAK: 0 LEAKNOW: 0 EXTRA: fbd241   CONSUMPTION=2112950
ID: 156xxxx230 UNKN_ONE:163 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 04352784 WRAP: 1 LEAK: 0 LEAKNOW: 0 EXTRA: 09730a   CONSUMPTION=2113000
ID: 156xxxx004 UNKN_ONE:163 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 00000015 WRAP: 7 LEAK: 0 LEAKNOW: 0 EXTRA: 2bbe7f   CONSUMPTION=11744052.7
ID: 156xxxx004 UNKN_ONE:163 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 00000015 WRAP: 7 LEAK: 0 LEAKNOW: 0 EXTRA: 2bbe7f   CONSUMPTION=11744052.7
ID: 157xxxx858 UNKN_ONE:163 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 13626284 WRAP: 1 LEAK: 0 LEAKNOW: 0 EXTRA: 542574   CONSUMPTION=3040350
ID: 157xxxx858 UNKN_ONE:163 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 13626284 WRAP: 1 LEAK: 0 LEAKNOW: 0 EXTRA: 542574   CONSUMPTION=3040350
ID: 157xxxx986 UNKN_ONE:163 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 10901784 WRAP: 1 LEAK: 0 LEAKNOW: 0 EXTRA: 820264   CONSUMPTION=2767900
ID: 157xxxx986 UNKN_ONE:163 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 10901784 WRAP: 1 LEAK: 0 LEAKNOW: 0 EXTRA: 820264   CONSUMPTION=2767900
ID: 157xxxx868 UNKN_ONE:163 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 08674784 WRAP: 1 LEAK: 0 LEAKNOW: 0 EXTRA: ccbc46   CONSUMPTION=2545200
ID: 157xxxx868 UNKN_ONE:163 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 08674784 WRAP: 1 LEAK: 0 LEAKNOW: 0 EXTRA: ccbc46   CONSUMPTION=2545200
ID: 157xxxx414 UNKN_ONE:163 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 07478784 WRAP: 1 LEAK: 0 LEAKNOW: 0 EXTRA: 486308   CONSUMPTION=2425600
ID: 157xxxx414 UNKN_ONE:163 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 07478784 WRAP: 1 LEAK: 0 LEAKNOW: 0 EXTRA: 486308   CONSUMPTION=2425600
ID: 157xxxx760 UNKN_ONE:163 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 07176784 WRAP: 1 LEAK: 0 LEAKNOW: 0 EXTRA: 97b772   CONSUMPTION=2395400
ID: 157xxxx760 UNKN_ONE:163 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 07176784 WRAP: 1 LEAK: 0 LEAKNOW: 0 EXTRA: 97b772   CONSUMPTION=2395400
ID: 157xxxx012 UNKN_ONE:163 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 16440000 WRAP: 0 LEAK: 0 LEAKNOW: 0 EXTRA: ee08a5   CONSUMPTION=1644000
ID: 157xxxx012 UNKN_ONE:163 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 16440000 WRAP: 0 LEAK: 0 LEAKNOW: 0 EXTRA: ee08a5   CONSUMPTION=1644000
ID: 157xxxx614 UNKN_ONE:163 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 05290068 WRAP: 2 LEAK: 0 LEAKNOW: 0 EXTRA: a2d451   CONSUMPTION=3884450
ID: 157xxxx614 UNKN_ONE:163 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 05290068 WRAP: 2 LEAK: 0 LEAKNOW: 0 EXTRA: a2d451   CONSUMPTION=3884450
ID: 157xxxx254 UNKN_ONE:163 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 07651568 WRAP: 2 LEAK: 0 LEAKNOW: 0 EXTRA: e1f9eb   CONSUMPTION=4120600
ID: 157xxxx254 UNKN_ONE:163 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 07651568 WRAP: 2 LEAK: 0 LEAKNOW: 0 EXTRA: e1f9eb   CONSUMPTION=4120600
ID: 157xxxx852 UNKN_ONE:163 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 11608068 WRAP: 2 LEAK: 0 LEAKNOW: 0 EXTRA: 51f2de   CONSUMPTION=4516250
ID: 157xxxx852 UNKN_ONE:163 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 11608568 WRAP: 2 LEAK: 0 LEAKNOW: 0 EXTRA: 6e265d   CONSUMPTION=4516300
ID: 157xxxx286 UNKN_ONE:163 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 07830568 WRAP: 2 LEAK: 0 LEAKNOW: 0 EXTRA: 3b154f   CONSUMPTION=4138500
ID: 157xxxx286 UNKN_ONE:163 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 07830568 WRAP: 2 LEAK: 0 LEAKNOW: 0 EXTRA: 3b154f   CONSUMPTION=4138500
[... cut ...]
ID: 157xxxx232 UNKN_ONE:163 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 09759784 WRAP: 1 LEAK: 0 LEAKNOW: 0 EXTRA: d39496   CONSUMPTION=2653700
ID: 157xxxx232 UNKN_ONE:131 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 09760284 WRAP: 1 LEAK: 0 LEAKNOW: 0 EXTRA: 1d5ffa   CONSUMPTION=97602840
ID: 157xxxx916 UNKN_ONE:163 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 08889784 WRAP: 1 LEAK: 0 LEAKNOW: 0 EXTRA: 19f619   CONSUMPTION=2566700
ID: 157xxxx916 UNKN_ONE:163 UNKN_TWO: 0 NOUSE: 00 CONSUMPTION: 08890784 WRAP: 1 LEAK: 0 LEAKNOW: 0 EXTRA: 07d13b   CONSUMPTION=2566800

zuckschwerdt, 2024-04-03:
Your findings seem plausible. A type-code /version in the first byte after the ID would be expected.

Best to look for the most active Neptune R900 users (https://github.com/merbanan/rtl_433/issues?q=r900) and ping them here, asking for their observations on the unknown 1-3 fields.

dockSquadron, 2024-04-03:
looks like a similar Pull Request over at RTLAMR regarding high bits of consumption and meter type: (https://github.com/bemasher/rtlamr/pull/270)

Can anyone else provide their observations regarding the unknown fields in the R900 protocol? Please provide some data from meters in your area to see if they match line above. Thanks! @robbat2 @zuckschwerdt @jruby411 @roger- @williamstlr

jruby411, 2024-04-03:
This is what I got from: **rtl_433 -C customary -f 912600155 -s 2000k -Y classic -R 228**

rtl_433 version 23.11 branch  at 202311281352 inputs file rtl_tcp RTL-SDR with TLS

New defaults active, use "-Y classic -s 250k" if you need the old defaults

Found Rafael Micro R820T tuner
[SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-04-03 18:54:06
model     : Neptune-R900 id        : 1568447122
unkn1     : 163          unkn2     : 4             nouse     : 0             backflow  : 0             consumption: 1962418      unkn3     : 0
leak      : 2            leaknow   : 0             extra     : 6d6fe4
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-04-03 18:54:11
model     : Neptune-R900 id        : 1568441166
unkn1     : 163          unkn2     : 4             nouse     : 0             backflow  : 0             consumption: 1053529      unkn3     : 0
leak      : 0            leaknow   : 0             extra     : 210435
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-04-03 18:54:15
model     : Neptune-R900 id        : 1568441994
unkn1     : 163          unkn2     : 4             nouse     : 0             backflow  : 0             consumption: 1340380      unkn3     : 0
leak      : 1            leaknow   : 0             extra     : 022021
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-04-03 18:54:16
model     : Neptune-R900 id        : 1568167740
unkn1     : 163          unkn2     : 4             nouse     : 0             backflow  : 0             consumption: 1023875      unkn3     : 0
leak      : 0            leaknow   : 0             extra     : 189bdc
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-04-03 18:54:19
model     : Neptune-R900 id        : 1568452830
unkn1     : 163          unkn2     : 4             nouse     : 0             backflow  : 0             consumption: 2953068      unkn3     : 0
leak      : 0            leaknow   : 0             extra     : 292188
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-04-03 18:54:22
model     : Neptune-R900 id        : 1568451124
unkn1     : 163          unkn2     : 4             nouse     : 0             backflow  : 0             consumption: 549135       unkn3     : 0
leak      : 0            leaknow   : 0             extra     : 81a356
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-04-03 18:54:29
model     : Neptune-R900 id        : 1568440884
unkn1     : 163          unkn2     : 4             nouse     : 0             backflow  : 0             consumption: 467153       unkn3     : 0
leak      : 2            leaknow   : 0             extra     : 2ab38e
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-04-03 18:54:36
model     : Neptune-R900 id        : 1568454006
unkn1     : 163          unkn2     : 4             nouse     : 0             backflow  : 0             consumption: 1626361      unkn3     : 0
leak      : 0            leaknow   : 0             extra     : 8fc220
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-04-03 18:54:45
model     : Neptune-R900 id        : 1568448864
unkn1     : 163          unkn2     : 4             nouse     : 0             backflow  : 0             consumption: 1174914      unkn3     : 0
leak      : 0            leaknow   : 0             extra     : 7b9153
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-04-03 18:55:03
model     : Neptune-R900 id        : 1568168338
unkn1     : 163          unkn2     : 4             nouse     : 1             backflow  : 0             consumption: 3816616      unkn3     : 0
leak      : 0            leaknow   : 0             extra     : f88f00
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-04-03 18:55:07
model     : Neptune-R900 id        : 1568441166
unkn1     : 163          unkn2     : 4             nouse     : 0             backflow  : 0             consumption: 1053529      unkn3     : 0
leak      : 0            leaknow   : 0             extra     : 210435
[... cut ...]
leak      : 1            leaknow   : 0             extra     : 69690e
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-04-03 18:55:12
model     : Neptune-R900 id        : 1568167740
unkn1     : 163          unkn2     : 4             nouse     : 0             backflow  : 0             consumption: 1023875      unkn3     : 0
leak      : 0            leaknow   : 0             extra     : 189bdc

jruby411, 2024-04-04:
I created a patch to follow the RTLAMR pull request 270 and this is what I get from my meters in the area.
Note: I changed the consumption calculation to use the 2 bits of unkn3. But, since I don't have any meters like that in my area, the numbers didn't change. And yes, the consumption for the current release needs to be divided by 10. I didn't change that either. That would be a breaking change.

See this patch: [701757c](https://github.com/jruby411/rtl_433/commit/701757c00790c13118cf6135976da0a77dc4de05)

./src/rtl_433 -C customary -f 912600155 -s 2000k -Y classic -R 228
rtl_433 version -128-NOTFOUND branch r900_non_bcd_wrap at 202404021454 inputs file rtl_tcp RTL-SDR SoapySDR with TLS

New defaults active, use "-Y classic -s 250k" if you need the old defaults

Found Rafael Micro R820T tuner
[SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-04-03 20:25:29
model     : Neptune-R900 id        : 1568440884
unkn1     : 10           metertype : 3             unkn2     : 4             nouse     : 0             backflow  : 0             consumption: 467153
unkn3     : 0            leak      : 2             leaknow   : 0             extra     : 2ab38e
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-04-03 20:25:43
model     : Neptune-R900 id        : 1568446236
unkn1     : 10           metertype : 3             unkn2     : 4             nouse     : 0             backflow  : 0             consumption: 1408668
unkn3     : 0            leak      : 0             leaknow   : 0             extra     : 8227c3
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-04-03 20:25:45
model     : Neptune-R900 id        : 1568448864
unkn1     : 10           metertype : 3             unkn2     : 4             nouse     : 0             backflow  : 0             consumption: 1174978
unkn3     : 0            leak      : 0             leaknow   : 0             extra     : 047d9e
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-04-03 20:25:47
model     : Neptune-R900 id        : 1568452830
unkn1     : 10           metertype : 3             unkn2     : 4             nouse     : 0             backflow  : 0             consumption: 2953282
unkn3     : 0            leak      : 0             leaknow   : 0             extra     : 8eea20
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-04-03 20:26:11
model     : Neptune-R900 id        : 1568441994
unkn1     : 10           metertype : 3             unkn2     : 4             nouse     : 0             backflow  : 0             consumption: 1340569
unkn3     : 0            leak      : 1             leaknow   : 0             extra     : baf281
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-04-03 20:26:18
model     : Neptune-R900 id        : 1568441730
unkn1     : 10           metertype : 3             unkn2     : 4             nouse     : 0             backflow  : 0             consumption: 816726
unkn3     : 0            leak      : 0             leaknow   : 0             extra     : f2fb8e
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-04-03 20:26:18
model     : Neptune-R900 id        : 1568451124
unkn1     : 10           metertype : 3             unkn2     : 4             nouse     : 0             backflow  : 0             consumption: 549480
unkn3     : 0            leak      : 0             leaknow   : 0             extra     : 292056
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-04-03 20:26:25
model     : Neptune-R900 id        : 1568440884
unkn1     : 10           metertype : 3             unkn2     : 4             nouse     : 0             backflow  : 0             consumption: 467153
unkn3     : 0            leak      : 2             leaknow   : 0             extra     : 2ab38e
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-04-03 20:26:35
model     : Neptune-R900 id        : 1568441166
unkn1     : 10           metertype : 3             unkn2     : 4             nouse     : 0             backflow  : 0             consumption: 1053531
unkn3     : 0            leak      : 0             leaknow   : 0             extra     : f79429
[... cut ...]
unkn3     : 0            leak      : 0             leaknow   : 0             extra     : 3cd73e
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-04-03 20:27:39
model     : Neptune-R900 id        : 1568452830
unkn1     : 10           metertype : 3             unkn2     : 4             nouse     : 0             backflow  : 0             consumption: 2953282
unkn3     : 0            leak      : 0             leaknow   : 0             extra     : 8eea20

gdt, 2024-06-03:
Where are we on this?  The title is confusing.   Is someone going to create an improvement PR?   (Speaking as ticket gardener, I'd like to have easily undersstandable tickets that will age well, or drive them to closure, and this is not  the first case because with 30s of effort I don't know which way is up.)

jruby411, 2024-06-03:
I created a quick beta patch, but never got any further feedback from dockSquadron. 

I am willing to continue working on this. But, I cannot test, since I don’t have any of these type of meters in my area. You can close this if you want to tidy this up. Someone can request this again if needed. 

I could create a pull request with what I have done. I don’t think it breaks existing functionality. But, it is untested at current.

gdt, 2024-06-03:
A pull request marked draft would be great.   It won't hurt if it isn't merged  but in all seriousness that gets the code out there for someone to try.    Who knows who will try it and see what they hear.

gdt, 2024-06-23:
Closing this in favor of the PR, because having an issue open when there's a PR doesn't help organize our thinking.  Please continue to work on the PR!

closed 2024-06-23 by gdt