merbanan/rtl_433 #2892: rtl_433 not working on older rtl-sdr device with new librtlsdr

acsawdey, 2024-04-03:
I have two SDR devices, a RTL-SDR Blog V4, and an older one I got from Adafruit (https://www.adafruit.com/product/1497).

What I am seeing is that rtl_433 will only work with the V4 if it is built using the new librtlsdr from their git, and it will only work with the old device if I build it with the "0.6" librtlsdr that raspbian or ubuntu 22.04 have. The other combinations, it builds and runs but I never receive any signals. rtl_adsp on the other hand works with either device and either library.

waltje, 2024-05-10:
I cannot get it to work at all with my V4.  Yes, using the new rtlsdr.dll driver it *sees* it, but it will not proceed to setting the frequency:

``C:\Program Files\RTL433>rtl_433-rtlsdr.exe
rtl_433 version 23.11 branch  at 202311281352 inputs file rtl_tcp RTL-SDR
Found Rafael Micro R828D tuner
RTL-SDR Blog V4 Detected
[SDR] Using device 0: RTLSDRBlog, Blog V4, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 250000.000414 Hz``

When started with the old rtlsdr.dll driver, it will report 'failed to set frequency' (as expected.)

zuckschwerdt, 2024-05-10:
Can you run with `-vvvv` to get debug messages? Also `-Y autolevel -Y magest -M level -M noise` to see what the input data looks like?

waltje, 2024-05-10:
I did.  Interestingly, in debug mode, it *does* show it is tuned to 433.92.

`rtl_433 version 23.11 branch  at 202311281352 inputs file rtl_tcp RTL-SDR
Found Rafael Micro R828D tuner
RTL-SDR Blog V4 Detected
[SDR] Using device 0: RTLSDRBlog, Blog V4, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 250000.000414 Hz`

in normal mode.  Using the old (original) driver, it says "Tuned to.." here.  I guess that comes from the driver then, and not from the application itself.

The debug log shows:

`[SDR] Found 1 device(s)
[SDR] trying device 0: RTLSDRBlog, Blog V4, SN: 00000001
[SDR] Using device 0: RTLSDRBlog, Blog V4, SN: 00000001, "Generic RTL2832U OEM"
[SDR] Sample rate set to 250000 S/s.
[Input] Bit detection level set to 0.0 (Auto).
[SDR] Tuner gain set to Auto.
[Input] Reading samples in async mode...
[SDR] rtlsdr_set_center_freq 433920000 = 0
[SDR] Tuned to 433.920MHz.
[acquire_thread] acquire_thread enter...
[Auto Level] Estimated noise level is -16.8 dB, adjusting minimum detection level to -13.8 dB
[Baseband] low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us`

so that looks good.

So, nevermind my comment on it not working.  It does, although you probably do want to include the *new* rtlsdr.dll file which also supports the V4 dongle.

waltje, 2024-05-14:
I think this can be closed.

I built the code from source, using the *new* libsdr, and it seems to work fine on both my V4 *and* V3 units.

zuckschwerdt, 2024-05-14:
Thanks for the feedback, yes, there is no clear indication that rtl_433 uses the libs in an incompatible way.

closed 2024-05-14 by zuckschwerdt