merbanan/rtl_433 #2893: BMW GEN 3 TPMS signal decoding

Billymazze, 2024-04-04:
Howdy,
I'm a complete noob to C language and decoder protocols, but I'm trying to make a decoder protocols for the flipper zero. I've successfully made one for abarth124 spider for TPMS app by Wosk, now I'm trying to make one for bmw tpms sensors. But I'm running into issues trying to get a sensible signal on URH. I have some signals I've captured with URH and the flipper zero Protoveiw app that I'll attach a [google drive link](https://drive.google.com/drive/folders/1ZM9ZmDwIUaQ74j1viKW6knrmj_pdXmgD?usp=drive_link) for anyone interested. Currently thinking I might be using the wrong record settings on URH. Current samples recorded from URH with an RTL-SDR (Realtek RTL2838UHIDIR) @433.92mhz with a sample rate of 1.0M, bandwidth of 1.0M, gain of 49, bias tree disabled, and direct sampling disabled. Just added new samples at a sample rate of 2.5M. I'm pretty sure the frequency and the flipper.sub signal is correct cause I can replay the signal on flipper zero Protoview view and trigger the tpms warning on the dash. Also ISTA doesn't appear provided a signal message breakdown for Gen 3 TPMS, but I'm pretty certain this message info is the same as the Gen 5 TPMS.

So far this is what I get in URH,

@1.0M sample rate recording (old)
cccccccd54cab4aaccab4d34acab5332ccd55552aaacd4cccb4d4
cccccccd54cab4aaccab4d34acab5332ccd55552aaacd4cccb4d4
cccccccd54cab4aaccab4d34acab5332ccd55552aaacd4cccb4d4
cccccccd54cab4aaccab4d34acab5332ccd55552aaacd4cccb4d4
cccccccd54cab4aaccab4d34acab5334ccd55552aaacaab5552d4

Any help is greatly appreciated, thanks.

ProfBoc75, 2024-04-05:
@Billymazze: 

Sounds to be MC signal. 

Can you zip your captured files and drop here in a comment. 

You can also drop files into https://triq.org/pdv/ and see what is the pulse signal that could give some information like the short and long pulse width.

From URH, in the Interpretation menu, what is the samples/symbol value and what is the modulation ?

For each capture, would be nice if you have all the information with the id, temp, pressure ... to help to decode the raw data.

Billymazze, 2024-04-06:
Here's my current recording samples, I added the pressure and if warning was on to some of the names. Also added the sample rate they were recorded at, appears to affect the samples/symbol. The one recorded at 1.0M appears as shown above at 50, the others at 2.5M appear as above at ~130 samples/symbol in URH. I will try to collect better signals with more info, tried going to a Discount tire to get the sensor ID's. But the technician said his scanner was unable to read the info, for now I'll try to post more signals with at least the current pressure, temperature (usually ~60-80*F) and if warning is active. (also any recordings below 32psi the warning is active)
[flipper zero Bmw_low.zip](https://github.com/merbanan/rtl_433/files/14891867/flipper.zero.Bmw_low.zip)
[bmw low.zip](https://github.com/merbanan/rtl_433/files/14891877/bmw.low.zip)

ProfBoc75, 2024-04-06:
First analysis:
- It's FSK signal, the pulse is around 53 or 54 µs in the Flipper Zero .sub files (clear text with the modulation and the pulse duration).
- From the URH records, @1Mhz , the samples/symbol = 50 ( 50 / 1Mhz  = 50 µs ) µs, it matchs F0 files. @2.5Mhz, the samples/symbol = 100 , but if you change the value to 125, you'll get the same kind of values and it matchs the 50 µs. ( 125 / 2.5Mhz = 50 µs ).
- looks like Manchester coded.

This is very close to other TPMS, with the FSK modulation and most pulse width = 52 µs.

Now I'm checking the decode part, MC, CRC, etc ...

ProfBoc75, 2024-04-06:
You can check all your complex16s file with rtl_433, but they need first to be renamed with extension .cs8 and following the `file name meta data` convention [here](https://github.com/merbanan/rtl_433/blob/ac1e4a8c5a36fb90e3b06c0f01cef00bb3b2614d/docs/IQ_FORMATS.md#file-name-meta-data):

    g001_433.92M_2500k.cs8 to g004_433.92M_2500k.cs8 for the files at 2.5M sample rate
    g005_433.92M_1000k.cs8 for the file at 1M sample rate

Then try with the flex decoder:

    rtl_433 -X "n=BMW_G3,m=FSK_PCM,s=53,l=53,r=1000,bits>=200" -Y autolevel -Y minmax *.cs8

Except the first file, all other have few rf signals.

Result filtered on codes (duplicates removed) in [bitbench ](https://triq.net/bitbench#c=%7B205%7D3266666aa655a556655a65a5999a2cb332aad54aaaccb3535350%20%5B%20g001_433.92M_2500k.cs8%20full%20pres%2032psi%20%5D&c=%7B207%7D3333fd9aa99569559956996965696a6999aaa55aaaa66a59634c%20%5B%20g002_433.92M_2500k.cs8%20low%2022psi%20%5D&c=%7B204%7D33333335532ad2ab32ad32d2cad1a9a666aa95692a66a5965a6%20%5B%20g002_433.92M_2500k.cs8%20low%2022psi%20%5D&c=%7B203%7D326009aa995692b32ad32d2c5a5a9a666aa956aaa99a965969a%20%5B%20g002_433.92M_2500k.cs8%20low%2022psi%20%5D&c=%7B204%7D33333332a62b4aacc56996965696a6999aaa55aaaa66a5965a6%20%5B%20g002_433.92M_2500k.cs8%20low%2022psi%20%5D&c=%7B206%7D33333335532ad2ab32ad32d2c5a5a9a666aa956aaa99a9659698%20%5B%20g002_433.92M_2500k.cs8%20low%2022psi%20%5D&c=%7B206%7D33333335532ad2ab32ad32d2cad2d4d333554ab2aa99a8cb9698%20%5B%20g002_433.92M_2500k.cs8%20low%2022psi%20%5D&c=%7B204%7D33333335532ad2ab315a65a595a5a9a666aa9569553351965a6%20%5B%20g002_433.92M_2500k.cs8%20low%2022psi%20%5D&c=%7B203%7D3333326aa6554aaccab4cb4b2b4b26999aaa55aaaa66a59658c%20%5B%20g002_433.92M_2500k.cs8%20low%2022psi%20%5D&c=%7B207%7D33333335532ad2ab32ad32d2cad2d4d333554ab5554cd4b2cb4c%20%5B%20g002_433.92M_2500k.cs8%20low%2022psi%20%5D&c=%7B207%7D33333335532ad2ab32ad32d2cb52d4d333554aad554d52d2ad4c%20%5B%20g003_433.92M_2500k.cs8%20low%20pres%2022psi%20%5D&c=%7B207%7D33333335532ad2ab32ad32d2cad52b2cccaab54aaab2caad4cb4%20%5B%20g003_433.92M_2500k.cs8%20low%20pres%2022psi%20%5D&c=%7B204%7D33333335532ad2ab32ad32d2caca56599955652aaacb2ab532c%20%5B%20g003_433.92M_2500k.cs8%20low%20pres%2022psi%20%5D&c=%7B208%7D33333335532ad2ab32ad32d2cab2d4d333aaa55aaaa6aa9a659a%20%5B%20g003_433.92M_2500k.cs8%20low%20pres%2022psi%20%5D&c=%7B207%7D33333335532ad2ab32ad32d2cad52b2cccaab54aaab2ca5a996a%20%5B%20g003_433.92M_2500k.cs8%20low%20pres%2022psi%20%5D&c=%7B205%7D333333355325a556655a65a595a8acb332aad52aaacb2ab532d0%20%5B%20g003_433.92M_2500k.cs8%20low%20pres%2022psi%20%5D&c=%7B204%7D33333335532ad2ab32ad32d2c9a5a9a4cd5515aaaa66a5965a6%20%5B%20g003_433.92M_2500k.cs8%20low%20pres%2022psi%20%5D&c=%7B205%7D33333332a655a556655a65a5966a565999556a95556564cd4b50%20%5B%20g004_433.92M_2500k.cs8%20low%20pres%20warn%20%5D&c=%7B207%7D33333335532ad2ab32ad32d2cb2d2b2cccaab54aaab32b52d534%20%5B%20g004_433.92M_2500k.cs8%20low%20pres%20warn%20%5D&c=%7B207%7D33333335532ad2ab32ad32d2cb2ad4d333554ab5554d351aa59a%20%5B%20g004_433.92M_2500k.cs8%20low%20pres%20warn%20%5D&c=%7B204%7D33333334a655a556655a65a5962b26999aaa55aaaa65a65a95a%20%5B%20g004_433.92M_2500k.cs8%20low%20pres%20warn%20%5D&c=%7B208%7D33333335532ad2ab32ad32d2cb4ad4d333554ab5554cb4cb52b5%20%5B%20g004_433.92M_2500k.cs8%20low%20pres%20warn%20%5D&c=%7B207%7D33333335532ad2ab32ad32d2cb4ad4d333554ab5554cb4cb52b4%20%5B%20g004_433.92M_2500k.cs8%20low%20pres%20warn%20%5D&c=%7B207%7D33333335532ad2ab32ad32d2cb4d2b2cccaab54aaab2aad4d54c%20%5B%20g004_433.92M_2500k.cs8%20low%20pres%20warn%20%5D&c=%7B205%7D33333335532ad2ab32ad32d2cb4d2b2cccaa6a95556555a35530%20%5B%20g004_433.92M_2500k.cs8%20low%20pres%20warn%20%5D&c=%7B207%7D33333335532ad2ab32ad32d2cb2ad4d333554ab5554d354d52cc%20%5B%20g004_433.92M_2500k.cs8%20low%20pres%20warn%20%5D&c=%7B206%7D33333335532ad2ab32ad32d29655a9a666aa956aaa9a6a9aa598%20%5B%20g004_433.92M_2500k.cs8%20low%20pres%20warn%20%5D&c=%7B206%7D33333335532ad2ab32ad32d2cb2ad4d333554ab52a9a6a9aa598%20%5B%20g004_433.92M_2500k.cs8%20low%20pres%20warn%20%5D&c=%7B205%7D33333335532ad2ab32ad32d2cb2ad14ccd552ad55534d5354b30%20%5B%20g004_433.92M_2500k.cs8%20low%20pres%20warn%20%5D&c=%7B208%7D33333335532ad2ab32ad32d2cb2d2b2cccaab54aaab32b52d535%20%5B%20g004_433.92M_2500k.cs8%20low%20pres%20warn%20%5D&c=%7B208%7D3333339aa9956955995699696596959666555aa5555995a96a9a%20%5B%20g004_433.92M_2500k.cs8%20low%20pres%20warn%20%5D&c=%7B203%7D3333266aa655a556655a634b2cb4acb332aaaa554b32b52d534%20%5B%20g004_433.92M_2500k.cs8%20low%20pres%20warn%20%5D&c=%7B206%7D3333266aa655a556655a65a5965a565999556a95556656a5aa68%20%5B%20g004_433.92M_2500k.cs8%20low%20pres%20warn%20%5D&c=%7B206%7D33333335532ad2ab32ad32d2cb2d2b2cccaab54aaab32b25aa68%20%5B%20g004_433.92M_2500k.cs8%20low%20pres%20warn%20%5D&c=%7B203%7D33333335532ac52ccab4cb4b2c69596662ab54aaab32b52d534%20%5B%20g004_433.92M_2500k.cs8%20low%20pres%20warn%20%5D&c=%7B205%7D33333335532ad2ab32ad32d2cb2c5659994ad52aaace56a5acd0%20%5B%20g004_433.92M_2500k.cs8%20low%20pres%20warn%20%5D&c=%7B207%7D3999999aa99552ab32ad32d2cb352b2cccaab54aaab2b53352d4%20%5B%20g004_433.92M_2500k.cs8%20low%20pres%20warn%20%5D&c=%7B206%7D33333335532ad2ab32ad32d2cb35165999556a9555656a66a5a8%20%5B%20g004_433.92M_2500k.cs8%20low%20pres%20warn%20%5D&c=%7B206%7D33333335532ac556655a65a5966a565999556a9555656a66a5a8%20%5B%20g004_433.92M_2500k.cs8%20low%20pres%20warn%20%5D&c=%7B207%7D33333335532ad2ab32ad34d2b2ad4ccb3355554aaab353332d34%20%5B%20g005_433.92M_1000k.cs8%20low%20warning%20on%20%5D&c=%7B207%7D [... cut ...] d34d2b2d54ccb3355554aaab34cd4aaac%20%5B%20g005_433.92M_1000k.cs8%20low%20warning%20on%20%5D&c=%7B207%7D33333335532ad2ab32ad34d2b2cab32cccaaaab5554d354cd334%20%5B%20g005_433.92M_1000k.cs8%20low%20warning%20on%20%5D&c=%7B207%7D33333335532ad2ab32ad34d2d552b334ccaaaab5554caab52d54%20%5B%20g005_433.92M_1000k.cs8%20low%20warning%20on%20%5D&f=8h%20&z=1&cw=4) to work with the data layout.

The signal is MC but need to find the good preamble / sync word and them find the CRC somewhere, hard part to find.

ProfBoc75, 2024-04-06:
Last progress based on this:

    rtl_433 -Y autolevel -Y minmax -X "n=BMW_G3,m=FSK_PCM,s=52,l=52,r=1000,decode_dm,bits>=90" *.cs8 2>&1 | grep "\{104\}"

RF signal:

    FSK, PCM, s=l=52 µs, Differential Manchester, inverted

Data layout{104} 13 x 8 :

    Byte Position 0  1  2  3  4  5  6  7  8  9 10 11 12
    Data Layout  PP[BB II II II II PP TT F1 F2 F3]CC CC 
    Sample       00 03 c7 5e 1d 15 01 64 0f bd f8 46 66

I guess, based on Gen4/5, to be confirmed: 

- PP:{8} preamble always 0x00, not used for the CRC16
- BB:{8} Brand model,
- II:{32} ID, 
- PP:{8}: Pressure,
- TT:{8}: Temp
- F1, F2, F3: Flags that could contain battery information, flat tire, lost of pressure ... 
- CC: CRC-16 bits, poly 0x1021, init 0x0000, xorout 0xdcca, [from previous 10 bytes]. 

[bitbench ](https://triq.net/bitbench#c=%7B104%7D7ffe1c50f1757f4df82103dccc%20%5B%20g001_433.92M_2500k.cs8%20full%20pres%2032psi%20%5D&c=%7B104%7Dfffc38a1e2eaca9bf0440796d7%20%5B%20g002_433.92M_2500k.cs8%20low%2022psi%20%5D&c=%7B104%7Dfffc38a1e2ead29bf042062a27%20%5B%20g003_433.92M_2500k.cs8%20low%20pres%2022psi%20%5D&c=%7B104%7Dfffc38a1e2eac89bf04406c275%20%5B%20g003_433.92M_2500k.cs8%20low%20pres%2022psi%20%5D&c=%7B104%7Dfffc38a1e2eada9bf04407928d%20%5B%20g004_433.92M_2500k.cs8%20low%20pres%20warn%20%5D&c=%7B104%7Dfffc38a1e2ead49bf044075d24%20%5B%20g004_433.92M_2500k.cs8%20low%20pres%20warn%20BAD%20CRC%20%5D&c=%7B104%7Dfffc38a1e2ead49bf044075d25%20%5B%20g004_433.92M_2500k.cs8%20low%20pres%20warn%20%5D&c=%7B104%7Dfffc38a1e2ead69bf044060987%20%5B%20g004_433.92M_2500k.cs8%20low%20pres%20warn%20%5D&c=%7B104%7Dfffc38a1e2ead89bf04406c62f%20%5B%20g004_433.92M_2500k.cs8%20low%20pres%20warn%20%5D&c=%7B104%7Dfffc38a1e2eada9bf04407928c%20%5B%20g004_433.92M_2500k.cs8%20low%20pres%20warn%20BAD%20CRC%5D&c=%7B104%7Dfffc38a1e2da627df004073fad%20%5B%20g005_433.92M_1000k.cs8%20low%20warning%20on%20%5D&c=%7B104%7Dfffc38a1e2da627bf004060815%20%5B%20g005_433.92M_1000k.cs8%20low%20warning%20on%20%5D&c=%7B104%7Dfffc38a1e2da687df004077903%20%5B%20g005_433.92M_1000k.cs8%20low%20warning%20on%20%5D&c=%7B104%7Dfffc38a1e2da6c7bf00406c7bd%20%5B%20g005_433.92M_1000k.cs8%20low%20warning%20on%20%5D&c=%7B104%7Dfffc38a1e2da827df0040704a1%20%5B%20g005_433.92M_1000k.cs8%20low%20warning%20on%20%5D&f=PREAMBLE%208h%20BRAND%208h%20ID%2032h%20PRESSURE%3F8d%20TEMP_F%3F8h%20FLAGS%20%3F8h%20FLAGS%3F%208h%20FLAGS%20%3F%208h%20CRC%208h8h&z=1&i=true&cw=4)updated

ProfBoc75, 2024-04-07:
I drafted a decoder, in order to confirm the conversions, I need more samples please, with accurate information, like, Brand, ID, Pressure, Temp.

Billymazze, 2024-04-07:
Here's some more accurate signals with pressure and temperature. For the temperature I just measured the temp of the rim, but the tire was ~80*F. I scheduled an appointment next week with another tire shop to try to get the brand and ID for the wheel I've been recording. Hopefully they have a scanner that can read it. Thanks again for the help and quick response
[bmw gen3.zip](https://github.com/merbanan/rtl_433/files/14895491/bmw.gen3.zip)

Billymazze, 2024-04-12:
Here's the information I got from the tire shops TPMS reader. I believe that number near the top is the sensor ID's, and the front right tire is where I've been capturing signals. Some of the older ones could be from the front left.

[image]
[image]
[image]

ProfBoc75, 2024-04-13:
Hi @Billymazze:

Very good, thanks to the IDs, now I'm able to align the data properly and a very nice CRC-16/XMODEM, poly 0x1021, init 0x0000.
The format is not the same as last gen4 and gen5 version.

    rtl_433 -Y autolevel -Y minmax -X "n=BMW_G3,m=FSK_PCM,s=52,l=52,r=1000,preamble=cccd,decode_dm,bits>=200" g011*.cs8 2>&1 | grep codes

    codes     : {89}1c50f175 85 45 f8 02 03 7342 8
    codes     : {89}1c50f175 85 44 f8 02 03 05f6 8
    codes     : {89}1c50f175 85 45 f8 02 03 7342 8
    codes     : {89}1c50f175 85 45 f8 02 03 7342 8
    codes     : {89}1c50f175 85 44 f8 02 03 05f6 8
    codes     : {89}1c50f175 85 44 f8 02 03 05f6 8
    codes     : {89}1c50f175 85 44 f8 02 03 05f6 8
    codes     : {89}1c50f175 85 45 f8 02 03 7342 8
    codes     : {89}1c50f175 85 45 f8 02 03 7342 8
    codes     : {89}1c50f175 85 45 f8 02 03 7342 8
    codes     : {89}1c50f175 85 45 f8 02 03 7342 8
    codes     : {89}1c50f175 85 44 f8 02 03 05f6 8
    codes     : {89}1c50f175 85 44 f8 02 03 05f6 8
    codes     : {89}1c50f175 85 45 f8 02 03 7342 8

Data layout:{89}

    [II II II II 11 22 33 44 55]CC CC 8
     1c 50 f1 75 85 45 f8 02 03 73 42 8

- II:{32} ID, 0x1c50f175 = 475066741 --> Same the picture
- 11 to 55: data to guess for temp, pressure, battery_ok, and any other flags ...
- CC:{16}: CRC-16/XMODEM, [from previous 9 bytes], poly 0x1021, init 0x0000
- 8: useless trailing bit ...

To be continued ...

Billymazze, 2024-04-13:
Wow, I've been bang my head on the desk for weeks trying to make since of this and you did it instantly. Amazing, thanks again for the help.

ProfBoc75, 2024-04-13:
> Wow, I've been bang my head on the desk for weeks trying to make since of this and you did it instantly. Amazing, thanks again for the help.

Someone said:
- Ever tried. Ever failed. No matter. Try again. Fail again. Fail better. 

We can add:
- Then finally succeed !

Anyway, I also spent a bit of time on it and was already very close within 1 bit and 1 inversion from my previous findings.

I converted your ID, shown on the photos, from decimal to hexa data, then I played with bitbench and the raw message to shift bit one by one until the Manchester decode show me the ID in hexa data value. This let me found the preamble = 0xcccccd

From that, I played with `reveng` to guess the CRC, I knew it was something with CRC-16 poly 0x1021 but the init value was not classic in my previous findings, so I already knew that I didn't get to good data layout, but here, it was 0x0000, so this one is the good one.

Now I'm updating my Excel table and the bitbench to guess the temp,  pressure and so on... And hopefully be able to propose a decoder soon, it's already drafted, just need to update it with this last finding on the rf signal decode part and then decode the data.

ProfBoc75, 2024-04-13:
@Billymazze:

From your values, I guess the pressure, and the temp. About the warming flags it's not clear so I let the flags to be guessed later.

I just create a PR to let you test and confirm my assumption please.

My last update:

```
Data layout{89} 11 x 8:

    Byte Position  0  1  2  3  4  5  6  7  8  9 10 11
    Data Layout  [II II II II PP TT F1 F2 F3]CC CC 8
    Sample        1c 50 f1 75 85 45 f8 02 03 73 42 8

- II:{32} ID, hexa 0x1c50f175 or decimal value 475066741
- PP:{8}: Tire pressure, PSI = (PP - 43) * 0.363 or kPa = ( PP - 43 ) * 2.5
- TT:{8}: Temperature in C offset 50 
- F1, F2, F3: Flags that could contain battery information, flat tire, lost of pressure ...
- CC: CRC-16 bits, poly 0x1021, init 0x0000 [from previous 9 bytes].
- 8: useless trailing bit
```

Try from rtl_433 feat-tpms_bmw_gen3 branch and let me know if pressure and temperature values are matching.

    rtl_433 -R 257 -Y minmax -C customary -s 1024k

Pressure and Temperature formulas can be adjusted accordingly.

If you want to test from existing older rtl_433 version, try:

    rtl_433 -s 1024k -Y minmax -X "n=BMW_G3,m=FSK_PCM,s=52,l=52,r=1000,preamble=cccd,decode_dm,bits>=190"

Collect all codes with 88 / 89 bits and import them into [bitbench](https://triq.net/bitbench#c=%7B88%7D1c50f1757f4df82103dccc%20%5B%20g001%20full%20pres%2032psi%20%20%5D&c=%7B89%7D1c50f175654df82203cb6b8%20%5B%20g002%20low%2022psi%20%20%5D&c=%7B89%7D1c50f175694df8210315138%20%5B%20g003%20low%20pres%2022psi%20%20%5D&c=%7B89%7D1c50f175644df82203613a8%20%5B%20g003%20low%20pres%2022psi%20%20%5D&c=%7B89%7D1c50f175654df82203cb6b8%20%5B%20g003%20low%20pres%2022psi%20%20%5D&c=%7B89%7D1c50f1756d4df82203c9468%20%5B%20g004%20low%20pres%20warn%20%20%5D&c=%7B89%7D1c50f1756a4df82203ae920%20%5B%20g004%20low%20pres%20warn%20%20%5D&c=%7B89%7D1c50f1756a4df82203ae928%20%5B%20g004%20low%20pres%20warn%20%20%5D&c=%7B89%7D1c50f1756b4df8220304c38%20%5B%20g004%20low%20pres%20warn%20%20%5D&c=%7B89%7D1c50f1756c4df8220363178%20%5B%20g004%20low%20pres%20warn%20%20%5D&c=%7B89%7D1c50f1756d4df82203c9460%20%5B%20g004%20low%20pres%20warn%20%20%5D&c=%7B89%7D1c50f16d313ef802039fd68%20%5B%20g005%20low%20warning%20on%20%20%5D&c=%7B89%7D1c50f16d313df80203040a8%20%5B%20g005%20low%20warning%20on%20%20%5D&c=%7B89%7D1c50f16d343ef80203bc818%20%5B%20g005%20low%20warning%20on%20%20%5D&c=%7B89%7D1c50f16d363df8020363de8%20%5B%20g005%20low%20warning%20on%20%20%5D&c=%7B89%7D1c50f16d413ef8020382508%20%5B%20g005%20low%20warning%20on%20%20%5D&c=%7B89%7D1c50f1753643f8020394378%20%5B%20g006%204psi%2027.58kpa%2090F%2032C-warn-on%20%5D&c=%7B89%7D1c50f1754544f80203364e8%20%5B%20g007%209psi%2062.05kpa%2090F%2032C-warn-on%20%5D&c=%7B89%7D1c50f1754443f80203cd328%20%5B%20g007%209psi%2062.05kpa%2090F%2032C-warn-on%20%5D&c=%7B89%7D1c50f1755444f8020398458%20%5B%20g008%2015psi%20103.42kpa%2090F%2032C-warn-on%20%5D&c=%7B89%7D1c50f1756544f802033efa8%20%5B%20g009%2021psi%20144.79kpa%2090F%2032C-warn-on%20%5D&c=%7B89%7D1c50f1756444f8020394ab8%20%5B%20g009%2021psi%20144.79kpa%2090F%2032C-warn-on%20%5D&c=%7B89%7D1c50f1758244f8020362228%20%5B%20g010%2032psi%20220.63kpa%2090F%2032C-warn-off%20%5D&c=%7B89%7D1c50f1758345f80203bec78%20%5B%20g010%2032psi%20220.63kpa%2090F%2032C-warn-off%20%5D&c=%7B89%7D1c50f1758445f80203d9138%20%5B%20g010%2032psi%20220.63kpa%2090F%2032C-warn-off%20%5D&c=%7B89%7D1c50f1758344f80203c8738%20%5B%20g010%2032psi%20220.63kpa%2090F%2032C-warn-off%20%5D&c=%7B89%7D1c50f1758345f80203bec70%20%5B%20g010%2032psi%20220.63kpa%2090F%2032C-warn-off%20%5D&c=%7B89%7D1c50f1758245f8020314968%20%5B%20g010%2032psi%20220.63kpa%2090F%2032C-warn-off%20%5D&c=%7B89%7D1c50f1758545f8020373428%20%5B%20g011%2032psi%20220.63kpa%2090F%2032C-warn-off%20%5D&c=%7B89%7D1c50f1758544f8020305f68%20%5B%20g011%2032psi%20220.63kpa%2090F%2032C-warn-off%20%5D&f=32d%20PRESSURE%3A%208d%20TEMP%3A%208d%20FLAGS%208b%20FLAGS%208b%20FLAGS%208b%2016h%20x%0A&z=1&cw=4), you still have to calculate, but it's easier to get figures.

Billymazze, 2024-04-13:
I'm still having trouble getting the RTL-433 working, do you have any recommendations for install instructions, and is it better if I run it on a Linux? Most instructions I find online are very vague.

ProfBoc75, 2024-04-13:
@Billymazze : you can follow this https://triq.org/rtl_433/BUILDING.html

Linux is easier to build and  run rtl_433.

Billymazze, 2024-04-14:
Well it took all day, but I got RTL-433 working. but every time I try to compile/build it with the updated RTL-433 version in the pull request I fails to launch. I highly drought it an issue on your end, I just can't get it to compile/build right. Any chance you can send me a precompiled version like [this one](https://github.com/merbanan/rtl_433/releases/download/23.11/rtl_433-win-msvc-x64-23.11.zip) from the merbanan latest release. I can launch and run those perfectly without issue in command prompt.

Billymazze, 2024-04-14:
I can run the Pre-Release Nightly versions, but they don't appear to not have added support for the Gen3 tpms. Current issues with linux is it refuses to run. I just get this error, like I didn't even install it. Might be an issue with WSL linux.
[image]

ProfBoc75, 2024-04-14:
@Billymazze : I'm also using WSL and Debian.

I guess your last cmake command failed because you should run cmake install under sudo:

    git clone https://github.com/merbanan/rtl_433.git
    cd rtl_433/
    git fetch origin
    git checkout feat-tpms_bmw_gen3
    cmake -B build
    sudo cmake --build build --target install

If not already done, you need also to install [usbipd ](https://learn.microsoft.com/en-us/windows/wsl/connect-usb)at MS Windows level to let attach the usb rtl-sdr dongle to WSL linux distribution.

**Then a session of WSL must be opened to let attach the usb device.**

From Windows Command Prompt (run as Administrator) enter `usbipd wsl list` (each time you plug the dongle or restart the pc):

```console
> usbipd wsl list
BUSID  VID:PID    DEVICE                                                        STATE
1-3    0bda:2838  RTL2838UHIDIR                                                 Not attached
1-7    322e:202c  USB2.0 HD UVC WebCam                                          Not attached
1-14   8087:0026  Intel(R) Wireless Bluetooth(R)                                Not attached
```

Then attach with `usbipd wsl attach --busid x-x`
Replace the x-x by the number from previous command line.

```console
> usbipd wsl attach --busid 1-3
usbipd: info: Using default WSL distribution 'Debian'; specify the '--distribution' option to select a different one.
```

_The message is because I have several WSL distributions installed,_

And because of this behavior, you will have to run rtl_433 under sudo also like you tried
or you have to change the permission with chmod:

Get the usb device num:

```console
$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Here Bus 001 and Device 003.

```console
$ ls -al /dev/bus/usb/001/003
crw------- 1 root root 189, 2 Apr 14 23:24 /dev/bus/usb/001/003

$ sudo chmod 666 /dev/bus/usb/001/003
$ ls -al /dev/bus/usb/001/003
crw-rw-rw- 1 root root 189, 2 Apr 14 23:27 /dev/bus/usb/001/003
```

Then rtl_433 should work without sudo.

I prefer sudo and I do not change the permission, because each time you unplug or restart wsl you will have to change the permission.

If you want, you can also update your linux distribution to add your user id into sudo group to avoid password each time you need sudo permission.

Billymazze, 2024-04-14:
Holy sh!t, your a genius! I can't believe you figured out that I was missing the sudo command just from that one screen shot. THANK YOU!, I'll get you some test result soon.

Billymazze, 2024-04-15:
Well I screwed it up again... I got it to run rtl_433, but forgot I needed to build a WSL 2 kernel in linux to make the usb work. I got excited and thought it was just my fault for messing up the linux distro, so I uninstalled it and got a fresh ubuntu install. But now I'm back to square one, something is wrong with my compiler/build. I followed your instructions, but every time I run the "sudo cmake --build build --target install" I get tons of errors. here's some screen shots, thinking I should open a new issue for this problem I'm having. Also should I try switching over to Debian and try again?
[image]
[image]
[image]
[image]

Billymazze, 2024-04-15:
I got it working again, this is what I did from a fresh install of Debian. Idk why it worked, but it did.
`sudo apt-get update`
`sudo apt-get upgrade`
`sudo apt-get install libtool libusb-dev librtlsdr-dev rtl-sdr build-essential cmake pkg-config`
`sudo apt install git`
`mkdir rtl`
`cd rtl`
`git clone https://github.com/merbanan/rtl_433.git`
`cd rtl_433/`
`git fetch origin`
`git checkout feat-tpms_bmw_gen3`
`mkdir build`
`cd build`
`sudo cmake ..`
`sudo make`

Billymazze, 2024-04-15:
I'm pretty sure my gauge is a little off, but the pressure readings seem pretty close. I'll try to get a new gauge next week. But the temperature is definitely off, tire temp was 25*C, tpms decoder was reading a constant 15*C. 
- 135kpa/132.5kpa
- 83kpa/82.5kpa
- 59kpa/60kpa
- 63kpa/62.5kpa
- 119kpa/117.5kpa
- 130kpa/130kpa
- 138kpa/140kpa
- 145kpa/145kpa
- 172kpa/175kpa
- 206kpa/202.5kpa
- 215kpa/215kpa
- 255kpa/255kpa
- my tire filler with electronic gauge VS/ tpms decoder pressure.

ProfBoc75, 2024-04-15:
@Billymazze : Good values !

Yes, mostly close and many are matching, this confirmed my assumption.
Waiting for Temp to be confirmed.

Billymazze, 2024-04-15:
I'd say temp is off by 10*c. During my testing it was reading 15*c in the decoder, while the actual temp of of the wheel was 25*c. Not sure how to test it, other than to drive around for a while and compare the decoder vs the wheel temp with my IR temp gun. Also I can capture some Gen2 bmw tpms signal if your interested in making another decoder.

ProfBoc75, 2024-04-15:
@Billymazze: I updated the decoder to change the temperature offset from 50 to 40. You can try with last commit:

Goto your rtl_433 folder then:

    git pull
    cd build
    make
    sudo make install

 And try ...

ProfBoc75, 2024-04-21:
@Billymazze : did you try ? 

Does someone test my pr and can give some feedback, please ?

Billymazze, 2024-04-21:
Sorry, I've been busy studying for finals. I'll give it shot later today.

Billymazze, 2024-04-22:
Had issues updating it, reinstalled Debian and rtl_433. but the temperature and pressure are perfect. Temp started at 20*c and increased to 25*c after driving around for a while. Matched exactly what I measured with my IR temp gun. Also pressures matched perfectly with my new pressure gauge. I'd say it ready to go, thanks again @ProfBoc75 for all the help.

closed 2024-04-23 by Billymazze