merbanan/rtl_433 #2895: Old version installing?

hessel-a, 2024-04-07:
Do I miss something? When running under Debian (bookworm). I run "apt-get install rtl-433" it installs release 22.11 . But I suspects release 23.11

So what do I do wrong.
Thanks

ProfBoc75, 2024-04-07:
This is the current stable version for Debian 12: https://packages.debian.org/search?keywords=rtl-433&searchon=names&suite=all&section=all

If you want a more recent version, you may update the apt source list (not recommended can make your system unstable) or follow the [build](https://github.com/merbanan/rtl_433/blob/master/docs/BUILDING.md) guide.
This will require few packages to be installed like RTL-SDR and/or SoapSDR ... and could take sometime to compile.

zuckschwerdt, 2024-04-07:
You can also check our [releases](https://github.com/merbanan/rtl_433/releases), there should be a build you can use as drop-in replacement (keep the apt package installed for the dependencies). Unzip and either run the build (`./rtl_433`) or install it to `/usr/local/bin` to override the apt one.

hessel-a, 2024-04-07:
Thanks. I learned something new. Did not expect that the version is 2 year behind.

gdt, 2024-06-02:
Answered question

closed 2024-06-02 by gdt