merbanan/rtl_433 #2902: Self Powered Switch Device 433mhz

adminy, 2024-04-15:
here is some logs from running `rtl_433 -A` and pressing the button a couple of times. No json, so its probably not a registered device yet. let me know how I can collect info for it and / or decode. thanks.

zuckschwerdt, 2024-04-15:
If you look at the "count" fields the numbers are too low. We expect more data, like 32 bits or even much more.
From what we know of self-powered devices the transmissions are short and fast.

Re-run with a higher sampe rate, e.g. `-s 1024k`
Also grab one sample, verify it looks promissing in https://triq.org/pdv/ and upload here as zip, see https://triq.org/rtl_433/ANALYZE.html

adminy, 2024-04-15:
[scans.tar.gz](https://github.com/merbanan/rtl_433/files/14984137/scans.tar.gz)

looks promising on the verify step, but not totally sure. still feels like its not consistent across scans. this is data from QX-305 only.

zuckschwerdt, 2024-04-15:
Actually a very simple OOK signal with 25 bits at 33/100 µs. This will decode the data:
`rtl_433 -R 0 -X 'n=name,m=OOK_PWM,s=33,l=100,g=150,r=1500'`

If it only has one function (single press on single button) then use this to only accept that button:
`rtl_433 -R 0 -X 'n=name,m=OOK_PWM,s=33,l=100,g=150,r=1500,match={25}573ace8,countonly'`

See https://github.com/merbanan/rtl_433/tree/master/conf example on how to write a documented conf file.
PR or paste us the finished file if you like to.

adminy, 2024-04-15:
I'm not sure that's it though ... https://logpaste.com/3X0Y7K6j, Also the single press command literally gives back nothing.

adminy, 2024-04-15:
https://github.com/merbanan/rtl_433/compare/master...adminy:rtl_433:patch-1 Something like this then?

closed 2024-04-16 by adminy

edco, 2024-05-05:
I have a QX-304 which works with qx-30x.conf. Unfortunately I just got the one, single gang button to test, so I can't compare the data from different devices/buttons, but would it be reasonable to assume this is a 1527 code? In that case I'm thinking we'd ignore the final bit, and split the remainder into 20 and 4 bits:
```
  bits       >= 24,
  unique,
  get         = get=@0:{20}:id,
  get         = get=@20:{4}:button
```
Configured that way my button returns a button code of 14(dec) the same as the sample output above, but the id value is different (the raw 25 bit data is 583c2e8). This config should make it easier to quickly setup multiple switches and potentially cater to multi-gang switches too.

zuckschwerdt, 2024-05-05:
Yes, that is reasonable. The EV1527 data bits (button code) would be the last 4 bits, and the sync is trailing.
You can also add `bits <= 25,` too.

Document your findings along with the changes and PR if you like.

andreaskoelsch, 2025-06-03:
I got a few QX-305 self-powered switches and was able to capture them using rtl_433 given the instructions above. However, my single-button switches sometimes report different data. How can I debug this? Here's an example of a single switch I pressed multiple times

```
{"time" : "2025-06-03 08:26:04", "model" : "QX-30X", "count" : 1, "num_rows" : 4, "len" : 25, "data" : "932a1e8", "id" : 602785, "button" : 9}
{"time" : "2025-06-03 08:26:06", "model" : "QX-30X", "count" : 1, "num_rows" : 3, "len" : 24, "data" : "26543d", "id" : 156995, "button" : 2}
{"time" : "2025-06-03 08:26:08", "model" : "QX-30X", "count" : 2, "num_rows" : 3, "len" : 24, "data" : "26543d", "id" : 156995, "button" : 2}
{"time" : "2025-06-03 08:26:10", "model" : "QX-30X", "count" : 2, "num_rows" : 7, "len" : 25, "data" : "932a1e8", "id" : 602785, "button" : 9}
{"time" : "2025-06-03 08:26:10", "model" : "QX-30X", "count" : 1, "num_rows" : 1, "len" : 25, "data" : "e665b28", "id" : 943707, "button" : 14}
{"time" : "2025-06-03 08:26:12", "model" : "QX-30X", "count" : 2, "num_rows" : 5, "len" : 25, "data" : "932a1e8", "id" : 602785, "button" : 9}
{"time" : "2025-06-03 08:26:13", "model" : "QX-30X", "count" : 1, "num_rows" : 5, "len" : 25, "data" : "932a1e8", "id" : 602785, "button" : 9}
{"time" : "2025-06-03 08:26:15", "model" : "QX-30X", "count" : 1, "num_rows" : 5, "len" : 25, "data" : "932a1e8", "id" : 602785, "button" : 9}
{"time" : "2025-06-03 08:26:18", "model" : "QX-30X", "count" : 1, "num_rows" : 1, "len" : 25, "data" : "932a1e8", "id" : 602785, "button" : 9}
{"time" : "2025-06-03 08:26:24", "model" : "QX-30X", "count" : 1, "num_rows" : 6, "len" : 25, "data" : "932a1e8", "id" : 602785, "button" : 9}
{"time" : "2025-06-03 08:26:27", "model" : "QX-30X", "count" : 1, "num_rows" : 8, "len" : 25, "data" : "932a1e8", "id" : 602785, "button" : 9}
{"time" : "2025-06-03 08:26:37", "model" : "QX-30X", "count" : 1, "num_rows" : 3, "len" : 25, "data" : "932a1e8", "id" : 602785, "button" : 9}
{"time" : "2025-06-03 08:26:41", "model" : "QX-30X", "count" : 1, "num_rows" : 5, "len" : 26, "data" : "c9950f4", "id" : 825680, "button" : 12}
{"time" : "2025-06-03 08:26:58", "model" : "QX-30X", "count" : 1, "num_rows" : 2, "len" : 24, "data" : "26543d", "id" : 156995, "button" : 2}
{"time" : "2025-06-03 08:27:02", "model" : "QX-30X", "count" : 2, "num_rows" : 34, "len" : 25, "data" : "932a1e8", "id" : 602785, "button" : 9}
{"time" : "2025-06-03 08:27:30", "model" : "QX-30X", "count" : 2, "num_rows" : 6, "len" : 26, "data" : "c9950f4", "id" : 825680, "button" : 12}
```

zuckschwerdt, 2025-06-03:
If you look at the codes in [a BitBench](https://triq.org/bitbench/#c=%7B25%7D932a1e8&c=%7B24%7D26543d&c=%7B24%7D26543d&c=%7B25%7D932a1e8&c=%7B25%7De665b28&c=%7B25%7D932a1e8&c=%7B25%7D932a1e8&c=%7B25%7D932a1e8&c=%7B25%7D932a1e8&c=%7B25%7D932a1e8&c=%7B25%7D932a1e8&c=%7B25%7D932a1e8&c=%7B26%7Dc9950f4&c=%7B24%7D26543d&c=%7B25%7D932a1e8&c=%7B26%7Dc9950f4&f=v&z=1&cw=4) you can see that a few bits are inserted or missing.
Try adding `repeats=3,` or such in the flex decoder.

andreaskoelsch, 2025-06-03:
Thanks for the quick response. If I set `repeats=3`, I get no detection. With `repeats=2`, detection is very unreliable. However, when the signal is detected, it is consistently the `932a1e8` one from above.

I'm just trying to figure out if the protocol is correct for my switches using the [instructions from here](https://triq.org/rtl_433/ANALYZE.html). When looking at a signal like [this](https://triq.org/pdv/#AAB011070100500138003000B8024803E427148555+AAB01C070100500138003000B8024803E42714829090A090B0B0C0A0A0A0B555+AAB01C070100500138003000B8024803E42714A09090A090B0B0C0A0A0A0B555+AAB018070100500138003000B8024803E42714A09090A090B0B09655) I wonder if long pulses should not be longer than 100 us?

andreaskoelsch, 2025-07-26:
This is just a comment for people searching for the model WHDZ01. They use the same protocol!