merbanan/rtl_433 #2905: Issues with a JYWDJ-009 device using esperanza_ews.c

iphands, 2024-04-17:
Hi folks. I've been using and loving rtl_433! Thanks!
I recently purchased a few  JYWDJ-009 devices to track temp and humidity.
https://www.aliexpress.us/item/3256806481245829.html

The price was great! and they seemed to be already working with rtl_433.
I have an issue though... The humidity is always reported as 0% :( and I need humidity data.

Added a small extra bit to print out the `b[]` byte array that esperanza_ews.c uses...
I see:
```
DEBUG: b[]: 15 14 26 00 09
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-04-17 02:47:49                    Protocol  : 35
model     : Esperanza-EWS ID        : 21
Channel   : 2            Battery   : 1             Temperature: 67.20 F      Humidity  : 0 %           Integrity : CRC
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -10.8 dB     SNR       : 23.6 dB       Noise     : -34.4 dB
[pulse_slicer_ppm] Esperanza EWS
codes     : {0}0, {0}0, {42}05450980024, {0}0, {42}05450980024, {0}0, {42}05450980024, {0}0, {42}05450980024, {0}0, {42}05450980024, {0}0, {42}05450980024, {0}0
```

Im not wondering ... is this some variant that needs to be decoded differently?
Or is device legit not emitting any humidity data?

I will upload a few cu8 files and whatever is needed for debugging. If someone can help me debug a bit, I'd be super happy to put up a PR and enable this device for myself and the community!

Thanks!

iphands, 2024-04-17:
Here is a capture with -S all I think the suggestions on pdv match closely to what we see in the existing decoder file:
[esperanza_cu8s.tar.gz](https://github.com/merbanan/rtl_433/files/15004559/esperanza_cu8s.tar.gz)
```
Pulses	79× 480.4 ±8.0 µs
Gaps	51× 2036.2 ±34.0 µs	23× 3990.6 ±32.0 µs	4× 8066.0 ±2.0 µs
Periods	23× 4471.1 ±38.0 µs	4× 8545.0 ±2.0 µs	51× 2516.7 ±36.0 µs
Timings	79× 480.4 ±8.0 µs	23× 3990.6 ±32.0 µs	5× 8453.6 ±970.0 µs	51× 2036.2 ±34.0 µs
```

I think this step in the process is mainly here to ensure that the `r_device` struct settings are proper eh?
```
 133 r_device const esperanza_ews = {
 134         .name        = "Esperanza EWS",
 135         .modulation  = OOK_PULSE_PPM,
 136         .short_width = 2000,
 137         .long_width  = 4000,
 138         .gap_limit   = 4400,
 139         .reset_limit = 9400,
```

Im a n00b though so there might be more I can do. Please give a nudge if you can.

iphands, 2024-04-17:
I was wondering it this device has more data to decode than the one that was originally used to write `esperanza_ews.c`.
But  given that the CRC check passes Im worried that this might not be the case. It seems like this CRC check would fail if there was data that the device sends but the decoder was ignoring/discarding the data... right?

iphands, 2024-04-17:
Unsure if its needed with the cu8 file... but here is the -vvvv pulse data
```
4
DEBUG: b[]: 05 47 09 80 03 00
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-04-17 03:13:56                    Protocol  : 35
model     : Esperanza-EWS ID        : 5
Channel   : 1            Battery   : 1             Temperature: 141.10 F     Humidity  : 0 %           Integrity : CRC
Modulation: ASK          Freq      : 434.0 MHz
RSSI      : -0.4 dB      SNR       : 32.7 dB       Noise     : -33.1 dB
[pulse_slicer_ppm] Esperanza EWS
codes     : {0}0, {0}0, {42}05470980030, {0}0, {42}05470980030, {0}0, {42}05470980030, {0}0, {42}05470980030, {0}0, {42}05470980030, {0}0, {42}05470980030, {0}0
Pulse data: 266 pulses
[  0] Pulse:  124, Gap: 2017, Period: 2141
[  1] Pulse:  118, Gap: 2020, Period: 2138
[  2] Pulse:  116, Gap:  509, Period:  625
[  3] Pulse:  116, Gap:  509, Period:  625
[  4] Pulse:  117, Gap:  509, Period:  626
[  5] Pulse:  117, Gap:  509, Period:  626
[  6] Pulse:  116, Gap:  508, Period:  624
[  7] Pulse:  118, Gap: 1012, Period: 1130
[  8] Pulse:  117, Gap:  509, Period:  626
[  9] Pulse:  116, Gap:  997, Period: 1113
[ 10] Pulse:  117, Gap:  509, Period:  626
[ 11] Pulse:  117, Gap: 1012, Period: 1129
[ 12] Pulse:  117, Gap:  509, Period:  626
[ 13] Pulse:  117, Gap:  508, Period:  625
[ 14] Pulse:  117, Gap:  509, Period:  626
[ 15] Pulse:  117, Gap: 1013, Period: 1130
[ 16] Pulse:  116, Gap:  997, Period: 1113
[ 17] Pulse:  117, Gap:  998, Period: 1115
[ 18] Pulse:  115, Gap:  509, Period:  624
[ 19] Pulse:  117, Gap:  524, Period:  641
[ 20] Pulse:  117, Gap:  509, Period:  626
[ 21] Pulse:  116, Gap:  509, Period:  625
[ 22] Pulse:  117, Gap:  997, Period: 1114
[ 23] Pulse:  117, Gap:  524, Period:  641
[ 24] Pulse:  117, Gap:  509, Period:  626
[ 25] Pulse:  116, Gap:  997, Period: 1113
[ 26] Pulse:  117, Gap:  997, Period: 1114
[ 27] Pulse:  117, Gap:  524, Period:  641
[ 28] Pulse:  117, Gap:  508, Period:  625
[ 29] Pulse:  117, Gap:  509, Period:  626
[ 30] Pulse:  117, Gap:  508, Period:  625
[ 31] Pulse:  118, Gap:  524, Period:  642
[ 32] Pulse:  116, Gap:  509, Period:  625
[ 33] Pulse:  117, Gap:  508, Period:  625
[ 34] Pulse:  117, Gap:  509, Period:  626
[ 35] Pulse:  117, Gap:  524, Period:  641
[ 36] Pulse:  117, Gap:  508, Period:  625
[ 37] Pulse:  118, Gap:  508, Period:  626
[ 38] Pulse:  117, Gap:  509, Period:  626
[ 39] Pulse:  116, Gap:  524, Period:  640
[ 40] Pulse:  118, Gap:  997, Period: 1115
[ 41] Pulse:  116, Gap:  997, Period: 1113
[ 42] Pulse:  117, Gap:  509, Period:  626
[ 43] Pulse:  117, Gap:  524, Period:  641
[ 44] Pulse:  117, Gap: 2018, Period: 2135
[ 45] Pulse:  118, Gap: 2020, Period: 2138
[ 46] Pulse:  116, Gap:  508, Period:  624
[ 47] Pulse:  117, Gap:  511, Period:  628
[ 48] Pulse:  115, Gap:  509, Period:  624
[ 49] Pulse:  117, Gap:  508, Period:  625
[ 50] Pulse:  117, Gap:  508, Period:  625
[ 51] Pulse:  118, Gap: 1012, Period: 1130
[ 52] Pulse:  117, Gap:  508, Period:  625
[ 53] Pulse:  117, Gap:  998, Period: 1115
[ 54] Pulse:  116, Gap:  509, Period:  625
[ 55] Pulse:  117, Gap: 1011, Period: 1128
[ 56] Pulse:  118, Gap:  509, Period:  627
[... 186 lines cut ...]
[243] Pulse:  116, Gap:  524, Period:  640
[244] Pulse:  117, Gap:  509, Period:  626
[245] Pulse:  117, Gap:  996, Period: 1113
[246] Pulse:  118, Gap:  997, Period: 1115
[247] Pulse:  116, Gap:  525, Period:  641
[248] Pulse:  116, Gap:  509, Period:  625
[249] Pulse:  117, Gap:  508, Period:  625
[250] Pulse:  117, Gap:  509, Period:  626
[251] Pulse:  117, Gap:  524, Period:  641
[252] Pulse:  117, Gap:  508, Period:  625
[253] Pulse:  117, Gap:  509, Period:  626
[254] Pulse:  117, Gap:  509, Period:  626
[255] Pulse:  117, Gap:  523, Period:  640
[256] Pulse:  117, Gap:  510, Period:  627
[257] Pulse:  116, Gap:  508, Period:  624
[258] Pulse:  118, Gap:  508, Period:  626
[259] Pulse:  117, Gap:  524, Period:  641
[260] Pulse:  117, Gap:  997, Period: 1114
[261] Pulse:  117, Gap:  997, Period: 1114
[262] Pulse:  117, Gap:  508, Period:  625
[263] Pulse:  117, Gap:  494, Period:  611
[264] Pulse:  116, Gap: 2020, Period: 2136
[265] Pulse:  117, Gap: 2501, Period: 2618
[pulse_slicer_ppm] Esperanza EWS
```

zuckschwerdt, 2024-04-17:
The `{42}` is the number of received bits. We shift the first two bits off and get 40 bits which are 5 bytes -- what you see in `b[]`.
Your example shows `DEBUG: b[]: 15 14 26 00 09` which has a `00` for humidity. Do all codes you get show that? Does the temperature decode correctly? Then maybe humidity is an option that is missing in the hardware yo got?

ProfBoc75, 2024-04-17:
Same findings for me, I guess the humidity part is missing from these devices, all samples are 0x00 at humidity position.

Open one box and see if you have humidity sensor soldered to confirm. Could be something like this:

[image]

the [bitbench](https://triq.net/bitbench#c=%7B42%7D%2000%2041%2065%2075%2001%20c%20%5B%20sample%20from%20rtl_433_tests%20%5D&c=%7B42%7D%2000%2041%20a5%2075%2003%204%20%5B%20sample%20from%20rtl_433_tests%20%5D&c=%7B42%7D%2000%2041%20e5%2071%2003%208%20%5B%20sample%20from%20rtl_433_tests%20%5D&c=%7B42%7D%2000%2042%2025%2071%2002%208%20%5B%20sample%20from%20rtl_433_tests%20%5D&c=%7B42%7D%2000%2042%2065%2071%2003%200%20%5B%20sample%20from%20rtl_433_tests%20%5D&c=%7B42%7D%2000%2042%20e5%206d%2002%200%20%5B%20sample%20from%20rtl_433_tests%20%5D&c=%7B42%7D%2000%2043%2025%206d%2002%204%20%5B%20sample%20from%20rtl_433_tests%20%5D&c=%7B42%7D%2000%2043%2065%206d%2003%20c%20%5B%20sample%20from%20rtl_433_tests%20%5D&c=%7B42%7D%2000%2043%20a5%2069%2000%200%20%5B%20sample%20from%20rtl_433_tests%20%5D&c=%7B42%7D%2000%2050%2029%2069%2002%208%20%5B%20sample%20from%20rtl_433_tests%20%5D&c=%7B42%7D%2000%2050%2069%2069%2003%200%20%5B%20sample%20from%20rtl_433_tests%20%5D&c=%7B42%7D%2000%2053%20a5%2069%2003%208%20%5B%20sample%20from%20rtl_433_tests%20%5D&c=&c=%7B42%7D%2000%2053%20e5%2069%2002%200%20%5B%20Sample%20from%20original%20decoder%20%5D&c=&c=%7B42%7D%2034%2085%204d%2080%2003%204%20%5B%20JYWDJ-009%20device%20g001_433.92M_250k.cu8%20%5D&c=%7B42%7D%2009%2006%200d%2080%2001%200%20%5B%20JYWDJ-009%20device%20g002_433.92M_250k.cu8%20%5D&c=%7B42%7D%2009%2006%208d%2080%2002%200%20%5B%20JYWDJ-009%20device%20g003_433.92M_250k.cu8%20%5D&c=&c=%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20%5B%20No%20JYWDJ-009%20device%20found%20in%20g003_433.92M_250k.cu8%20%5D&c=&c=%7B42%7D%2009%2007%200d%2080%2003%20c%20%5B%20JYWDJ-009%20device%20g005_433.92M_250k.cu8%20%5D&f=ID%3A%208h%20%3F%202b%20CHANNEL%3A%202d%20TEMP%3A%204h%204h%204h%20HUM%3A%204h%204h%20FLAGS%3A%204b%20CRC%204h&z=1&s=-2&cw=4) with the samples, notice that g004.... doesn't contain esperanza ews signal, but something else.

iphands, 2024-04-17:
> Same findings for me, I guess the humidity part is missing from these devices, all samples are 0x00 at humidity position.
> Open one box and see if you have humidity sensor soldered to confirm. Could be something like this:

Yeah I opened one first thing and could not see any sensor. I was unsure if there might be some super tiny SMC thing that would do it... but I guess that was just wishful thinking.

I asked the seller if they can provide any technical details on the humidity sensor component though.
Im going to try and have the company refund/exchange and hopefully change their listing unless they can prove the sensor hw is present. It would be awesome have a reliable source for these devices with humidity... hopefully they comply/change.

Closing this :(

closed 2024-04-17 by iphands