merbanan/rtl_433 #2908: Motorized blinds 433 remote

adminy, 2024-04-19:
```
Detected OOK package    2024-04-19 22:45:49
Analyzing pulses...
Total count:  451,  width: 615.21 ms            (629976 S)
Pulse width distribution:
 [ 0] count:    1,  width: 1040 us [1040;1040]  (1065 S)
 [ 1] count:  313,  width:  378 us [376;381]    ( 387 S)
 [ 2] count:  127,  width:  722 us [721;723]    ( 739 S)
 [ 3] count:   10,  width: 4769 us [4768;4771]  (4883 S)
Gap width distribution:
 [ 0] count:   11,  width: 1492 us [1491;1494]  (1528 S)
 [ 1] count:  310,  width:  716 us [713;731]    ( 733 S)
 [ 2] count:  119,  width:  333 us [327;346]    ( 341 S)
 [ 3] count:   10,  width: 7836 us [7699;8283]  (8024 S)
Pulse period distribution:
 [ 0] count:    1,  width: 2531 us [2531;2531]  (2592 S)
 [ 1] count:  429,  width: 1083 us [1049;1110]  (1109 S)
 [ 2] count:   10,  width: 8489 us [8422;9005]  (8693 S)
 [ 3] count:   10,  width: 6262 us [6261;6263]  (6412 S)
Pulse timing distribution:
 [ 0] count:    1,  width: 1040 us [1040;1040]  (1065 S)
 [ 1] count:  432,  width:  365 us [327;381]    ( 374 S)
 [ 2] count:  437,  width:  717 us [713;731]    ( 734 S)
 [ 3] count:   10,  width: 4769 us [4768;4771]  (4883 S)
 [ 4] count:   11,  width: 1492 us [1491;1494]  (1528 S)
 [ 5] count:   10,  width: 7836 us [7699;8283]  (8024 S)
 [ 6] count:    1,  width: 47706 us [47706;47706]       (48851 S)
Level estimates [high, low]:  15967,     29
RSSI: -0.1 dB SNR: 27.4 dB Noise: -27.5 dB
Frequency offsets [F1, F2]:     410,      0     (+6.4 kHz, +0.0 kHz)
Guessing modulation: No clue...
view at https://triq.org/pdv/#AAB03907010410016D02CC12A005D41E9BBA5A8492A1929292A1A1A1A19292A19292929292929292929292A19292A192929292A1929292A1929292A555+AAB03907070410016D02CC12A005D41E9BBA5AB492A1929292A1A1A1A19292A19292929292929292929292A19292A192929292A1929292A1929292A555+AAB03907020410016D02CC12A005D41E9BBA5AB492A1929292A1A1A1A19292A19292929292929292929292A19292A192929292A1929292A1A1A1A19555+AAB03907010410016D02CC12A005D41E9BBA5AB492A1929292A1A1A1A19292A19292929292929292929292A19292A192929292A1929292A1A1A1A19655

Detected OOK package    2024-04-19 22:45:51
Analyzing pulses...
Total count:  328,  width: 443.87 ms            (454520 S)
Pulse width distribution:
 [ 0] count:    1,  width: 1199 us [1199;1199]  (1228 S)
 [ 1] count:  216,  width:  378 us [376;381]    ( 387 S)
 [ 2] count:  104,  width:  722 us [721;723]    ( 739 S)
 [ 3] count:    7,  width: 4769 us [4768;4770]  (4883 S)
Gap width distribution:
 [ 0] count:    8,  width: 1493 us [1492;1495]  (1529 S)
 [ 1] count:  216,  width:  715 us [713;731]    ( 732 S)
 [ 2] count:   96,  width:  332 us [327;347]    ( 340 S)
 [ 3] count:    7,  width: 7702 us [7701;7705]  (7887 S)
Pulse period distribution:
 [ 0] count:    1,  width: 2692 us [2692;2692]  (2757 S)
 [ 1] count:  312,  width: 1082 us [1050;1109]  (1108 S)
 [ 2] count:    7,  width: 8425 us [8423;8427]  (8627 S)
 [ 3] count:    7,  width: 6262 us [6262;6264]  (6412 S)
Pulse timing distribution:
 [ 0] count:    9,  width: 1460 us [1199;1495]  (1495 S)
 [ 1] count:  312,  width:  364 us [327;381]    ( 373 S)
 [ 2] count:  320,  width:  718 us [713;731]    ( 735 S)
 [ 3] count:    7,  width: 4769 us [4768;4770]  (4883 S)
 [ 4] count:    7,  width: 7702 us [7701;7705]  (7887 S)
 [ 5] count:    1,  width: 47696 us [47696;47696]       (48841 S)
Level estimates [high, low]:  15959,     24
RSSI: -0.1 dB SNR: 28.2 dB Noise: -28.3 dB
Frequency offsets [F1, F2]:     -60,      0     (-0.9 kHz, +0.0 kHz)
Guessing modulation: No clue...
view at https://triq.org/pdv/#AAB037060105B3016C02CD12A01E16BA508092A1929292A1A1A1A19292A19292929292929292929292A19292A192929292A192A192A192A192A455+AAB037060605B3016C02CD12A01E16BA50B092A1929292A1A1A1A19292A19292929292929292929292A19292A192929292A192A192A192A192A455+AAB037060105B3016C02CD12A01E16BA50B092A1929292A1A1A1A19292A19292929292929292929292A19292A192929292A192A192A192A192A555

Detected OOK package    2024-04-19 22:45:52
Analyzing pulses...
Total count:  451,  width: 614.40 ms            (629150 S)
Pulse width distribution:
 [ 0] count:    1,  width: 1020 us [1020;1020]  (1044 S)
 [ 1] count:  297,  width:  378 us [376;381]    ( 387 S)
[... 103 lines cut ...]
 [ 1] count:  429,  width: 1081 us [1050;1110]  (1107 S)
 [ 2] count:   10,  width: 8488 us [8421;9002]  (8692 S)
 [ 3] count:   10,  width: 6262 us [6260;6263]  (6412 S)
Pulse timing distribution:
 [ 0] count:    1,  width:  952 us [952;952]    ( 975 S)
 [ 1] count:  432,  width:  363 us [328;381]    ( 372 S)
 [ 2] count:  437,  width:  718 us [712;732]    ( 735 S)
 [ 3] count:   10,  width: 4768 us [4767;4770]  (4882 S)
 [ 4] count:   11,  width: 1493 us [1492;1495]  (1529 S)
 [ 5] count:   10,  width: 7836 us [7699;8280]  (8024 S)
 [ 6] count:    1,  width: 47696 us [47696;47696]       (48841 S)
Level estimates [high, low]:  15986,     17
RSSI: -0.1 dB SNR: 29.7 dB Noise: -29.8 dB
Frequency offsets [F1, F2]:    -195,      0     (-3.0 kHz, +0.0 kHz)
Guessing modulation: No clue...
view at https://triq.org/pdv/#AAB039070103B8016B02CD129F05D51E9BBA508492A1929292A1A1A1A19292A19292929292929292929292A19292A192929292A19292A1A19292A1A555+AAB039070703B8016B02CD129F05D51E9BBA50B492A1929292A1A1A1A19292A19292929292929292929292A19292A192929292A19292A1A19292A1A555+AAB039070203B8016B02CD129F05D51E9BBA50B492A1929292A1A1A1A19292A19292929292929292929292A19292A192929292A19292A1A1A1A1929555+AAB039070103B8016B02CD129F05D51E9BBA50B492A1929292A1A1A1A19292A19292929292929292929292A19292A192929292A19292A1A1A1A1929655
```

Says its a simple OOK signal. 

Not sure how to get high and low from analysis yet, I am guessing something like this? `rtl_433 -R 0 -X 'n=name,m=OOK_PWM,s=28,l=1600,g=715,r=1083'`

please tell me where the numbers are coming from so that I can learn next time how to add new devices myself.

Thank you!

zuckschwerdt, 2024-04-20:
Short and Long are the most prevalent pulse widths. Look at the PDV link, you'll see it's 360 and 720.
Also look here https://triq.org/rtl_433/PULSE_FORMATS.html

closed 2024-04-20 by zuckschwerdt

adminy, 2024-04-20:
Aaaaah, I get it now! Thanks a million!
So its the most counts for long, short and gaps.

what about reset? What should that be?

is this something like total possible duration for the signal?

zuckschwerdt, 2024-04-20:
Reset should be longer than the longest expected gap. rtl_433 will abort reading data when such a gap is encountered.
Since longest gap in a data packet here is ~720 and inter-packet gap is ~1500 you might want `gap=1000,reset=2000` here to read packets as rows in a single bitbuffer. Or use just `reset=1000` to read each packet to it's own bitbuffer.

adminy, 2024-04-20:
so you say its best to put gaps higher but not too high. But if you're sending too many packets doesn't all the extras in a gap accumulate over time to form incorrect parsing? Or does rtl_433 account for that? 
also the reset would be.

```
        reset 
____________________
 ___ __________ ___
 gap   packet   gap
```
which would be the length of packet + the 2 gaps + a little extra or just larger than the largest gap?

one more thing, does the parser not get confused that the gap is the same length here as the long pulse width? or its able to tell them apart?

Thanks so much @zuckschwerdt, you've helped me understand enormously!

zuckschwerdt, 2024-04-20:
In OOK you get pulses and gaps. Each pulse has a "normal" gap after it. The gap and reset option say: if a gap after a pulse is too long then abort. With the "gap" option this abort means "open a new row" with the "reset" option it means "end the bitbuffer".
In other words, the `gap` option says: maximum allowed gap for data must be shorter than this, otherwise break into new row. The `reset` option says: maximum allowed gap for data must be shorter than this, otherwise end receiving.
It does not concern with pulses or packet length, just the current gap the slicer is looking at.

adminy, 2024-04-20:
[image]

Does it make sense to make the gap a little higher to just get 1 row instead of two? (1500 in this case)

I know this is kind of silly, but I was testing around to see if it can pack all the data sent in just 1 row. I guess the disadvantage of that would be the delay. but its a nice way to see what code got transmitted how many times at a glance.

zuckschwerdt, 2024-04-20:
> Does it make sense to make the gap a little higher to just get 1 row instead of two? (1500 in this case)

Those are two bitbuffers / transmissions. The first has 1 row the second 11 rows. Rows are the packets of data in a transmission. I.e. "repeats" here. Entirely depends on your application if you want to see repeats as rows in the same transmission or every data packet as it's own transmission.

> Also the last 3 seem to be different code, is that part of what the remote sends or is it just data loss?

Data loss surely.

> to see if it can pack all the data sent in just 1 row. […] nice way to see what code got transmitted how many times at a glance.

That's usually the intention of getting multiple rows in one "chunk", to count and compare for reliability.