merbanan/rtl_433 #2909: device_class seems to be wrong in rtl_433_mqtt_hass.py example

amordicus, 2024-04-20:
Thanks for this extremely useful code.
Reading the code in rtl_433_mqtt_hass.py I came across the following code where device_class should be **energy** not power.
    "energy_kWh": {
        "device_type": "sensor",
        "object_suffix": "kwh",
        "config": {
            "device_class": "power",
            "name": "Energy",
            "unit_of_measurement": "kWh",
            "value_template": "{{ value|float }}",
            "state_class": "measurement"
        }
I think this can become a problem if someone uses the "Integration - Riemann sum integral" helpers in Home Assistant? This is a guess,I can be wrong perhaps it could cause some other problem on the long run.
I cannot confirm nor test this because I don't have an energy sensor to test with. There's probably a way to test it without one but I'm just starting to look into using the RTL_433.

zuckschwerdt, 2024-04-20:
You are correct. And "voltage_V" should be "voltage" not "power". Maybe "power" was the only class available two years aga?
PR the change if you like.

amordicus, 2024-04-20:
I think you might be correct that it may have been the only class at the time from what I see browsing old information on the web. I'm worried that it may break the current user's configuration when they update their rtl_433_mqtt_hass.py?

gdt, 2024-04-20:
Don't worry.   We have to fix bugs, and that means people are going to have to deal.  It is madness to leave bugs forever because someone might have trouble.

commit 1296bf02 (2024-04-26): Change HA script power to energy and UV index to float

Closes #2909
commit a6fc37df (2024-04-26): Change HA script power to energy and UV index to float

Closes #2909
commit 376ca519 (2024-04-26): Change HA script power to energy and UV index to float (#2913)

Closes #2909
closed 2024-04-26 by zuckschwerdt