merbanan/rtl_433 #2912: [feature request]: Frequency hoping logging

PovilasID, 2024-04-25:
Hey,

I use the `-H` pram to log couple off different devices but when I look at the log (tried kv and just log) I have not noticed a notation for a frequency switch and while configuring the system I am then not sure if it not picking the signal up is just now listening on different frequency, so adding log output to indicate a that is not on this frequency would be very nice.

There is a possibility that it is an option just one have not seen. In that case

ProfBoc75, 2024-04-25:
@PovilasID : add `-M level` option will give you the signal details like frequency.

PovilasID, 2024-04-25:
@ProfBoc75 : Yah but only if it picks something on that frequency  it does not inform you if it started listening on it and just have not heard anything yet.

PovilasID, 2024-04-26:
Actually... with `-v` or or `-vv` or `-vvv` I am getting less logs not more... without `-v` I get `Autolevel` into the log and though it does not indicate the frequency if there is a block of log with level readjustment it good indicator that there is a change in frequency but if I add  any of `-v` options I get nothing.

Here is what the command looks like:
```
rtl_433 -vvv -H 60 -f 868.3M -f 868.9M -M protocol -M level -M noise -M stats -M bits -Y autolevel -F
mqtt://localhost:1883 -F kv
```

Here is what the logs look like:
```
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-04-26 12:20:30                    Protocol  : 172
model     : Bresser-6in1 id        : 14305bc2
channel   : 0            Battery   : 1             Temperature: 12.4 C       Humidity  : 58            Sensor type: 1
Wind Gust : 0.5 m/s      Wind Speed: 0.5 m/s       Direction : 338           UV        : 0.0
Startup   : 1            Flags     : 0             Integrity : CRC
Modulation: FSK          Freq1     : 868.2 MHz     Freq2     : 868.4 MHz
RSSI      : -14.2 dB     SNR       : 14.0 dB       Noise     : -28.2 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-04-26 12:21:20                    Protocol  : 104
model     : Wireless-MBus Mode      : T            Manufacturer: BMT         ID        : 3245354
Version   : 23           Device Type: 0x06         Device Type String: Warm Water                      Control   : 0x44
Data Length: 37          Data      : 2244b4095453240317068c00717adf0000000c1388680100046d0a2b1a340f0300000000000a9c
Integrity : CRC
Control Info: 0x8C       Access number: 0x00       Device Type: 0x00         Configuration Word: 0x0000
Min Volume flow[0]: 1398016.436 m3/min             Min Energy[0]: 9176599.000 Wh                        Energy[0]: 0.000 Wh
 Energy[0]: 0.000 MJ     Max Energy[0]: 262145.000 J                         Min Energy[1]: 0.000 kJ
Modulation: FSK          Freq1     : 868.9 MHz     Freq2     : 869.0 MHz
RSSI      : -0.7 dB      SNR       : 24.3 dB       Noise     : -25.1 dB
```
These are both caught on different frequencies aka before and after the hop and there is nothing in the log indicating that there was a frequency changed. 

rtl_433 version 23.11 (2023-11-28)

zuckschwerdt, 2024-04-26:
That command looks ok. The version 23.11 is the last stable version before the logging changed and does not support the current detailed logging. A current version would look like this:

[image]

PovilasID, 2024-04-26:
Grate!
Thanks for your patience. 
Will be waiting for package update!