merbanan/rtl_433 #2914: Intermittent issues decoding Rubicson wireless thermometer

hsand, 2024-04-26:
I have some intermittent issues decoding the signal of a Rubicscon wireless thermometer, product link below.

https://www.kjell.com/no/produkter/hjem-fritid/termometer-og-hygrometere/utendorstermometere/rubicson-tradlost-termometer-for-inne-og-utetemperatur-p49107

The thermometer can decode with the `Bresser-3CH` protocol, but appears to be very dependent on a strong signal. I've tried to rule out issues with my SDR, and experienced the same behavior with a USRP and antenna that was tuned to 434MHz with a VNA. Additionally, the cheap display that came with the thermometer seems to vastly outperform rtl_433 and a high quality SDR...

I've also experienced that when the thermometer is exposed to direct sunlight, and the temperature reading shoots up (visible on the display it came with) rtl_433 fails to decode entirely.

Not my areas of expertise, but would it be possible that this model has some pulse timing variations from the device the decoder was originally developed for? And sometimes it works, but sometimes when it heats up or the stars otherwise get misaligned, the timings are just outside of the range for rtl_433 to function correctly? ie. clock drifting with heat?

I've attached a zip with samples so somebody can take a look. I live in a noisy area, and there's some other devices in there as well, but the first sample and any other that decodes with the Bresser protocol are relevant. I also captured the samples at a time when decoding was working well, but can try to get some more of a time when it's not decoding properly if needed.

[samples.zip](https://github.com/merbanan/rtl_433/files/15133695/samples.zip)

zuckschwerdt, 2024-04-26:
The Bresser (ID 12) signal is nearly out-of-band, check e.g. g011, g021, g028, g037 in https://triq.org/pdv/
Temperature drift can reasonably shift it further out.

You need to tune a little higher, e.g. `-f 433.95M` if you still want to monitor other 433.92M signals. But even `-f 434M` should work.

hsand, 2024-04-26:
Thanks for the tip! I'll give that a try.

The thought did cross my mind about it transmitting out of band, and I tried increasing the sample rate to 1024k, but that actually seemed to end up decreasing performance. Any thoughts on what was happening?

zuckschwerdt, 2024-04-26:
For 433M the default discriminator `-Y classic` works best at 250k, for e.g. the 868M the `-Y minmax` is used and works good at 1024k. It should basically work regardless but performance is decreased as the algorithms are tuned for some sample rate.

hsand, 2024-04-27:
Changing the frequency worked 👍

closed 2024-04-27 by hsand