merbanan/rtl_433 #2920: Compilation warning Wpragmas on Ubuntu

kenwawa01, 2024-05-06:
Hi!

When compiling on Ubuntu 24.04 I get the following compilation warning:

`[ 93%] Building C object src/CMakeFiles/data.dir/data.c.o
/home/myuser/Projects/rtl_433/src/data.c:156:32: warning: unknown option after ‘#pragma GCC diagnostic’ kind [-Wpragmas]
  156 | #pragma GCC diagnostic ignored "-Wunknown-warning-option"
      |                                ^~~~~~~~~~~~~~~~~~~~~~~~~~
`

I'm wondering if it's possible to do sth with that as it doesn't look elegant...

Might it have any impact on the generated code as in my case when typing

`rtl_433 -R 0 -X "n=MYFIRSTTEST,m=FSK_PCM,s=26.042,l=26.042,r=1000" -r mylocalfile`

I get empty output (with no codes analysis as expected)

`rtl_433 version 23.11 (2023-11-28) inputs file rtl_tcp RTL-SDR with TLS Disabling all device decoders. [Input] Test mode active. Reading samples from file: g001_433.92M_250k.cu8 
`

Despite the input file is from the official test repository and other user got the normal ouptut, it means with codes (while I don''t have any codes)"

`codes     : {108}0002803b86c945024b48019173f
...
codes     : {108}fff2803b86c945024b48019202e
...
codes     : {107}000500770d928a04969003265be
`

zuckschwerdt, 2024-05-06:
That warning message is known. Ironically it's a command ("pragma") to ignore another warning message. But older compilers warn that they can't handle the command.

Analysis is done with the `-A` option. The `-X` option is a flex decoder. No idea if it could or should work on the file. Maybe add `-Y autolevel -Y magest -M level -M noise` to inspect and adjust signal levels.

kenwawa01, 2024-05-06:
Unfortunately I' not good in pragmas thus I can't recommend here anything... :(

Concerning my problem with flex decoder, I'm using these files as example: https://github.com/merbanan/rtl_433_tests/tree/master/tests/continental_tpms/01

Unfortunately also adding your options doesn't help. Currently I have

`rtl_433 -R 0 -X "n=BMW,m=FSK_PCM,s=52,l=52,r=900,preamble=2,bits>=106,bits<=108"-r g001_433.92M_250k.cu8 -Y autolevel -Y magest -M level -M noise
`

and I am not able to get anything from that. Could you provide your output when typing this command as above for the first [g001_433.92M_250k.cu8](https://github.com/merbanan/rtl_433_tests/blob/master/tests/continental_tpms/01/g001_433.92M_250k.cu8) file please?

kenwawa01, 2024-05-06:
Well I spent a lot of time and sth had to be changed or there is some error in code.

Ex. I take this issue https://github.com/merbanan/rtl_433/issues/2384

@zuckschwerdt  You write that for the file https://github.com/Gliebig/rtl_433_tests/blob/master/tests/EEZ%20Tire%20TPMS/Model%20618E/Tests/g071_433.88M_1024k.cu8 when typing

`rtl_433 -R 0 -X 'n=name,m=OOK_MC_ZEROBIT,s=50,l=50,r=120,invert' g071_433.88M_1024k.cu8`

you get

`{80}ffff 81 0d177e 99 46 0000`

while me, for the same file and the same command (with all the same parameters), I get

`rtl_433 version 23.11 (2023-11-28) inputs file rtl_tcp RTL-SDR with TLS
Disabling all device decoders.
[Input] Test mode active. Reading samples from file: g039_433.88M_1024k.cu8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.014513s
model     : name         count     : 1             num_rows  : 1             rows      : 
len       : 1            data      : 8
codes     : {1}8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.014513s
model     : name         count     : 1             num_rows  : 1             rows      : 
len       : 2            data      : 8
codes     : {2}8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.015858s
model     : name         count     : 1             num_rows  : 1             rows      : 
len       : 3            data      : a
codes     : {3}a
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.015858s
model     : name         count     : 1             num_rows  : 1             rows      : 
len       : 1            data      : 8
codes     : {1}8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.016471s
model     : name         count     : 1             num_rows  : 1             rows      : 
len       : 2            data      : 8
codes     : {2}8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.018839s
model     : name         count     : 1             num_rows  : 1             rows      : 
len       : 3            data      : 8
codes     : {3}8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.019172s
model     : name         count     : 1             num_rows  : 1             rows      : 
len       : 2            data      : 8
codes     : {2}8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.021903s
model     : name         count     : 1             num_rows  : 1             rows      : 
len       : 2            data      : 8
codes     : {2}8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.021903s
model     : name         count     : 1             num_rows  : 1             rows      : 
len       : 2            data      : 8
codes     : {2}8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.021903s
model     : name         count     : 1             num_rows  : 1             rows      : 
len       : 1            data      : 8
codes     : {1}8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.022866s
model     : name         count     : 1             num_rows  : 1             rows      : 
[... 3028 lines cut ...]
time      : @0.893665s
model     : name         count     : 1             num_rows  : 1             rows      : 
len       : 1            data      : 8
codes     : {1}8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.894757s
model     : name         count     : 1             num_rows  : 1             rows      : 
len       : 1            data      : 8
codes     : {1}8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.894757s
model     : name         count     : 1             num_rows  : 1             rows      : 
len       : 2            data      : 8
codes     : {2}8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.897332s
model     : name         count     : 1             num_rows  : 1             rows      : 
len       : 2            data      : 8
codes     : {2}8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @1.023865s
model     : name         count     : 1             num_rows  : 1             rows      : 
len       : 1            data      : 8
codes     : {1}8
`

kenwawa01, 2024-05-06:
Well I spent a lot of time and sth had to be changed or there is some error in code.

Ex. I take this issue https://github.com/merbanan/rtl_433/issues/2384

@zuckschwerdt  You write that for the file https://github.com/Gliebig/rtl_433_tests/blob/master/tests/EEZ%20Tire%20TPMS/Model%20618E/Tests/g071_433.88M_1024k.cu8 when typing

`rtl_433 -R 0 -X 'n=name,m=OOK_MC_ZEROBIT,s=50,l=50,r=120,invert' g071_433.88M_1024k.cu8`

you get

`{80}ffff 81 0d177e 99 46 0000`

while me, for the same file and the same command (with all the same parameters), I get

`
rtl_433 version 23.11 (2023-11-28) inputs file rtl_tcp RTL-SDR with TLS
Disabling all device decoders.
[Input] Test mode active. Reading samples from file: g039_433.88M_1024k.cu8
                                     
time      : @0.014513s
model     : name         count     : 1             num_rows  : 1             rows      : 
len       : 1            data      : 8
codes     : {1}8
                                     
time      : @0.014513s
model     : name         count     : 1             num_rows  : 1             rows      : 
len       : 2            data      : 8
codes     : {2}8
                                     
time      : @0.015858s
model     : name         count     : 1             num_rows  : 1             rows      : 
len       : 3            data      : a
codes     : {3}a
                                     
time      : @0.015858s
model     : name         count     : 1             num_rows  : 1             rows      : 
len       : 1            data      : 8
codes     : {1}8
                                     
time      : @0.016471s
model     : name         count     : 1             num_rows  : 1             rows      : 
len       : 2            data      : 8
codes     : {2}8
                                     
time      : @0.018839s
model     : name         count     : 1             num_rows  : 1             rows      : 
len       : 3            data      : 8
codes     : {3}8
                                     
time      : @0.019172s
model     : name         count     : 1             num_rows  : 1             rows      : 
len       : 2            data      : 8
codes     : {2}8
                                     
time      : @0.021903s
model     : name         count     : 1             num_rows  : 1             rows      : 
len       : 2            data      : 8
codes     : {2}8
                                     
time      : @0.021903s
model     : name         count     : 1             num_rows  : 1             rows      : 
len       : 2            data      : 8
codes     : {2}8
                                     
time      : @0.021903s
model     : name         count     : 1             num_rows  : 1             rows      : 
len       : 1            data      : 8
codes     : {1}8
                                     
time      : @0.022866s
[... 3030 lines cut ...]
model     : name         count     : 1             num_rows  : 1             rows      : 
len       : 1            data      : 8
codes     : {1}8
                                     
time      : @0.894757s
model     : name         count     : 1             num_rows  : 1             rows      : 
len       : 1            data      : 8
codes     : {1}8
                                     
time      : @0.894757s
model     : name         count     : 1             num_rows  : 1             rows      : 
len       : 2            data      : 8
codes     : {2}8
                                     
time      : @0.897332s
model     : name         count     : 1             num_rows  : 1             rows      : 
len       : 2            data      : 8
codes     : {2}8
                                     
time      : @1.023865s
model     : name         count     : 1             num_rows  : 1             rows      : 
len       : 1            data      : 8
codes     : {1}8

`

zuckschwerdt, 2024-05-07:
Your version of rtl_433 might be too old. I'm not aware of changes to the demod, but the current master decodes the mentioned file well.

kenwawa01, 2024-05-07:
@zuckschwerdt Thank you for answering. I've downloaded my version from the official repository: https://github.com/merbanan/rtl_433 as master first and when having those problems also as nightly build. Both behave in the same way. I'm wondering if it might be any problem with some library in Ubuntu 24.04 which is either too old or too new. Any suggestions please? Which version (branch/tag) should I use for compilation?

zuckschwerdt, 2024-05-07:
There are no libraries used in the decoding. Your "rtl_433 version 23.11 (2023-11-28)" is the last stable release. The current (nightly) would be something like "rtl_433 version 23.11-115-g8be3ef15 branch master at 202405022024"

But I've just tested 23.11 and it works fine:
```
./build/src/rtl_433 -R 0 -X 'n=name,m=OOK_MC_ZEROBIT,s=50,l=50,r=120,invert' g071_433.88M_1024k.cu8
rtl_433 version 23.11 branch  at 202311281352 inputs file rtl_tcp RTL-SDR SoapySDR with TLS
Disabling all device decoders.
[Input] Test mode active. Reading samples from file: g071_433.88M_1024k.cu8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.041087s
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 232          data      : ffffffffffffffffffffffffffffffffffffffda4644894d32a70c8920
codes     : {232}ffffffffffffffffffffffffffffffffffffffda4644894d32a70c8920
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.075282s
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 80           data      : ffff810d177e99460000
codes     : {80}ffff810d177e99460000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.093727s
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 80           data      : ffff810d177e99460000
codes     : {80}ffff810d177e99460000
```

We are using different files though? I don't know what g039_433.88M_1024k.cu8 is and the Continental repo files just don't work.

kenwawa01, 2024-05-07:
@zuckschwerdt Magic... Very strange thing.

Being sure that I do sth wrong with my compilation, I decided to uninstall and clean everything.

Than for Ubuntu 24.04 I run 

`sudo apt-get install rtl-433`

to install it from the official repository. I got the version 23.11 but... it behaves exactly as the same version I compiled manually thus I don't get the result as you. :(

I tried to investigate more and with Ubuntu 22.04 (=older) on a different computer (!!!) I did the same steps (=install from the official repository). I got 21.12-1 version and I also didn't get the expected result although here the number of codes is limited to 3 but very short ones:

`...
codes     : {1}8
...
codes     : {1}8
...
codes     : {1}8
...
`
For me that small experiment shows that there is some kind of general problem with versions for Ubuntu. Probably there is sth wrong with compilation but not sure what.

What operating system do you use and in what version?

kenwawa01, 2024-05-07:
@zuckschwerdt  Ok, I think I got the problem. It seems not to be related with rtl_433 itself but with... the way of downloading the files. Depending how I do this in web browser, I get the same result as you or... not...

The world is sometimes crazy... My fault! :( :( :( Sorry for creating this confusion. :(

ProfBoc75, 2024-05-07:
Hi @kenwawa01 

One time, I did a mistake and wrongly downloaded cu8 files that were html files, so not working, did you:
`git clone https://github.com/Gliebig/rtl_433_tests.git` or download files with another method ?

Title of this issue could be changed, it's not related to Continental TPMS as you are testing EEZ files.

From my up to date Ubuntu 22.04 running under Windows WSL2.
Using last version of rtl_433 (compiled by following the Build guidance), same command and same file as you:

    rtl_433 -R 0 -X 'n=name,m=OOK_MC_ZEROBIT,s=50,l=50,r=120,invert' g071_433.88M_1024k.cu8

I got the expected result:

[image]

And same file properly decoded:

[image]

kenwawa01, 2024-05-08:
@ProfBoc75 Yeah we were writing our comments in almost exactly the same moment... Yeah the way of downloading files is guilty here... (or rather my brain :( ). Still good it's solved and thanks for help.

ProfBoc75, 2024-05-08:
And it's funny because I did the same mistake for the same [EEZ issue 2384](https://github.com/merbanan/rtl_433/issues/2384#issuecomment-1435766186)

kenwawa01, 2024-05-09:
Thanks to all for help. I believe that this issue is still valuable as others who are just entering into project, might have the same problem. We are only humans...

closed 2024-05-09 by kenwawa01