merbanan/rtl_433 #2923: Transmissions not read after hop

Jocala, 2024-05-10:
rtl_433 version 23.11, linux mint, RTL-SDR Blog V4.  

command: /usr/local/bin/rtl_433 -H 120 -f 433M -f 915M -C customary

I'm attempting to read a LaCrosse-TX141THBv2 temp sensor at 433M and a LaCrosse LTV-Pool-INT (TFA-Marbella) pool thermometer at 915M.  I can read either separately, but when I attempt to hop between the two frequencies I get multiple: 

bitbuffer_add_bit: Warning: row count limit (50 rows) reached.  

Reception seems to cease after this.

zuckschwerdt, 2024-05-10:
That's not an error but just a waring that noise looks like data and then exceeds the reasonable amount of data to read.
Perhaps levels are not ideal. Add `-Y autolevel -Y magest -M level -M noise` and check what noise and level readings you get.

Jocala, 2024-05-10:
I'm obviously a noob, thanks for looking at this.

rtl_433 -Y autolevel -Y magest -M level -M noise  -H 120 -f 433M -f 915M -C customary

With this command-line I get:

[Auto Level] Current noise level -2.5 dB, estimated noise -2.7 dB
[repeats]
[Auto Level] Estimated noise level is -27.5 dB, adjusting minimum detection level to -24.5 dB
[Auto Level] Current noise level -27.9 dB, estimated noise -27.9 dB
[repeats]

First transmission received:

time      : 2024-05-10 11:56:06
model     : TFA-Marbella
 id        : 690574
counter   : 7
Temperature: 84.6 F  
Integrity : CRC
Modulation: FSK 
Freq1     : 915.0 MHz
Freq2     : 915.0 MHz
RSSI      : -12.9 dB
SNR       : 15.9 dB
Noise     : -28.8 dB

bitbuffer_add_bit: Warning: row count limit (50 rows) reached
[repeats]

I never see a transmission from the LaCrosse-TX141THBv2.

zuckschwerdt, 2024-05-10:
A noise level below -20 dB is good, that noise level close to 0 dB is a jammed band or recpetion problems.
What change do you see when hopping occurs? Use `-v` to get hopping notifications.

Jocala, 2024-05-10:
[SDR] Tuned to 433.000MHz.
[Auto Level] Current signal level -2.1 dB, estimated noise -21.9 dB
[Auto Level] Current noise level -2.1 dB, estimated noise -2.4 dB
<snip>
bitbuffer_add_bit: Warning: row count limit (50 rows) reached
<snip>
SDR] Tuned to 915.000MHz.
bitbuffer_add_bit: Warning: row count limit (50 rows) reached
[Auto Level] Estimated noise level is -16.6 dB, adjusting minimum detection level to -13.6 dB
[Auto Level] Estimated noise level is -18.0 dB, adjusting minimum detection level to -15.0 dB
<snip>
No transmissions from 433Mhz noted when both -f 915 and -f 433 are used.

ProfBoc75, 2024-05-11:
@Jocala : can you try `433.92M` instead of `433M`

gdt, 2024-06-02:
Calling this a probably answered question, unless the submitter would like to respond/restate/etc.

closed 2024-06-23 by gdt