merbanan/rtl_433 #2924: Dewenwils BH-V series AC outlet switches

waltje, 2024-05-10:
Hello:

Looks like the BH-V series transmitters (I have 5-channel units, but they also exist in 1, 2 and 3-channel models)
are not supported yet.

These use a standard PT2260-series encoder, but with a twist: their oscillator is set to a very short period (4.7M ohm
reistors, I believe that ends up being 168usec or so?)

Debug output available.

zuckschwerdt, 2024-05-10:
Can you grab a sample file and upload here as zip? https://triq.org/rtl_433/ANALYZE.html

zuckschwerdt, 2024-05-10:
Looks good. The `rtl_433 -A` suggestions will work, e.g. use a flex decoder with
`-X 'n=name,m=OOK_PWM,s=150,l=500,g=800,r=6000,t=50,bits=25'`
for an output of `{25}eaaacc8` and `{25}eaaac38`

Write a documented conf file like these https://github.com/merbanan/rtl_433/tree/master/conf and PR if you like.

waltje, 2024-05-10:
I will.
I also obtained a remote with a different code (mine is 0301, new one has 0304) just to verify my decoder.  I will PR it once that remote is in, and has been verified.
Fred

waltje, 2024-05-14:
PR opened.

closed 2024-05-14 by waltje