merbanan/rtl_433 #2925: GERMA (Chinese) 4-Button Keychain Remote

waltje, 2024-05-10:
These are samples recorded for the Germa 4-button keychain remote, as found on AliExpress (https://www.aliexpress.us/item/3256805955378503.html) and Amazon (https://www.amazon.com/Control-Duplicator-Electric-Transmitter-Keychain/dp/B0C6LBCQSY/ref=sr_1_3) for next to nothing.
[GERMA_4Button_433_Remote.zip](https://github.com/merbanan/rtl_433/files/15279846/GERMA_4Button_433_Remote.zip)

zuckschwerdt, 2024-05-11:
The signal is somewhat unstable but works. It's also at 434.02M to 434.04M so tuning a little hight would be better.
This should work: `rtl_433 -f 434.0M -X 'n=name,m=OOK_PWM,s=240,l=740,g=1000,r=8000,t=100'`

Codes in the samples are:
- `{25}2e2a8b8`
- `{25}2e2a878`
- `{25}2e2a8e8`
- `{25}2e2a8d8`

which is expected for a EV1527 encoder. Adapt one of our EV1527 conf files, e.g. something like
```
        bits>=24,
        bits<=25,
        repeats>=3,
        invert,
        get=@0:{20}:code:[859479:REMOTE-A 123456:REMOTE-B],
        get=@20:{4}:button:[1:A 2:B 3:AB 4:C 5:AC 6:BC 7:ABC 8:D 9:AD 10:BD 11:ABD 12:CD 13:ACD 14:BCD 15:ALL],
        unique
```

waltje, 2024-05-14:
This will require some more testing.
I received three more sets of these, from a different supplier (red LED and slightly differently colored keys),and those DO get recognized by default.

 
 

   

        unique

gdt, 2024-06-02:
@waltje Please don't quote email from github in comments.  It is very hard to follow and just adds noise.

What's next?  Is this an answered question, or is someone(tm) going to contribute a PR that documents this for others?

waltje, 2024-06-02:
> @waltje Please don't quote email from github in comments. It is very hard to follow and just adds noise.
OK, boss.

closed 2024-06-03 by zuckschwerdt

waltje, 2024-06-03:
> Closed #2925 as completed.
?Message ID: ***@***.***>

zuckschwerdt, 2024-06-03:
Closed as this is not a todo item for rtl_433. Feel free to ask for further help if needed or add a PR with your findings if you like. Please don't comment chat-like, we can barely handle the volume of issues.