merbanan/rtl_433 #2928: 1/Sqrt(2) modulation support.?

erosenst, 2024-05-16:
Hello, where I should add modulation support code for squareroot2 modulation.? (I have not yet looked your codes at all).

I have esp32 + sym115 chip which I drive with 1/sqrt2 modulation and like to check it go air well with rtl_433.

1/Sqrt(2) modulation is quite simple and if it extend over binary range (1/sqrt2 - sqrt2/1) can be used direct float number from MCU (can help if MCU have not FPU). So receiver FPU can calculate float numbers.. I have been used @80Mhz/div2 / 5 x (41/58 -  58/41) ratios for bit=0 - bit=1. Delay 205 ~3.075ms = short,  290 ~4.35ms = long
Rule are: divider length is how long is signal low and divident amount is how long signal up. In signal first divident then divisor.

Start/end condition = 9th 'bit' is two long: up long followed down long. And every ninth 'bit' is long/long = 1. When pack start 
 there may be preample + up long. Two following shorts (short/short=1) newer used. 

Say code A:
---___--___---__--___--___--___--___--___---__---___
bit:  S     0     1     0      0     0      0     0     1       S

And when decode to float, 41/58=0.70689 & 58/41=1.14146:
1.0;0.71;1.41;0.71;0.71;0.71;0.71;0.71;1.41;1.0
B:S    0      1      0       0      0      0      0      1        S

So is there some ready code or should I try extend modulation section with new one.? If someone know this modulation other name can point it also. I just take it from my head so don't know is it new or not..

Thank you

zuckschwerdt, 2024-05-16:
I'm not sure "modulation" is the right name. Modulation is the modification of a carrier to add data. Usually FSK, ASK or PSK (we support 2-FSK, 2-ASK = OOK, but not PSK).

It seems like a coding scheme, but I didn't grasp what you write about timings. Is this PWM maybe? If so, is it analog or digital? Please compare to https://triq.org/rtl_433/PULSE_FORMATS.html

erosenst, 2024-05-16:
Hello, it seems to be fixed width PWM with every 9th pair is same up/down time and are separator. (seems my example have corrupt, need change up/down chars to different. Forgot also 9th pair).
 Timings are short: ~3ms, long 4.35ms.  Bits are inverted vs. your example.
https://triq.org/rtl_433/PULSE_FORMATS.html#fixed-bit-width-pwm

So there is ready driver, I have only little success with it. Packet is some 2kb but I got only 50bits and they are wrong. Wrong options..? Need sharper spectrum RTL as default 2Mhz.?

 UUUUdddUUUddddUUUUdddUUUddddUUUddddUUUddddUUUddddUUUddddUUUUdddUUUUdddd
bit:           X                0                 1                0               0                 0               0               0                 1                  X
U=up d=down, one char ~1us

zuckschwerdt, 2024-05-16:
Which modulation are you using, is it OOK? 2000 bits is a lot to discriminate (find if the demod is pulse or gap) correctly. What frequency? Try the other discriminator perhaps, either `-Y classic` or `-Y minmax`.

What timing "same up/down time" are the syncs /9th bit? Is it 3.6/3.6? I.e. `s=3000,l=4350,y=3600` and likely `,t=200,r=5000` or such?

If your bit time is 7.35ms then your packet is 15 seconds? That is way to much air time to process in one go for rtl_433 and very likely in violation of regulatory rules (there is a maximum radio occupancy rule, varying by country though).

If you really need that much data I'd advice a fast FSK transmission.

erosenst, 2024-05-16:
It is syn115 ASK,. It is over mfg limit 10kbs by 10x but their limit is moustly from receiver part. Need check can syn115 do that speed, seems that it do but need be sure so RTL-SDR may help. ?
https://www.rhydolabz.com/documents/33/SYN113-SYN115-datasheet-version-1-1-.0.pdf
Osciloscope show nice pulses in to transmitter. 

 Packet max is ~2kB ~450ms typical air time. It have 40byte header so some payload need also carry. 
 Timing sync bits are ~4.35us / ~4.35us so long/long. short/short is not used. Think long/long timings is more visible than short/short.
 EU regulations, I do not know rules but Tx power is quite low, short distance ~10-15m cell, usually inside room. 315Mhz version.
Bit time come 8x 3.075us + 10x 4.35us = 68.1us -> 544.8us/byte -> ~1s air time per 2kB, but I see typical ~450us air time so pack have not grown so big (2kB is max buffer).
 If syn115 don't work then need replace radio, some speed drop maybe ~40% I can do also.

zuckschwerdt, 2024-05-16:
Ah ok, that's one of those ultra low cost/quality superheterodyne things.

Pulse timings in µs not ms then ;) half a second does not sound so bad if utilization is low (e.g. transmit with a few minutes pause between).

The timings are unusally short for OOK. Even with fast FSK we usually have 50µs (TPMS), in rare cases (kinetic switch) 8µs. That's half-bit time with Manchester coding usually. You'll need 2000k sample rate at least and `-Y minmax`.

But usually with that many bits to transfer you'd use a capable FSK chip like the CC1101 or any of the similar ones.

zuckschwerdt, 2024-06-05:
Not a todo issue for rtl_433, closing.

closed 2024-06-05 by zuckschwerdt