merbanan/rtl_433 #2929: RFE: Support for decoding Hoymiles HMS inverter messages

LorbusChris, 2024-05-18:
I'd like to be able to use my RTL-SDR dongle to decode messages from my Hoymiles HMS solar inverter.

There are two implementations for codecs using the CMT2300A chip included in these two projects:
- https://github.com/tbnobody/OpenDTU ([dir](https://github.com/tbnobody/OpenDTU/tree/master/lib/Hoymiles))
- https://github.com/lumapu/ahoy ([dir](https://github.com/lumapu/ahoy/tree/main/src/hms))

Related issues:
- https://github.com/tbnobody/OpenDTU/issues/1993
- https://github.com/lumapu/ahoy/issues/1630

zuckschwerdt, 2024-05-18:
Do you have any information what configuration or protocol the CMT2300A would be using? I.e. timings, coding and data payload structure?

LorbusChris, 2024-05-18:
I haven't dug in, but is it this that you're looking for?
https://github.com/tbnobody/OpenDTU/blob/1f1227fa104cda435df7132ddfa011524a262554/lib/CMT2300a/cmt2300a_params_860.h#L7-L86

ap20132022, 2024-07-08:
I'm using openDTU for this purpose. It would be great, if it works with RTL-SDR

closed 2025-07-04 by gdt