merbanan/rtl_433 #2930: Isolating protocol 113 does not include EcoWitt WH40

imca-potato, 2024-05-19:
Greetings!

I have been trying to decode with my Ecowitt WH40 sensor together with my Ambient Weather WH31 Temperature and Humidity sensor. I want to save on some computer resources, so I intend to isolate them using `-R 113`, which I assume will achieve the "saving computer resources" part. However, when I do that, the WH40 is not being decoded. It does get decoded if I remove the tag, however I would prefer to have it not decode all other signals (which I assume consumes resources). Can someone help clarify this to me?

Thank you for your help!

ProfBoc75, 2024-05-19:
@imca-potato : 

When you tried without tag but with `-M protocol` are both decoded as protocol 113 ? 
What are your rtl_433 command arguments/options/tags ?

From the cu8 samples at [rtl_433_tests/tests/EcoWitt-WH40](https://github.com/merbanan/rtl_433_tests/tree/5f836200660f931b3225e92103ee65a80306cef5/tests/EcoWitt-WH40), both sensors are properly decoded with the `-R 113` argument.

Are both sensors on the same frequency range ? Test with `-M level` to get the frequency, then refine it with `-f xxx.xxM`

Notice that at 433.92M, the sample rate is by default 250kHz so the frequency range is narrower than at 868M/915M with a sample rate at 1000kHz. The reason why you may need to adjust the frequency if you're working around 433.92M, could be `-f 433.6M` or `-f 434.2M` for example or keep default frequency but increase the sample rate `-s 1024k` but this could have a computer performance impact.

zuckschwerdt, 2024-05-19:
The biggest resource savings are not to decode FSK frames (by only selecting OOK protocols). The WH40 uses FSK so you can't do that. But you can use `-Y squelch` to skip decoding all "quiet" frames.
As mentioned use `-M level` and `-M noise -Y autolevel` to watch signal levels.

imca-potato, 2024-05-21:
Greetings!

> When you tried without tag but with `-M protocol` are both decoded as protocol 113 ?

I have tried decoding with the `-M protocol` tag and indeed both WH31E and WH40 are protocol 113.

> What are your rtl_433 command arguments/options/tags ?

The entire command was `rtl_433 -d "" -f 915M -s 1000k -M level -M protocol -R 113`. I did forget to mention I was using the PlutoSDR, hence the `-d ""`.

> Are both sensors on the same frequency range ?

Yes, both are at 915 MHz.

> But you can use -Y squelch to skip decoding all "quiet" frames.

Thank you for this, I will try it.

So far, I am using the command without the `-R 113` tag, yet it does not decode the WH40 all the time. I've tried increasing the gain manually, even up to 60, yet no luck. I have yet to try it with an RTL-SDR, though. I am sure that the transmitter is working, as indicated by its blinking every time it transmits data.

Once again, thank you for the suggestions. I am still having trouble with WH40 but I will update this as soon as something good happens.

gdt, 2024-06-02:
This was a question about how -R works, and that's been resolved.

closed 2024-06-02 by gdt