merbanan/rtl_433 #2935: Handling a new device that is almost the same as an existing one

jpmv27, 2024-05-24:
I recently purchased [this device](https://www.aliexpress.com/item/33054004675.html) from AliExpress.
It's not identified with a Make or Model, but seems to be the same as [this ORIA WA50](https://www.amazon.ca/ORIA-Refrigerator-Thermometer-Wireless-Restaurants/dp/B07B9QF64N/) on Amazon.

The sensors are decoded with protocol 47 and recognized as "Conrad-S3318P".

In fact, there are a few differences between my sensors and what is decoded:
- my sensors only report temperature, not humidity (humidity is reported as 0)
- my sensors do not have a button
- the base and sensors do not support a channel selection; sensor 1 reports channel "2" and sensor 2 reports channel "3"

I will probably just live with this, but I was wondering how such a situation could be handled if I wanted to add a custom decoder for these sensors.

ProfBoc75, 2024-05-24:
Salut @jpmv27 : You can add a data condition at the result , since humidity = 0% it's not human or into the desert or no humidity  sensor ?
[Here](https://github.com/merbanan/rtl_433/blob/86abae0550eb134d2fad1badead623fd3a800356/src/devices/s3318p.c#L105)

    "humidity",         "Humidity",    DATA_COND, humidity != 0, DATA_FORMAT, "%u %%", DATA_INT, humidity,

You may have to open the temperature sensor to get a reference on the PCB for this kind of no name / rebrand sensor.

Enrich the comment section and append your reference into the name [here](https://github.com/merbanan/rtl_433/blob/86abae0550eb134d2fad1badead623fd3a800356/src/devices/s3318p.c#L128), up to you.

jpmv27, 2024-05-25:
Merci @ProfBoc75 

Thank you for the hint. This is very clever. I am starting to discover the flexibility of the software.

jpmv27, 2024-05-25:
A couple of questions.

1) What is the relationship between the fields in data_make and those in output_fields? Does it matter that humidity will still be listed in output_fields even though it is conditional in data_make?
2) I presume there is conversion happening somewhere, since the temperature is received from the sensor in degrees F but is output in degrees C. Correct?

zuckschwerdt, 2024-05-25:
The `output_fields` is just a hint to the CSV output, it defines the possible columns.
There is a conversion option `-C si` (`native`, `customary`) or `convert si` in a conf file.

gdt, 2024-06-23:
@jpmv27 I am going to consider this resolved by the mentioned merge, unless you would like to explain what needs doing still, relative to up-to-date master.  Adding feedback tag with 7-day timer.

jpmv27, 2024-06-23:
Yes, resolved. Thanks

closed 2024-06-23 by jpmv27