merbanan/rtl_433 #2942: trh 433.92 radologger decoding

trefnis13, 2024-05-27:
If someone is interested in decoding signal from one of my T/RH sensors working on 433.92.
There of a 6 data-packets recorded with RTL-SDR v4 in URH software. Each data-packet contains sensor ID (it should be 1) and raw values of temperature (around 950) and relative humidity (around 1500). If someone would like to take up this challenge I can provide more details. 
[sample3_RTL-SDR-20240525_195429-433_92MHz-1MSps-1MHz.zip](https://github.com/merbanan/rtl_433/files/15458441/sample3_RTL-SDR-20240525_195429-433_92MHz-1MSps-1MHz.zip)

zuckschwerdt, 2024-05-27:
What units are 950 and 1500? Does not sound like temperature or %rh.

Looks like FSK, but one freq is dead center, which is not good. Can you offset by 50 to 100 kHz and record again.

Also, is that signal post-processed? The dynamic range seems strange. Or is that a very low manual gain setting?

trefnis13, 2024-05-27:
Thanks!! Yes, you may be right as this sensor has two modes and when I recorded this sample it was set to physical units i.e. deg C and % of rel. humidity, so in this case values should be around 24C and 45%. As for the dynamic range you are also right, I didn't have a good antenna so the range is poor. I will record new sample with your suggestions tomorrow as I left my dongle at work.

zuckschwerdt, 2024-06-04:
> [sdr_set_center_freq] Failed to set center freq.

That is bad. Should not happen. Try to unplug and replug the SDR. Check Zadig is installed correctly.

trefnis13, 2024-06-05:
Hi,
it looks like it is somehow working since e.g. this command
rtl_433 -Y autolevel -Y magest -M level -M noise -M time:usec -v
resulted with some reasonably looking results

[image: image.png]

trefnis13, 2024-06-17:
I have another sample if someone is interested - I hope this is better.
[RTL-SDR-20240611_155633-434_6MHz-2_5MSps-2_5MHz.zip](https://github.com/user-attachments/files/15873484/RTL-SDR-20240611_155633-434_6MHz-2_5MSps-2_5MHz.zip)

zuckschwerdt, 2024-06-17:
Rename the file to e.g. `RTL-SDR-20240611_155633_434.6M_2500k.complex16s` (important meta data is `_434.6M_`and `_2500k_`), then load in https://triq.org/spectrogram-next/

Set (mouse wheel) Gain to 0 dB and Range to e.g. 10 dB, now you can see the original signal. The FSK-looking lines at +/-200kHz are just aliases -- the signal is clipping (too loud).

Looks like very narrow and fast FSK maybe.
[image]

Zoom in (mouse wheel) to max and observe the very funny waveform in the I/Q display (lower part). I don't know what that is, is it just clipping FSK near DC?

[image]

Also, why is that signal at 434.6 MHz exactly? That's too far from 433.92 MHz. Do you always see the signal at the center frequency? Can you get it to be *not* at the center (and also quieter, not clipping)?

trefnis13, 2024-06-18:
Hi,
I played little with recording parameters and there is another data file with a sequence of 15 data-packets of temperature (in deg C) and rel. humidity (in %) recorded from the sensor with ID nr 1 with specific values attributed to specific packet.
[image]
[15xID1-434_052MHz-1MSps-1MHz.zip](https://github.com/user-attachments/files/15888842/15xID1-434_052MHz-1MSps-1MHz.zip)

zuckschwerdt, 2024-06-18:
Much improved. The signal looks really nice, the -1 dB is optimal. E.g. zoomed in on the first packet we see the FSK clearly

[image]

A packet seems to be ~200 ms. Basic symbols seem to be 1000 / 1000 µs pulse/gap and 2000/2000 µs pulse/gap after a 2x 1000/1000 preamble and a 3000/3000 sync symbol.

With a quick test rtl_433 doesn't seem to work well on that signal though :/

ProfBoc75, 2024-06-18:
@zuckschwerdt , @trefnis13 : 

Same approach for me, I renamed the complex16s to g001_434.052M_1000k.cs8 and I got more results than keeping current name.

From previous samples and URH, I saw that it's a Manchester coded signal.

I tried this:

    rtl_433 -X "n=test,m=FSK_PCM,s=1000,l=1000,r=10000,preamble=55c" -Y minmax -Y magest -Y filter=100 -M level g001_434.052M_1000k.cs8
[image]

Then this:

    rtl_433 -X "n=test,m=FSK_PCM,s=1000,l=1000,r=10000,preamble=55c,decode_dm" -Y minmax -Y magest -Y filter=100 -M level g001_434.052M_1000k.cs8
[image]

Not sure about decode_dm but just to try ...
Then another try with:

    rtl_433 -X "n=test,m=FSK_MC_ZEROBIT,s=1000,l=1000,r=10000" -Y minmax -Y magest -Y filter=100 -M level g001_434.052M_1000k.cs8
[image]

Some answers to start bitbench ...

ProfBoc75, 2024-06-18:
Little update, with a preamble = 0x55c6 and decode_dm , data is starting with 0x01.... So could be the sensor id.

ProfBoc75, 2024-06-22:
@trefnis13:

I'm not able to get the temp or humidity values, sounds like your samples are partially done, not all the same duration.
Same, it's not possible to get the id, crc or checksum values.

Can you try to get more codes but using rtl_433 ?

    rtl_433 -X "n=test,m=FSK_PCM,s=1000,l=1000,r=10000"

And as you did, share the temp/hum values along the codes.

If you have bad or too many answers, try to add filters like this:

    rtl_433 -X "n=test,m=FSK_PCM,s=1000,l=1000,r=10000,bits>=50"
    rtl_433 -X "n=test,m=FSK_PCM,s=1000,l=1000,r=10000,bits>=80,preamble=55c"

Adjust the bits value and the preamble (0x55c6  , 0x55c65, 0x55c655 ... ).