merbanan/rtl_433 #2943: New device Ecowitt water sensor, PN WN34D

Mart124, 2024-05-27:
Hello,

Ecowitt does a water temperature sensor, PN [WN34D](https://www.ecowitt.com/shop/goodsDetail/294), [user manual](https://osswww.ecowitt.net/uploads/20230919/WN34D%20Manual.pdf).

Here are then some samples from it : [WN34D.zip](https://github.com/merbanan/rtl_433/files/15458982/WN34D.zip)

g001_868.34M_250k.cu8 : 20.3°C
g001_868.34M_1000k.cu8 : 25.7°C
g002_868.34M_250k.cu8 : -4.9°C
g002_868.34M_1000k.cu8 : -9.0°C

It is decoded by protocol 221, but with a 40°C offset for positive values, and another one for negative values...

Could we have a look to properly support it please ?

Thank you very much 👍

ProfBoc75, 2024-05-27:
Hi @Mart124:

Yes, there is a gap at the data layout for the Temp.

The current decoder could be updated accordingly.

Current data layout:

    TYPE 8h ID 24h NOT_USED 5h TEMP_RAW 11s  NOT_USED 1b BAT 7d CRC 8h CHECKSUM 8h TRAIL_BYTE 12h 2x

New one:

    TYPE 8h ID 24h SUB_TYPE 4h TEMP_RAW 12s  NOT_USED 1b BAT 7d CRC 8h CHECKSUM 8h TRAIL_BYTE 12h 2x
 
    If SUB_TYPE = 0   = WN34L, Current decoder, temp = (Temp_raw - 400) * 0.1f
    If SUB_TYPE = 0x4 = WN34D, No Temp offset , temp = Temp_raw * 0.1f

Here the [bitbench](https://triq.net/bitbench#c=%5B%20WN34D%20%5D&c=%7B85%7D340064d341014f97932090%20%5B%20g001_868.34M_1000k.cu8%20%3A%2025.7%C2%B0C%2078.26%20F%20%5D&c=%7B86%7D340064d340cb4f955a2010%20%5B%20g001_868.34M_250k.cu8%20%20%3A%2020.3%C2%B0C%2065.54%20F%20%5D&c=%7B85%7D340064d34fa64f31e02090%20%5B%20g002_868.34M_1000k.cu8%20%3A%20-9.0%C2%B0C%2015.8%20%20F%20%5D&c=%7B86%7D340064d34fcf4fa77f2010%20%5B%20g002_868.34M_250k.cu8%20%20%3A%20-4.9%C2%B0C%2023.18%20F%20%5D&c=&c=%5B%20WN34L%20%5D&c=%7B85%7D34003bcd021d446a092090%20%5B%2014.1C_915M_1000k.cu8%20%5D&c=%7B85%7D34003bcd028c44d3e12090%20%5B%2025.2C_915M_1000k.cu8%20%5D&c=%7B84%7D34003bcd03e84443ae202%20%20%5B%2060.0C_915M_1000k.cu8%20%5D&c=%7B85%7D34003bcd02ad44fb2a2090%20%5B%2083.3F_915M_1000k.cu8%20(28.5%20%C2%B0C)%20%5D&f=TYPE%208h%20ID%2024h%20SUB_TYPE%204h%20TEMP%2012s%201b%20BAT%207d%20CRC%208h%20SUM%208h%20TRAIL_BYTE%2012h%202x&z=1&cw=4)

ProfBoc75, 2024-05-28:
I want to confirm my assumption, can someone confirm the sub_type approach as we don't have lot of samples to replay.

Mart124, 2024-05-28:
Hi @ProfBoc75

First, thank you very much for the debug, and for the PR ! Really glad :)
Also glad to see that it does not require a full new decoder, but an update only.

> I want to confirm my assumption, can someone confirm the sub_type approach as we don't have lot of samples to replay.

Do you want some more samples from the WN34D variant ? Or from the L/S ones ?
I only have the D here...

Thank you again 👍

ProfBoc75, 2024-05-28:
Hi @Mart124 : 

Just try with `-R 221:v` and report few `codes : {72}xxxx...` results, please, it's enough for me.

like this: (from WN34L samples at rtl_433_tests)

    {72}34003bcd021d446a09
    {72}34003bcd028c44d3e1
    {72}34003bcd03e84443ae
    {72}34003bcd02ad44fb2a

or WN34D: (your CU8 files)

    {72}340064d341014f9793
    {72}340064d340cb4f955a
    {72}340064d34fa64f31e0
    {72}340064d34fcf4fa77f

Mart124, 2024-05-28:
Here are some codes from WN34D, but without your PR (I was not able to find a compiled version from the CI, not sure it does) :
```
codes     : {72}340064d3411b4f172d
value : 28.3

codes     : {72}340064d340e14fa782
value : 22.5

codes     : {72}340064d340dc4f552b
value : 22

codes     : {72}340064d340d74f4f20
value : 21.5

codes     : {72}340064d340d24f3804
value : 21

codes     : {72}340064d340ce4fe2aa
value : 20.6
```

Hope this is what you are looking for, of course feel free if needed 👍

ProfBoc75, 2024-05-28:
[bitbench ](https://triq.net/bitbench#c=%5B%20WN34D%20%5D&c=%7B85%7D340064d341014f97932090%20%5B%20g001_868.34M_1000k.cu8%20%3A%2025.7%C2%B0C%2078.26%20F%20%5D&c=%7B86%7D340064d340cb4f955a2010%20%5B%20g001_868.34M_250k.cu8%20%20%3A%2020.3%C2%B0C%2065.54%20F%20%5D&c=%7B85%7D340064d34fa64f31e02090%20%5B%20g002_868.34M_1000k.cu8%20%3A%20-9.0%C2%B0C%2015.8%20%20F%20%5D&c=%7B86%7D340064d34fcf4fa77f2010%20%5B%20g002_868.34M_250k.cu8%20%20%3A%20-4.9%C2%B0C%2023.18%20F%20%5D&c=&c=%7B72%7D340064d3411b4f172d%20%5B%2028.3%20%5D&c=%7B72%7D340064d340e14fa782%20%5B%2022.5%20%5D&c=%7B72%7D340064d340dc4f552b%20%5B%2022%20%20%20%5D&c=%7B72%7D340064d340d74f4f20%20%5B%2021.5%20%5D&c=%7B72%7D340064d340d24f3804%20%5B%2021%20%20%20%5D&c=%7B72%7D340064d340ce4fe2aa%20%5B%2020.6%20%5D&c=&c=%5B%20WN34L%20%5D&c=%7B85%7D34003bcd021d446a092090%20%5B%2014.1C_915M_1000k.cu8%20%5D&c=%7B85%7D34003bcd028c44d3e12090%20%5B%2025.2C_915M_1000k.cu8%20%5D&c=%7B84%7D34003bcd03e84443ae202%20%20%5B%2060.0C_915M_1000k.cu8%20%5D&c=%7B85%7D34003bcd02ad44fb2a2090%20%5B%2083.3F_915M_1000k.cu8%20(28.5%20%C2%B0C)%20%5D&f=TYPE%208h%20ID%2024h%20SUB_TYPE%204h%20TEMP%2012s%201b%20BAT%207d%20CRC%208h%20SUM%208h%20TRAIL_BYTE%2012h%202x&z=1&cw=4) with your last codes, sounds good, Thx.

closed 2024-05-30 by ProfBoc75