merbanan/rtl_433 #2948: Confusing battery_ok conundrum

DigiH, 2024-05-30:
This has been bugging me for ages, but with my ON/OFF consistency PR I though I finally might as well get it off my chest.

**battery_ok**, when being used as a binary 0/1, on/off in decoders could be such a nice proper binary for all of them, also when creating HA discovery messages, for a very nice

[image]

Unfortunately, instead it uses a very awkward transformation of

```
"battery_ok": {
        "device_type": "sensor",
        "object_suffix": "B",
        "config": {
            "device_class": "battery",
            "name": "Battery",
            "unit_of_measurement": "%",
            "value_template": "{{ float(value) * 99 + 1 }}",
            "state_class": "measurement",
            "entity_category": "diagnostic"
        }
    },
```

causing 0/off to be 1% and 1/on to be 100%. 

This has already caused some YouTube reviewers to mention that there must be something wrong in the code, as their battery showed 1% , but the weather station was still running fine - that didn't make sense to them.

As there seem to be decoders which actually do produce a float for their battery level, my suggestion would be to rename their battery key to something like **battery_level** to then apply the above or any other required transformation, but to leave all **battery_ok** instances as they are, a nicely interpretable binary, which, when applied correctly in a discovery message, nicely produces the above for HA, and for all other controllers a binary 0/1 also makes a lot more sense than 1 and 100.

zuckschwerdt, 2024-05-30:
We did document that `battery_ok` is a level between 0 and 1, with most devices not using any steps in between though.
https://triq.org/rtl_433/DATA_FORMAT.html#common-device-data

I'm not sure why the mapping for HA is 1% to 100%. It might be to indicate low-but-not-dead or some problem where 0 would not show?

It is a unified value (0–1) for all types of devices though. The extra bit is `battery_mV`, where supported.

DigiH, 2024-05-30:
> I'm not sure why the mapping for HA is 1% to 100%. It might be to indicate low-but-not-dead or some problem where 0 would not show?

Because the above pasted discovery implementation for "battery_ok" only ever uses the
`"value_template": "{{ float(value) * 99 + 1 }}",`
for all of the sensors, creating a 1% to 100% only for all "battery_ok" discoveries., even if they are actually correct binary o/1 values.

So I still think that two separate keys "battery_ok" and "battery_level" would be the clearest solution, or alternatively a check for float or integer would be required to actually correctly create discovery messages for the currently ambiguous battery_ok definition you linked to above.

I have to admit though, that I'm coming from OpenMQTTGateway, where the above discovery routine has been taken over 1:1.

zuckschwerdt, 2024-05-30:
I actually meant the "why" (the "how" is clear, I wrote that as a cheap example in #1016), I can't remember any particular reason.

If there is some foo to fix it in the script or the template that would be preferred. It took years to carefully get all decoders to the "battery_ok" standard without deprecating too much too soon along the way. I don't think we'll ever touch that "standard" again.

DigiH, 2024-05-30:
> I actually meant the "why"

To have an actual sensible battery LOW/OK discovery for HA discoveries for correct binary battery_ok decoders, and proper 0% - 100% with all the possible steps in between for all float battery indicating decoders, instead of only 100% and 1% for all of them.

As actually described in
https://triq.org/rtl_433/DATA_FORMAT.html#common-device-data
but there it only describes on how it should be reported, but unfortunately doesn't make any distinction between float or binary when being read for discovery purposes.

This would avoid user confusion for all general users as well as YouTube review users.

> It took years to carefully get all decoders to the "battery_ok" standard without deprecating too much too soon along the way.

I just wished then that there would have been a distinction between proper float battery levels and LOW/OK binary ones from the beginning, not having them pooled together under one "battery_ok", so that the majority of binary LOW/OK statuses would actually be discovered correctly and only ever not as either 100% and 1% only.

DigiH, 2024-05-30:
As I personally do not use discovery at all, all my rtl_433 devices have been properly implemented to use correct binary LOW/OK statuses - as I do not have any which report a float. 

So with any reluctance to touch this issue, I'm fine with closing this issue and a continued awkward mismatching discovery. Up to all the other users to wrangle with the current output ;)

DigiH, 2024-05-30:
Ok, I cumbersomely fixed this for OpenMQTTGateway, now correctly discovering binary battery_ok statuses as such, which is actually the majority of decoders, looking like 
[image]
and
[image]

while still discovering, and therefore displaying, all float battery statuses as sensors with a rounded integer and the % unit, i. e. 1% - 100%

Closing this for rtl_433, as auto-discovery doesn't seem to be a high priority

closed 2024-05-30 by DigiH

zuckschwerdt, 2024-05-31:
That looks very nice.
You are right, there is no maintainer for the rtl_433_mqtt_hass.py script currently.

DigiH, 2024-05-31:
Unfortunately this is still very awkward to do, with different types populating the same property key in rtl_433. 

That is likely also why the rtl_433_mqtt_hass.py script and the resulting HA rtl_433 MQTT Auto Discovery Add-on only handle the simple unified properties which are always floats for all decoders, like temperature_C, temperature_F, humidity, moisture etc. but completely ignoring all other existing ambiguous property keys. 

Just some examples to show where my frustration is coming from 😜 

**UV and/or UVI keys for several weather stations** - also broken in rtl_433 MQTT Auto Discovery
https://community.openmqttgateway.com/t/ha-sensor-missing-for-fineoffset-wh65b-weather-station/3223

**Nexa Security, only discovered with the single universal RSSI key**, but no other properties at all - also broken in rtl_433 MQTT Auto Discovery
https://community.openmqttgateway.com/t/missing-discovery-messages-for-nexa-433mhz-devices/3305

From your name and profile here I think you might also speak German, so at 6:20 of - actually this issue here - also broken in rtl_433 MQTT Auto Discovery
https://www.youtube.com/watch?v=KbRNd1DyRbg

And some issue discussions here, where missing keys are asked for to be included in the auto-discovery. This will only be possible though when selectively conditioning these keys with their applicable type decoders, as otherwise all remaining keys in other decoders with that particular name will be discovered, but being of different types they will always only ever show up as **Unknown** in the wrongly discovered UI - possibly even worse than not being discovered at all.

This is also what I had to do with the "state" key when implementing the Nexa Security mentioned above. Only allowing "state" to be discovered as a binary_sensor for the Nexa-Security, Brennenstuhl-RCS2044, Proove-Security and Waveman-Switch. With the last one actually not yet working, because of
https://github.com/merbanan/rtl_433/pull/2946

So because of these different types being included in the same name keys it is currently virtually impossible, or only possible with very meticulous and long work and a pain to maintain, to have auto-discovery working nicely for all of rtl_433 ☹️

gdt, 2024-05-31:
To really fix this, I think we need to have a document with a normative data dictionary that says what keys mean and types, and then change the decoders that don't line up with that, separating per-decoder keys that just tell it like they got it (bit fidelity) and keys that are in a standard namespace.   The elephant in the room: are we willing to break backwards compat to fix things, or aren't we? I think we should be willing.

DigiH, 2024-05-31:
> The elephant in the room: are we willing to break backwards compat to fix things, or aren't we? I think we should be willing.

That is always the big worry ;) and I also commented this for the yet to be merged battery_ok fix for OpenMQTTGateway. But if a complete and well done auto-discovery for rtl_433 is desired at all, breaking backwards compatibility is unavoidable.

My vote would also be yes, as more and more HA users, and other controllers picking up the HA auto-discovery convention, rely on auto-discovery a lot.

DigiH, 2024-05-31:
Quite frankly, that is also why I started with logging this issue and the property casing consistency PR, as both are braking, but not majorly breaking, to see what the responses would be.

And property casing consistency is one vital base requirement for all possible other changes towards a better auto-discoverabilty for any binary string properties.

So both might be a starting test bed for any such changes.

sheilbronn, 2025-03-30:
I've been following this discussion because I was interested on how to deal with this inconsistency in my rtl_433 wrapper script ([rtl2mqtt](https://github.com/sheilbronn/rtl2mqtt)).
FYI: I used the following premises fpr MQTT/HASS autodiscovery announcements for the rtl_433 output in battery_ok:

- If value is 0 or 1: announce it as a binary_sensor (with 1 as the payload for ON, and 0 for OFF)
- Any other numeric value between 0 and 1: Announce it as a measurement sensor.

(If anybody is interested I can also post samples for these announcements here.)

DigiH, 2025-03-30:
Hi @sheilbronn 

While this distinction was also fun my head when trying to get the battery status discovered fine, I ended up manually going through the included decoders and then differentiating by the models ;) as

> If value is 0 or 1: announce it as a binary_sensor (with 1 as the payload for ON, and 0 for OFF)

while 0 is pretty obviously a binary_sensor, and would never realistically come with numeric measurement sensors, if the 1 differentiated itself as an integer compared to a possible 1.0 for numeric sensors. And with this initial full battery scenario, likely at the beginning of taking new devices into service and discovering them, if this integer vs. float was there to differentiate. 

Can you confirm any such 1 vs. 1.0 difference?

sheilbronn, 2025-03-30:
Hi @DigiH 
> Can you confirm any such 1 vs. 1.0 difference?

I can't confirm this difference (don't have a device emitting something different than 0 or 1).

To be explicit, the regular expressions in my (bash) code currently are:
``` sh
_battok=$(cExtractJsonVal battery_ok)
_bHasBatteryOK=$(  [[ $_battok =~ ^[01]$ ]] && e1 ) # 0=LOW;1=HIGH from https://triq.org/rtl_433/DATA_FORMAT.html#common-device-data
_bHasBatteryOKVal=$( [[ $_battok =~ ^[01].[0-9]+$ ]] && e1 ) # or some value in between
```

DigiH, 2025-03-30:
> I can't confirm this difference (don't have a device emitting something different than 0 or 1).

Same here ;) so I ended up with the surefire, but very cumbersome, model filtering.

sheilbronn, 2025-03-30:
Hi @zuckschwerdt , @gdt ,
this discussion with @DigiH inspired me to an idea:
Would you accept changes to those decoders, that might also emit a battery_ok value other than "0" or "1",
to emit the 0 and the 1 as **0.0** and **1.0** in the JSON for the MQTT?
With this a textual analysis of the battery_ok value, e.g. by rtl_433_mqtt_hass.py, could distiguish between _binary_sensors_ and _measurements_!
The advantage of this would be that it is a non-breaking change....

zuckschwerdt, 2025-03-30:
> changes to those decoders, that might also emit a battery_ok value other than "0" or "1", to emit the 0 and the 1 as **0.0** and **1.0**

They should already do that. If there is a range then the value is a float and should encode with a decimal point (and three digits in JSON).
You can grep for some samples with e.g. `rg 'battery_ok" : .\.' rtl_433_tests/tests/**/*.json` -- one such sample is `rtl_433_tests/tests/kedsum/kedsum-62.7f-61h.json`

A list of non-integer battery_ok keys can be found with e.g. `rg 'battery_ok.*DATA_' src/devices | rg -v DATA_INT` and should be: govee.c, inkbird_ith20r.c, fineoffset_ws90.c, archos_tbh.c, kedsum.c, fineoffset_wh55.c, fineoffset_wh46.c, tpms_ave.c, fineoffset_wh31l.c, ambientweather_wh31e.c, fineoffset_ws80.c, fineoffset_wn34.c, fineoffset_wh45.c, fineoffset.c.

gdt, 2025-03-30:
I thought we were going to regularize around battery_ok being a binary sensor and battery or battery_level being a 0-100 numeric sensor.  I think it's asking for trouble to put two kinds of semantics in one name.

sheilbronn, 2025-03-30:
> zuckschwerdt: ... They should already do that. If there is a range then the value is a float and should encode with a decimal point (and three digits in JSON). ...

Sounds almost exactly like what I thought about. Thx!

> gdt: I thought we were going to regularize around battery_ok being a binary sensor and battery or battery_level being a 0-100 numeric sensor. I think it's asking for trouble to put two kinds of semantics in one name.

Yes, Probably most here would agree on that but it's not more worse than before and at least autodiscovery will work fine for now.

zuckschwerdt, 2025-06-16:
note: this is continued in #3103