merbanan/rtl_433 #2950: Acurite 6002m not working

GHSTUDIO1, 2024-05-30:
I replaced an older acurite temp/humidity sensor with their newer model, the 6002m.....however the 6002m is not being decoded by RTL_433.  I am running with no protocol entries and I tried entering all of the acurite ones as separate protocols.....

RTL-433 shows it's up and waiting patiently for input...and node-red has connected to rtl_433.

Is there an updated protocol or which one should I use???

gdt, 2024-05-30:
I'm using  a 6002 and it works fine.   This seems like a discussion not a bug report, and it sounds like you need some back and forth guidance.  Others might be fine with using issues with that, but I personally avoid that, so this is my last issue comment, and happy to see you in discussions or the mailinglist.

GHSTUDIO1, 2024-05-31:
I wasn’t sure if it was a bug, so I posted.  And the resolution is it’s
apparently not a bug.  Okay.

gdt, 2024-05-31:
No, I meant that it seems to me that you are having trouble and need help; if that's what you think then feel free to open a discussion or ask on the mailinglist.  If you can make a case there's a bug, that's fine too, but you left out all the details that would be necessary for that.

ProfBoc75, 2024-05-31:
@GHSTUDIO1 : try `rtl_433 -R 40:v` or vv to get more information and analyze the result which could explain why not working. If no result try `rtl_433 -v` / -vv / -vvv ... Increase verbose level until you get some results that could help you find the reason and may be identify a possible improvement or decoder update follow/track by the current issue.

rct, 2024-05-31:
Acurite's SKUs/Model numbers can be a bit difficult to track since there are multiple numbers for the same device. Be aware that the 06002M was originally known and listed in rtl_433 as the 592TXR or "Tower Sensor".  There are quite a few Acurite devices with the same message format under protocol 40 that are quite a few years old. 

Given that they are supposed to be compatible with the Acurite Access and various displays, it is pretty unlikely that they've changed the message format. 

So given that, it would be easier to help you here (or in discussions) if you provided more information:

* Does anything else (Acurite display) receive and decode the new device?
* Does your rtl_433 set up correctly receive any other devices?
* Do you have protocol 40 enabled?  Startup messages and debug output as suggested above by @ProfBoc75 would be helpful.   Note: Acurite does make devices that are decoded by other protocols (606, 609, fridge/fieezer, etc. so your old set up might have worked with a different protocol enabled.

gdt, 2024-06-23:
answered question with discussion trailed off.   So far no reason to think this is a bug.   @GHSTUDIO1 Feel free to open a "Discussion" or ask on the mailinglist if you need help.

closed 2024-06-23 by gdt