merbanan/rtl_433 #2955: Fineoffset/Ecowitt WH51 every second message is received twice

bz177, 2024-06-02:
I noticed in my logs that I get every second message from my Ecowitt WH51 twice. 

Looking at several double received messages, the time stamps of those second message arrive around 35900 µs after the first message, one time after 36800µs. 

It seems that the device sends every second message either twice or with a different duration so the second message don't get filtered out. 

Some example outputs:

```
[rtl_433] {"time" : "2024-06-02 19:50:13.241046", "protocol" : 142, "model" : "Fineoffset-WH51", "id" : "0d83d0", "battery_ok" : 0.889, "battery_mV" : 1500, "moisture" : 44, "boost" : 0, "ad_raw" : 237, "mic" : "CRC", "mod" : "FSK", "freq1" : 868.260, "freq2" : 868.318, "rssi" : -0.115, "snr" : 37.258, "noise" : -37.373}
[rtl_433] {"time" : "2024-06-02 19:51:23.239424", "protocol" : 142, "model" : "Fineoffset-WH51", "id" : "0d83d0", "battery_ok" : 0.889, "battery_mV" : 1500, "moisture" : 44, "boost" : 0, "ad_raw" : 237, "mic" : "CRC", "mod" : "FSK", "freq1" : 868.260, "freq2" : 868.318, "rssi" : -0.111, "snr" : 36.013, "noise" : -36.124}
[rtl_433] {"time" : "2024-06-02 19:51:23.275304", "protocol" : 142, "model" : "Fineoffset-WH51", "id" : "0d83d0", "battery_ok" : 0.889, "battery_mV" : 1500, "moisture" : 44, "boost" : 0, "ad_raw" : 237, "mic" : "CRC", "mod" : "FSK", "freq1" : 868.258, "freq2" : 868.318, "rssi" : -0.113, "snr" : 42.031, "noise" : -42.144}
[rtl_433] {"time" : "2024-06-02 19:52:33.237758", "protocol" : 142, "model" : "Fineoffset-WH51", "id" : "0d83d0", "battery_ok" : 0.889, "battery_mV" : 1500, "moisture" : 44, "boost" : 0, "ad_raw" : 237, "mic" : "CRC", "mod" : "FSK", "freq1" : 868.261, "freq2" : 868.318, "rssi" : -0.103, "snr" : 37.270, "noise" : -37.373}
[rtl_433] {"time" : "2024-06-02 19:53:43.235712", "protocol" : 142, "model" : "Fineoffset-WH51", "id" : "0d83d0", "battery_ok" : 0.889, "battery_mV" : 1500, "moisture" : 44, "boost" : 0, "ad_raw" : 237, "mic" : "CRC", "mod" : "FSK", "freq1" : 868.260, "freq2" : 868.319, "rssi" : -0.130, "snr" : 35.025, "noise" : -35.154}
[rtl_433] {"time" : "2024-06-02 19:53:43.271612", "protocol" : 142, "model" : "Fineoffset-WH51", "id" : "0d83d0", "battery_ok" : 0.889, "battery_mV" : 1500, "moisture" : 44, "boost" : 0, "ad_raw" : 237, "mic" : "CRC", "mod" : "FSK", "freq1" : 868.264, "freq2" : 868.319, "rssi" : -0.124, "snr" : 35.999, "noise" : -36.124}
[rtl_433] {"time" : "2024-06-02 19:54:53.243116", "protocol" : 142, "model" : "Fineoffset-WH51", "id" : "0d83d0", "battery_ok" : 0.889, "battery_mV" : 1500, "moisture" : 44, "boost" : 0, "ad_raw" : 237, "mic" : "CRC", "mod" : "FSK", "freq1" : 868.261, "freq2" : 868.317, "rssi" : -0.126, "snr" : 33.567, "noise" : -33.693}
[rtl_433] {"time" : "2024-06-02 19:56:03.240876", "protocol" : 142, "model" : "Fineoffset-WH51", "id" : "0d83d0", "battery_ok" : 0.889, "battery_mV" : 1500, "moisture" : 44, "boost" : 0, "ad_raw" : 237, "mic" : "CRC", "mod" : "FSK", "freq1" : 868.262, "freq2" : 868.317, "rssi" : -0.118, "snr" : 37.255, "noise" : -37.373}
[rtl_433] {"time" : "2024-06-02 19:56:03.276772", "protocol" : 142, "model" : "Fineoffset-WH51", "id" : "0d83d0", "battery_ok" : 0.889, "battery_mV" : 1500, "moisture" : 44, "boost" : 0, "ad_raw" : 237, "mic" : "CRC", "mod" : "FSK", "freq1" : 868.265, "freq2" : 868.320, "rssi" : -0.102, "snr" : 39.031, "noise" : -39.134}
[rtl_433] {"time" : "2024-06-02 19:57:13.231906", "protocol" : 142, "model" : "Fineoffset-WH51", "id" : "0d83d0", "battery_ok" : 0.889, "battery_mV" : 1500, "moisture" : 44, "boost" : 0, "ad_raw" : 237, "mic" : "CRC", "mod" : "FSK", "freq1" : 868.262, "freq2" : 868.319, "rssi" : -0.116, "snr" : 39.018, "noise" : -39.134}
[rtl_433] {"time" : "2024-06-02 19:58:23.238571", "protocol" : 142, "model" : "Fineoffset-WH51", "id" : "0d83d0", "battery_ok" : 0.889, "battery_mV" : 1500, "moisture" : 44, "boost" : 0, "ad_raw" : 237, "mic" : "CRC", "mod" : "FSK", "freq1" : 868.263, "freq2" : 868.318, "rssi" : -0.117, "snr" : 37.256, "noise" : -37.373}
[rtl_433] {"time" : "2024-06-02 19:58:23.274483", "protocol" : 142, "model" : "Fineoffset-WH51", "id" : "0d83d0", "battery_ok" : 0.889, "battery_mV" : 1500, "moisture" : 44, "boost" : 0, "ad_raw" : 237, "mic" : "CRC", "mod" : "FSK", "freq1" : 868.261, "freq2" : 868.319, "rssi" : -0.121, "snr" : 36.002, "noise" : -36.124}
[rtl_433] {"time" : "2024-06-02 19:59:33.229154", "protocol" : 142, "model" : "Fineoffset-WH51", "id" : "0d83d0", "battery_ok" : 0.889, "battery_mV" : 1500, "moisture" : 44, "boost" : 0, "ad_raw" : 237, "mic" : "CRC", "mod" : "FSK", "freq1" : 868.262, "freq2" : 868.318, "rssi" : -0.133, "snr" : 35.991, "noise" : -36.124}
[rtl_433] {"time" : "2024-06-02 20:00:43.227948", "protocol" : 142, "model" : "Fineoffset-WH51", "id" : "0d83d0", "battery_ok" : 0.889, "battery_mV" : 1500, "moisture" : 44, "boost" : 0, "ad_raw" : 237, "mic" : "CRC", "mod" : "FSK", "freq1" : 868.263, "freq2" : 868.318, "rssi" : -0.134, "snr" : 37.239, "noise" : -37.373}
[rtl_433] {"time" : "2024-06-02 20:00:43.263836", "protocol" : 142, "model" : "Fineoffset-WH51", "id" : "0d83d0", "battery_ok" : 0.889, "battery_mV" : 1500, "moisture" : 44, "boost" : 0, "ad_raw" : 237, "mic" : "CRC", "mod" : "FSK", "freq1" : 868.260, "freq2" : 868.319, "rssi" : -0.120, "snr" : 33.573, "noise" : -33.693}
```

zuckschwerdt, 2024-06-02:
Those look like genuine messages. rtl_433 does not de-duplicate in any way, those messages can be used to verify values and good reception.
Closing as this is not something we can do anything about. Feel free to discuss further if you need some help.

closed 2024-06-02 by zuckschwerdt

gdt, 2024-06-02:
That's not a bug; rtl_433 intentionally decodes what is sent.

See #2640.