merbanan/rtl_433 #2961: Add support for ThermoPro TP829B

AryehGielchinsky, 2024-06-09:
I'd like help adding the [ThermoPro TP829B](https://buythermopro.com/product/tp829/) meat thermometer. I'm somewhat new to radio decoding and RTL_433. I've read through https://triq.org/rtl_433/.

I have my rtl_sdr positioned an inch from the transmitter with the antennas unscrewed.

`rtl_433 -f 915.1M -S unknown` produces the attached files with probe1 reading 77F and the other 3 probes disconnected.
[77F_probe1.zip](https://github.com/user-attachments/files/15751685/77F_probe1.zip)

```
(base) aryeh@desktop:~$ rtl_433 -A g001_915.1M_1000k.cu8 -s 1000k
rtl_433 version 23.11-120-g8d35eb3e branch master at 202405210902 inputs file rtl_tcp RTL-SDR
[Input] Test mode active. Reading samples from file: g001_915.1M_1000k.cu8
Detected FSK package	@0.022479s
Analyzing pulses...
Total count:   74,  width: 26.67 ms		(26673 S)
Pulse width distribution:
 [ 0] count:    1,  width:    0 us [0;0]	(   0 S)
 [ 1] count:   52,  width:  102 us [93;106]	( 102 S)
 [ 2] count:   10,  width:  204 us [202;209]	( 204 S)
 [ 3] count:    5,  width:  307 us [306;312]	( 307 S)
 [ 4] count:    4,  width:  412 us [408;416]	( 412 S)
 [ 5] count:    1,  width:  719 us [719;719]	( 719 S)
 [ 6] count:    1,  width:   12 us [12;12]	(  12 S)
Gap width distribution:
 [ 0] count:    1,  width:   34 us [34;34]	(  34 S)
 [ 1] count:   59,  width:  103 us [98;107]	( 103 S)
 [ 2] count:    9,  width:  205 us [202;209]	( 205 S)
 [ 3] count:    2,  width:  563 us [508;618]	( 563 S)
 [ 4] count:    1,  width:  306 us [306;306]	( 306 S)
 [ 5] count:    1,  width: 6003 us [6003;6003]	(6003 S)
Pulse period distribution:
 [ 0] count:    1,  width:   34 us [34;34]	(  34 S)
 [ 1] count:   40,  width:  205 us [200;210]	( 205 S)
 [ 2] count:   18,  width:  307 us [305;314]	( 307 S)
 [ 3] count:    6,  width:  409 us [404;412]	( 409 S)
 [ 4] count:    3,  width:  647 us [613;717]	( 647 S)
 [ 5] count:    1,  width:  822 us [822;822]	( 822 S)
 [ 6] count:    3,  width:  515 us [515;515]	( 515 S)
 [ 7] count:    1,  width: 6109 us [6109;6109]	(6109 S)
Pulse timing distribution:
 [ 0] count:    1,  width:    0 us [0;0]	(   0 S)
 [ 1] count:  111,  width:  102 us [93;107]	( 102 S)
 [ 2] count:   19,  width:  204 us [202;209]	( 204 S)
 [ 3] count:    6,  width:  307 us [306;312]	( 307 S)
 [ 4] count:    5,  width:  431 us [408;508]	( 431 S)
 [ 5] count:    2,  width:  668 us [618;719]	( 668 S)
 [ 6] count:    1,  width:   12 us [12;12]	(  12 S)
 [ 7] count:    1,  width:   34 us [34;34]	(  34 S)
 [ 8] count:    1,  width: 6003 us [6003;6003]	(6003 S)
 [ 9] count:    1,  width:    0 us [0;0]	(   0 S)
Level estimates [high, low]:  15933,      9
RSSI: -0.1 dB SNR: 32.5 dB Noise: -32.6 dB
Frequency offsets [F1, F2]:   -1128,  -9558	(-17.2 kHz, -145.8 kHz)
Guessing modulation: No clue...
```

Uploading the samples to https://triq.org/pdv/, it looks like there are 2 frequencies, so FSK. There are a bunch of evenly spaced 1s and 0s at the beginning, which indicates PCM Non-return-to-Zero (which is what the website guesses).

The length of the 1s and 0s at the beginning are about 100us. Setting the pdv site `short` and `long` to 100us each I get
`Bits: {259} 55 55 55 55 55 55 55 69 2A 96 EA 04 17 97 7F 6E ED DE DD E8 D2 D2 D2 D2 D2 00 00 00 00 00 00 00 0`

When probe1 is at 76F I get
`Bits: {258} 55 55 55 55 55 55 55 69 2A 96 EA 04 17 97 47 6E ED DE DD B1 D2 D2 D2 D2 C4 00 00 00 00 00 00 00 0`

Am I heading the in right direction? Any advice on how to decode those Bits?

I've tried `rtl_433 -f 915M -F json -X "n=tp829b,m=FSK_PCM,s=105,l=105,r=5500,preamble=d2aa2dd4"` based on a [similar device](https://github.com/merbanan/rtl_433/pull/2882/commits/eb8ca7f8d6c3a412ea92d44abba3528d8ee20974), but didn't get any output.

Also, what are the red vertical lines in the chart below?
[image]

Thanks

zuckschwerdt, 2024-06-09:
PCM 100 µs seems correct. You don't have that sync-word though (preamble= parameter), try this to get raw data:
`-X "n=tp829b,m=FSK_PCM,s=100,l=100,r=2000,preamble=aaaa,bits>=100"`

Try to collect resulting data codes while slowly changing the temperature to observe the field where it's encoded.

AryehGielchinsky, 2024-06-09:
`rtl_433 -f 915M -F json -X "n=tp829b,m=FSK_PCM,s=100,l=100,r=2000,preamble=aaaa,bits>=100"  | jq --unbuffered -r '.codes[0]'`

Probe1 started at 85F, dropped to 76F.
```
{203}aaaaaaaaaad2552dd4082f317eddeddedd56d2d2d2d2d200000
{203}aaaaaaaaaad2552dd4082f314eddeddeddded2d2d2d2d200000
{203}aaaaaaaaaad2552dd4082f311eddeddedd77d2d2d2d2d200000
{203}aaaaaaaaaad2552dd4082f30feddeddedd53d2d2d2d2d200000
{203}aaaaaaaaaad2552dd4082f30ceddeddedddbd2d2d2d2d200000
{203}aaaaaaaaaad2552dd4082f30aeddeddeddfad2d2d2d2d200000
{204}aaa9555555692a96ea0417983f6ef6ef6ee0696969696900000
{205}aaaaaaaaaad2552dd4082f304eddeddedd48d2d2d2d2d2000000
{203}aaaaaaaaaad2552dd4082f302eddeddedd69d2d2d2d2d200000
{206}aaaaaaaaaad2552dd4082f300eddeddedd99d2d2d2d2d2000000
{203}aaaaaaaaaad2552dd4082f2feeddeddedd06d2d2d2d2d200000
{203}aaaaaaaaaad2552dd4082f2fceddeddeddf6d2d2d2d2d200000
{203}aaaaaaaaaad2552dd4082f2faeddeddeddd7d2d2d2d2d200000
{208}aaaaaaaaaad2552dd4082f2f8eddeddedd27d2d2d2d2d2000000
{203}aaaaaaaaaad2552dd4082f2f6eddeddedd95d2d2d2d2d200000
{203}aaaaaaaaaad2552dd4082f2f4eddeddedd65d2d2d2d2d200000
{203}aaaaaaaaaad2552dd4082f2f2eddeddedd44d2d2d2d2d200000
{203}aaaaaaaaaad2552dd4082f2f1eddeddeddccd2d2d2d2d200000
{212}aaaaaaaaaad2552dd4082f2f0eddeddeddb4d2d2d2d2d20000000
{203}aaaaaaaaaad2552dd4082f2efeddeddedde8d2d2d2d2d200000
{203}aaaaaaaaaad2552dd4082f2ededdeddedd18d2d2d2d2d200000
{203}aaaaaaaaaad2552dd4082f2eceddeddedd60d2d2d2d2d200000
{203}aaaaaaaaaad2552dd4082f2eceddeddedd60d2d2d2d2d200000
{203}aaaaaaaaaad2552dd4082f2ebeddeddedd39d2d2d2d2d200000
{203}aaaaaaaaaad2552dd4082f2ebeddeddedd39d2d2d2d2d200000
{203}aaaaaaaaaad2552dd4082f2eaeddeddedd41d2d2d2d2d200000
```

AryehGielchinsky, 2024-06-09:
`rtl_433 -f 915M -F json -X "n=tp829b,m=FSK_PCM,s=100,l=100,r=2000,preamble=aaaa,bits>=100"  | jq --unbuffered -r '.codes[0]'`

Start: all 4 probes between 90-91F
End: all 4 probes between 79-80F

```
{233}aaaaaaaaaad2552dd4082f33733533c33771d2d2d2d2d20000000000000
{219}aaaaaaaaaad2552dd4082f33733533c33771d2d2d2d2d2000000000
{203}aaaaaaaaaad2552dd4082f33733533c33771d2d2d2d2d200000
{225}aaaaaaaaaad2552dd4082f33733533c33771d2d2d2d2d200000000000
{203}aaaaaaaaaad2552dd4082f33733533c33771d2d2d2d2d200000
{203}aaaaaaaaaad2552dd4082f33733633c33748d2d2d2d2d200000
{203}aaaaaaaaaad2552dd4082f33733633c33748d2d2d2d2d200000
{203}aaaaaaaaaad2552dd4082f33733633b3370ad2d2d2d2d200000
{203}aaaaaaaaaad2552dd4082f334331337333b4d2d2d2d2d200000
{211}aaaaaaaaaad2552dd4082f33132e33332f87d2d2d2d2d20000000
{203}aaaaaaaaaad2552dd4082f32e32a33032d8ed2d2d2d2d200000
{216}aaaaaaaaaad2552dd4082f32d32732e32a41d2d2d2d2d200000000
{203}aaaaaaaaaad2552dd4082f32b32432c3282dd2d2d2d2d200000
{203}aaaaaaaaaad2552dd4082f32832232832522d2d2d2d2d200000
{203}aaaaaaaaaad2552dd4082f327320327322dbd2d2d2d2d200000
{203}aaaaaaaaaad2552dd4082f32531e3263225fd2d2d2d2d200000
{203}aaaaaaaaaad2552dd4082f32331c32432024d2d2d2d2d200000
{203}aaaaaaaaaad2552dd4082f32131b32231d53d2d2d2d2d200000
{203}aaaaaaaaaad2552dd4082f32031932031cb8d2d2d2d2d200000
{203}aaaaaaaaaad2552dd4082f31e31831f31bc6d2d2d2d2d200000
{203}aaaaaaaaaad2552dd4082f31b31631d319d1d2d2d2d2d200000
{203}aaaaaaaaaad2552dd4082f31a31631c31893d2d2d2d2d200000
{203}aaaaaaaaaad2552dd4082f31931431a316f3d2d2d2d2d200000
{203}aaaaaaaaaad2552dd4082f3173123193157dd2d2d2d2d200000
{203}aaaaaaaaaad2552dd4082f3173103183138dd2d2d2d2d200000
{203}aaaaaaaaaad2552dd4082f31530f3173125fd2d2d2d2d200000
{203}aaaaaaaaaad2552dd4082f31330e3163109ad2d2d2d2d200000
{203}aaaaaaaaaad2552dd4082f31130c31530fc9d2d2d2d2d200000
{203}aaaaaaaaaad2552dd4082f31030c31330d00d2d2d2d2d200000
{203}aaaaaaaaaad2552dd4082f30f30a31230d53d2d2d2d2d200000
{203}aaaaaaaaaad2552dd4082f30d30931030c27d2d2d2d2d200000
{203}aaaaaaaaaad2552dd4082f30c30831030aebd2d2d2d2d200000
{222}aaaaaaaaaad2552dd4082f30b30730f3098bd2d2d2d3a40000000000
{202}aaaaaaaaaad2552dd4085e61260a61a611d1a5a5a5a5a400000
{208}aaaaaaaaaad2552dd4082f30830430c30603d2d2d2d2d2000000
{203}aaaaaaaaaad2552dd4082f30730230a3054dd2d2d2d2d200000
{203}aaaaaaaaaad2552dd4082f30630130930328d2d2d2d2d200000
{207}aaaaaaaaaad2552dd4082f304300307301c5d2d2d2d2d2000000
{203}aaaaaaaaaad2552dd4082f3032fe305300e8d2d2d2d2d200000
{222}aaaaaaaaaad2552dd4082f3022fd3032fea5d2d2d2d2d20000000000
{203}aaaaaaaaaad2552dd4082f3012fb3032fd96d2d2d2d2d200000
{203}aaaaaaaaaad2552dd4082f3002fa3022fcc3d2d2d2d2d200000
{203}aaaaaaaaaad2552dd4082f2ff2f93012fa39d2d2d2d2d200000
```

ProfBoc75, 2024-06-09:
Hi @zuckschwerdt , @AryehGielchinsky , may have a look to #2877 , could be current decoder to be updated with this model.

Preamble is very closed, d2552dd4 here and d2aa2dd4 for TB28B. 
Data is shorter here also. Not yet check all the data layout

AryehGielchinsky, 2024-06-09:
Hi @ProfBoc75 

The TB829B is manufactured by the same company as the TP28B, but I think there are some differences.
From a practical view:
1) TP28B has 2 meat probes, while the TB829B has 4.
2) TP28B has some settings like temperature warnings (temp too high or low) that can be set at the transmitter, while the TB829B has those settings on the receiver.

From a technical view:
1) The TP28B temperatures are in 16 bit BCD, while I think the TB829B temperatures are some kind of 3 digit hex.
2) The TP28B transmits the hi and low desired temperature range, while the TB829B doesn't (see 2 above).

I added bitbench links to my 2 comments above after posting them. This is the format I see.
`CONSTANT:hhhhhhhhhhhhhhhhhhhhhh TEMPONE:hhh TEMPTWO:hhh TEMPTHREE:hhh TEMPFOUR:hhh SOMEVAR:hh CONSTANT:hhhhhhhhhhhhhhhhhhhhhhh`

`CONSTANT=aaaaaaaaaad2552dd4082f`
`TEMPONE=3 digit hex, edd if probe disconnected`
`TEMPTWO=3 digit hex, edd if probe disconnected`
`TEMPTHREE=3 digit hex, edd if probe disconnected`
`TEMPFOUR=3 digit hex, edd if probe disconnected`
`SOMEVAR=2 digit hex` (maybe a checksum)
`CONSTANT=d2d2d2d2d200000`

How did you know the preamble is `d2552dd4` and not `d2552dd4082f`?

ProfBoc75, 2024-06-09:
> How did you know the preamble is `d2552dd4` and not `d2552dd4082f`?

- By experience and compare to other devices. 

Your Temp value are in °C, offset 500, scale 10.

I'm still working on the checksum part, not a CRC from reveng.

My [bitbench](https://triq.net/bitbench#c=082f33733533c33771&c=082f33733633c33748&c=082f33733633b3370a&c=082f334331337333b4&c=082f33132e33332f87&c=082f32e32a33032d8e&c=082f32d32732e32a41&c=082f32b32432c3282d&c=082f32832232832522&c=082f327320327322db&c=082f32531e3263225f&c=082f32331c32432024&c=082f32131b32231d53&c=082f32031932031cb8&c=082f31e31831f31bc6&c=082f31b31631d319d1&c=082f31a31631c31893&c=082f31931431a316f3&c=082f3173123193157d&c=082f3173103183138d&c=082f31530f3173125f&c=082f31330e3163109a&c=082f31130c31530fc9&c=082f31030c31330d00&c=082f30f30a31230d53&c=082f30d30931030c27&c=082f30c30831030aeb&c=082f30b30730f3098b&c=085e61260a61a611d1&c=082f30830430c30603&c=082f30730230a3054d&c=082f30630130930328&c=082f304300307301c5&c=082f3032fe305300e8&c=082f3022fd3032fea5&c=082f3012fb3032fd96&c=082f3002fa3022fcc3&c=082f2ff2f93012fa39&f=ID%20%3F%3A%2016h%20TEMPONE%3A12d%20TEMPTWO%3A12d%20TEMPTHREE%3A12d%20TEMPFOUR%3A12d%20CHECKSUM%20%3F%3A8h&z=1&cw=4)

I removed the preamble and the trailing bytes.

ProfBoc75, 2024-06-09:
Got it, the checksum is a Galois bit reflect, gen = 0x98, key = 0x55, final XOR 00

ProfBoc75, 2024-06-10:
@AryehGielchinsky : Are you able to get some values with low battery?

AryehGielchinsky, 2024-06-10:
@ProfBoc75 Thank you for all your helping adding this device. I'll try to get some low batteries (or maybe use some resistors to fake a low battery) in the next few days. I'll also see if the other buttons (C/F) change anything. Anything else you can suggest in terms of testing?

ProfBoc75, 2024-06-10:
@AryehGielchinsky : yes, If you play with the batteries, this will also allow us to see if the sensor id changes, the first 2 bytes of the message, my assumption. But I saw from other Thermopro temp sensors that the ID is one byte then one byte for flags like battery low, channel and button, could be the same here.

AryehGielchinsky, 2024-06-13:
@ProfBoc75 

- Removing and replacing batteries (10 second delay) doesn't change anything.
- Turning on the backlight on the transmitter doesn't change anything.
- Switched transmitter back and forth from C to F changes the 3rd character. "082f" when transmitter is in F, "080f" when in C. See data below.

Only Probe 1 attached. Ambiant temperature about 78F/26C. Switched transmitter back and forth from C to F.
```
rtl_433 -f 915M -F json -X "n=tp829b,m=FSK_PCM,s=100,l=100,r=2000,preamble=d2552dd4,bits>=100"  | jq --unbuffered -r '.codes[0]'
{131}080f2f5eddeddedddbd2d2d2d2d200000
{131}080f2f5eddeddedddbd2d2d2d2d200000
{131}080f2f5eddeddedddbd2d2d2d2d200000
{162}082f2f5eddeddedd1dd2d2d2d2d20000000000000
{131}082f2f5eddeddedd1dd2d2d2d2d200000
{131}082f2f5eddeddedd1dd2d2d2d2d200000
{140}082f2f5eddeddedd1dd2d2d2d2d20000000
{131}080f2f5eddeddedddbd2d2d2d2d200000
{140}080f2f6eddeddedd53d2d2d2d2d20000000
{131}080f2f6eddeddedd53d2d2d2d2d200000
{131}082f2f6eddeddedd95d2d2d2d2d200000
{131}082f2f6eddeddedd95d2d2d2d2d200000
{143}082f2f6eddeddedd95d2d2d2d2d200000000
```

AryehGielchinsky, 2024-06-13:
The back of my device has the following info:
- FCC ID: 2AATP-TP29B
- Model: TP829B

I've contacted 2 people I found in [this thread (Thermopro TP829 / TP-29)](https://groups.google.com/g/rtl_433/c/wdzXxoz4Ehk) and asked them to post the output of the following command to this issue:

`rtl_433 -f 915M -F json -X "n=tp829b,m=FSK_PCM,s=100,l=100,r=2000,preamble=d2552dd4,bits>=100"  | jq --unbuffered -r '.codes[0]'` 

This may help with the first few characters.

ProfBoc75, 2024-06-13:
@AryehGielchinsky : Thx, So 0x08 could be the ID of the sensor and 0x0F or 0x2F some flags, the 0x2 is already link to °C / F from the other tp28b model.

    - flags & 0x2000 => Display in Celcius, otherwise Fahrenheit 

But this doesn't change the raw temp values. 

Waiting for feedback from the group to go further.

AryehGielchinsky, 2024-06-21:
@ProfBoc75 I haven't heard back from anyone in the thread I posted (I tried posting to the thread, and private messages), I think we should continue moving forward.

I pulled and checked out your branch `feat-thermopro_tp829b` and rebuilt today. Here is my output.

`4 probes, set to Fahrenheit`
```
(base) aryeh@desktop:~/repos/rtl_433$ rtl_433 -f 915.1M
rtl_433 version 23.11-133-ga0512ccf branch feat-thermopro_tp829b at 202406101936 inputs file rtl_tcp RTL-SDR

New defaults active, use "-Y classic -s 250k" if you need the old defaults

Detached kernel driver
Found Rafael Micro R820T tuner
[SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2024-06-21 12:55:55
model     : ThermoPro-TP829b                       id        : 082f
Temperature 1: 23.3 C    Temperature 2: 25.4 C     Temperature 3: 25.7 C     Temperature 4: 24.9 C     Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2024-06-21 12:56:03
model     : ThermoPro-TP829b                       id        : 082f
Temperature 1: 23.4 C    Temperature 2: 25.3 C     Temperature 3: 25.7 C     Temperature 4: 24.8 C     Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2024-06-21 12:56:23
model     : ThermoPro-TP829b                       id        : 082f
Temperature 1: 23.3 C    Temperature 2: 25.2 C     Temperature 3: 25.6 C     Temperature 4: 24.7 C     Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2024-06-21 12:56:31
model     : ThermoPro-TP829b                       id        : 082f
Temperature 1: 23.3 C    Temperature 2: 25.2 C     Temperature 3: 25.6 C     Temperature 4: 24.7 C     Integrity : CHECKSUM
```

`probe4 only, set to Celsius`
```
(base) aryeh@desktop:~/repos/rtl_433$ rtl_433 -f 915.1M
rtl_433 version 23.11-133-ga0512ccf branch feat-thermopro_tp829b at 202406101936 inputs file rtl_tcp RTL-SDR

New defaults active, use "-Y classic -s 250k" if you need the old defaults

Detached kernel driver
Found Rafael Micro R820T tuner
[SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2024-06-21 12:58:35
model     : ThermoPro-TP829b                       id        : 080f
Temperature 4: 24.6 C    Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2024-06-21 12:58:39
model     : ThermoPro-TP829b                       id        : 080f
Temperature 4: 24.6 C    Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2024-06-21 12:58:47
model     : ThermoPro-TP829b                       id        : 080f
Temperature 4: 24.7 C    Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2024-06-21 12:59:07
model     : ThermoPro-TP829b                       id        : 080f
Temperature 4: 24.7 C    Integrity : CHECKSUM
```

Temperature readings look good. 2 questions:
1) The id value has the C/F flag in it, would it make sense to break it out?
2) The transmitter transmits every 4 seconds, but you can see from the timestamps that I'm not catching every transmit, is that expected? When I use the command below, I do get a ping every 4 seconds. Any way I can debug this?
`rtl_433 -f 915M -F json -X "n=tp829b,m=FSK_PCM,s=100,l=100,r=2000,preamble=d2552dd4,bits>=100"  | jq --unbuffered -r '.codes[0]'`

Thanks

zuckschwerdt, 2024-06-21:
You can increase verbosity (`-v`, `-vv`, …) to see what rtl_433 and the decoders are doing. But only enable one decoder (`-R 259`) to limit the amount of logging. Or just switch that one decoder verbose (`-R 259:v`, `-R 259:vv`, …).

AryehGielchinsky, 2024-06-21:
Probe1 only@ about 77F
```
(base) aryeh@desktop:~$ rtl_433 -f 915.1M -R 259:vv
rtl_433 version 23.11-133-ga0512ccf branch feat-thermopro_tp829b at 202406101936 inputs file rtl_tcp RTL-SDR

New defaults active, use "-Y classic -s 250k" if you need the old defaults

Detached kernel driver
Found Rafael Micro R820T tuner
[SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
[thermopro_tp829b_decode] Checksum error, calculated 9, expected c1
[pulse_slicer_pcm] ThermoPro TP829b Meat Thermometer 4 coated probes
codes     : {253}555554aaaaaa55a4aa5ba810bcbb3b77b77b75834b4a8d2d2000000000000000
[thermopro_tp829b_decode] Checksum error, calculated 11, expected c1
[pulse_slicer_pcm] ThermoPro TP829b Meat Thermometer 4 coated probes
codes     : {257}55555555555555692a96ea04172eceddeddedd60d2d2d2d2d2000000000000000
[thermopro_tp829b_decode] Checksum error, calculated dd, expected c1
[pulse_slicer_pcm] ThermoPro TP829b Meat Thermometer 4 coated probes
codes     : {258}55555555555555692a96ea041797676deddedd60d2d2d2d2d2000000000000000
[thermopro_tp829b_decode] Checksum error, calculated 6a, expected 83
[pulse_slicer_pcm] ThermoPro TP829b Meat Thermometer 4 coated probes
codes     : {251}55555555555555692a96ea041797676eedbdbac1a5a5a5a5a40000000000000
[thermopro_tp829b_decode] Checksum error, calculated 28, expected b0
[pulse_slicer_pcm] ThermoPro TP829b Meat Thermometer 4 coated probes
codes     : {252}5555555552aaaad2552dd4082f2ecedef6ef6eb069696969520000000000000
[thermopro_tp829b_decode] MSG
codes     : {72}082f2eceddeddedd60
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2024-06-21 15:30:07
model     : ThermoPro-TP829b                       id        : 082f
Temperature 1: 24.8 C    Integrity : CHECKSUM
[pulse_slicer_pcm] ThermoPro TP829b Meat Thermometer 4 coated probes
codes     : {257}55555555555555692a96ea041797676ef6ef6eb0696968d2d2000000000000000
[thermopro_tp829b_decode] Checksum error, calculated fc, expected 31
[pulse_slicer_pcm] ThermoPro TP829b Meat Thermometer 4 coated probes
codes     : {259}52aaad55555555692a96ea0417976f6ef6ef6d18d2d2d2d2d2000000000000000
[thermopro_tp829b_decode] Checksum error, calculated 44, expected 83
[pulse_slicer_pcm] ThermoPro TP829b Meat Thermometer 4 coated probes
codes     : {250}555555554aaaaad2552dd4082f2ecdbbdbbdb5834b4b4b4b480000000000000
[thermopro_tp829b_decode] Checksum error, calculated d8, expected c1
[pulse_slicer_pcm] ThermoPro TP829b Meat Thermometer 4 coated probes
codes     : {251}55555555555555692a96ea041796ceddeddedd60d2d1a5a5a40000000000000
[thermopro_tp829b_decode] Sync word not found
[pulse_slicer_pcm] ThermoPro TP829b Meat Thermometer 4 coated probes
codes     : {258}5555555555555568552dd4082f2f676ef6ef6eb06969696969000000000000000
[thermopro_tp829b_decode] Sync word not found
[pulse_slicer_pcm] ThermoPro TP829b Meat Thermometer 4 coated probes
codes     : {253}5555555555555562552dd4082f2eceddeddedd60d2d2d2d2d900000000000000
[thermopro_tp829b_decode] Checksum error, calculated d1, expected b0
[pulse_slicer_pcm] ThermoPro TP829b Meat Thermometer 4 coated probes
codes     : {254}55555555555555692a96ea041797676ef6efb75834b4b4b4b480000000000000
[thermopro_tp829b_decode] MSG
codes     : {72}082f2eceddeddedd60
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2024-06-21 15:30:35
model     : ThermoPro-TP829b                       id        : 082f
Temperature 1: 24.8 C    Integrity : CHECKSUM
[pulse_slicer_pcm] ThermoPro TP829b Meat Thermometer 4 coated probes
codes     : {258}55555555555555692a96ea041797676ef6ef6eb062d2d2d2d2000000000000000
[thermopro_tp829b_decode] Checksum error, calculated bb, expected c1
[pulse_slicer_pcm] ThermoPro TP829b Meat Thermometer 4 coated probes
codes     : {252}55555555555555692a96ea041797676ef5dedd60d2d2d2d2d20000000000000
```

AryehGielchinsky, 2024-06-21:
These codes are every 4 seconds
```
(base) aryeh@desktop:~$ rtl_433 -f 915M -F json -X "n=tp829b,m=FSK_PCM,s=100,l=100,r=2000,preamble=d2552dd4,bits>=100"  | jq --unbuffered -r '.codes[0]'
rtl_433 version 23.11-133-ga0512ccf branch feat-thermopro_tp829b at 202406101936 inputs file rtl_tcp RTL-SDR

New defaults active, use "-Y classic -s 250k" if you need the old defaults

Use "-F log" if you want any messages, warnings, and errors in the console.
Detached kernel driver
Found Rafael Micro R820T tuner
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
{143}082f2eceddeddedd60d2d2d2d2d200000000
null
{131}082f2eceddeddedd60d2d2d2d2d200000
null
{139}082f2eceddeddedd60d2d2d2d2d20000000
null
{131}082f2eceddeddedd60d2d2d2d2d200000
null
{131}082f2ebeddeddedd39d2d2d2d2d200000
null
{147}082f2ebeddeddedd39d2d2d2d2d2000000000
null
{131}082f2ebeddeddedd39d2d2d2d2d200000
null
{131}082f2ebeddeddedd39d2d2d2d2d200000
null
{131}082f2ebeddeddedd39d2d2d2d2d200000
null
{131}082f2ebeddeddedd39d2d2d2d2d200000
```

ProfBoc75, 2024-06-21:
Hi @AryehGielchinsky:

I need to check at the checksum level, since I have some difficulties there to have the good formula, the order of the data are reversed and for each byte the order of bit are reserved too.

Yes, I will split the 2 bytes ID in one byte and the second byte as Flags, with Display °C / Display F

Give some times (days ?) to correct my PR.

AryehGielchinsky, 2024-06-21:
@ProfBoc75 sure no problem, take your time. Let me know when to test again .

Thank you for all you've done to get this device in.

ProfBoc75, 2024-06-21:
About the check, the code is good.

- Your messages are durty, try to stay at `-f 915M` instead of 915.1M , at this frequency the sample rate is 1Mhz so your sensor will be in the range.
- Check your dongle, your antenna, may be you have some noise at RF level which can explain the errors.

Just to explain, the first code with checksum errors:

    {253}555554aaaaaa55a4aa5ba810bcbb3b77b77b75834b4a8d2d2000000000000000

after sync preamble 0xd2552dd4 we have this:

       08 5e 5d 9d bb db bd ba c1 a5 a5 46 96 90 00 00 00 00 00 00
    ID 08 FLAGS 5e ONE 5d9 TWO dbb THREE dbb FOUR dba CHECK c1

ID ok
Flags not ok, expected 0x2f (display F) or 0x0f (display °C) 
the rest not ok also, expected 0xedd for empty probes.

if we right shift 1 bit we have:

    ID 04 FLAGS 2f ONE 2ec TWO edd THREE edd FOUR edd CHECK 60

ID KO, expected 0x08, the flags and the rest look ok, I did not verify the check, but here, 1 bit shifted and the result can't be validated.

For other signal/messages, this is the same, the beginning of the message could be good , like 0x082F ... but not the rest and the reverse, may be the frame is not stable during the transmit, may be battery, may be noise ... lot of reasons, welcome to RF world.

I saw another error with sync not found, in that case the preamble can be shorted to 0x552dd4 instead of 0xd2552dd4 and will have more success (hope so), less sensitive to noise, the rest will be controlled with the checksum.

ProfBoc75, 2024-06-21:
Finally, I had time to make the update before the match ... please review my PR, and try !

- Sensor ID now 1 byte
- Display Unit C or F
- preamble reduced to 3 bytes for better success.

gdt, 2024-06-23:
@AryehGielchinsky  We just had a merge; please test with current master and either close or explain what needs to be done still.  Adding feedback with 7 day timer.

AryehGielchinsky, 2024-06-23:
@ProfBoc75 Good call on my frequency being off. That was the issue.

I switched to master and pulled. The data looks good! I'm catching every transmission, and temperature and `Display Unit` are correct. Thank you for adding this device.

```
(base) aryeh@desktop:~$ rtl_433 -f 915M -R 259
rtl_433 version 23.11-140-g367378fa branch master at 202406230725 inputs file rtl_tcp RTL-SDR

New defaults active, use "-Y classic -s 250k" if you need the old defaults

Detached kernel driver
Found Rafael Micro R820T tuner
[SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2024-06-23 23:22:28
model     : ThermoPro-TP829b                       id        : 08
Display Unit: Celcius    Temperature 1: 25.4 C     Temperature 2: 26.6 C     Flags     : f             Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2024-06-23 23:22:32
model     : ThermoPro-TP829b                       id        : 08
Display Unit: Celcius    Temperature 1: 25.5 C     Temperature 2: 26.6 C     Flags     : f             Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2024-06-23 23:22:36
model     : ThermoPro-TP829b                       id        : 08
Display Unit: Celcius    Temperature 1: 25.5 C     Temperature 2: 26.6 C     Flags     : f             Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2024-06-23 23:22:40
model     : ThermoPro-TP829b                       id        : 08
Display Unit: Celcius    Temperature 1: 25.5 C     Temperature 2: 26.6 C     Flags     : f             Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2024-06-23 23:22:44
model     : ThermoPro-TP829b                       id        : 08
Display Unit: Celcius    Temperature 1: 25.5 C     Temperature 2: 26.6 C     Flags     : f             Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2024-06-23 23:22:48
model     : ThermoPro-TP829b                       id        : 08
Display Unit: Celcius    Temperature 1: 25.5 C     Temperature 2: 26.6 C     Flags     : f             Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2024-06-23 23:22:52
model     : ThermoPro-TP829b                       id        : 08
Display Unit: Celcius    Temperature 1: 25.5 C     Temperature 2: 26.6 C     Flags     : f             Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2024-06-23 23:22:56
model     : ThermoPro-TP829b                       id        : 08
Display Unit: Fahrenheit Temperature 1: 25.5 C     Temperature 2: 26.6 C     Flags     : f             Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2024-06-23 23:23:00
model     : ThermoPro-TP829b                       id        : 08
Display Unit: Fahrenheit Temperature 1: 25.6 C     Temperature 2: 26.7 C     Flags     : f             Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2024-06-23 23:23:04
[... cut ...]
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2024-06-23 23:23:28
model     : ThermoPro-TP829b                       id        : 08
Display Unit: Celcius    Temperature 1: 34.8 C     Temperature 2: 26.7 C     Flags     : f             Integrity : CHECKSUM
```

closed 2024-06-23 by AryehGielchinsky

tabnul, 2024-12-08:
Any idea on whether the TP829 (non B) works as well?
It seems that the TP829 and TP829B are different devices, also listed separately on the thermopro website. 

Original post describes TP829B but links to TP829 product om Thermopro website.
TP829B correct link would be [Link](https://buythermopro.com/product/thermopro-tp829b-1000-ft-wireless-meat-thermometer-with-4-meat-probes-for-grilling-and-smoking/)

AryehGielchinsky, 2024-12-12:
I'm not sure. You could always try. If it doesn't you could work with the people in this repo, they were very helpful.

To clarify for anyone reading this in the future, the device added in this issue is ThermoPro Model TP829B, FCCID 2AATP-TP29B.

[image]

tabnul, 2024-12-18:
I can confirm that the ' normal' tp829 is just the tp829b . just received it.

not able to receive data yet but that is another story.

zuckschwerdt, 2024-12-18:
When everything is confirmed working please remind us to update the infos at https://github.com/merbanan/rtl_433/blob/master/src/devices/thermopro_tp82xb.c with your findings that it's "TP829(B)".

tabnul, 2024-12-18:
thanks, I guess no change needed.

I did have trouble detecting it at first. Very occasionally I got response. 
only after setting ' -Y classic -s 250k ' it started working consistently. Most probably related to my cheap Chinese DVB dongle but for the record , my command:

`docker run --device /dev/bus/usb/001/004 hertzg/rtl_433:24.10-alpine-3.21 -f 915M -R 259 -Y classic -s 250k`

cptfbi, 2025-01-04:
Im trying to get this to work so that I can bring it into Home Assistant.  Any way I can help?  Glad to see there is still progress on this.

warwagon1979, 2026-10-23:
I had been trying to get the ThermoPro TP829B wireless thermometer to work in home assistant.

The problem was that the temperatures would take forever to show up in Home Assistant, and when they finally did, the readings were completely wrong. I'm talking temperatures like 400°F or even 800°F when the probe was sitting at room temperature.

Tonight, I spent quite a while troubleshooting it with ChatGPT. After testing different frequencies, sample rates, SDR gain settings, antennas, and even analyzing raw radio recordings, we finally got it working!

Now the temperatures are accurate and updating approximately every four seconds, just like they should.

I figured I'd share the settings that worked for me in case someone else is fighting the same problem.

I'm using the rtl_433 (next) Home Assistant add-on with a Nooelec NESDR SMArt v5 USB receiver.

```
# rtl_433 configuration — ThermoPro TP829B test
# Receiver: Nooelec NESDR SMArt v5
# Select the 915 MHz receiver

device :00000002

# MQTT connection

output mqtt://192.168.1.216:1883,user=(Your Username),pass=(Your Password),retain=0

# Timestamp information

report_meta time:iso:usec:tz

# Show decoded readings in Home Assistant add-on logs

output kv

# Radio configuration
frequency 915.0M
sample_rate 1024k
gain 0
pulse_detect filter=400000
pulse_detect classic

protocol -59
protocol -60
protocol -82
protocol -88
protocol -89
protocol -90
protocol -95
protocol -110
protocol -123
protocol -140
protocol -156
protocol -168
protocol -180
protocol -186
protocol -201
protocol -203
```

Happy temping.