merbanan/rtl_433 #2963: Add model/id to an output_tag?

jtowe1, 2024-06-09:
I'm trying to create custom output tags to put in the mqtt message, but the `[/model]` and `[/id]` variable don't seem to be translating to the actual model and id (they do in the `output` line, but not in the `output_tag` line.

```
# These two translate model and id correctly
output mqtt://10.0.0.245:1883,events=rtl_433[/model][/id],retain=1
output mqtt://10.0.0.245:1883,events=homeassistant/sensor[/model][/id]/config,retain=1

# These don't translate model and id
output_tag state_topic=rtl_433[/model][/id]
output_tag value_template={{ value_json.contact_open }}
output_tag name=Honeywell-[/id]
```

zuckschwerdt, 2024-06-09:
The `output_tag` mechanism is run before any concrete output (like MQTT) and has no access to those substitutions.

Are you trying to insert some type of auto-configuration information with the events? Did you notice https://github.com/merbanan/rtl_433/blob/master/examples/rtl_433_mqtt_hass.py

jtowe1, 2024-06-10:
Yeah.  Was hoping that rtl_433 could send those itself.  I guess i'll need to run a script like that also to create the config topics.  Thanks for linking it!

closed 2024-06-10 by jtowe1

gdt, 2024-06-10:
FWIW, I think it's best to -F syslog to rtl_433_mqtt_relay and then use the hass script for configuration data.   i'm not even a fan of rtl_433 having mqtt builtin.

As ticket gardener, thank you for closing your own ticket now that it's concluded.  I was just about to hit close.

jtowe1, 2024-06-10:
As a follow up for future folks, running `rtl_433_mqtt_hass.py` worked flawlessly.